emsesp/EMS-ESP #323: Device detection enhancement

lendy007, 2020-02-05:
Due to issues discussed in #311 where RC30 thermostat was properly recognized only during boiler power cycle or execution of `autodetect scan` in telnet following enhancements were proposed by @proddy which seems to be quite useful in multiple scenarios:

- `devices `shows the known devices recognized by EMS-ESP, and which are on/off e.g. the heating circuits
- `devices detect` will ask the Boiler/Master which devices he has connected and will also try and connect to them
- `devices detect force` will do a force check on all known devices in the EMS-ESP library and see which ones acknowledge. Then add them to the device list.
- `devices save` will save the known devices to EMS-ESP so on next reboot it can quickly connect to them.

proddy, 2020-02-05:
lol, i created one too #322. I'll remove it and use this

proddy, 2020-02-05:
On boot EMS-ESP sends a query to the Master (boiler) asking to show us which EMS devices it has registered on its bus. Then it queries each one for their details like the version number, product id etc.

The idea is for this feature is to remember the last set of known EMS-ESP devices and load them first when booting up EMS-ESP instead of waiting for the Master to respond.  Note this would only be available via Telnet and not the web, as its a power-user function and only needed for those strange situations when Thermostats can't be found (e.g. #311 as stated above)

PhillyGilly, 2020-02-07:
This power user enhancement would have helped when I was testing my MM200 in hc1/hc2/wwc1/wwc2 when I had to power cycle EMS-ESP to force it to “forget” previous configurations.

proddy, 2020-02-07:
> This power user enhancement would have helped when I was testing my MM200 in hc1/hc2/wwc1/wwc2 when I had to power cycle EMS-ESP to force it to “forget” previous configurations.

a plain `autodetect` wipes the previous config and does a re-scan.

The probem lendy007 was having is that sometimes his Thermostat wouldn't be detected by EMS-ESP, only after rebooting the actual boiler. It's a bit of an edge case

lendy007, 2020-02-18:
Hi @proddy - any idea when this enhancement could possibly get into firmware roadmap?

Thank you

proddy, 2020-02-18:
I haven't looked into it yet.  Been busy trying to get 1.9.5 stable with issues #298 and #326 

are you still having the issue when your thermostat is not detected after a power-cycle and need to do a manual `autodetect` to find it again?

lendy007, 2020-02-18:
Yes, after each reboot RC30 isn't detected and I need to visit telnet and perform `autodetect scan`

proddy, 2020-02-18:
just to be 100% sure, after a reboot, can you go into telnet, `log v` and `boiler read 7` and send the output of the lines with type 0x07 in it, e.g. :

```
log v
boiler read 7

Requesting type UBADevices(0x07) from dest 0x08
(12:52:43) Sending read of type 0x07 to 0x08, telegram: 0D 88 07 00 20 (CRC=C8)
(12:52:43) Boiler -> Me, type 0x07, telegram: 08 0D 07 00 03 80 00 00 00 00 00 00 00 00 00 00 00 (CRC=8E) #data=13
(12:52:43) Boiler -> All, type 0x07, telegram: 08 00 07 00 23 80 00 00 00 00 00 00 00 00 00 00 00 (CRC=DB) #data=13
(12:52:44) Boiler -> All, type 0x07, telegram: 08 00 07 00 03 80 00 00 00 00 00 00 00 00 00 00 00 (CRC=6B) #data=13
```

I just want to confirm something before I change all the code for this single edge case.

lendy007, 2020-02-18:
Sure, this is the content:

```
log v

System Logging set to Verbose
boiler read 7

Requesting type UBADevices(0x07) from dest 0x08
(13:01:31) Sending read of type 0x07 to 0x08, telegram: 0B 88 07 00 20 (CRC=A8)
(13:01:31) UBAMaster -> Me, type 0x07, telegram: 08 0B 07 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=7C) #data=15
(13:01:35) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 14 00 FD 64 19 09 11 25 00 00 94 00 94 80 00 FF FF FF 2D 48 00 00 00 01 00 (CRC=26) #data=25
(13:01:35) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 0A 00 93 00 93 00 00 00 01 00 (CRC=54) #data=10
(13:01:36) 0x10 -> All, type 0x3E, telegram: 10 00 3E 00 00 00 1D 00 9D 00 00 14 1E 2C 00 00 64 00 14 (CRC=04) #data=15
(13:01:37) 0x10 -> UBAMaster, type 0x1A, telegram: 10 08 1A 00 14 64 64 00 (CRC=70) #data=4
(13:01:37) 0x10 -> UBAMaster, type 0x35, telegram: 10 08 35 00 01 01 (CRC=00) #data=2
```

proddy, 2020-02-18:
ok, I added this functionality. Only used the word scan instead of detect.

So in your case after you have done a `devices scan deep` do a `devices save`. 

`info` should show you what is saved, e.g. look at the last line....

```
info

EMS-ESP system status:
  System logging set to None
  LED: on, Listen mode: off
  1 external temperature sensor found
  Boiler: enabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: enabled, Shower Alert: disabled

EMS Bus status:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=61, # CRC errors=13
  Saved known device IDs: 08 09 17
```

please provide feedback.

lendy007, 2020-02-18:
Works like a charm! Thank you very much :)

Just small comment to help section - should be added as devices save, because it is mentioned there just as `save `and that didn't work for me. I then checked in your comment above that command should be `devices save`

closed 2020-02-18 by lendy007