emsesp/EMS-ESP #325: Home Assistant displayed values

PhillyGilly, 2020-02-10:
I want to display working time in hours in my HA.
I have the following lines in the sensor.yaml file:

- platform: mqtt
  state_topic: 'home/ems-esp/boiler_data'
  name: 'DHW Working Time'
  unit_of_measurement: 'hr'
  value_template: '{{ state_attr(value_json.wWWorkM) | (float / 60) | round(0) }}'

but my overview shows:

[image]

Any suggestions? TIA

proddy, 2020-02-10:
do you want it rounded to the nearest hour?

proddy, 2020-02-10:
https://community.home-assistant.io/t/create-output-from-time-sensor-in-hours-and-minutes-instead-of-7-43-hours/143689 ?

```
{% set hours = state_attr(value_json.wWWorkM) | float %}
{% set minutes = ((hours % 1) * 60) | int %}
{% set hours = (hours - (hours % 1)) | int %}
{{ '%02i:%02i'%(hours, minutes) }}
```

PhillyGilly, 2020-02-11:
Thanks.  Yes I just wanted a value in whole hours.

I couldn't make your example work but I this one did.
https://community.home-assistant.io/t/time-output-in-minutes-want-it-to-hours-minutes/102909/3

- platform: mqtt
  state_topic: 'home/ems-esp/boiler_data'
  name: 'DHW Working Time'
  unit_of_measurement: 'hr'
  value_template: >
    {% set mins = value_json.wWWorkM %}
    {% set hours = ((mins | int /  60) | string).split('.')[0] %}
    {{hours}}  
[image]

closed 2020-02-11 by PhillyGilly