emsesp/EMS-ESP #326: Version 1.9.5b29 instable?

LineF, 2020-02-12:
Hi,
since some days I'm using v1.9.5b29 with some external DS18B20 sensors.
Sometimes when the ESP is starting up it restarts itself one or more times.

In this example it restarted 4 times:

ESP8266 System status:

 [APP] EMS-ESP version: 1.9.5b29
 [APP] MyESP version: 1.2.26
 [APP] Build timestamp: 2020-02-09 20:01:29
 [APP] Uptime: 0 days 0 hours 0 minutes 36 seconds
 [APP] System Load: 0%
 [WIFI] WiFi Hostname: ems-esp
 [WIFI] WiFi IP: 192.168.1.137
 [WIFI] WiFi signal strength: 100%
 [WIFI] WiFi MAC: 5C:CF:7F:8B:EF:69
 [MQTT] is connected (heartbeat enabled)
 [MQTT] # failed topic publishes: 1
 [NTP] Local Time is 20:13:06 CET  (1581538386)
 [SYSTEM] System is Stable
 [SYSTEM] Board: PLATFORMIO_NODEMCUV2
 [SYSTEM] CPU frequency: 80 MHz
 [SYSTEM] SDK version: 2.2.2-dev(38a443e)
 [SYSTEM] CPU chip ID: 0x8BEF69
 [SYSTEM] Core version: 2_6_2
 [SYSTEM] Boot version: 31
 [SYSTEM] Boot mode: 1
 [SYSTEM] Last reset reason: Exception
 [SYSTEM] Last reset info: Fatal exception:9 flag:2 (EXCEPTION) epc1:0x401053c0 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000003 depc:0x00000000
 [SYSTEM] Restart count: 4
 [SYSTEM] # TCP disconnects: 0
 [SYSTEM] rtcmem status: blocks:2 addr:0x60001280
 [SYSTEM] rtcmem 00: 1163087990
 [SYSTEM] rtcmem 01: 65540
 [FLASH] Flash chip ID: 0x1640E0
 [FLASH] Flash speed: 40000000 Hz
 [FLASH] Flash mode: DIO
 [FLASH] Flash size (CHIP): 4194304
 [FLASH] Flash size (SDK): 4194304
 [FLASH] Flash Reserved: 4096
 [MEM] Firmware size: 618576
 [MEM] Max OTA size: 2519040
 [MEM] OTA Reserved: 16384
 [MEM] Free Heap: 22952 bytes initially | 11872 bytes used (51%) | 11080 bytes free (48%)

The pings to the device looked like that during the restart phase(s):
Antwort von 192.168.1.137: Bytes=32 Zeit=2ms TTL=255
Antwort von 192.168.1.137: Bytes=32 Zeit=2ms TTL=255
Antwort von 192.168.1.137: Bytes=32 Zeit=2ms TTL=255
Antwort von 192.168.1.137: Bytes=32 Zeit=3ms TTL=255
Antwort von 192.168.1.137: Bytes=32 Zeit=2ms TTL=255
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.1.137: Bytes=32 Zeit=2425ms TTL=255
Antwort von 192.168.1.137: Bytes=32 Zeit=4ms TTL=255
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.1.20: Zielhost nicht erreichbar.
Antwort von 192.168.1.137: Bytes=32 Zeit=1966ms TTL=255
Antwort von 192.168.1.137: Bytes=32 Zeit=14ms TTL=255
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.1.137: Bytes=32 Zeit=2092ms TTL=255
Antwort von 192.168.1.137: Bytes=32 Zeit=11ms TTL=255
Antwort von 192.168.1.137: Bytes=32 Zeit=3ms TTL=255
Antwort von 192.168.1.137: Bytes=32 Zeit=3ms TTL=255
Antwort von 192.168.1.137: Bytes=32 Zeit=4ms TTL=255
Antwort von 192.168.1.137: Bytes=32 Zeit=2ms TTL=255
Antwort von 192.168.1.137: Bytes=32 Zeit=39ms TTL=255
Antwort von 192.168.1.137: Bytes=32 Zeit=11ms TTL=255
Antwort von 192.168.1.137: Bytes=32 Zeit=2ms TTL=255
Antwort von 192.168.1.137: Bytes=32 Zeit=6ms TTL=255
Antwort von 192.168.1.137: Bytes=32 Zeit=2ms TTL=255

Additional to that I get sporadic reboots (in average one per day).
There were 7 external sensors attached.

BTW: when I increase the sensors to 9, I get more problems:
The sensors aren't reported by mqtt any more and I can't logon to the Web-UI: the password input field doesn't pop up...

Version 1.9.5b11 worked four weeks without a single reboot (and without external sensors).

Kind regards,
Martin

proddy, 2020-02-12:
@LineF thanks for reporting this. There are reports of crashes and I'm pretty sure its in the MQTT code (either the latest TCP libraries or my code). If disable MQTT do you still get the restarts?

I'm thinking of adding back publish_time=0 which turns off all the automatic MQTT publishing. If I can narrow down the root cause I can fix it.

LineF, 2020-02-12:
I tried over 20 restarts with mqtt disabled. None of them got multiple restarts.

Then I typed the command "set mqtt_enable on" and with sending the command the ESP rebooted. The value "enabled on" was saved without the need of the "save" command.

The reboots after enabling mqtt got multiple reboots again...

proddy, 2020-02-12:
@LineF do you have a mixer (MM) by any chance?

LineF, 2020-02-12:
no, sorry - I have no mixer device

proddy, 2020-02-12:
if you try the latest 1.9.5b31 and `set publish_time 0` which should prevent any MQTT being sent (only receive) does it still crash?

ps. thanks for the coffee

LineF, 2020-02-12:
no, publish_time=0 in 1.9.5b31 doesn't solve the problem.
Still multiple reboots with that.

proddy, 2020-02-12:
i made a mistake with publish_time - just did a new build of b31 so make sure you get the latest. Also travis is broken for some reason which I need to look into, so build manually if you can.

LineF, 2020-02-12:
normally I build myself with platformio. I will test again.

LineF, 2020-02-12:
no, again multiple reboots

proddy, 2020-02-12:
damn. ok. so something else is causing the crashes. Any ideas from the logs?

LineF, 2020-02-12:
I haven't activated logs yet.
What does happen, if my 7 sensors don't fit in the mqtt payload? Is there an overwrite at the end of the field or does the json lib handle overflows? - just an idea.
Another thing I saw: when issuing multiple "system" commands each time the free mem goes a little bit down...
Maybe there are memory leaks or overwriters which may cause in the end the crash?

proddy, 2020-02-12:
there are a few mem leaks but nothing that should cause a crash. mind you, I have a v2 close to ready of EMS-ESP which has a new Telnet and WebUI so any of my free time goes into that project. I'm only doing small fixed on 1.9.x. The current version of EMS-ESP was just a hacked hobby project I did over christmas I never thought it would be useful for others so didn't really spend a lot of time on writing clean code.

LineF, 2020-02-12:
Yes, I saw v2 on github, but didn't try that.
I'm impressed on your coding skills! The whole project seems being clear structured and changes are quick implemented by you. Congratulations!
I know, memory leaks are horrible to find most times...

proddy, 2020-02-12:
seriously? its horrible coding. most of it is in plain C for speed and optimizations. Now I understand the EMS protocol better I've started reprogramming the core into C++ using classes, templates and the SDL libraries to be more secure, thread safe to avoid interrupt locks which will make it easier to read and extend by others. Also the web has been re-written to use typescript instead of JavaScript with stronger encryption. I just need a weekend to tie it all together. As soon as I can fix these crahes with 1.9.5 it'll be full-time on v2.

LineF, 2020-02-12:
For now I went back to 1.9.5b11.
I noticed that the behaviour of "publish_time=0" changed from "publish whenever necessary" to "never publish"...
I personally prefer the first (older) variant.

LineF, 2020-02-12:
But it's no problem for me to switch to the new version for testing...

proddy, 2020-02-13:
> I noticed that the behaviour of "publish_time=0" changed from "publish whenever necessary" to "never publish"...
> I personally prefer the first (older) variant.

The problem I found with the automatic publishing was that it was too frequent, almost every second something would trigger a change and this caused the MQTT to flood. I'll add it back as an option anyway and use a publish_time setting of "-1" to mean off.

proddy, 2020-02-13:
ok, implemented publish_time=0 for automatic and publish_time=-1 for off.

now to fix the stability problems, would be good to stress test with publish_time=-1 to see if it's any better. This will help narrow down on which section of the code is caused the buffer overflow.

LineF, 2020-02-13:
I tested restarts with publish_time=10 and =0.
In both cases I got multiple restarts.
After an autodetect sometimes the sensors aren't published, at other times the boiler values aren't published (publish_time set to 10).
Now I will test with publish_time=-1

LineF, 2020-02-13:
With publish_time=-1 I get multiple reboots, too...

LineF, 2020-02-13:
I just set publish_time = 0 and then sensors stopped being published.
Seems that there is another bug when executing commands then some type of publishing gets supressed...
After reboot all is working ok again.

proddy, 2020-02-14:
@LineF , thanks for checking. with b32 I can't reproduce what you reported. publish_time=0 does send sensor data, publish_time=-1 sends nothing and publish_time=10 sends everything every 10 seconds. All with no crashes.

If you're getting crashes with publish_time=-1 then the cause is not in the MQTT code, which is good. Something else is wrong and we need to find what is causing it. Does it happen when you just boot and leave it running, without any MQTT commands, Telnet or Web commands?

LineF, 2020-02-16:
I let this version now run alone for some days (with sensors and with mqtt publishing) to see its stability. But today I got the first reboot at a time where I didn't do anything with the EMS gateway...
Looking more for that...
I'll let you know in some days.
Martin

LineF, 2020-02-17:
I've got two reboots last night:
[image]
I'll go back to 1.9.5b11 (this one was stable for weeks) to see if it is still stable with my attached sensors. If not, then I have a problem with my sensors...

proddy, 2020-02-17:
that's not good. What was your `publish_time` set too? Also if you type in `system` does it show:

`[MQTT] # failed topic publishes: 0`

Perhaps I need to add that to the heartbeat MQTT payload too at some point.

@majdzik84 is getting restarts too, all the time when MQTT is trying to publish. See #298. But I just don't know what is causing it. I'm not seeing the same behavior here on my test systems.

Let me know how you get on.

LineF, 2020-02-17:
publish time was set to zero.
old version already running - no more debugging with new version...
I saw sporadically publishing errors when I was connected to the telnet console...
Will test now old version - you will hear from me.

proddy, 2020-02-18:
I added some coded a while back to auto-retry when a mqtt publish failed. I think that is what is causing the resets. See if this still happens with b35

LineF, 2020-02-18:
due to the hierarchical structure of the sensors topic I can't use the new version at the moment. I first have to find out how to configure my grafana for hierarchical topics or even use telegraf to convert the json formats...

proddy, 2020-02-18:
how were you getting the values into Grafana in the first place? Was it using Home Assistant or Domoticz to push data to InfluxDB?

LineF, 2020-02-18:
Hi,
im quick testing your new version.
I get many mqtt publish failures - but only for the sensors data:
[image]

Getting the hierarchical sensors data into my grafana is a big thing for me at the moment...

proddy, 2020-02-18:
there were always publish failures on some systems (I could never reproduce on mine though) so I built in a retry mechanism a while ago which I think caused the watch-dog resets. The latest 1.9.5b35 has it removed. 

Could you send me the raw MQTT sensor data so I can use it on my test system?

As for Grafana, how are you getting the data into InfluxDB - via a python script or via another application?

LineF, 2020-02-18:
How can I get the _RAW_ mqtt data?
EMS-ESP sends it to my mosqitto broker (armbian/debian linux). There I have iobroker installed which has node red functionality built in. In this node red I fetch the mqtt data and send it directly to the influx-db:
[image]
In the influxdb the json data is the same structure as sent by EMS-ESP and can be interpreted directly by a grafana select (from db)  statement (but only the non-nested data).

At the moment I'm trying to learn node red json message handling, so that I can convert the hierarchical json structure to the old flat one...

LineF, 2020-02-18:
Do you mean this one as _raw_ data?:
home/heating/ems-esp/sensors {"sensor1":{"id":"28D45A79A2190310","temp":30.88},"sensor2":{"id":"287A2D79A216037B","temp":35.69},"sensor3":{"id":"28F52C79A200033E","temp":61.94},"sensor4":{"id":"280D4179A2160356","temp":26.81},"sensor5":{"id":"28ED0C79A216037A","temp":33.56},"sensor6":{"id":"28BD4079A216039C","temp":52.75},"sensor7":{"id":"28472279A21903AB","temp":41.31},"sensor8":{"id":"281F2479A21603A0","temp":13.25},"sensor9":{"id":"28FF6779A2190336","temp":22.94}}

LineF, 2020-02-18:
BTW: I had to increase for this the mqtt payload buffer from 400 to 800...

LineF, 2020-02-19:
I'm now back on 1.9.5b35 (I can decode the nested sensors message in the meanwhile) and will let it run for a while.
As I can say after the first few minutes:
I had again a multiple restart:
[image]
Normally there is a message "clean restart after flashing'... or something like that.

The "restart" command doesn't clear the last reset reason. So I can't see clearly, if the last "restart" command worked in a "clean"way. Perhaps you can "restart" the ESP with a real "reset" command so that the cpu registers are cleared from the "exception reset" status...

Second, I see again many errors publishing sensor data:
[image]

With the setting "publish_time=0" at least some of the messages arrive at my mqtt broker (about half of them).
With the setting "publish_time=10" only the first few messages after a restart were correctly sent, then _all_ messages result in an error and I get no more sensor messages at my mqtt broker.

proddy, 2020-02-19:
I'll see if I can reproduce. Are you using mqtt_qos of 0? What happens if its 1 or 2 ?

LineF, 2020-02-19:
Yes, I'm using 0.
At the moment it's better with "publish_time=0". The errors have now nearly gone away without doing anything else...
Will try qos with 1 and 2...

Another thing I saw with the new version:
[image]

Never saw redetecting device 0x10 before... got this message now two times within 15-20 minutes.

LineF, 2020-02-19:
mqtt_qos=1 :
publish_time=10: only the first few messages were sent, then all further sends got an error.
publish_time=0:
first reboot: first few messages were sent, then all further got an error
second reboot: most messages are sent, only few give an error

The behaviour seems undeterministic...

Another problem I found: In the Web-UI I get no password window to log on - may be another internal memory problem??

proddy, 2020-02-19:
I fixed the "Detected device" issue - thanks for reporting.

I've also been testing with your sensor data and on my setup it's sending out the MQTT correctly with 0 errors. Also no restarts. I've been through the code hundreds of times and still don't know what is causing this.

I've added the #mqtt failures to the MQTT heartbeat to give an easy warning when it  happens e.g.
`heartbeat={"rssid":88,"load":1,"uptime":251,"freemem":50,"tcpdrops":0,"mqttpublishfails":0}`

I'll leave it running overnight and see what happens.

LineF, 2020-02-19:
testing b36...
EMS detection seems to be still problematic:
[image]
sometimes after reboot some of my EMS devices aren't detected. So it was here too.
I then sent a "devices scan" which normally finds my devices - now I had to send a "devices scan deep" to find them...
And you can see that 0x10 was detected more often.

With mqtt_payload size of 600 I get a incomplete sensor list:
[image]
Was your list with 9 sensors ok?

I'll watch now for the mqtt transmition errors...

LineF, 2020-02-19:
seems that I have to do now a "devices scan deep" after each reboot

LineF, 2020-02-19:
still have mqtt send errors.
Now I had problems in mode "publish_time=0" too: first I had only sporadic errors and after some minutes the errors came with each message.

But I think it can't be a problem with my network because then other messages than sensor data should also run into some send errors.

proddy, 2020-02-19:
I'm fixing the devices scan issue. I haven't changed the MQTT at all so its probably something on your end.

proddy, 2020-02-21:
@LineF made a b38 which hopefully will the MQTT publish errors and also optimizes the device scan. Could you do a quick check when you have time?

LineF, 2020-02-21:
ok, new version is running... let's see...

LineF, 2020-02-21:
since 1 hour no mqtt send fails (publish_time=10). memory parameters are stable:
[image]

proddy, 2020-02-21:
nice. I haven't seen any issues on my side either. Nice to see a little more precious memory has free'd up. 

BTW those SYSTEM messages - I didn't know they were printed to the telnet console. Should I remove them?

LineF, 2020-02-21:
Yes.
For debugging now ok, but for production annoying.

proddy, 2020-02-22:
I left my test system on overnight and saw 5 mqtt publish failures (this is in the heartbeat msg and also seen in `system` command). I've increased the retry. Still I think 5 from the 300,000 messages is not bad on a noisy wifi.

LineF, 2020-02-22:
still no retransmits neither reboots - very good till  now:
[image]

LineF, 2020-02-24:
b38 running now for three days - no reboots - no tcp drops - no mqtt resends!
Perfect!

proddy, 2020-02-24:
phew! thats good news. Just hope b42 is as stable. I made a lot of changes over the weekend and usually I end up breaking something. The latest build has optimized MQTT and I managed to squeeze out more free memory.

proddy, 2020-02-24:
I'll close this one now (as the title refers to b29). If we see errors again we can always re-open. Thanks for stress testing Martin

closed 2020-02-24 by proddy