emsesp/EMS-ESP #327: External sensor names should be aliased

LineF, 2020-02-12:
Hi,

it would be nice if a specific sensor (identified by its hardware ID) could be given an alias name. This name then should be used in the mqtt telegram. The ID - name assignment should be saved in flash memory.

This would solve the problem that the sensors get a new numbering scheme after adding or removing some sensors.

At the moment the sensors are saved in my DB with the names temp_x. After adding new sensors the names are mixed and I have in my DB the same name but temperature from a completely other sensor. So all my graphs get messed up...

Kind regards,
Martin

proddy, 2020-02-12:
that would be a good enhancement. Would it help (as a temporary solution) if we used the internal h/w ID as it is unique?

LineF, 2020-02-12:
yes, that would help.
Then I could use the IDs in my graphs.
But be aware: changing the names will affect other users, too...

proddy, 2020-02-12:
I made a small change in 1.9.5b31 to use the sensor address as the primary ID. Kinda makes sense.

LineF, 2020-02-12:
I see only 6 adresses of my 7 sensors in my db...
mabe mqtt payload field is too small now for the long adresses?

proddy, 2020-02-12:
probably. buffer was 200 bytes. I made it 600.

LineF, 2020-02-12:
increasing the buffer from 200 to 600 bytes (in line 577) doesn't help - I still get only 6 sensors:

home/heating/ems-esp/sensors {"287A2D79A216037B":30,"28F52C79A200033E":66.38,"280D4179A2160356":27.75,"28ED0C79A216037A":42.13,"28BD4079A216039C":59.31,"28472279A21903AB":50.69}

LineF, 2020-02-12:
mqtt limit? maximum publish size?

proddy, 2020-02-12:
does `info` list all 7 ?

LineF, 2020-02-12:
yes

LineF, 2020-02-12:
External temperature sensors:
  Sensor #1 type:DS18B20 id:287A2D79A216037B temperature: 36.68 C
  Sensor #2 type:DS18B20 id:28F52C79A200033E temperature: 66.44 C
  Sensor #3 type:DS18B20 id:280D4179A2160356 temperature: 27.94 C
  Sensor #4 type:DS18B20 id:28ED0C79A216037A temperature: 42.25 C
  Sensor #5 type:DS18B20 id:28BD4079A216039C temperature: 59.38 C
  Sensor #6 type:DS18B20 id:28472279A21903AB temperature: 50.75 C
  Sensor #7 type:DS18B20 id:28FF6779A2190336 temperature: 22.62 C

proddy, 2020-02-12:
Ok. I know where to look now.

proddy, 2020-02-13:
let me know if this is still a problem with the latest dev build.

LineF, 2020-02-13:
at the moment I'm testing...

LineF, 2020-02-13:
all 9 sensors are now published correctly by mqtt.

proddy, 2020-02-14:
ok good. I'll leave it like this for now. Customizing MQTT is a good feature to have and I've thought about it before. Perhaps using an additional mapping file so it's fixed at compile time since looking up each MQTT topic, translating it and seeing if it's needed is quite CPU intensive.

LineF, 2020-02-16:
ok, publishing the sensor values by their sensor address is ok for now. I get unique adressable values and I can set an alias name in my graphs.

Thanks for the quick solution here!

Maybe you can make the number of maximum sensors more flexible by calculating the needed space for the variables by a constant MAX_DS_SENSORS or someting like that...

Martin

proddy, 2020-02-17:
there is no max #sensors. Each sensor is stored in a dynamic array (C++ vector). If you're worried the MQTT payload will become too large then I added a check into the code to make sure there is no buffer overflow. Currently, it supports up to 12 sensors which I expect is more than enough.

bbqkees, 2020-02-18:
Would it be an idea to also keep the 'sensor 1' etc in the payload as well?
A long string of f.i. {"28AD1C26060000FD":30.19,"286D9127060000CA":27.06} makes it harder to identify each sensor.

proddy, 2020-02-18:
good idea, I'll add it.

proddy, 2020-02-18:
@LineF @bbqkees if we choose to include the sensor number it will turn the MQTT into a nested JSON as having a single key/value pair wouldn't work. So the payload would end up looking like:

`{"sensor1":{"id":"28FF47AC90160444","temp":19.88},"sensor2":{"id":"3AFH49ACA0130873","temp":21.16}}`

or

```
{
	"sensor1": {
		"id": "28FF47AC90160444",
		"temp": 19.88
	},
	"sensor2": {
		"id": "3AFH49ACA0130873",
		"temp": 21.16
	}
}
```

which would require a little more logic parsing on the receiving end, whether it's straight into Grafana, Home Assistant or node-red.

you still ok with this?

bbqkees, 2020-02-18:
The nested parsing is already necessary for decoding the HC1 to HC4 thermostat topics, so this should not be an issue for the sensors topic.
If you go the nested way you can maintain that structure even if you start added other non-DS18B20 sensors in a future firmware.

proddy, 2020-02-18:
ok, implemented this in b35

proddy, 2020-02-22:
@LineF  I added a new setting called `mqtt_nestedjson` . When set to off it will send each sensor separately via MQTT. This should make it easier for your iobroker to parse into influxdb/grafana

e.g.
before: `sensors={"sensor1":{"id":"28FF47AC90160444","temp":20.5}}`
after: `sensor1={"id":"28FF47AC90160444","temp":20.5}`

let me know if it works for you.

MichaelDvP, 2020-02-22:
@proddy  Both versions use the numbering, which changes if, after a restart, a sensor is missing or a new sensor is attached. To identify the correct sensor  we have to walk through all sensor# and check the id to be shure it is the same sensor as before. 
Your first version with  `sensors={"28FF47AC90160444":20.5}` always points to same sensor.
I prefer if we can choose between one version with numbering and one without.  
BTW: after is `sensors1=` not `sensor1=`

LineF, 2020-02-22:
I agree. This is a good suggestion. But may fail or must be adapted if other sensor types are attached.
But for now... ok.

proddy, 2020-02-23:
ok, I'll add a third option to go back to `sensors={"28FF47AC90160444":20.5, "28FF47AC90160445":23.5,"28FF47AC90160448":18.2}`

but will this make a compile-time option (using -D in platformio.ini) especially for you power users as adding more settings will make EMS-ESP more complicated.

proddy, 2020-02-23:
did this for b40 (will upload soon). Flag is  -DSENSOR_MQTT_USEID

closed 2020-02-23 by proddy