emsesp/EMS-ESP #330: Support for Enviline

VincentSC, 2020-02-22:
**Is your feature request related to a problem? Please describe.**
This is the output that the dashboard gives. I read different data from the boiler and heatpump. Thermostat seems to be correct.
[image]
Note: with the 1.95 dev-version, it does not find the "heat pump" and does not accidentally add support.

**Describe the solution you'd like**
I'd like to have support for this device - got a programming and Linux background. What steps should I follow? I saw #248, but would like some assistance to get started.

VincentSC, 2020-02-23:
What I'd like to see if the telegrams are understood or not. Is there a way to translate a telegram from the command line?

info:
```
EMS-ESP system stats:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: enabled, Shower Alert: enabled

EMS Bus stats:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=4165, # CRC errors=0
  Tx: Last poll=2.237 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: Nefit Enviline (DeviceID:0x08 ProductID:172 Version:01.16)
  Hot tap water: running
  Central heating: off
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water selected temperature: ? C
  Warm Water desired temperature: ? C
  Warm Water current temperature: ? C
  Warm Water current tap water flow: ? l/min
  Warm Water # starts: ? times
  Warm Water 3-way valve: off
  Selected flow temperature: 23 C
  Current flow temperature: 21.2 C
  Return temperature: ? C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: ? bar
  System service code:
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Outside temperature: 7.6 C
  Boiler temperature: ? C
  Pump modulation: 0 %
  Burner # starts: 0 times
  Total burner operating time: 0 days 0 hours 0 minutes
  Total heat operating time: 0 days 0 hours 0 minutes
  Total UBA working time: 84 days 9 hours 42 minutes

Heat Pump stats:
  Heat Pump module: HeatPump Module (DeviceID:0x38 ProductID:200 Version:40.04)
  Pump modulation: 0 %
  Pump speed: ? %

Thermostat stats:
  Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID:0x10 ProductID:158 Version:13.07)
  Thermostat time is 08:29:56 23/02/2020
  Heating Circuit 1
   Current room temperature: 19.8 C
   Setpoint room temperature: 19.5 C
   Mode is set to manual
   Day Mode is set to day
```

Log v:
```
(08:27:20) UBAMaster -> All, type 0x03A3, telegram: 08 00 FF 00 03 A3 00 0B 00 00 00 00 00 00 00 00 00 1F 00 00 00 00 00 00 00 00 08 00 00 00 00 (CRC=0A) #data=25
(08:27:25) Gateway -> Thermostat, type 0x0101, telegram: 48 90 FF 1D 01 01 A6 (CRC=3F) #data=1
[... 119 lines cut ...]
(08:28:03) Sending read of type 0x18 to 0x08, telegram: 0B 88 18 00 20 (CRC=D4)
(08:28:03) UBAMaster -> Me, telegram: 08 0B 18 00 (CRC=5C)
(08:28:03) Gateway -> Thermostat, telegram: 48 90 F9 00 11 FF 01 9B 00 (CRC=C9)
(08:28:03) Thermostat -> Gateway, type 0x019B, telegram: 10 48 F9 00 FF 01 9B 00 4F 00 00 00 01 00 00 00 01 00 03 00 07 00 03 00 02 (CRC=D2) #data=17
(08:28:03) UBAMaster -> All, type 0x039F, telegram: 08 00 FF 00 03 9F 00 00 0A 8C 08 34 00 00 07 08 00 00 07 08 00 00 00 00 00 00 00 00 00 00 00 (CRC=65) #data=25
n(08:28:03) UBAMaster -> All, type 0x03A0, telegram: 08 00 FF 00 03 A0 0E 10 01 51 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=FE) #data=25
(08:28:03) UBAMaster -> All, type 0x03A2, telegram: 08 00 FF 00 03 A2 00 01 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=48) #data=25
(08:28:04) Sending read of type 0x19 to 0x08, telegram: 0B 88 19 00 20 (CRC=D0)
(08:28:04) UBAMaster -> Me, telegram: 08 0B 19 00 (CRC=5E)
(08:28:04) UBAMaster -> All, type 0x03A3, telegram: 08 00 FF 00 03 A3 00 0B 00 00 00 00 00 00 00 00 00 1F 00 00 00 00 00 00 00 00 08 00 00 00 00 (CRC=0A) #data=25
(08:28:04) Sending read of type 0x33 to 0x08, telegram: 0B 88 33 00 20 (CRC=78)
(08:28:04) UBAMaster -> Me, telegram: 08 0B 33 00 (CRC=0A)
(08:28:04) Sending read of type 0x16 to 0x08, telegram: 0B 88 16 00 20 (CRC=EC)
(08:28:04) UBAMaster -> Me, telegram: 08 0B 16 00 (CRC=40)
(08:28:05) Sending read of type 0x14 to 0x08, telegram: 0B 88 14 00 20 (CRC=E4)
(08:28:05) UBAMaster -> Me, type 0x14, telegram: 08 0B 14 00 01 DA C4 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=15) #data=27
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
(08:28:06) Gateway -> Thermostat, type 0x0101, telegram: 48 90 FF 03 01 01 A6 (CRC=CF) #data=1
(08:28:06) Thermostat -> Gateway, telegram: 10 48 FF 03 01 A6 (CRC=C9)
lo(08:28:06) UBAMaster -> All, type 0x07, telegram: 08 00 07 00 01 01 00 00 00 00 01 00 01 00 00 00 00 (CRC=81) #data=13
g(08:28:07) Gateway -> Thermostat, type 0x0101, telegram: 48 90 FF 04 01 01 A6 (CRC=F7) #data=1
(08:28:07) Thermostat -> Gateway, telegram: 10 48 FF 04 01 A6 (CRC=D5)
 n(08:28:07) Gateway -> Thermostat, type 0x0201, telegram: 48 90 FF 00 02 01 A6 (CRC=DB) #data=1
```

VincentSC, 2020-02-23:
So when I want to start to find the "hot tap water on/off" telegram like is done in #309, what should I do? https://emsesp.github.io/docs/#/Deciphering-EMS-Plus is not really complete here, as it assumes too many things.

proddy, 2020-02-23:
Looks like your boiler is using quite different EMS+ messages that I haven't seen before and thus not yet supported by EMS-ESP, e.g. 0x101, 0x201, 0x39F, 0x3A0. The data is in those somewhere so with some trial and error we should be able to deduct what is what, And then you can help build up the library and documentation in the wiki. The EMS protocols are not publically shared unfortunately so we need to decypher ourselves.

I see `0xE4` coming in which is some basic info from the boiler and being read correctly, however, the crucial data you need is in `0xE5` which I don't see.

To get this working I would start with
- concentrate on only the boiler (UBAMAster) data for now
- find out which telegrams are sent frequently (every 10 seconds, every 60 seconds)
- figure out which of these contain the boiler information. There should be 2 or 3 like Slow, Fast and WWParameter
- look for settings on your thermostat, app, boiler front-panel for things like boiler flow temp and see if you can find them back in the telegrams

There is no feature to show real-time what each of the telegrams contain but could be easily added by just adding myDebug() statements to the process function calls in `ems.cpp`

VincentSC, 2020-02-23:
Thanks for assisting! Bummer it's not supported yet, but at least we got a really new boiler, hehehe. 

While reading through documents about EMS+, I found out that I needed to look for telegram-types, not the full telegrams. So the below is with the "old thinking", but I keep them here as LibreOffice crashed. I'll add a new reply with the statistics for the telegram types, before going into steps 3 and 4.

Ok, got some statistics from UBAMAster to All in about 4.5 minutes:

27 times, regular 6x per minute:
- type 0x039F, telegram: 08 00 FF 00 03 9F 00 00 0A 8C 08 34 00 00 07 08 00 00 07 08 00 00 00 00 00 00 00 00 00 00 00 (CRC=65) #data=25
- type 0x03A0, telegram: 08 00 FF 00 03 A0 0E 10 01 50 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=1D) #data=25

10 times, each minute at :00 and 14-16 seconds after each minute:
- type 0x07, telegram: 08 00 07 00 01 01 00 00 00 00 01 00 01 00 00 00 00 (CRC=81) #data=13

once a minute, regular:
- type 0x07, telegram: 08 00 07 00 09 01 00 00 00 00 01 00 01 00 00 00 00 (CRC=AD) #data=13
- type 0xE5, telegram: 08 00 E5 00 00 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=9E) #data=27
- type 0x0394, telegram: 08 00 FF 31 03 94 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=9B) #data=25
- type 0x0394, telegram: 08 00 FF 18 03 94 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=F5) #data=25
- type 0x0395, telegram: 08 00 FF 18 03 95 00 00 00 80 00 00 01 AD 00 00 00 00 00 00 00 00 00 00 00 95 00 00 00 8B 00 (CRC=0C) #data=25
- type 0x0395, telegram: 08 00 FF 30 03 95 00 00 00 09 00 00 00 00 00 00 05 B0 00 00 00 00 00 00 06 45 00 00 04 5E 00 (CRC=2C) #data=25
- type 0x0395, telegram: 08 00 FF 48 03 95 00 00 01 52 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=AE) #data=25

(Did this by cleaning up the logs in notepad++: removed the lines without timestamp, made sure there are delimiters everywhere, etc. Then turned into spreadsheet and did some sorting, where I added an extra delimiter for the ones without a type, among other things. Thinking of fixing this in the code to make it more easy to parse, or make a simple counter for the 10-second and 60-second telegrams).

I left out the irregular ones, but kept the [edited log file](https://github.com/proddy/EMS-ESP/files/4242345/original-20200223.log).

VincentSC, 2020-02-23:
Type | Count
-- | --
type 0x0006 | 3
type 0x0012 | 3
type 0x0101 | 15
type 0x0201 | 5
type 0x038D | 57
type 0x038F | 56
type 0x0394 | 15
type 0x0395 | 20
type 0x039F | 27
type 0x03A0 | 27
type 0x03A3 | 27
type 0x07 | 15
type 0x14 | 5
type 0xBF | 1
type 0xD1 | 5
type 0xE3 | 27
type 0xE4 | 53
type 0xE9 | 27

proddy, 2020-02-24:
The interesting ones look like E3,E4,E9 and 39F, 38D, 38F

did you manage to get a reference point? like a boiler temps that we can try and find back in these telegrams?

VincentSC, 2020-02-24:
38D and 38F are fired when I change something in the temperature. In the logs you see there are multiple of these close together - I'll leave them for later. So I think that I did not do the Excel-exercise well, and these are not regular at all.

Now the boiler temperature "jumped" from 50.3 to 50.2. Watched E4. There was no 50. The numbers stat where not changing all the time, none changed there. As the temperature did not change, this could take quite some time. I'm too lazy for that...

So second approach, guessing it's some HEX. Then is it 50.2 (502=01F6) or 50.20 (5020=139C)? E9 was showing 01 F6. Watched the temperature for 50.0 in other logs and bingo!

50.0°C in E9: "08 00 E9 00 37 **01 F4** 01 E7 00 00 00 00 41 3C 00 00 00 00 00 00 00 00 00 00 00 00 37 00 00 00"

proddy, 2020-02-24:
could be! temps are stored as signed 2-byte integers divided by 10 so 0x01F4 is indeed 50.0

what may be handy is watching a telegram, e.g. `log w e9` will only show E9 telegrams.

VincentSC, 2020-02-24:
I know how to log/watch, but not where to pay attention to. :/

Temp went to 49.8, E9 changed to 01 F2. So I'm sure it's that.

Context I missed: the length of the telegrams is different. There is also a short E4, but that one is from the gateway to the UBAMaster - can be ignored. Seems that there is no FAST telegram anymore. E4, E3, E9, 0x039F, 0x03A0 and 0x03A3 are sent each 10 seconds in the same order, so some kind of mega-telegram in 6 parts:
```
(20:47:33) UBAMaster -> All, type 0xE4, telegram: 08 00 E4 00 10 20 30 48 00 CB 16 00 CF 00 00 02 00 02 B2 00 00 00 D2 00 00 00 00 00 D4 00 CF (CRC=40) #data=27
(20:47:33) UBAMaster -> All, type 0xE3, telegram: 08 00 E3 00 01 00 01 00 00 00 00 00 00 00 00 00 CF 00 64 55 00 00 00 00 00 00 00 00 00 00 00 (CRC=F4) #data=27
(20:47:34) UBAMaster -> All, type 0xE9, telegram: 08 00 E9 00 37 01 F6 01 ED 00 00 00 00 41 3C 00 00 00 00 00 00 00 00 00 00 00 00 37 00 00 00 (CRC=77) #data=27
(20:47:34) UBAMaster -> All, type 0x039F, telegram: 08 00 FF 00 03 9F 00 00 0A 8C 08 34 00 00 07 08 02 58 07 08 00 00 00 00 00 00 00 00 00 00 00 (CRC=53) #data=25
(20:47:34) UBAMaster -> All, type 0x03A0, telegram: 08 00 FF 00 03 A0 0E 10 01 51 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=FE) #data=25
(20:47:34) UBAMaster -> All, type 0x03A3, telegram: 08 00 FF 00 03 A3 00 0B 00 00 00 00 00 00 00 00 00 1F 00 00 00 00 00 00 00 00 08 00 00 00 00 (CRC=0A) #data=25

```
and 2 hours later:
```
(22:40:29) UBAMaster -> All, type 0x07, telegram: 08 00 07 00 01 01 00 00 00 00 01 00 01 00 00 00 00 (CRC=81) #data=13
(22:40:34) UBAMaster -> All, type 0xE4, telegram: 08 00 E4 00 10 20 30 48 00 CB 16 00 E1 00 00 02 00 02 B2 00 00 00 FF 00 00 00 00 01 02 00 E1 (CRC=E0) #data=27
(22:40:34) UBAMaster -> All, type 0xE3, telegram: 08 00 E3 00 01 00 01 00 00 00 00 00 00 00 00 00 E1 00 64 55 00 00 00 00 00 00 00 00 00 00 00 (CRC=C2) #data=27
(22:40:34) UBAMaster -> All, type 0xE9, telegram: 08 00 E9 00 37 01 F0 01 C8 00 00 00 00 41 3C 00 00 00 00 00 00 00 00 00 00 00 00 37 00 00 00 (CRC=9E) #data=27
(22:40:35) UBAMaster -> All, type 0x039F, telegram: 08 00 FF 00 03 9F 00 00 0A 8C 08 34 00 00 07 08 02 58 07 08 00 00 00 00 00 00 00 00 00 00 00 (CRC=53) #data=25
(22:40:35) UBAMaster -> All, type 0x03A0, telegram: 08 00 FF 00 03 A0 0E 10 01 4F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=7F) #data=25
(22:40:35) UBAMaster -> All, type 0x03A3, telegram: 08 00 FF 00 03 A3 00 0B 00 00 00 00 00 00 00 00 00 1F 00 00 00 00 00 00 00 00 08 00 00 00 00 (CRC=0A) #data=25
```
Is this also with normal EMS+? Are the lengths of all these different? Is it correct, that the last three 39F, 3A0 and 3A3 are new?

So let's focus on these 6? Assuming that the 0x01XX are temperatures, still a lot to decipher. What would be a good next experiment?

Is there a way to watch a selection of telegram types at once? `log w E4 E9` did not work.

VincentSC, 2020-02-26:
As probably I need to start from scratch, I need to make a project plan. Just need to know where I have to start from scratch and where I can build further.

@proddy Do you have a publicly editable overview in i.e. Google sheets for the various infos and the telegrams? Also, do you have written down tactics per info on how to find it? And is there a way to watch multiple telegrams at the same time?

Plan:
- Build an overview in G.sheets of what is available for EMS and EMS+ 1.0, so I know what to look for in EMS+ 2.0
- Design tactics for finding each of the infos
- Improve log-system, so I can better parse it
- Watch a specific device
- Add a new log-type that watches all telegrams from a certain device and prints only the elements which changed (or parse the logs)
- Find out everything  :)

Feedback?

proddy, 2020-02-27:
I don't really have a plan. It typically ends up with a lot of trial and error and your approach sounds solid. When I find something I do put it up on the wiki. 

Most EMS+ messages are pretty consistent across devices so its just a matter of finding the patterns. A good start would be looking at how Bosch/Junkers HT3 does it. See [here](https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT3-Bus_Telegramme.pdf).

VincentSC, 2020-02-27:
Ok, I'll set it up then. I'll take the data from the code `ems.cpp` and hopefully you have info that can be added. We need to be ready when EMS+ 3.0 comes in. :)

Yeah, what I understood is that "Buderus" is behind the bus for all Bosch heaters?

Unfortunately I'm not able to understand that document. I don't see any telegrams I know, so I'm not sure what I can get out of this document. Can you explain what you think I can learn from it?

I see the "calculation types" there. Are these also used on Bosch/Nefit?

Looked to the other doc too. Norberts1's analyser is a more advanced version of the statistics-thingy I want (watching which variables change in the telegrams). Now thinking that using mqtt for sending all the telegrams would be easier.

PS: I got a full time job, so will take my time for this project

closed 2020-09-23 by proddy