emsesp/EMS-ESP #332: Version 1.9.5b41

PhillyGilly, 2020-02-24:
Just uploaded latest firmware.
After auto re-boot there is no Dashboard screen and the version isn't shown in the menu.
Also  the update firmware menu it reported no internet connection.
[image]
I have rolled back to 1.9.5b38

proddy, 2020-02-24:
just made a fix (was reported in #298) in b42. Hopefully that would have fixed the web issue

PhillyGilly, 2020-02-24:
[image]
Updated using the espota.py and that's good.
I'm keep updating as I'm looking out for the rev to incorporate my code fragments from issue #267 :-)

proddy, 2020-02-24:
I forgot about the SM200 !

closed 2020-02-24 by proddy