emsesp/EMS-ESP #333: moduline 200 thermostat not recognized ?

TaliesinDS, 2020-02-25:
It shows `unknown? (DeviceID: 0x17, ProductID: 80, Version: 01.02)`

I am not getting any values from the thermostat.

I can't set or see set temp.

Everything else seems to work fine.

I am running the latest dev build, tried a few others before, all the same issue.

MrWillibald, 2020-02-25:
I have a similar issue with a Sieger eS 72 (Moduline 200). Today I could see a device `unkwon? (DeviceID: 0x17, ProductID: 113, Version: 02.03`.
I tried to add it to the devices, but unfortunately I have some issues with my hardware interface so I couldn't confirm it.

proddy, 2020-02-25:
@TaliesinDS The Moduline 200 wasn't officially supported (see https://emsesp.github.io/docs/#/Supported-EMS-Devices). 

But I've added it to the code. Please check if it works. In the latest dev 1.9.5b43.

proddy, 2020-02-25:
@MrWillibald same with the es 72

TaliesinDS, 2020-02-25:
It recognizes the thermostat and displays temp and set temp now.

Thanks for the quick fix! :)

TaliesinDS, 2020-02-27:
Setting temp from domoticz doesnt seem to do anything.

I get data in domoticz from hc1 but sending a command to set the thermostat temperature does nothing.

From what i can see the mqtt message is correct but nothing happens on the ems gateway side ?
this is the log in domoticz:
`
2020-02-27 02:37:40.077 (EMS bus Wi-Fi Gateway) onCommand called for Unit 112: Parameter 'Set Level', Level: 15.5

2020-02-27 02:37:40.077 (EMS bus Wi-Fi Gateway) MqttClient::Publish home/ems-esp/thermostat_cmd_temp1 (15.5) `

proddy, 2020-02-27:
support for writing to a Moduline 200/ES72 hasn't been coded yet. We'll need to work together to add it. I'll send over some telegrams for you to try

TaliesinDS, 2020-02-27:
sound good, i'll do what i can :)

proddy, 2020-02-28:
with logging on, what does `thermostat temp 20` do? I think it should work?

If that doesn't work try it manually with `send 0B 17 B0 04 28`

yes for the telegram type 0xB0 the value 27 is hex and the temperature multiplied by 2. So indeed 19.5 degrees. In the example about 28 hex (0x28) is thus 20 degrees

TaliesinDS, 2020-02-28:
`System Logging set to Thermostat only
thermostat temp 20

Setting new thermostat temperature to 20.0 for heating circuit 1
-> Write failed. Compared set value 0x28 with received value of 0xC1
-> Write didn't work, retrying (1/2)...
-> Write failed. Compared set value 0x28 with received value of 0xC1
-> Write didn't work, retrying (2/2)...
-> Write failed. Compared set value 0x28 with received value of 0xC1
-> Write failed. Giving up, removing from queue
(23:16:48.662) Thermostat -> All, RC10StatusMessage(0xB1), telegram: 17 00 B1 00                                                           04 26 00 A9 00 00 00 A9 (#data=8)
(23:16:48.900) Thermostat -> Boiler, UBAFlags(0x35), telegram: 17 08 35 00 11 00                                                           (#data=2)
`

`
send 0B 17 B0 04 28

Unknown command or wrong number of arguments. Use ? for help.
send 0B 17 B0 04 28

(23:19:35.335) Sending raw: 0B 17 B0 04 28
(23:19:45.474) Thermostat -> All, RC10Set(0xB0), telegram: 17 00 B0 00 00 FF 00 20 28 02 01 (#data=7)
(23:19:50.362) Thermostat -> All, RC10StatusMessage(0xB1), telegram: 17 00 B1 00 04 28 00 A9 00 00 00 A9 (#data=8)
(23:19:50.600) Thermostat -> Boiler, UBAFlags(0x35), telegram: 17 08 35 00 11 00 (#data=2)
Fetching data from EMS devices
Publishing MQTT data for: thermostat boiler
(23:20:01.385) Thermostat -> Me, telegram: 17 0B 91 00
(23:20:02.013) Thermostat -> Me, telegram: 17 0B A8 00
(23:20:02.264) Thermostat -> Me, telegram: 17 0B 06 00
`

Sending hex worked !
funny enough i tried that yesterday but got the same unknown command error, 2nd time today it worked.
I guess it's a problem with the terminal and not the code?

proddy, 2020-02-28:
I need to see the telegram. can you do the `thermostat temp 20` again but this time with log enabled.

proddy, 2020-02-28:
Can you try the latest dev build (b46). I have the Moduline 200 incorrectly specified as an RC20 and not an RC10

TaliesinDS, 2020-02-28:
With version 1.9.5b46 it says "EMS Bus Connected but Tx is not working." on the EMSD dashboard.

proddy, 2020-02-28:
ok good. Wanted to see the send command, the last line in:
```
Setting new thermostat temperature to 20.0 for heating circuit 1
(00:02:13.577) Sending write of type 0xB0 to 0x17, telegram: 0B 17 B0 04 28
```

that corrupt telegram is not good. it means the send didn't work. Try a few times using the `thermostat temp` command with different temperatures to see if its working but checking the physical thermostat display.

Also the output of `info` would be helpful to see the # CRC errors and the poll timing

TaliesinDS, 2020-02-28:
sending a command for a different setpoint than what is currently set seems to work.
it also changes on the physical thermostat:

`thermostat temp 18

Setting new thermostat temperature to 19.0 for heating circuit 1
(00:20:58.232) Thermostat -> All, Type(0xA2), telegram: 17 00 A2 00 00 00 00 00 00 00 00 00 00 00 (#data=10)
(00:20:59.727) Sending write of type 0xB0 to 0x17, telegram: 0B 17 B0 04 26
(00:20:59.901) Sending validate of type 0xB0 to 0x17, telegram: 0B 97 B0 04 01
-> Validate confirmed, last Write to 0x17 was successful
Requesting type RC10StatusMessage(0xB1) from dest 0x17`

`[TELNET] Connected to EMS-ESP version 1.9.5b46. Type ? for commands.
info

EMS-ESP system status:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: d                                      isabled
  Shower Timer: disabled, Shower Alert: disabled

EMS Bus status:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=125, # CRC errors=34
  Tx: Last poll=2.760 seconds ago, # successful write requests=2

Boiler data:
  Boiler: Nefit Proline (DeviceID: 0x08, ProductID: 122, Version: 03.09)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: on
  Warm Water comfort setting: Intelligent
  Warm Water selected temperature: 60 C
  Warm Water desinfection temperature: 70 C
  Warm Water Circulation active: off
  Warm Water current temperature: ? C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 131618 times
  Warm Water active time: 39 days 4 hours 25 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 65 C
  Current flow temperature: 61.6 C
  Return temperature: 60.6 C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 57 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: 1.8 bar
  System service code: 0Y (204)
  Heating temperature setting on the boiler: 65 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 50 %
  Boiler temperature: 0.0 C
  Warm water storage temperature1: ? C
  Warm water storage temperature2: ? C
  Exhaust temperature: 49.9 C
  Pump modulation: 50 %
  Burner # starts: 186795 times
  Total burner operating time: 549 days 1 hours 48 minutes
  Total heat operating time: 509 days 21 hours 23 minutes
  Total UBA working time: 3614 days 5 hours 18 minutes

Thermostat data:
  Thermostat: Moduline 200 (DeviceID: 0x17, ProductID: 80, Version: 01.02)
  Thermostat time is ?
  Heating Circuit 1
   Current room temperature: 17.9 C
   Setpoint room temperature: 19.0 C
`

proddy, 2020-02-28:
ok, looks like it works, It's not perfect, you do have some CRC errors (which is normal) but its close. 

BTW for the putty settings see https://emsesp.github.io/docs/#/Running-and-Monitoring?id=recommended-putty-settings

TaliesinDS, 2020-02-28:
Thanks, im guessing local line editing is what's messing with me.

proddy, 2020-02-28:
nice. so if you're good please close this issue and feel free to open new ones if errors pop up.

closed 2020-02-28 by TaliesinDS