emsesp/EMS-ESP #334: Sieger ES72 thermostat not sending version info on request

MrWillibald, 2020-02-26:
In my flat I have a Sieger HG 15 W boiler with a single Sieger ES72 room thermostat in the living room. The boiler is recognized as Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID: 0x08, ProductID: 95, Version: 13.08) with HT3 Controller (DeviceID: 0x09, ProductID: 95, Version: 13.08). I use the EMS+ tx mode on the ems-esp.
I can send version information requests and get back the desired information from the boiler. But the ES72 thermostat does not send anything on request. Here is what I found out about the thermostat so far:
- It looks like a Moduline 200
- The PCB says it is a Buderus Moduline 100/200 V1.3
- The thermostat sends RC20 status messages 0xAE
- After restarting the thermostat it identifies itself as Sieger ES72 (DeviceID: 0x17, ProductID: 113, Version: 02.03) ( thanks @proddy for the implementation)
- The thermostat does not respond to any request sent by ems-esp
As far as I understand a regular Moduline 200 should respond to requests made by ems-esp. So does anybody else have a Sieger ES72 and can confirm this behavior?

[image]

proddy, 2020-02-26:
@MrWillibald one way to see which telegrams are being sent is to watch the thermostat messages (log t) and then change the temperature on the thermostat and capture the logs.

proddy, 2020-02-26:
BTW is the title correct since you stated earlier that EMS-ESP does recognize this as a Moduline 200/ES72

MrWillibald, 2020-02-26:
> BTW is the title correct since you stated earlier that EMS-ESP does recognize this as a Moduline 200/ES72

It recognizes the ES72 but only after powering off and on the thermostat which is not a common mode of operation. Sending anything actively through EMS-ESP to the thermostat is not answered. Also 'devices scan' does not identify the thermostat.

proddy, 2020-02-26:
ok, I've seen this problem before in #323 

Next time it detects the thermostat correctly, do a `devices save` and it'll always remember it, even after reboot. It's a bit of a workaround for some thermostats that do not always broadcast their existance to the boiler/uba master.

MrWillibald, 2020-02-26:
(15:42:12) Thermostat -> Boiler, (0x1A), telegram: 17 08 1A 00 1E 64 64 00 (#data=4)
(15:42:26) Thermostat -> Boiler, (0x35), telegram: 17 08 35 00 11 11 (#data=2)
(15:42:43) Thermostat -> All, (0xAE), telegram: 17 00 AE 00 80 12 2A 00 D0 00 00 64 (#data=8)
Changing temperature at the ES72
(15:42:50) Thermostat -> All, (0xAD), telegram: 17 00 AD 0D 2E (#data=1)
(15:42:50) Thermostat -> Boiler, (0x1A), telegram: 17 08 1A 00 36 64 64 00 (#data=4)
(15:43:26) Thermostat -> Boiler, (0x35), telegram: 17 08 35 00 11 11 (#data=2)
(15:43:43) Thermostat -> All, (0xAE), telegram: 17 00 AE 00 80 12 2E 00 D0 00 00 64 (#data=8)
(15:43:51) Thermostat -> Boiler, (0x1A), telegram: 17 08 1A 00 38 64 64 00 (#data=4)
(15:44:26) Thermostat -> Boiler, (0x35), telegram: 17 08 35 00 11 11 (#data=2)
(15:44:44) Thermostat -> All, (0xAE), telegram: 17 00 AE 00 80 12 2E 00 D0 00 00 64 (#data=8)
(15:44:51) Thermostat -> Boiler, (0x1A), telegram: 17 08 1A 00 39 64 64 00 (#data=4)
(15:45:27) Thermostat -> Boiler, (0x35), telegram: 17 08 35 00 11 11 (#data=2)
(15:45:44) Thermostat -> All, (0xAE), telegram: 17 00 AE 00 80 12 2E 00 D0 00 00 64 (#data=8)
(15:45:51) Thermostat -> Boiler, (0x1A), telegram: 17 08 1A 00 3A 64 64 00 (#data=4)
Changing back temperature
(15:46:01) Thermostat -> Boiler, (0x1A), telegram: 17 08 1A 00 00 00 64 00 (#data=4)
(15:46:02) Thermostat -> All, (0xAD), telegram: 17 00 AD 0D 28 (#data=1)
After a while
(15:48:45) Thermostat -> All, (0xAE), telegram: 17 00 AE 00 80 12 28 00 D0 00 00 00 (#data=8)
(15:49:02) Thermostat -> Boiler, (0x1A), telegram: 17 08 1A 00 00 00 00 00 (#data=4)
(15:49:27) Thermostat -> Boiler, (0x35), telegram: 17 08 35 00 11 11 (#data=2)

 So all the thermostat does is changing the boiler set temperature and the warmwater and boiler power demand (according to https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme).

By studying the [manual of the ES72](http://www.sieger.net/images/Downloads/Sieger_Archivunterlagen/Regelungstechnik/1_Programm_eSTAR/Raumstation_eS_72/be_raumstation_estar_es72.pdf) I found out that there a two modes the ES72 can operate in. One is without a ES73 and one with ES73. When an ES73 is present the ES72 only acts as kind of remote control. In the single ES72 mode (which I run) no heating circuits are configured and the ES72 only does the things I logged.

proddy, 2020-02-26:
We can pull some of the data out of 0xAE and 0xAD but you'll need to help me with providing the actual values.
`0xAD` has one temperature byte, e.g. your `17 00 AD 0D 28` is 20C degrees
`0xAE` has multiple values, e.g. your `17 00 AE 00 80 12 2E 00 D0 00 00 64`  has 18C, 23C and 20.8C

can you try and see if these numbers match what is displayed on the thermostat?

proddy, 2020-02-26:
can you try the latest dev build and see if the setpoint and current temps are correctly displayed?

MrWillibald, 2020-02-26:
Thank you it is displayed correctly.
Do you think it will be possible to modify the current setpoint via ems-esp?

proddy, 2020-02-27:
yes, we need to work together to enable writing. Same as #333

MrWillibald, 2020-02-27:
Ah thanks, I'll be happy to help

MrWillibald, 2020-03-03:
As an update, I can now write to the thermostat after getting the board from bbqkees. It sends the version info now on request.
But trying your instructions from #333 has no success with the ES72.

thermostat temp 1 22
Setting new thermostat temperature to 22.0 for heating circuit 1
(15:55:47) Sending write of type 0xA8 to 0x17, telegram: 0B 17 A8 1C 2C 
(15:55:47) Sending validate of type 0xA8 to 0x17, telegram: 0B 97 A8 1C 01 
-> Write failed. Compared set value 0x2C with received value of 0xC1
-> Write didn't work, retrying (1/2)...
(15:55:47) Boiler -> All, UBADevices(0x07), telegram: 08 00 07 00 0B 80 00 00 00 00 00 00 00 00 00 00 00 00 00 (#data=15)
(15:55:48) Sending write of type 0xA8 to 0x17, telegram: 0B 17 A8 1C 2C 
(15:55:48) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 36 01 9C 3F 00 01 23 A4 C0 01 47 80 00 80 00 FF FF FF 30 59 00 CC 00 00 00 (#data=25)
(15:55:48) Boiler -> 0x13, Type(0x05), telegram: 08 13 05 22 00 (#data=1)
(15:55:48) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 08 00 34 00 31 01 47 80 00 81 00 04 01 00 00 64 34 00 4C 2E 00 (#data=17)
(15:55:49) Sending validate of type 0xA8 to 0x17, telegram: 0B 97 A8 1C 01 
-> Write failed. Compared set value 0x2C with received value of 0xC1
-> Write didn't work, retrying (2/2)...
(15:55:49) Sending write of type 0xA8 to 0x17, telegram: 0B 17 A8 1C 2C 
(15:55:49) Sending validate of type 0xA8 to 0x17, telegram: 0B 97 A8 1C 01 
-> Write failed. Compared set value 0x2C with received value of 0xC1
-> Write failed. Giving up, removing from queue

proddy, 2020-03-03:
forgetting the errors in the log for a minute, did it  actually work and change the temperature on your thermostat?

MrWillibald, 2020-03-04:
The temperature on the thermostat is not changed.

pitulek, 2020-04-08:
try this

0B 17 AD 03 --- 00 NIGHT , 01 - DAY MODE, 02 AUTO MODE
0B 17 AD 0B ---- 07 CHANGE SET PROGRAM FOR AUTO IN THIS CASE NO 7 01-09
0B 17 AD 01 25 -- CHANGE TEMP NIGHT 19,5C
0B 17 AD 02 2A -- CHANGE TEMP DAY 21C

MrWillibald, 2020-04-09:
@pitulek wow thank you very much for your effort in #361 that actually works. I see you spend more time investigating the messages for the RC20. I could never find out what the AD message is.

closed 2020-04-09 by MrWillibald

proddy, 2020-04-14:
thanks @pitulek for the info. I've added this also to EMS-ESP 2.0 to fully support the ES72