emsesp/EMS-ESP #335: FR100 Thermostat

hansstulp, 2020-02-27:
Hello, i can't write data to my Bosch/Junkers FR100 Thermostat.

proddy, 2020-02-27:
Write is currently only support for the Junkers Thermostats FW100, FR10 an FW120 mainly because a user helped test it. When you try and write with EMS-ESP to your FR100 it will show you in the console that "write is not supported". 

I'll be happy to add support for this enhancement but you'll also need to do a bit of work. First of all make sure you're comfortable with the telnet console commands and logging.

hansstulp, 2020-02-27:
With a little help it will be possible.

hansstulp, 2020-02-27:
These device IDs are on the EMS Bus: 0x08 0x09 0x10
and 3 were recognized by EMS-ESP as:
 Nefit Proline (DeviceID:0x08 ProductID:122 Version:03.09)
 BC10 Base Controller (DeviceID:0x09 ProductID:114 Version:01.07)
 Junkers FR100 (DeviceID:0x10 ProductID:107 Version:15.07)

EMS-ESP system stats:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: disabled, Shower Alert: disabled

EMS Bus stats:
  Bus is connected, protocol: HT3
  Rx: # successful read requests=27, # CRC errors=0
  Tx: Last poll=2.398 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: Nefit Proline (DeviceID:0x08 ProductID:122 Version:03.09)
  Hot tap water: off
  Central heating: active
  Warm Water activated: on
  Warm Water circulation pump available: on
  Warm Water comfort setting: Eco
  Warm Water selected temperature: 60 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: ? C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 29427 times
  Warm Water active time: 7 days 19 hours 2 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 65 C
  Current flow temperature: 49.3 C
  Return temperature: 41.7 C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 75 %
  Burner current power: 23 %
  Flame current: 6.3 uA
  System pressure: 1.9 bar
  System service code:  (200)
  Heating temperature setting on the boiler: 70 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 50 %
  Boiler temperature: 0.0 C
  Pump modulation: 50 %
  Burner # starts: 142341 times
  Total burner operating time: 726 days 12 hours 25 minutes
  Total heat operating time: 718 days 17 hours 23 minutes
  Total UBA working time: 3645 days 1 hours 45 minutes

Thermostat stats:
  Thermostat: Junkers FR100 (DeviceID:0x10 ProductID:107 Version:15.07)
  Thermostat time is 22:22:00 27/02/2020
  Heating Circuit 1
   Current room temperature: 19.9 C
   Setpoint room temperature: 20.0 C
   Mode is set to auto
   Day Mode is set to day

hansstulp, 2020-02-27:
System Logging set to Thermostat only
Publishing boiler data via MQTT
loPublishing boiler data via MQTT
gPublishing boiler data via MQTT
 vFetching data from EMS devices
(22:28:51) Thermostat -> Me, type 0x06, telegram: 90 0B 06 00 14 02 16 1B 19 3B 04 00 (CRC=53) #data=8
Publishing boiler data via MQTT
Publishing boiler data via MQTT
Publishing boiler data via MQTT
Publishing boiler data via MQTT
** Warning, we missed the bus - Rx non-idle!
Publishing boiler data via MQTT
Publishing boiler data via MQTT
Publishing boiler data via MQTT
Publishing boiler data via MQTT
(22:29:00) Thermostat -> All, type 0x06, telegram: 90 00 06 00 14 02 16 1B 1A 1D 04 00 (CRC=74) #data=8
(22:29:01) Thermostat -> UBAMaster, type 0x16, telegram: 90 88 16 00 02 (CRC=AF) #data=1
(22:29:01) UBAMaster -> Thermostat, type 0x16, telegram: 88 10 16 00 FF 46 (CRC=98) #data=2
(22:29:02) Thermostat -> All, type 0x006F, telegram: 90 00 FF 00 00 6F 03 02 00 C8 00 C6 (CRC=0C) #data=6
Publishing thermostat data via MQTT
Publishing boiler data via MQTT
Publishing boiler data via MQTT
(22:29:12) Thermostat -> UBAMaster, type 0x1A, telegram: 90 08 1A 00 3F (CRC=C6) #data=1
(22:29:12) Thermostat -> UBAMaster, type 0x23, telegram: 90 08 23 00 3F (CRC=22) #data=1
Publishing boiler data via MQTT
Publishing boiler data via MQTT
Publishing boiler data via MQTT
Publishing boiler data via MQTT
Publishing boiler data via MQTT
Publishing boiler data via MQTT
(22:29:35) Thermostat -> UBAMaster, type 0x02, telegram: 90 88 02 00 0A (CRC=F7) #data=1
(22:29:35) UBAMaster -> Thermostat, type 0x02, telegram: 88 10 02 00 7A 03 09 00 00 00 00 00 00 01 (CRC=66) #data=10
(22:29:37) Thermostat -> UBAMaster, type 0x1A, telegram: 90 08 1A 00 40 (CRC=B9) #data=1
(22:29:37) Thermostat -> UBAMaster, type 0x23, telegram: 90 08 23 00 40 (CRC=5D) #data=1
Publishing boiler data via MQTT
Publishing boiler data via MQTT
(22:29:45) Thermostat -> UBAMaster, type 0x1A, telegram: 90 08 1A 00 41 (CRC=B8) #data=1
(22:29:45) Thermostat -> UBAMaster, type 0x23, telegram: 90 08 23 00 41 (CRC=5C) #data=1
Publishing boiler data via MQTT
Publishing boiler data via MQTT
Fetching data from EMS devices
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Publishing thermostat data via MQTT
(22:29:50) Thermostat -> UBAMaster, type 0x35, telegram: 90 08 35 00 11 (CRC=54) #data=1
(22:29:50) Thermostat -> UBAMaster, type 0x1A, telegram: 90 08 1A 00 41 (CRC=B8) #data=1
(22:29:50) Thermostat -> UBAMaster, type 0x1A, telegram: 90 08 1A 02 00 00 03 (CRC=DC) #data=3
(22:29:50) Thermostat -> UBAMaster, type 0x23, telegram: 90 08 23 00 41 64 00 (CRC=A1) #data=3
(22:29:50) Thermostat -> UBAMaster, type 0x1A, telegram: 90 08 1A 04 03 (CRC=F2) #data=1
(22:29:51) Thermostat -> Me, type 0x06, telegram: 90 0B 06 00 14 02 16 1B 1A 3B 04 00 (CRC=4B) #data=8
Publishing boiler data via MQTT
Publishing boiler data via MQTT
Publishing boiler data via MQTT
Publishing boiler data via MQTT
** Warning, we missed the bus - Rx non-idle!
Publishing boiler data via MQTT
(22:29:53) Thermostat -> UBAMaster, type 0x1A, telegram: 90 08 1A 00 42 (CRC=BB) #data=1
(22:29:53) Thermostat -> UBAMaster, type 0x23, telegram: 90 08 23 00 42 (CRC=5F) #data=1
Publishing boiler data via MQTT
Publishing boiler data via MQTT
Publishing boiler data via MQTT
(22:30:00) Thermostat -> All, type 0x06, telegram: 90 00 06 00 14 02 16 1B 1B 1C 04 00 (CRC=78) #data=8
(22:30:02) Thermostat -> UBAMaster, type 0x1A, telegram: 90 08 1A 00 43 (CRC=BA) #data=1
(22:30:02) Thermostat -> UBAMaster, type 0x23, telegram: 90 08 23 00 43 (CRC=5E) #data=1
(22:30:02) Thermostat -> UBAMaster, type 0x1A, telegram: 90 08 1A 00 42 (CRC=BB) #data=1
(22:30:02) Thermostat -> UBAMaster, type 0x23, telegram: 90 08 23 00 42 (CRC=5F) #data=1
(22:30:02) Thermostat -> UBAMaster, type 0x16, telegram: 90 88 16 00 02 (CRC=AF) #data=1
(22:30:02) UBAMaster -> Thermostat, type 0x16, telegram: 88 10 16 00 FF 46 (CRC=98) #data=2
(22:30:02) Thermostat -> All, type 0x006F, telegram: 90 00 FF 00 00 6F 03 02 00 C8 00 C6 (CRC=0C) #data=6
Publishing thermostat data via MQTT
Publishing boiler data via MQTT
Publishing boiler data via MQTT
(22:30:10) Thermostat -> UBAMaster, type 0x1A, telegram: 90 08 1A 00 44 (CRC=BD) #data=1
(22:30:10) Thermostat -> UBAMaster, type 0x23, telegram: 90 08 23 00 44 (CRC=59) #data=1
Publishing boiler data via MQTT
Publishing boiler data via MQTT
Publishing boiler data via MQTT
(22:30:18) Thermostat -> UBAMaster, type 0x1A, telegram: 90 08 1A 00 45 (CRC=BC) #data=1
(22:30:18) Thermostat -> UBAMaster, type 0x23, telegram: 90 08 23 00 45 (CRC=58) #data=1
(22:30:26) Thermostat -> UBAMaster, type 0x1A, telegram: 90 08 1A 00 46 (CRC=BF) #data=1
(22:30:26) Thermostat -> UBAMaster, type 0x23, telegram: 90 08 23 00 46 (CRC=5B) #data=1
Publishing boiler data via MQTT
Publishing boiler data via MQTT

proddy, 2020-02-28:
try this
- upgrade to the latest 1.9.5dev version
- go to telnet, turn on logging (`log v`)
- `thermotstat temp 20`
- send back what you see in the logs. We're looking for what telegrams are sent and what the response is. It should work.

hansstulp, 2020-02-28:
Write not supporters yet for this type of thermostat.

proddy, 2020-02-28:
oh, forgot about that. fixed in b46. please try again

hansstulp, 2020-02-28:
System Logging set to Verbose
(14:35:42) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 41 02 8C 4B 22 09 01 25 40 80 00 80 00 02 1C 00 4C 13 00 00 00 C8 00 02 00 (#data=25)
(14:35:42) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B DD 00 72 F4 00 (#data=17)
thermostat temp 21

Setting new thermostat temperature to 21.0 for heating circuit 1
(14:35:46) Sending write of type 0x65 to 0x10, telegram: 8B 10 65 00 2A
(14:35:47) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 40 (#data=1)
(14:35:47) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 40 (#data=1)
(14:35:49) Sending validate of type 0x65 to 0x10, telegram: 8B 90 65 00 01
-> Write failed. Compared set value 0x2A with received value of 0x02
-> Write didn't work, retrying (1/2)...
(14:35:49) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(14:35:49) Sending write of type 0x65 to 0x10, telegram: 8B 10 65 00 2A
(14:35:50) Thermostat -> Boiler, UBAFlags(0x35), telegram: 90 08 35 00 11 (#data=1)
(14:35:50) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 40 (#data=1)
(14:35:50) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 02 00 00 03 (#data=3)
(14:35:50) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 40 64 00 (#data=3)
(14:35:51) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 04 03 (#data=1)
(14:35:51) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(14:35:52) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 40 02 8C 4B 1F 09 01 25 40 80 00 80 00 02 1E 00 4A 13 00 00 00 C8 00 02 00 (#data=25)
(14:35:52) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B DD 00 72 F4 00 (#data=17)
(14:35:57) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 40 02 89 4B 1D 09 01 25 40 80 00 80 00 02 1F 00 47 13 00 00 00 C8 00 02 00 (#data=25)
(14:36:02) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B DD 00 72 F4 00 (#data=17)
(14:36:02) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 40 02 83 4B 1C 09 01 25 40 80 00 80 00 02 20 00 45 13 00 00 00 C8 00 02 00 (#data=25)
(14:36:02) Thermostat -> Boiler, UBAParametersMessage(0x16), telegram: 90 88 16 00 02 (#data=1)
(14:36:02) Boiler -> Thermostat, UBAParametersMessage(0x16), telegram: 88 10 16 00 FF 46 (#data=2)
(14:36:03) Thermostat -> All, JunkersStatusMessage_HC1(0x006F), telegram: 90 00 FF 00 00 6F 03 02 00 C8 00 C6 (#data=6)
Fetching data from EMS devices
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
Publishing MQTT data for: thermostat boiler
(14:36:04) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 41 (#data=1)
(14:36:04) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 41 (#data=1)
(14:36:05) Sending read of type 0x06 to 0x10, telegram: 8B 90 06 00 20
(14:36:06) Thermostat -> Me, RCTime(0x06), telegram: 90 0B 06 00 14 02 0E 1C 21 08 05 00 (#data=8)
(14:36:06) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(14:36:06) Sending read of type 0x18 to 0x08, telegram: 8B 88 18 00 20
(14:36:06) Boiler -> Me, UBAMonitorFast(0x18), telegram: 88 0B 18 00 41 02 7F 4B 1C 09 01 25 40 80 00 80 00 02 20 00 45 13 00 00 00 C8 00 02 00 (#data=25)
(14:36:06) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 41 02 7F 4B 1C 09 01 25 40 80 00 80 00 02 20 00 45 13 00 00 00 C8 00 02 00 (#data=25)
(14:36:06) Sending read of type 0x19 to 0x08, telegram: 8B 88 19 00 20
(14:36:06) Boiler -> Me, UBAMonitorSlow(0x19), telegram: 88 0B 19 00 80 00 00 00 02 16 00 00 00 32 02 2C 32 0F F8 87 00 00 00 0F CC AA 01 B9 3E 00 00 (#data=27)
(14:36:07) Sending read of type 0x33 to 0x08, telegram: 8B 88 33 00 20
(14:36:07) Boiler -> Me, UBAParameterWW(0x33), telegram: 88 0B 33 00 08 FF 3C FB 00 28 FF 02 46 D8 FF (#data=11)
(14:36:07) Sending read of type 0x16 to 0x08, telegram: 8B 88 16 00 20
(14:36:07) Boiler -> Me, UBAParametersMessage(0x16), telegram: 88 0B 16 00 FF 46 4B 00 06 FA 0A 01 01 64 32 00 (#data=12)
(14:36:07) Sending read of type 0x14 to 0x08, telegram: 8B 88 14 00 20
(14:36:08) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(14:36:12) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 40 (#data=1)
(14:36:12) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 40 (#data=1)
(14:36:12) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 41 02 7D 4B 1C 09 01 25 40 80 00 80 00 02 22 00 45 13 00 00 00 C8 00 02 00 (#data=25)
(14:36:12) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B DD 00 72 F4 00 (#data=17)
(14:36:14) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 3F (#data=1)
(14:36:14) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 3F (#data=1)

proddy, 2020-02-28:
just ignoring the errors in the log for a minute, did it work? is the thermostat at 21 degrees?

proddy, 2020-02-28:
also set the thermostat to a temp, capture the logs, set it to another temp and capture the logs again. we need to find which telegrams have changed.

hansstulp, 2020-02-28:
the thermostat is not changed. I will try it again.

hansstulp, 2020-02-28:
log

Unknown command or wrong number of arguments. Use ? for help.
log t

System Logging set to Thermostat only
(15:50:40) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 35 (#data=1)
(15:50:40) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 35 (#data=1)
(15:50:48) Thermostat -> Boiler, UBAFlags(0x35), telegram: 90 08 35 00 11 (#data=1)
(15:50:48) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 35 (#data=1)
(15:50:48) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 02 00 00 03 (#data=3)
(15:50:48) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 35 64 00 (#data=3)
(15:50:48) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 04 03 (#data=1)
(15:50:53) Thermostat -> All, RCTime(0x06), telegram: 90 00 06 00 14 02 0F 1C 30 17 05 00 (#data=8)
(15:50:56) Thermostat -> Boiler, Version(0x02), telegram: 90 88 02 00 0A (#data=1)
(15:50:56) Boiler -> Thermostat, Version(0x02), telegram: 88 10 02 00 7A 03 09 00 00 00 00 00 00 01 (#data=10)
(15:50:59) Thermostat -> Boiler, UBAParametersMessage(0x16), telegram: 90 88 16 00 02 (#data=1)
(15:51:00) Boiler -> Thermostat, UBAParametersMessage(0x16), telegram: 88 10 16 00 FF 46 (#data=2)
(15:51:00) Thermostat -> All, JunkersStatusMessage_HC1(0x006F), telegram: 90 00 FF 00 00 6F 03 02 00 C8 00 C8 (#data=6)
(15:51:04) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 36 (#data=1)
(15:51:04) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 36 (#data=1)
Fetching data from EMS devices
(15:51:20) Thermostat -> Me, RCTime(0x06), telegram: 90 0B 06 00 14 02 0F 1C 30 17 05 00 (#data=8)
(15:51:26) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 37 (#data=1)
(15:51:27) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 37 (#data=1)
(15:51:37) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 38 (#data=1)
(15:51:37) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 38 (#data=1)
(15:51:45) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 39 (#data=1)
(15:51:45) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 39 (#data=1)
Thermostat temp 19

Unknown command or wrong number of arguments. Use ? for help.
(15:51:53) Thermostat -> All, RCTime(0x06), telegram: 90 00 06 00 14 02 0F 1C 31 17 05 00 (#data=8)
(15:51:57) Thermostat -> Boiler, Version(0x02), telegram: 90 88 02 00 0A (#data=1)
(15:51:57) Boiler -> Thermostat, Version(0x02), telegram: 88 10 02 00 7A 03 09 00 00 00 00 00 00 01 (#data=10)
(15:51:57) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 3A (#data=1)
(15:51:58) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 3A (#data=1)
(15:52:00) Thermostat -> Boiler, UBAParametersMessage(0x16), telegram: 90 88 16 00 02 (#data=1)
(15:52:00) Boiler -> Thermostat, UBAParametersMessage(0x16), telegram: 88 10 16 00 FF 46 (#data=2)
(15:52:00) Thermostat -> All, JunkersStatusMessage_HC1(0x006F), telegram: 90 00 FF 00 00 6F 03 02 00 C8 00 C8 (#data=6)
(15:52:02) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 37 (#data=1)
(15:52:02) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 37 (#data=1)
(15:52:08) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 38 (#data=1)
(15:52:08) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 38 (#data=1)
(15:52:10) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 37 (#data=1)
(15:52:10) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 37 (#data=1)
thermostat temp 19

Setting new thermostat temperature to 19.0 for heating circuit 1
-> Write failed. Compared set value 0x26 with received value of 0x02
-> Write didn't work, retrying (1/2)...
-> Write failed. Compared set value 0x26 with received value of 0x02
-> Write didn't work, retrying (2/2)...
-> Write failed. Compared set value 0x26 with received value of 0x02
-> Write failed. Giving up, removing from queue
Fetching data from EMS devices
Publishing MQTT data for: thermostat boiler
(15:52:21) Thermostat -> Me, RCTime(0x06), telegram: 90 0B 06 00 14 02 0F 1C 31 17 05 00 (#data=8)
thermostat temp 21

Setting new thermostat temperature to 21.0 for heating circuit 1
(15:52:49) Thermostat -> Boiler, UBAFlags(0x35), telegram: 90 08 35 00 11 (#data=1)
(15:52:49) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 37 (#data=1)
(15:52:49) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 02 00 00 03 (#data=3)
(15:52:49) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 37 64 00 (#data=3)
(15:52:49) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 04 03 (#data=1)
(15:52:53) Thermostat -> All, RCTime(0x06), telegram: 90 00 06 00 14 02 0F 1C 32 16 05 00 (#data=8)
(15:52:58) Thermostat -> Boiler, Version(0x02), telegram: 90 88 02 00 0A (#data=1)
(15:52:58) Boiler -> Thermostat, Version(0x02), telegram: 88 10 02 00 7A 03 09 00 00 00 00 00 00 01 (#data=10)
(15:52:59) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 38 (#data=1)
(15:52:59) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 38 (#data=1)
(15:53:01) Thermostat -> Boiler, UBAParametersMessage(0x16), telegram: 90 88 16 00 02 (#data=1)
(15:53:01) Boiler -> Thermostat, UBAParametersMessage(0x16), telegram: 88 10 16 00 FF 46 (#data=2)
(15:53:01) Thermostat -> All, JunkersStatusMessage_HC1(0x006F), telegram: 90 00 FF 00 00 6F 03 02 00 C8 00 C8 (#data=6)
(15:53:06) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 37 (#data=1)
(15:53:06) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 37 (#data=1)
(15:53:07) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 36 (#data=1)
(15:53:07) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 36 (#data=1)
Fetching data from EMS devices
(15:53:22) Thermostat -> Me, RCTime(0x06), telegram: 90 0B 06 00 14 02 0F 1C 32 16 05 00 (#data=8)

proddy, 2020-02-28:
sorry, it needs to be verbose logging `log v` !

hansstulp, 2020-02-28:
[TELNET] exiting telnet session
[TELNET] Connected to EMS-ESP version 1.9.5b46. Type ? for commands.
log v

System Logging set to Verbose
(16:16:12) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B DD 00 72 F4 00 (#data=17)
(16:16:13) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 40 01 BB 4B 17 09 01 25 40 80 00 80 00 01 6A 00 3E 12 00 00 00 C8 00 02 00 (#data=25)
thermostat temp 19

Setting new thermostat temperature to 19.0 for heating circuit 1
(16:16:18) Sending write of type 0x65 to 0x10, telegram: 8B 10 65 00 26
(16:16:18) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(16:16:18) Sending validate of type 0x65 to 0x10, telegram: 8B 90 65 00 01
(16:16:18) Thermostat -> Boiler, UBAParametersMessage(0x16), telegram: 90 88 16 00 02 (#data=1)
(16:16:18) Boiler -> Thermostat, UBAParametersMessage(0x16), telegram: 88 10 16 00 FF 46 (#data=2)
(16:16:19) Thermostat -> All, JunkersStatusMessage_HC1(0x006F), telegram: 90 00 FF 00 00 6F 03 02 00 C8 00 C7 (#data=6)
(16:16:19) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
Fetching data from EMS devices
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
Publishing MQTT data for: thermostat boiler
(16:16:21) Sending read of type 0x06 to 0x10, telegram: 8B 90 06 00 20
(16:16:21) Thermostat -> Me, RCTime(0x06), telegram: 90 0B 06 00 14 02 10 1C 0D 30 05 00 (#data=8)
(16:16:21) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(16:16:21) Sending read of type 0x18 to 0x08, telegram: 8B 88 18 00 20
(16:16:21) Boiler -> Me, UBAMonitorFast(0x18), telegram: 88 0B 18 00 40 01 BA 4B 17 09 01 25 40 80 00 80 00 01 68 00 3E 12 00 00 00 C8 00 02 00 (#data=25)
(16:16:22) Thermostat -> Boiler, Version(0x02), telegram: 90 88 02 00 0A (#data=1)
(16:16:22) Boiler -> Thermostat, Version(0x02), telegram: 88 10 02 00 7A 03 09 00 00 00 00 00 00 01 (#data=10)
(16:16:22) Sending read of type 0x19 to 0x08, telegram: 8B 88 19 00 20
(16:16:22) Boiler -> Me, UBAMonitorSlow(0x19), telegram: 88 0B 19 00 80 00 00 00 01 A1 00 00 00 32 02 2C 39 0F F8 BD 00 00 00 0F CC E0 01 B9 45 00 00 (#data=27)
(16:16:22) Sending read of type 0x33 to 0x08, telegram: 8B 88 33 00 20
(16:16:22) Boiler -> Me, UBAParameterWW(0x33), telegram: 88 0B 33 00 08 FF 3C FB 00 28 FF 02 46 D8 FF (#data=11)
(16:16:22) Sending read of type 0x16 to 0x08, telegram: 8B 88 16 00 20
(16:16:22) Boiler -> Me, UBAParametersMessage(0x16), telegram: 88 0B 16 00 FF 46 4B 00 06 FA 0A 01 01 64 32 00 (#data=12)
(16:16:23) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 40 01 BA 4B 17 09 01 25 40 80 00 80 00 01 68 00 3E 12 00 00 00 C8 00 02 00 (#data=25)
(16:16:23) Boiler -> All, UBAMonitorSlow(0x19), telegram: 88 00 19 00 80 00 00 00 01 A1 00 00 00 32 02 2C 39 0F F8 BD 00 00 00 0F CC E0 01 B9 45 00 00 (#data=27)
(16:16:23) Sending read of type 0x14 to 0x08, telegram: 8B 88 14 00 20
(16:16:23) Boiler -> Me, UBATotalUptimeMessage(0x14), telegram: 88 0B 14 00 50 1B BE (#data=3)
(16:16:23) Boiler -> All, UBAMaintenanceStatusMessage(0x1C), telegram: 88 00 1C 00 00 3B C7 48 00 00 00 00 00 00 00 (#data=11)
(16:16:23) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B DD 00 72 F4 00 (#data=17)
(16:16:23) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(16:16:24) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 3F (#data=1)
(16:16:24) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 3F (#data=1)
(16:16:24) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(16:16:33) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 3F 01 BA 4B 17 09 01 25 40 80 00 80 00 01 67 00 3E 12 00 00 00 C8 00 02 00 (#data=25)
(16:16:33) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B DD 00 72 F4 00 (#data=17)
(16:16:43) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 3F 01 B9 4B 17 09 01 25 40 80 00 80 00 01 6A 00 3E 12 00 00 00 C8 00 02 00 (#data=25)
(16:16:43) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B DD 00 72 F4 00 (#data=17)
(16:16:44) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 3E (#data=1)
(16:16:44) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 3E (#data=1)
(16:16:48) Thermostat -> All, RCTime(0x06), telegram: 90 00 06 00 14 02 10 1C 0E 12 05 00 (#data=8)
thermostat temp 21

Setting new thermostat temperature to 21.0 for heating circuit 1
(16:16:52) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 3E 01 B9 4B 17 09 01 25 40 80 00 80 00 01 75 00 3E 12 00 00 00 C8 00 02 00 (#data=25)
(16:16:53) Sending write of type 0x65 to 0x10, telegram: 8B 10 65 00 2A
(16:16:53) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B DD 00 72 F4 00 (#data=17)
[... cut ...]
(16:17:34) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 3C 01 DF 4B 17 09 01 25 40 80 00 80 00 01 90 00 3E 12 00 00 00 C8 00 02 00 (#data=25)
(16:17:36) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 3C 01 E0 4B 17 09 01 25 40 80 00 80 00 01 90 00 3E 13 00 00 00 C8 00 02 00 (#data=25)
(16:17:38) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 3C 01 DF 4B 17 09 01 25 40 80 00 80 00 01 8F 00 3E 12 00 00 00 C8 00 02 00 (#data=25)
(16:17:42) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 3C 01 DF 4B 17 09 01 25 40 80 00 80 00 01 8F 00 3E 13 00 00 00 C8 00 02 00 (#data=25)

proddy, 2020-02-28:
I see the problem. The thermostat is not responding to the 0x65 telegram type like the other Junkers boilers. What does `thermostat read 65` and `thermostat read 6F` come back with? We'll need to do a bit of ping-pong trial and error here.

hansstulp, 2020-02-28:
That's not a problem. I'm a ping-ponger. 

System Logging set to None
log v

System Logging set to Verbose
(19:13:20) Thermostat -> Boiler, UBAFlags(0x35), telegram: 90 08 35 00 11 (#data=1)
Fetching data from EMS devices
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(19:13:20) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 39 (#data=1)
(19:13:20) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 02 00 00 03 (#data=3)
(19:13:20) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 39 64 00 (#data=3)
(19:13:20) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 04 03 (#data=1)
(19:13:21) Sending read of type 0x06 to 0x10, telegram: 8B 90 06 00 20
(19:13:21) Thermostat -> Me, RCTime(0x06), telegram: 90 0B 06 00 14 02 13 1C 0A 2A 05 00 (#data=8)
(19:13:21) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(19:13:21) Sending read of type 0x18 to 0x08, telegram: 8B 88 18 00 20
(19:13:21) Boiler -> Me, UBAMonitorFast(0x18), telegram: 88 0B 18 00 39 02 0B 4B 00 01 01 20 40 80 00 80 00 02 0C 00 00 13 00 00 00 CC 00 00 00 (#data=25)
(19:13:21) Sending read of type 0x19 to 0x08, telegram: 8B 88 19 00 20
(19:13:21) Boiler -> Me, UBAMonitorSlow(0x19), telegram: 88 0B 19 00 80 00 00 00 01 D6 00 00 00 32 02 2C 43 0F F9 30 00 00 00 0F CD 52 01 B9 4E 00 00 (#data=27)
(19:13:21) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 39 02 0C 4B 00 01 01 20 40 80 00 80 00 02 0B 00 00 13 00 00 00 CC 00 00 00 (#data=25)
(19:13:22) Boiler -> All, UBAMonitorSlow(0x19), telegram: 88 00 19 00 80 00 00 00 01 D6 00 00 00 32 02 2C 43 0F F9 30 00 00 00 0F CD 52 01 B9 4E 00 00 (#data=27)
(19:13:22) Boiler -> All, UBAMaintenanceStatusMessage(0x1C), telegram: 88 00 1C 00 00 3B C7 48 00 00 00 00 00 00 00 (#data=11)
(19:13:22) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B DE 00 72 F5 00 (#data=17)
(19:13:22) Sending read of type 0x33 to 0x08, telegram: 8B 88 33 00 20
(19:13:23) Boiler -> Me, UBAParameterWW(0x33), telegram: 88 0B 33 00 08 FF 3C FB 00 28 FF 02 46 D8 FF (#data=11)
(19:13:23) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(19:13:23) Sending read of type 0x16 to 0x08, telegram: 8B 88 16 00 20
(19:13:23) Boiler -> Me, UBAParametersMessage(0x16), telegram: 88 0B 16 00 FF 46 4B 00 06 FA 0A 01 01 64 32 00 (#data=12)
(19:13:23) Sending read of type 0x14 to 0x08, telegram: 8B 88 14 00 20
(19:13:24) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
thermostat read 65

Requesting type (0x65) from dest 0x10
(19:13:27) Sending read of type 0x65 to 0x10, telegram: 8B 90 65 00 20
(19:13:27) Thermostat -> Me, telegram: 90 0B 65 00
(19:13:27) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(19:13:28) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(19:13:31) Thermostat -> Boiler, UBAParametersMessage(0x16), telegram: 90 88 16 00 02 (#data=1)
(19:13:31) Boiler -> Thermostat, UBAParametersMessage(0x16), telegram: 88 10 16 00 FF 46 (#data=2)
(19:13:32) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 39 02 0A 4B 00 01 01 20 40 80 00 80 00 02 0A 00 00 13 00 00 00 CC 00 00 00 (#data=25)
(19:13:32) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B DE 00 72 F5 00 (#data=17)
(19:13:32) Thermostat -> All, JunkersStatusMessage_HC1(0x006F), telegram: 90 00 FF 00 00 6F 03 02 00 C8 00 C7 (#data=6)
(19:13:42) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 39 02 09 4B 00 01 01 20 40 80 00 80 00 02 0A 00 00 13 00 00 00 CC 00 00 00 (#data=25)
(19:13:42) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B DE 00 72 F5 00 (#data=17)
(19:13:42) Thermostat -> All, RCTime(0x06), telegram: 90 00 06 00 14 02 13 1C 0B 0C 05 00 (#data=8)
thermostat read 6F

Requesting type JunkersStatusMessage_HC1(0x6F) from dest 0x10
(19:13:48) Sending read of type 0x6F to 0x10, telegram: 8B 90 6F 00 20
(19:13:48) Thermostat -> Me, telegram: 90 0B 6F 00
(19:13:48) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(19:13:49) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(19:13:49) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 38 (#data=1)
(19:13:49) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 38 (#data=1)
(19:13:51) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 38 02 08 4B 00 01 01 20 40 80 00 80 00 02 0A 00 00 13 00 00 00 CC 00 00 00 (#data=25)
(19:13:52) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B DE 00 72 F5 00 (#data=17)
(19:13:52) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 38 02 08 4B 00 01 01 20 40 80 00 80 00 02 09 00 00 13 00 00 00 CC 00 00 00 (#data=25)
[... cut ...]
(19:14:12) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B DE 00 72 F5 00 (#data=17)
(19:14:12) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 37 02 05 4B 00 01 01 20 40 80 00 80 00 02 06 00 00 13 00 00 00 CC 00 00 00 (#data=25)
(19:14:14) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 36 (#data=1)
(19:14:14) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 36 (#data=1)
(19:14:15) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 37 (#data=1)
(19:14:15) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 37 (#data=1)

norberts1, 2020-02-29:
Hi, let me help you a bit for that junker-devices as good as I can do.
For setting up temperatur and working-mode try following messages:

I. Setup temperatures with Message-ID:357 (for Heating-Circuit1)
1.Setup temperatur for temperatur-niveau:'Normal'
 required message for heating-circuit 1 (HC1) and 'Normal':
  8x 10 FF 11 00 65 <desired-temperatur-value * 2> CRC Break
  example for heating-circuit1 and requested temperatur of: 22.5 degree and service-key
  **8B 10 FF 11 00 65 2D** CRC Break
2.Setup temperatur for temperatur-niveau:'Sparen'
 required message for heating-circuit 1 (HC1) and 'Sparen':
  8x 10 FF 10 00 65 <desired-temperatur-value * 2> CRC Break
  example for heating-circuit1 and requested temperatur of: 21.5 degree and service-key
  **8B 10 FF 10 00 65 2B** CRC Break
3.Setup temperatur for temperatur-niveau:'Frost'
 required message for heating-circuit 1 (HC1) and 'Frost':
  8x 10 FF 0F 00 65 <desired-temperatur-value * 2> CRC Break
  example for heating-circuit1 and requested temperatur of: 20.5 degree and service-key
  **8B 10 FF 0F 00 65 29** CRC Break
   **x := any Sender-device-ID like: Service-key:=B or modem:=0D**

Requesting that Messages is done with:
  8B 90 FF 11 00 65 CRC Break
---------------------------------------
II. Setup working status/mode with Message-ID:377 (for Heating-Circuit1)
1.Setup Mode for for heatinc-circuit1
 required message:
  8x 10 FF 04 00 79 <desired-mode-value> CRC Break
  example for heating-circuit1 and mode:'auto'
  **8B 10 FF 04 00 79 04** CRC Break

desired-mode-value are (german definitions, translate if required):
 0 := undefined
 1 := Frost
 2 := Sparen
 3 := Normal
 4 := Auto
Use also my message-description on github:
 https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT3-Bus_Telegramme.pdf
Extracted telegrams for above messages attached here:
[image]
[image]
It hopefully helps a bit to get it working!

proddy, 2020-02-29:
thanks @norberts1 ! This is really helpful in so many ways. I think I'll just cut & paste that into my wiki ;)

I now fully understand why my code didn't work for Junkers. I was sending HT3 telegrams as EMS 1.0. For example, to change the temperature the telegram would be like `B8 10 65 11 2A` when it should have been `B8 10 FF 11 00 65 2A`.

Luckily its only 1 line I need to change but before I do that can you @hansstulp confirm this works:

`send B8 10 FF 11 00 65 2A` (change HC1 to 21 degrees C)

hansstulp, 2020-02-29:
send B8 10 FF 11 00 65 2A

Unknown command or wrong number of arguments. Use ? for help.
send B8 10 FF 11 00 65 2A

info

EMS-ESP system status:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: disabled, Shower Alert: disabled

EMS Bus status:
  Bus is connected, protocol: HT3
  Rx: # successful read requests=71, # CRC errors=0
  Saved known device IDs: 08 09 10
  Tx: Last poll=2.398 seconds ago, # successful write requests=0

Boiler data:
  Boiler: Nefit Proline (DeviceID: 0x08, ProductID: 122, Version: 03.09)
  Hot tap water: off
  Central heating: active
  Warm Water activated: on
  Warm Water circulation pump available: on
  Warm Water comfort setting: Eco
  Warm Water selected temperature: 60 C
  Warm Water desinfection temperature: 70 C
  Warm Water Circulation active: off
  Warm Water current temperature: ? C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 29429 times
  Warm Water active time: 7 days 19 hours 0 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 45 C
  Current flow temperature: 44.9 C
  Return temperature: 37.5 C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 75 %
  Burner current power: 23 %
  Flame current: 6.3 uA
  System pressure: 1.9 bar
  System service code:  (200)
  Heating temperature setting on the boiler: 70 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 50 %
  Boiler temperature: 0.0 C
  Warm water storage temperature1: ? C
  Warm water storage temperature2: ? C
  Exhaust temperature: 40.9 C
  Pump modulation: 50 %
  Burner # starts: 142461 times
  Total burner operating time: 727 days 6 hours 28 minutes
  Total heat operating time: 719 days 11 hours 28 minutes
  Total UBA working time: 3646 days 16 hours 12 minutes

Thermostat data:
  Thermostat: Junkers FR100 (DeviceID: 0x10, ProductID: 107, Version: 15.07)
  Thermostat time is 13:41:22 29/02/2020
  Heating Circuit 1
   Current room temperature: 20.1 C
   Setpoint room temperature: 20.0 C
   Mode is set to manual

proddy, 2020-02-29:
I need to see the logging. `log v` then `send B8 10 FF 11 00 65 2A` so I can see what is sent and what the thermostat responds with

norberts1, 2020-02-29:
Hi @proddy. B8 ist wrong. It must be: 8B as first byte. This first byte is for the sender/source indication and the second one for the receiver/target.
The first byte on Junkers then has to be 8B (hex):
  MSB must be set: 80 and the device-ID will be added -->>
   8B --> if the service-key is used as device-ID
or
  8D --> if the modem-ID is used
Some other ID's are also possible.

hansstulp, 2020-02-29:
log v

System Logging set to Verbose
(14:37:43) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B D4 00 72 F5 00 (#data=17)
(14:37:44) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 2A 01 A0 4B 00 01 01 20 40 80 00 80 00 01 A1 00 00 13 00 00 00 CC 00 00 00 (#data=25)
send 8B 10 FF 11 00 65 2A

(14:37:50) Sending raw: 8B 10 FF 11 00 65 2A
(14:37:50) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(14:37:51) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(14:37:53) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 2A 01 A1 4B 00 01 01 20 40 80 00 80 00 01 A2 00 00 13 00 00 00 CC 00 00 00 (#data=25)
(14:37:54) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B D4 00 72 F5 00 (#data=17)
Fetching data from EMS devices
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
Publishing MQTT data for: thermostat boiler
(14:37:58) Sending read of type 0x06 to 0x10, telegram: 8B 90 06 00 20
(14:37:58) Thermostat -> Me, RCTime(0x06), telegram: 90 0B 06 00 14 02 0E 1D 23 0B 06 00 (#data=8)
(14:37:58) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(14:37:59) Sending read of type 0x18 to 0x08, telegram: 8B 88 18 00 20
(14:37:59) Boiler -> Me, UBAMonitorFast(0x18), telegram: 88 0B 18 00 2A 01 A1 4B 00 01 01 20 40 80 00 80 00 01 A2 00 00 13 00 00 00 CC 00 00 00 (#data=25)
(14:37:59) Sending read of type 0x19 to 0x08, telegram: 8B 88 19 00 20
(14:37:59) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(14:38:01) Sending read of type 0x33 to 0x08, telegram: 8B 88 33 00 20
(14:38:01) Boiler -> Me, UBAParameterWW(0x33), telegram: 88 0B 33 00 08 FF 3C FB 00 28 FF 02 46 D8 FF (#data=11)
(14:38:01) Sending read of type 0x16 to 0x08, telegram: 8B 88 16 00 20
(14:38:01) Boiler -> Me, UBAParametersMessage(0x16), telegram: 88 0B 16 00 FF 46 4B 00 06 FA 0A 01 01 64 32 00 (#data=12)
(14:38:01) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(14:38:02) Sending read of type 0x14 to 0x08, telegram: 8B 88 14 00 20
(14:38:02) Boiler -> Me, UBATotalUptimeMessage(0x14), telegram: 88 0B 14 00 50 20 C3 (#data=3)
(14:38:02) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(14:38:03) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 2A 01 A1 4B 00 01 01 20 40 80 00 80 00 01 A2 00 00 13 00 00 00 CC 00 00 00 (#data=25)
(14:38:04) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B D4 00 72 F5 00 (#data=17)

proddy, 2020-02-29:
> Hi @proddy. B8 ist wrong. It must be: 8B as first byte. This first byte is for the sender/source indication and the second one for the receiver/target.
> The first byte on Junkers then has to be 8B (hex):
> MSB must be set: 80 and the device-ID will be added -->>
> 8B --> if the service-key: 0B is used as device-ID
> or
> 8D --> if the modem-ID: 0D is used
> Some other ID's are also possible.

yup, thanks. was just a typo.

proddy, 2020-02-29:
@hansstulp looks like the ` 8B 10 FF 11 00 65 2A` is not being sent correctly. Can you try with tx_mode 2 and 3. You'll need to restart the EMS after each. Start with 3 (HT3) and then 2 (EMS+). This changes the timing of the UART signals.

hansstulp, 2020-02-29:
It was in HT3

hansstulp, 2020-02-29:
This is mode 1
[FS] Custom config loaded (264 bytes)
[WEB] Web server started
[UART] Rx/Tx connection established
[WIFI] Connecting to SSID: FRITZ!Box 7581 GG...
[WIFI] Connected to SSID FRITZ!Box 7581 GG (hostname: ems-esp, IP: 192.168.178.50)
[OTA] Listening to firmware updates on ems-esp.local:8266
[SYSTEM] Disabling serial port communication
[NTP] NTP internet time enabled via server pool.ntp.org with timezone 2
[MQTT] Connecting to MQTT...
[NTP] Internet time: 14:06:44 UTC on 29/2. Local time: 15:06:44 CET
[MQTT] MQTT connected
[SYSTEM] Free Heap: 23560 bytes initially | 8584 bytes used (36%) | 14976 bytes free (63%)
New EMS device recognized as a Bosch Boiler: Nefit Proline (DeviceID: 0x08, ProductID: 122, Version: 03.09)
New EMS device recognized as a Controller: BC10 Base Controller (DeviceID: 0x09, ProductID: 114, Version: 01.07)
07)uctID: 107, Version: 15.s a Bosch Thermostat: Junkers FR100 (DeviceID: 0x10,
[TELNET] Connected to EMS-ESP version 1.9.5b46. Type ? for commands.
log v

System Logging set to Verbose
send 8B 10 FF 11 00 65 2A

(15:08:04) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 31 01 E0 4B 19 09 01 25 40 80 00 80 00 01 91 00 42 13 00 00 00 C8 00 02 00 (#data=25)
(15:08:04) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B D4 00 72 F5 00 (#data=17)
(15:08:04) Sending raw: 8B 10 FF 11 00 65 2A
(15:08:04) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(15:08:05) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(15:08:06) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 30 (#data=1)
(15:08:06) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 30 (#data=1)
(15:08:12) Thermostat -> Boiler, Version(0x02), telegram: 90 88 02 00 0A (#data=1)
(15:08:12) Boiler -> Thermostat, Version(0x02), telegram: 88 10 02 00 7A 03 09 00 00 00 00 00 00 01 (#data=10)
(15:08:14) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 30 01 E2 4B 19 09 01 25 40 80 00 80 00 01 92 00 42 13 00 00 00 C8 00 02 00 (#data=25)
(15:08:14) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B D4 00 72 F5 00 (#data=17)

hansstulp, 2020-02-29:
In EMS Dashboard : EMS Bus Connected but Tx is not working.

hansstulp, 2020-02-29:
This is with EMS+. EMS and HT3 are ok.

proddy, 2020-02-29:
ok leave it at HT3 for now. Still strange why your FR100 doesn't respond to EMS commands. Perhaps the telegram structure is wrong. Need to think about it and ask around. @norberts1 any ideas?

norberts1, 2020-02-29:
Hi, I'll think it works already in HT3-mode.
You can find the command-result (2A) in:
(14:37:59) Boiler -> Me, UBAMonitorFast(0x18), telegram: 88 0B 18 00 2A 01 A1 4B 00 01 01 20 40 80 00 80 00 01 A2 00 00 13 00 00 00 CC 00 00 00 (#data=25)
2A is for your desired temp and the
01 A1 after that is the current flow-temp as: 417/10-->> 41.7 degree.
You can test it in HT3 - mode using other desired-temps and watch what is happen.

proddy, 2020-02-29:
normal EMS behavior is that when you send write command via EMS to a device it would respond with an acknowledgment (1=0, 4=fail) and also a response telegram. @hansstulp I would just try with a different value, something crazy like 30 degrees (8B 10 FF 11 00 65 3C) and see if this number is shown on the thermostat. If not, we need to look closely at the signal and get the big guns out.

hansstulp, 2020-02-29:
[WIFI] Connected to SSID FRITZ!Box 7581 GG (hostname: ems-esp, IP: 192.168.178.50)
[OTA] Listening to firmware updates on ems-esp.local:8266
[SYSTEM] Disabling serial port communication
[NTP] NTP internet time enabled via server pool.ntp.org with timezone 2
[MQTT] Connecting to MQTT...
[NTP] Internet time: 14:20:22 UTC on 29/2. Local time: 15:20:22 CET
[MQTT] MQTT connected
[SYSTEM] Free Heap: 23560 bytes initially | 8584 bytes used (36%) | 14976 bytes free (63%)
New EMS device recognized as a Bosch Boiler: Nefit Proline (DeviceID: 0x08, ProductID: 122, Version: 03.09)
New EMS device recognized as a Controller: BC10 Base Controller (DeviceID: 0x09, ProductID: 114, Version: 01.07)
07)uctID: 107, Version: 15.s a Bosch Thermostat: Junkers FR100 (DeviceID: 0x10,
 on 29/2. Local time: 16:20:23 CET
3 CET C on 29/2. Local time: 17:20:2
[TELNET] Connected to EMS-ESP version 1.9.5b46. Type ? for commands.
log v

System Logging set to Verbose
send 8B 10 FF 11 00 65 3C(17:53:03) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 28 01 5E 4B 00 01 01 20 40 80 00 80 00 01 60 00 00 13 00 00 00 CC 00 00 00 (#data=25)
(17:53:03) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B D4 00 72 F5 00 (#data=17)
(17:53:09) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 28 01 5D 4B 00 01 01 20 40 80 00 80 00 01 60 00 00 13 00 00 00 CC 00 00 00 (#data=25)
(17:53:13) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B D4 00 72 F5 00 (#data=17)
(17:53:13) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 28 01 5D 4B 00 01 01 20 40 80 00 80 00 01 60 00 00 13 00 00 00 CC 00 00 00 (#data=25)
Fetching data from EMS devices
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(17:53:19) Sending read of type 0x06 to 0x10, telegram: 8B 90 06 00 20
(17:53:20) Thermostat -> Me, RCTime(0x06), telegram: 90 0B 06 00 14 02 11 1D 32 20 06 00 (#data=8)
(17:53:20) Sending read of type 0x18 to 0x08, telegram: 8B 88 18 00 20
(17:53:20) Boiler -> Me, UBAMonitorFast(0x18), telegram: 88 0B 18 00 28 01 5D 4B 00 01 01 20 40 80 00 80 00 01 5F 00 00 13 00 00 00 CC 00 00 00 (#data=25)
(17:53:20) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(17:53:20) Sending read of type 0x19 to 0x08, telegram: 8B 88 19 00 20
(17:53:20) Boiler -> Me, UBAMonitorSlow(0x19), telegram: 88 0B 19 00 80 00 00 00 01 69 00 00 00 32 02 2C 8C 0F FB 3A 00 00 00 0F CF 66 01 B9 97 00 00 (#data=27)
(17:53:20) Sending read of type 0x33 to 0x08, telegram: 8B 88 33 00 20
(17:53:20) Boiler -> Me, UBAParameterWW(0x33), telegram: 88 0B 33 00 08 FF 3C FB 00 28 FF 02 46 D8 FF (#data=11)
(17:53:21) Sending read of type 0x16 to 0x08, telegram: 8B 88 16 00 20
(17:53:21) Boiler -> Me, UBAParametersMessage(0x16), telegram: 88 0B 16 00 FF 46 4B 00 06 FA 0A 01 01 64 32 00 (#data=12)
(17:53:21) Sending read of type 0x14 to 0x08, telegram: 8B 88 14 00 20
(17:53:22) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(17:53:23) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 28 01 5C 4B 00 01 01 20 40 80 00 80 00 01 5F 00 00 13 00 00 00 CC 00 00 00 (#data=25)
(17:53:23) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B D4 00 72 F5 00 (#data=17)

norberts1, 2020-02-29:
Hi @proddy , yes the response with an acknowledgment (1=0, 4=fail) should be there.
@hansstulp, in your last log the desired temperatur command: 8B 10 FF 11 00 65 3C was not executed.
For that reason please check your FR100 the factory build date (FD:=FertigungsDatum). This is written inside the frontpanel of your thermostat. Hopefully it is younger than 9.2008 := FD889.
If not, then bad new's and the commands for setting temperatur or modes are not supported with your FD100.
So, please check that FD-code and compare it with that table, see attachment:
[image]
[image]

hansstulp, 2020-02-29:
It is de 987.

norberts1, 2020-02-29:
@hansstulp it's OK. From 7.2009 and it should work.
Then we should try other command's tomorrow morning if you have time for that.

norberts1, 2020-03-01:
@hansstulp I'll think you have **one** heating-circuit in your system and that FR100 as single modul connected to your heater. Then is time to check the configuration of your FR100. Mainly that heating-circuit configuration is important. Default value should be:1, but also 2-10 is possible.
You can check this in the expert configuration menu:systemconfiguration-->>heatingcircuit codingvalue.
If this differ to the default value:1, then that is the reason for no response to the previous commands.
Please check it.

hansstulp, 2020-03-01:
It is value 1.

norberts1, 2020-03-01:
@hansstulp OK, then we can try to change that current mode/status of your heater-system. Please control also that display on your FR100 for any changes and stay in verbose-mode for outputs (perhaps you can truncate them a bit to that interesting telegrams).
Please check that mode-switch inside your FR100 before that test starts. It should have 'auto'-mode.
Starting tests:
1. setting up mode 'frost'
command: 8B 10 FF 04 00 79 01
2. setting up mode 'sparen'
command: 8B 10 FF 04 00 79 02
3. setting up mode 'heizen'
command: 8B 10 FF 04 00 79 03
4. setting up mode 'auto'
command: 8B 10 FF 04 00 79 04
That display on your FR100 should change and that 'NC' symbol is displayed. This attached pictures are made with FW100, but it works like to your FR100.
[image]
[image]
[image]
That text is in german and can differ to your display.
Good luck.

hansstulp, 2020-03-01:
[TELNET] Connected to EMS-ESP version 1.9.5b46. Type ? for commands.
log v

System Logging set to Verbose

send 8B 10 FF 04 00 79 01

(17:40:19) Sending raw: 8B 10 FF 04 00 79 01
(17:40:19) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(17:40:20) Thermostat -> All, Type(0x0079), telegram: 90 00 FF 00 00 79 01 08 10 4B 01 0A 10 00 04 00 01 (#data=11)
(17:40:20) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
Fetching data from EMS devices
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
Publishing MQTT data for: thermostat boiler
(17:40:22) Sending read of type 0x06 to 0x10, telegram: 8B 90 06 00 20
(17:40:22) Thermostat -> Me, RCTime(0x06), telegram: 90 0B 06 00 14 03 11 01 25 0D 00 00 (#data=8)
(17:40:22) Sending read of type 0x18 to 0x08, telegram: 8B 88 18 00 20
(17:40:22) Boiler -> Me, UBAMonitorFast(0x18), telegram: 88 0B 18 00 05 01 C4 00 00 00 00 40 40 80 00 80 00 01 D4 00 00 10 00 00 00 CB 00 00 00 (#data=25)
(17:40:22) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(17:40:23) Sending read of type 0x19 to 0x08, telegram: 8B 88 19 00 20
(17:40:23) Boiler -> Me, UBAMonitorSlow(0x19), telegram: 88 0B 19 00 80 00 00 00 01 DD 00 00 00 00 02 2C DB 0F FD 4D 00 00 00 0F D1 79 01 B9 E5 00 00 (#data=27)
(17:40:23) Sending read of type 0x33 to 0x08, telegram: 8B 88 33 00 20
(17:40:23) Boiler -> Me, UBAParameterWW(0x33), telegram: 88 0B 33 00 08 FF 3C FB 00 28 FF 02 46 D8 FF (#data=11)
(17:40:23) Sending read of type 0x16 to 0x08, telegram: 8B 88 16 00 20
(17:40:23) Boiler -> Me, UBAParametersMessage(0x16), telegram: 88 0B 16 00 FF 46 4B 00 06 FA 0A 01 01 64 32 00 (#data=12)
(17:40:24) Sending read of type 0x14 to 0x08, telegram: 8B 88 14 00 20
(17:40:24) Boiler -> Me, UBATotalUptimeMessage(0x14), telegram: 88 0B 14 00 50 27 21 (#data=3)
(17:40:24) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(17:40:25) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 05 01 C4 00 00 00 00 40 40 80 00 80 00 01 D4 00 00 10 00 00 00 CB 00 00 00 (#data=25)
(17:40:26) Boiler -> All, UBAMonitorSlow(0x19), telegram: 88 00 19 00 80 00 00 00 01 DD 00 00 00 00 02 2C DB 0F FD 4D 00 00 00 0F D1 79 01 B9 E5 00 00 (#data=27)
(17:40:26) Boiler -> All, UBAMaintenanceStatusMessage(0x1C), telegram: 88 00 1C 00 00 3B C7 48 00 00 00 00 00 00 00 (#data=11)
(17:40:26) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 01 00 00 01 00 00 2B D4 00 72 F6 00 (#data=17)
(17:40:26) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(17:40:27) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 05 01 C4 00 00 00 00 40 40 80 00 80 00 01 D4 00 00 10 00 00 00 CB 00 00 00 (#data=25)
(17:40:35) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 01 00 00 01 00 00 2B D4 00 72 F6 00 (#data=17)
(17:40:36) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 05 01 C4 00 00 00 00 40 40 80 00 80 00 01 D4 00 00 10 00 00 00 CB 00 00 00 (#data=25)
(17:40:37) Thermostat -> Boiler, UBAParametersMessage(0x16), telegram: 90 88 16 00 02 (#data=1)
(17:40:37) Boiler -> Thermostat, UBAParametersMessage(0x16), telegram: 88 10 16 00 FF 46 (#data=2)
(17:40:37) Thermostat -> All, JunkersStatusMessage_HC1(0x006F), telegram: 90 00 FF 00 00 6F 01 01 00 32 00 CA (#data=6)
send 8B 10 FF 04 00 79 02

(17:40:44) Sending raw: 8B 10 FF 04 00 79 02
(17:40:44) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(17:40:44) Thermostat -> All, Type(0x0079), telegram: 90 00 FF 00 00 79 01 08 10 4B 02 0A 10 00 04 00 01 (#data=11)
(17:40:45) Thermostat -> All, JunkersStatusMessage_HC1(0x006F), telegram: 90 00 FF 00 00 6F 02 01 00 50 00 CA (#data=6)
(17:40:45) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 04 02 (#data=1)
(17:40:45) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(17:40:45) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 05 01 C4 00 00 00 00 40 40 80 00 80 00 01 D4 00 00 10 00 00 00 CB 00 00 00 (#data=25)
(17:40:46) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 01 00 00 01 00 00 2B D4 00 72 F6 00 (#data=17)
(17:40:46) Thermostat -> Boiler, Version(0x02), telegram: 90 88 02 00 0A (#data=1)
(17:40:46) Boiler -> Thermostat, Version(0x02), telegram: 88 10 02 00 7A 03 09 00 00 00 00 00 00 01 (#data=10)
(17:40:55) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 05 01 C3 00 00 00 00 40 40 80 00 80 00 01 D4 00 00 10 00 00 00 CB 00 00 00 (#data=25)
(17:40:56) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 01 00 00 01 00 00 2B D4 00 72 F6 00 (#data=17)
send 8B 10 FF 04 00 79 03

(17:41:00) Sending raw: 8B 10 FF 04 00 79 03
(17:41:00) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(17:41:01) Thermostat -> All, Type(0x0079), telegram: 90 00 FF 00 00 79 01 08 10 4B 03 0A 10 00 04 00 01 (#data=11)
[... cut ...]
(17:41:22) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(17:41:23) Sending read of type 0x18 to 0x08, telegram: 8B 88 18 00 20
(17:41:23) Boiler -> Me, UBAMonitorFast(0x18), telegram: 88 0B 18 00 05 01 C3 00 00 00 00 40 40 80 00 80 00 01 D4 00 00 10 00 00 00 CB 00 00 00 (#data=25)
(17:41:23) Sending read of type 0x19 to 0x08, telegram: 8B 88 19 00 20
(17:41:24) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)

the thermostat display have changed

proddy, 2020-03-01:
so writing direct commands to the FR100 seems to be working for changing the modes (except there are no EMS telegram responses). Are you still sure changing the temperature like with `8B 10 FF 11 00 65 3C` is not working?

hansstulp, 2020-03-01:
[TELNET] Connected to EMS-ESP version 1.9.5b46. Type ? for commands.
log v

System Logging set to Verbose
send 8B 10 FF 11 00 65 3C

(18:21:54) Sending raw: 8B 10 FF 11 00 65 3C
(18:21:54) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(18:21:55) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(18:21:56) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 46 02 B8 4B 1A 09 01 25 40 80 00 80 00 02 65 00 44 14 00 00 00 C8 00 02 00 (#data=25)
(18:21:56) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 01 00 00 01 00 00 2B D4 00 72 F6 00 (#data=17)
(18:21:57) Thermostat -> Boiler, UBAFlags(0x35), telegram: 90 08 35 00 11 (#data=1)
(18:21:57) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 00 46 (#data=1)
(18:21:57) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 02 00 00 02 (#data=3)
(18:21:57) Thermostat -> Boiler, Type(0x23), telegram: 90 08 23 00 46 64 00 (#data=3)
(18:21:58) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 90 08 1A 04 02 (#data=1)
(18:21:58) Thermostat -> All, RCTime(0x06), telegram: 90 00 06 00 14 03 12 01 13 05 00 00 (#data=8)

proddy, 2020-03-01:
@hansstulp forget the log responses. Did it actually change the temperature? Is your house warmer? :) Did the FR100 thermostat show the new temperature (which was 30 degrees!)

norberts1, 2020-03-01:
Hi @hansstulp, @proddy regarding that temp-setup command it's  required to send another command. I have tested it on my environment in dummy mode (I haven't a FW100 anymore) and following command should be send to setup 22.5 degrees in 'normal' heating-mode:
8B 10 FF 07 00 79 2D
Please check the result on the FR100 Display (perhaps you have to dig into the menues to find that value for: 22.5 and 'normal' mode.
After that please switch back to that temperatur you normaly uses as eg. 21.5 degrees
8B 10 FF 07 00 79 2B
I'll hope it works with it:-)

hansstulp, 2020-03-01:
send 8B 10 FF 11 00 65 3C did not change the display and did not change my room temp :)

norberts1, 2020-03-01:
@hansstulp our comments are send at the same time. Please check it with that above mentioned telegrams.

hansstulp, 2020-03-01:
It has changed the temp. It was set to 23.0 degrees and not 21.5 degrees but 22.0 degrees.

proddy, 2020-03-01:
@norberts1 I never really understood the difference between types 0x65-0x6E and 0x79-0x82 with Junkers but if it works for the FW100 then its great. Will it be the same for the other thermostats or just this type of older models?

norberts1, 2020-03-01:
@proddy it will work for any of that Fxyz-group, without FR50 and that one's older then 9.2008.
@hansstulp great, we got it now!
Some modul's are using 1 degree steps, others 0.5.
So, following sequences can also be done 
(keep in mind that this changes are available also next day, you have to set them back to the old values if you dont want that changes)
ps: also is required that the temp's for that niveau's must raising upwards, ment:
'frost' must have less temp then 'sparen'
'sparen must have less temp then 'heizen'

1. set temp for 'frost'-niveau to 15 degree
8B 10 FF 05 00 79 1E
2. set temp for 'sparen'-niveau to 20 degree
8B 10 FF 06 00 79 28
3. set temp for 'heizen'-niveau to 22 degree
8B 10 FF 07 00 79 1E
You can check and control this on that FR100 display, if you like :-)

hansstulp, 2020-03-01:
Where can i find information about Netcom?

hansstulp, 2020-03-01:
@norberts1 All changes have been made.

norberts1, 2020-03-01:
@hansstulp Netcom is an old name for one modul (NetCom100) used to control the Junkers-heater per phone. I had one but isn't up do date anymore. The displays on that Fxyz-controllers results on that modul, but that text wasn't never changed.
Later on they have replaced that old modul with that MBLan or MBLan2 (Junkers) or KM200 (Buderus). This moduls are available today for controlling that heatersystem.
They are quit expensive. So it's better to replace them with proddy's SW/HW.

norberts1, 2020-03-02:
@proddy regarding the lack of acknowledgement I have seen some context in that telegrams typ(0x07)
Example:
(18:21:54) Sending raw: 8B 10 FF 11 00 65 3C
(18:21:54) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 **0B 01** 00 00 00 00 00 00 00 00 00 00 00 (#data=13)  <<--- That one is OK
(18:21:55) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 **03 01** 00 00 00 00 00 00 00 00 00 00 00 (#data=13)  <<--- That one is faulty
This message tells us, which device-IDs are currently on the bus. Every high-bit in the payload is the available device-ID starting with device-ID 8.
0B is then: 0000 1011 -->> active device-IDs then (decimal): ---- 11 - 9 8 or hex: ---- B - 9 8
01 is then: 0000 0001 -->> active device-ID 2.Byte(decimal): ---- --- 16 or hex: ---- --- 10
Result: Device-ID's (hex): 8,9,B and 10 are currently available on the bus.
In the faulty-condition there is the B -->> Service-Key not available any more. Perhaps the responses to the bus from the EMS-adapter aren't handled correctly.

proddy, 2020-03-02:
@norberts1 what I mean by acknowledgment is that the EMS protocol should always respond to a read/write request, even if it can't perform the operation. So in the console, I would be expecting to see messages like "Thermostat -> Me", just like what you see with the boiler. I'm guessing the FR100 uses a very old version of the protocol. It's fine though, just a bit weird.

I'm aware of the 0x07 telegram. This is what I use to determine which devices are on the bus. The 0x0B (11) is always intermittent (on and off) because I haven't built in the official acknowledgment handshake. It's not really needed and not a problem since my code just ignores it.

When @hansstulp has confirmed the temp settings work with 0x79 (e.g. `send 8B 10 FF 05 00 79 1E`) I'll modify the code. The confusing thing is that I know with the FW100 and FW120 models the type is 0x65 for Heating Circuit 1 and confirmed by other users to work ??

hansstulp, 2020-03-02:
[TELNET] Connected to EMS-ESP version 1.9.5b47. Type ? for commands.
log v t

System Logging set to Verbose
send 8B 10 FF 05 00 79 1E

(11:07:20) Sending raw: 8B 10 FF 05 00 79 1E
(11:07:20) Thermostat -> All, Type(0x0079), telegram: 90 00 FF 05 00 79 0A (#data=1)
(11:07:20) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 05 01 B4 00 00 00 00 20 40 80 00 80 00 01 B5 00 00 13 00 00 00 CB 00 00 00 (#data=25)
(11:07:21) Boiler -> All, UBAMonitorSlow(0x19), telegram: 88 00 19 00 80 00 00 00 01 A9 00 00 00 32 02 2D 0E 0F FE E1 00 00 00 0F D3 09 01 BA 18 00 00 (#data=27)
(11:07:21) Boiler -> All, UBAMaintenanceStatusMessage(0x1C), telegram: 88 00 1C 00 00 3B C7 48 00 00 00 00 00 00 00 (#data=11)
(11:07:21) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B D8 00 72 F6 00 (#data=17)
(11:07:21) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(11:07:30) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 05 01 B1 00 00 00 00 20 40 80 00 80 00 01 B1 00 00 13 00 00 00 CB 00 00 00 (#data=25)
(11:07:31) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B D8 00 72 F6 00 (#data=17)
(11:07:37) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 05 01 AF 00 00 00 00 00 40 80 00 80 00 01 AF 00 00 10 00 00 00 CB 00 00 00 (#data=25)
(11:07:40) Thermostat -> Boiler, UBAParametersMessage(0x16), telegram: 90 88 16 00 02 (#data=1)
(11:07:40) Boiler -> Thermostat, UBAParametersMessage(0x16), telegram: 88 10 16 00 FF 46 (#data=2)
(11:07:40) Thermostat -> All, JunkersStatusMessage_HC1(0x006F), telegram: 90 00 FF 00 00 6F 02 02 00 96 00 C3 (#data=6)
(11:07:41) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B D8 00 72 F6 00 (#data=17)
(11:07:41) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 05 01 AF 00 00 00 00 00 40 80 00 80 00 01 AF 00 00 10 00 00 00 CB 00 00 00 (#data=25)
(11:07:43) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 05 01 AF 00 00 00 00 40 40 80 00 80 00 01 AF 00 00 10 00 00 00 CB 00 00 00 (#data=25)
(11:07:50) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 3C 80 00 80 00 21 00 00 01 00 00 2B D8 00 72 F6 00 (#data=17)
(11:07:51) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 05 01 AF 00 00 00 00 40 40 80 00 80 00 01 B0 00 00 10 00 00 00 CB 00 00 00 (#data=25)

I see no changes on the thermostat

norberts1, 2020-03-02:
@proddy regarding the 'acknowledgment' I'll think the FR100 send also that 1/0 responses to any command like the FW100 I owned but I'm also not sure.
Yes, the telegram-handling is a bit confusing. So what I really have implemented in my code for setting the temp is to send more then one message to anyone how can use/need it on the bus (available or not) in following sequence
1. 8x **10** FF <y> 00 65 <desired-temp *2>
2. 8x **18** FF <y> 00 65 <desired-temp *2>
3. 8x **10** FF <z> 00 79 <desired-temp *2>
where: 
 x:=device ID for the sender e.g. (B)-->>service-key
 y:=offset depending on that desired temp-niveau and message
 z:=offset depending on that desired temp-niveau and message
As you can see I'll send that messages also to that 0x18 device-ID, that is/was my remote-controller 'FB10' and later the 'FB100' mounted in my living-room.
They both required this if they are on the bus. Today they aren't on the bus any more, but I handle it like: fire and forget.
Perhaps is one solution to handle this in EMS-ESP too.

norberts1, 2020-03-02:
@hansstulp with this telegram you want to change the 'frost'-temp level to 15 degree.
(8B 10 FF 05 00 79 1E), but this is not yet displayed cause you are (or the FR100) is running in 'auto' using that programmed temp-level -->> 'heating' I'm sure and not 'frost'!
In this mode 'auto' the display will show the current room-temperature and not the 'frost'.
So you have to dig into the menue's or to switching the mode-switch in the FR100 to 'frost' to see that new 'frost'-level of 15 degrees.
Details see manual.

hansstulp, 2020-03-02:
No, i didn't changes the frost temp. I have 5, 15 and 20 set. Now i can't changes the temperature.

norberts1, 2020-03-02:
@hansstulp yesterday you wrote:
 It has changed the temp. It was set to 23.0 degrees and not 21.5 degrees but 22.0 degrees.
What's different today?

hansstulp, 2020-03-02:
I have changed the temp manually to 5,17,20 and now it's working.

hansstulp, 2020-03-02:
I would like to switch by Domotiz between "sparen" and "heizen". And also set de temp manually higher or lower.

[20 more replies]

closed 2020-03-10 by proddy