emsesp/EMS-ESP #339: Bosch easycontrol

edwin6856, 2020-03-01:
Bosch easycontrol thermostat is not working. 
Is it possible to make it work?

Thanks
Edwin

proddy, 2020-03-01:
If you mean being able to change parameters on the EasyControl as a thermostat then sadly no, it's too difficult as these type of thermostats don't "talk" EMS. Which is why it's not supported by EMS-ESP and clearly documented in the wiki at https://emsesp.github.io/docs/#/Supported-EMS-Devices and https://emsesp.github.io/docs/#/wish-list.

But if this is similar to what @curantil discovered in #45 then lets take another look at that.

bbqkees, 2020-03-09:
Bosch has the KM200 module, which also controls connected thermostats via the EMS bus.
The Bosch EasyControl thermostat is their own high-end web-enabled thermostat. 
And its not supported by the KM200.
This is also an indication this thermostat is just not configurable via the EMS bus. The thermostat itself might still read telegrams on the bus to decode the boiler data and send telegrams to the boiler to control the burner but that's likely it.

proddy, 2020-05-16:
There are some earlier posts of people using python and home assistant to control the Easy. The Easy receives its commands via a cloud service hosted by Bosch, and not via EMS

closed 2020-05-16 by proddy