emsesp/EMS-ESP #341: Failed to "Setting boiler warm water temperature" 1.9.5b49

stravinci, 2020-03-05:
I tried to set warm water temeprature level. Log is below.
After that ESP was restarted and here are list of detected devices:
```
These 2 were recognized by EMS-ESP:
 Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID: 0x08, ProductID: 95, Version: 23.12)
 Controller: HT3 Controller (DeviceID: 0x09, ProductID: 95, Version: 23.12)
```
And here is log from rescan:
```
devices scan

Requesting EMS bus master for its device list and scanning for external sensors...
[EMS] Detected new EMS Device with ID 0x08. Fetching version information...
[EMS] Detected new EMS Device with ID 0x09. Fetching version information...
[EMS] Detected new EMS Device with ID 0x10. Fetching version information...
New EMS device recognized as a Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID: 0x08, ProductID: 95, Version: 23.12)
New EMS device recognized as a Controller: HT3 Controller (DeviceID: 0x09, ProductID: 95, Version: 23.12)
New EMS device recognized as a Thermostat: Junkers FW120 (DeviceID: 0x10, ProductID: 192, Version: 53.02)
```

```
System Logging set to Verbose
Setting boiler warm water temperature to 46 C
Publishing MQTT data for: boiler
(00:07:38.086) Sending write of type 0x33 to 0x08, telegram: 8B 08 33 02 2E
(00:07:38.170) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (#data=15)
(00:07:38.384) Boiler -> All, UBAParameterWW(0x33), telegram: 88 00 33 00 08 FF 31 00 00 00 00 02 4B 00 FF FF (#data=12)
(00:07:38.615) Sending validate of type 0x33 to 0x08, telegram: 8B 88 33 02 01
-> Write failed. Compared set value 0x2E with received value of 0x31
-> Write didn't work, retrying (1/2)...
(00:07:38.771) Sending write of type 0x33 to 0x08, telegram: 8B 08 33 02 2E
(00:07:38.914) Boiler -> All, UBAParameterWW(0x33), telegram: 88 00 33 00 08 FF 31 00 00 00 00 02 4B 00 FF FF (#data=12)
(00:07:39.176) Sending validate of type 0x33 to 0x08, telegram: 8B 88 33 02 01
-> Write failed. Compared set value 0x2E with received value of 0x31
-> Write didn't work, retrying (2/2)...
(00:07:39.331) Sending write of type 0x33 to 0x08, telegram: 8B 08 33 02 2E
(00:07:39.412) Boiler -> All, UBAParameterWW(0x33), telegram: 88 00 33 00 08 FF 31 00 00 00 00 02 4B 00 FF FF (#data=12)
(00:07:39.710) Sending validate of type 0x33 to 0x08, telegram: 8B 88 33 02 01
-> Write failed. Compared set value 0x2E with received value of 0x31
-> Write failed. Giving up, removing from queue
(00:07:40.194) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (#data=15)

```

How can I help you to fix it?

proddy, 2020-03-05:
Hi, strange it should work. The telegram looks correct. I'll need to check with Norbert, our Junkers/HT3 expert

stravinci, 2020-03-05:
Ok I will wait for your response.
Another think - it is ok that system find 3 nodes(0x08, 0x09, 0x10)? I have Junkers Smart Cerapur + thermostat FW120.

proddy, 2020-03-05:
the 3 devices is normal. your boiler has a built in controller (0x09) which is the front-panel.

norberts1, 2020-03-05:
@stravinci @proddy that command: `8B 08 33 02 2E` seems to be OK for that desired hot-water temperature. It's OK for 'reading' that value on offset:02, but I'm not sure for 'writing' for that FWxyz-Thermostat's. 
I had a FW100 until 2018 and I tried to change that DHW too. But with less luck.
What I know is: there are some preconditions required as:
1. The hot-water max.value selected in your boiler (at that poti) must be higher than your wanted temperature.
2. That DHW-program setting in your FW120 for timed controlled hotwater-level setting must at least disabled, cause that will fight against your wanted temperature.
3. Some times that available telegrams with it's values are only foreseen for 'reading' and not 'writing'. So what we can do is, try and error (as usual).
4. There are some other telegrams doing compareable things with that DHW desired temperature. That ones have to be checked.
Following telegrams are then known for that:
1. `8B 08 33 02 2E` <<-- (type:33, MesID:51) failed on FR120, check already done
2. `8B 08 1B 00 2E` <<-- (type:1B, MesID:27) check required
3. `8B 08 34 00 2E` <<-- (type:34, MesID:52) check required
@stravinci can you check this on your system?
I haven't that FWxyz controller anymore on my system, so I can't do.
Please enable that verbose-mode to get some outputs from it, but most important is to watch your system (at your FW100) for the responses.

stravinci, 2020-03-05:
@norberts1 
1. max temperature was set to 49
2. I'm not sure if it is disabled.
I see in "info" command:
`Warm Water activated: on`
But it is not triggered by FW120 in him I think I can enable/disable it using dedicated button.
Here are logs:
`(00:34:23.299) Sending raw: 8B 08 1B 00 2E
(00:34:25.793) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 31 02 3E 64 00 02 23 64 C0 01 DC 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (#data=25)
(00:34:27.012) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 31 02 3C 64 00 02 23 64 C0 01 DD 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (#data=25)
(00:34:27.252) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 31 01 DC 80 00 A9 00 00 01 00 01 64 6A 01 9D 17 00 (#data=17)`

`(00:32:42.857) Sending raw: 8B 08 33 02 2E
(00:32:43.171) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 31 02 49 64 48 0A 33 65 C0 01 D5 80 00 80 00 FF FF FF 00 00 00 00 36 02 18 (#data=25)
(00:32:43.374) Boiler -> All, UBAParameterWW(0x33), telegram: 88 00 33 00 08 FF 31 00 00 00 00 02 4B 00 FF FF (#data=12)
(00:32:44.369) Thermostat -> Boiler, UBAParametersMessage(0x16), telegram: 90 88 16 00 02 (#data=1)
(00:32:44.390) Boiler -> Thermostat, UBAParametersMessage(0x16), telegram: 88 10 16 00 FF 3A (#data=2)`
`(00:35:44.742) Sending raw: 8B 08 34 00 2E
(00:35:46.694) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (#data=15)
(00:35:46.924) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 31 02 42 64 46 0A 33 65 C0 01 CC 80 00 80 00 FF FF FF 00 00 00 00 35 02 18 (#data=25)
(00:35:47.133) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 31 01 CC 80 00 A9 00 00 01 33 01 64 6B 01 9D 18 00 (#data=17)`

- At FW120 and "info" command I didn't see any changes.
`Warm Water selected temperature: 49 C`

proddy, 2020-03-06:
try disconnecting the FW120 and trying the command again

norberts1, 2020-03-06:
@stravinci please check that configuration in your FW120 which type of programcontrol on hotwater generation it is. That configuration can be:
C1: controlling is done together with your heating-program. 
Then we have two different hotwater-level for heating-niveau: heizen and heating-niveau:sparen.
C2: controlling of hotwater generation with separate program.
It would be good to know which one is configured in your FW120.
Q1: That value for setup on 49 C is done where?
Is this from program-control or from that poti-setup in the boiler?
If we know this then we can do a test strategy together with another command to get this working:
This test sequence could be:
T1: setup of maximum DHW to 55 C in the boiler with that poti inside.
T2: setup of DHW-program to a fixed value less then T3-value like: 30 C for the hole day.
T3: command the desired hotwater-level like 40 C and look for what is happen.
I have to figure out that not yet tested command and then we can do it again (try and error).

stravinci, 2020-03-06:
@proddy disconnection of FW120 does not help

@norberts1 
Q1 yes, it is set by poti-setup at boiler, I didn't find similar settings in FW120
My configuration was set to manual (C2 in your msg)- time dependent. I switch it to C1, and now I set Save(night) mode using FW120.
T1: I set max value for tap water using poti to 60 degree.
After above settings I tried 3 commands:
```
send 8B 08 1B 00 2E

send 8B 08 34 00 2E

send 8B 08 33 02 2E
```
There are no changes.

T2, T3 I need your instruction how to set it.

stravinci, 2020-03-06:
And one more info - it is not important for me to set it via HA or any another way, I only report this issue for you, but if it is important for you I can still help with testing.

norberts1, 2020-03-06:
@stravinci , thanks for that. Any help and hand can bring it forward:-)
It seems a bit more complicated as we want, but do what we can.
At first we need some preconditions:
P1: please switch of that 'ECO'-button on your boiler. I'll think that has one.
P2: set that max temp-value to higher value e.g. 60 C. You did it already.
P3: set that temp-value on DHW-program to a lower value as you want to command like 25 C.
P4. check that current DHW temp-level on the FW120.
Now is time to send this command:
`8B 08 35 01 00 00 <desired temp>`
e.g. for 40 C
`8B 08 35 01 00 00 28`
Please get that required infos from the FW120. Additional you can press that button on the thermostat named like: 'Warmwasser sofort' or something with that watertap-icon for requesting that DHW-warmup sequence.
Hopefully we get that informations we want.
@proddy with that verbose-mode we are getting **any** telegrams in RAW-mode?

stravinci, 2020-03-06:
Ok, precondition status:
Switch to C2 - done
P1 Eco mode at boiler - done
P2 max temp - done
P3 - when I press DHW button at F120 then I got only confirmation message and I can only click "Ok" to close that message. I don't know how can I modify DHW temperature.

norberts1, 2020-03-07:
@stravinci P1 -->> Eco Mode := **OFF** !
P3 -->> if you did that configuration C2 already (separate DHW-program) then you are able to select that DHW WW-program on your FW120.
Attached you will find some screenshots of that manual (sorry, in german), but you have that in your language. So I'll translate it as good as I can do:
[image]
After configuration C2 (separate DHW-program) then you are able to select that 'Warmwasser-Programm'.
To get that WW-program available (C2) then you have to select that as: 'Separate Programme' in the menues.
Dig into that WW-program and adjust that default desired WW-tempvalues for every day to your new value e.g. 30 C.
Default settings is like:
[image]
That values depend a bit at your type of boiler you have. 'WW-generating with storage' or 'combi-device'.
But anyway, you should find that menu-part and adjust it as wonted.
If you start the test with command:
`8B 08 35 01 00 00 <desired temp>` then you should select a higher 'desired-temp' then is already inside that ww-storage. Only then the boiler will start to heating up that DHW.
e.g. 56 C
`8B 08 35 01 00 00 38`
 You can press the DHW-button additional to force that heating of WW.

stravinci, 2020-03-07:
Ok, I look at my manual of FW120, there is information similar to your. As I understand in my case I will not have setting of temperature, because I don't have external storage for water. Maybe you can disable this feature for my configuration - dual function boiler without external storage for hot water.

proddy, 2020-03-07:
Do you know of a reliable way to determine if there is an external warm water storage available?

norberts1, 2020-03-08:
@stravinci you have a Cerapur Smart one. Which type (ZSB,ZWB...) is it?
Your system could be then like that picture:
[image]
But anyway which type your system is, most important is "it has WW- generation" and you can set the maximum temperature inside the boiler (you did already).
So please send that command-line for e.g. 56 C on any of that current/selected WW-configuration :
`8B 08 35 01 00 00 38`
and see what's happen.

stravinci, 2020-03-08:
@proddy below is output from "info" command. Maybe hot water circuit pump is not applicable in my setup? I don't know how to determinate this.
```
EMS-ESP system status:
  System logging set to None
  LED: off, Listen mode: off
  Boiler: enabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: d                                                                             isabled
  Shower Timer: enabled, Shower Alert: enabled

EMS Bus status:
  Bus is connected, protocol: HT3
  Rx: # successful read requests=9, # CRC errors=4
  Tx: Last poll=1.183 seconds ago, # successful write requests=0

Boiler data:
  Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester                                                                              Greenstar i/Generic HT3 (DeviceID: 0x08, ProductID: 95, Version: 23.12)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water circulation pump type: charge pump
  Warm Water circulation pump freq: 2x3min
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 48 C
  Warm Water desinfection temperature: 75 C
  Warm Water circulation active: off
  Warm Water current temperature: 50.5 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 106118 times
  Warm Water active time: 63 days 13 hours 23 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 40 C
  Current flow temperature: 32.2 C
  Return temperature: ? C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 76 %
  Burner current power: 0 %
  Flame current: 3200.0 uA
  System pressure: ? bar
  System service code:  (0)
  Heating temperature setting on the boiler: 58 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 10 %
  Outside temperature: 3.9 C
  Boiler temperature: ? C
  Warm water storage temperature1: 50.7 C
  Warm water storage temperature2: ? C
  Exhaust temperature: ? C
  Pump modulation: 100 %
  Burner # starts: 229236 times
  Total burner operating time: 326 days 11 hours 35 minutes
  Total heat operating time: 262 days 22 hours 12 minutes
  Total UBA working time: 1673 days 15 hours 34 minutes

Thermostat data:
  Thermostat: Junkers FW120 (DeviceID: 0x10, ProductID: 192, Version: 53.02)
  Thermostat time is 20:09:42 08/03/2020
  Heating Circuit 1
   Current room temperature: 23.3 C
   Setpoint room temperature: 21.5 C
   Mode is set to auto
```

proddy, 2020-03-08:
not sure either. What is your ultimate objective, to use EMS-ESP to set the warm water temperature? If this is something that the thermostat can do or the boiler can do via the front-panel then it should be possible via EMS+ calls. I'm a bit lost on this thread to what the goal is ;-)

stravinci, 2020-03-08:
@norberts1 my setup:
- Junkers Smart Cerapur SWB 28-3C
- FW120
- External temperature sensor
Nothing else.

Regarding to your check instruction - I set 60 degree and in info command I see:
`Warm Water selected temperature: 60 C`

And now sent telegram:
```
System Logging set to Verbose
(00:10:21.590) Thermostat -> Boiler, Version(0x02), telegram: 90 88 02 00 0A (#data=1)
(00:10:21.614) Boiler -> Thermostat, Version(0x02), telegram: 88 10 02 00 5F 17 0C 00 00 00 00 00 00 00 (#data=10)
send 8B 08 35 01 00 00 38

(00:10:23.939) Sending raw: 8B 08 35 01 00 00 38
(00:10:24.671) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (#data=15)
(00:10:24.901) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 28 01 42 4C 00 01 03 00 C0 01 8A 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (#data=25)
(00:10:25.110) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 38 01 8A 80 00 80 00 00 01 00 01 65 83 01 9E 86 00 (#data=17)
(00:10:25.370) Boiler -> All, UBAMonitorSlow(0x19), telegram: 88 00 19 00 00 27 80 00 80 00 FF FF 01 00 03 7F 76 07 2C 7A 00 00 00 05 C6 F6 01 E0 F0 80 00 (#data=27)
(00:10:26.520) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 28 01 42 4C 00 01 03 00 C0 01 8A 80 00 80 00 FF FF FF 00 00 00 00 00 02 18 (#data=25)
(00:10:27.517) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 28 01 43 4C 00 01 03 20 C0 01 89 80 00 80 00 FF FF FF 00 00 00 00 00 02 18 (#data=25)
(00:10:28.514) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 28 01 43 4C 00 01 03 24 C0 01 89 80 00 80 00 FF FF FF 00 00 00 00 00 02 18 (#data=25)
(00:10:29.541) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 28 01 43 4C 00 01 03 2D C0 01 89 80 00 80 00 FF FF FF 00 00 00 00 00 02 18 (#data=25)
(00:10:30.538) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 28 01 43 4C 2F 09 03 2D C0 01 89 80 00 80 00 FF FF FF 00 00 00 00 00 02 18 (#data=25)
(00:10:32.692) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 28 01 42 4C 2F 09 03 25 C0 01 89 80 00 80 00 FF FF FF 00 00 00 00 00 02 18 (#data=25)
(00:10:33.498) Thermostat -> All, RCTime(0x06), telegram: 90 00 06 00 14 03 14 08 14 0C 00 00 (#data=8)
(00:10:33.768) Thermostat -> Boiler, UBAParametersMessage(0x16), telegram: 90 88 16 00 02 (#data=1)
(00:10:33.782) Boiler -> Thermostat, UBAParametersMessage(0x16), telegram: 88 10 16 00 FF 3A (#data=2)
(00:10:33.881) Thermostat -> All, JunkersStatusMessage_HC1(0x006F), telegram: 90 00 FF 00 00 6F 03 02 00 D7 00 E8 F3 34 00 (#data=9)
(00:10:34.649) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 28 01 49 4C 2F 09 03 25 C0 01 89 80 00 80 00 FF FF FF 00 00 00 00 00 02 18 (#data=25)
(00:10:34.858) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 38 01 89 80 00 80 00 00 01 00 01 65 83 01 9E 86 00 (#data=17)
log(00:10:35.682) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 28 01 4F 4C 2F 09 03 25 C0 01 89 80 00 80 00 FF FF FF 00 00 00 00 00 02 18 (#data=25)
 n

System Logging set to None

```

stravinci, 2020-03-08:
@proddy my first use case to use EMS is monitoring, based on this maybe I will want to do some operations  but as I wrote earlier setting Warm Water is on low priority.

I want to setup valve on floor loops and radiators then use EMS to be sure that boiler will turn off heating and pump to do not broke it.

norberts1, 2020-03-08:
@stravinci thanks for your test.
It seems that we have got what we want.
your command was:
`send 8B 08 35 01 00 00 **38**` and the response is:
`(00:10:25.110) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 **38** 01 ...` and
`(00:10:34.858) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 **38** 01 ...`
@proddy it seems that command works for that FW120 !?
Anyway, the goal could be: somebody want to have this future.

proddy, 2020-03-08:
thanks @norberts1 ! I had already the command `boiler wwonetime on/off` implemented but only tested on a Buderus RC20 thermostat. The values I used were:
```
 on  = 0x22 = ‭00100010‬
 off = 0x02 = 00000010
```
where in your case you used:
```
 on  = 0x38 = ‭00111000
```

so guessing its bit5 as you described in your document.

The only thing I was missing is to handle both the first and second warm water systems.

‬

norberts1, 2020-03-09:
@proddy no, isn't bit 5. The command has to be seen as:
8B := MSB=1 | `<Servcie-Key>`
08 := device-ID--> boiler with it's electronic
35 := type or MessageID=53
0**1** := **Offset** for this Message
00 := Value for forced 'WW-Status2' -->> 53_**1**_y
00 := Value for forced 'Reduzierung Sollwert' -->> 53_2_0
**38** := Value for desired WW-temperature -->> 53_3_0
That 0x38 := 56C desired WW-temperature is that at least we want to have.
Other values could be tested as well.
I'll think it whasn't foreseen to have two different desired WW-temperature's between first and second warm water system.
Keep in mind that this commands/messages are a bit older one's, older then 10 years. I'm thinking on that commands/types < 0xF0 (less then 240).
That is the reason for extending that command-range (EMSxy)  with that unlucky `0xFF <offset> EMS(High-byte) EMS(Low-byte)`  for Junkers/Buderus/Bosch.
In that 'extended' range it is perhaps a setup available for the first and second WW system, I don't know.
Attached that command-sequence we talking about:
[image]

norberts1, 2020-03-10:
@proddy regarding that used Offset=x it's a very common feature used by Junkers and I'm sure Buderus too. So it's possible to send that command above in an other version like:
8B := MSB=1 | `<Servcie-Key>`
08 := target device-ID--> boiler with it's electronic
35 := type or MessageID=53
03 := Offset for this Message
38 := Value for desired WW-temperature -->> 53_3_0
So this command starts with that offset:=3 -->>(Byte#7 in that complete message above) and end's with `<CRC> and <break>`.
Perhaps it's a good idea to check this too.

proddy, 2020-03-10:
yes, offset works in the same way.

The thing is, in EMS-ESP the command I use to set the boiler warm water temp uses telegram type 0x33 (UBAParameterWW). For example to change the boiler temp to 56C it would be:

`8B 08 33 02 38`

offset 2 is the DWH temperature. I took it from this wiki a long time ago: https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme#ubaparameterww

What's the difference between that and your `B8 08 35 03 28` ?

norberts1, 2020-03-10:
@proddy you mentioned the wrong sequence:  `B8 08 35 03 28`, 
it has to be (see above):
`8B 08 35 03 <desired temp>` or `8B 08 35 01 00 00 <desired temp>`.

`...offset 2 is the DWH temperature for 0x33` -->> yes, it's like I wrote it to my document, but it will not work for Junkers for that Fxyz (see test-results above).
So it seems that type=0x33 (MsgID=51) is only foreseen for 'broadcasting' that values and that type=0x35 (MsgID=53) for writing/command values (Junkers specific perhaps).
A bit strange anyway!

proddy, 2020-03-10:
oops, yes excuse the typo but you know what I mean :-)

On Buderus writing to 0x33 is used to change the WW parameters directly on the boiler and used in many situations in my code. It's independent of any thermostat attached.

I haven't tried if writing to 0x35 does the same. I was just a little confused on the difference between the two.

proddy, 2020-03-10:
BTW @norberts1  on a different note do you know of an EMS telegram to alter the reset state as in #306.

norberts1, 2020-03-10:
@proddy regarding that #306 I found also that message:0x05 with many parameters set-/select-able.
But it seems also a bit dangerous telegram using it, if somebody don't know what he is doing.
At least to reset that error-memory there is one placeholder-Byte for it in that message.
What I found is:
```
payload-byte   MsgID  Meaning
0 & 1          5_0_0  Reset-Mode (Manuel, others)
.
.
12             5_12_0  Reset that Error-memory
...
```
So that resulting (untested) command could be:
`8B 00 05 0C <reset-flag>`
I don't know regarding that value of reset-flag, but it should be: 1, or 0xFF.

proddy, 2020-05-16:
implemented in v2, but needs testing.

closed 2020-05-16 by proddy

stravinci, 2020-05-16:
Hi, I need to build v2 or it was released anywhere?

proddy, 2020-05-16:
still in alpha and being tested (you'll find it in the v2 branch)