emsesp/EMS-ESP #342: 1.9.5b49 - MQTT Heartbeat changed

giovanne123, 2020-03-05:
Is it right in new version 1.9.5bxx that within the Heartbeat the e.g. Firmware and IP are no longer send via MQTT? (and MQTTdisconnects gets mqttpublishfails) 

MyEsp.cpp:
EMS-ESP version: 1.9.5b49

```
        rootHeartbeat["rssid"]            = getWifiQuality();
        rootHeartbeat["load"]             = getSystemLoadAverage();
        rootHeartbeat["uptime"]           = _getUptime();
        rootHeartbeat["freemem"]          = mem_available;
        rootHeartbeat["tcpdrops"]         = _getSystemDropoutCounter();
        rootHeartbeat["mqttpublishfails"] = _mqtt_publish_fails;
```

EMS-ESP version: 1.9.4
```
        rootHeartbeat["version"]         = _app_version;
        rootHeartbeat["IP"]              = WiFi.localIP().toString();
        rootHeartbeat["rssid"]           = getWifiQuality();
        rootHeartbeat["load"]            = getSystemLoadAverage();
        rootHeartbeat["uptime"]          = _getUptime();
        rootHeartbeat["freemem"]         = mem_available;
        rootHeartbeat["MQTTdisconnects"] = _getSystemDropoutCounter();
```

If yes, I have to adjust my Homeassistant config.

proddy, 2020-03-05:
yes, that's correct. I think this is the most important information. Sending the IP and version every minute is not really useful and waste of wifi traffic. `tcpdrops` are any network outages (wifi, mqtt) and `mqttpublishfails` was new as lots of people were seeing MQTT messages failing to be published. If you have any ideas of what else to add let me know!

giovanne123, 2020-03-05:
It's ok for me not to waste the wifi traffic. 

Doc: https://emsesp.github.io/docs/#/MQTT

Maybe it is an idea to send the **Firmware** and **IP** in the "**status**-topic" (or is it "start"? I currently receive status but in linked doc above it is marked as start!?) instead of in "**heartbeat**-topic". Than it is send not that often. It was helpful before to see the used version also directly in HA (esp. when updating often beta version ;-)

PhillyGilly, 2020-03-06:
Here is my update Home Assistant tab 
[image]
from sensor.yaml and lovelace UI 
[194b49.txt](https://github.com/proddy/EMS-ESP/files/4297024/194b49.txt)

proddy, 2020-03-06:
ok, how about the `start` topic so at boot time it'll send for example payload `{"status":"online","version":"1.9.5b49"}`

bbqkees, 2020-03-06:
Yes perfect.

proddy, 2020-03-06:
it'll be in b50

giovanne123, 2020-03-06:
1.9.5b50:
was it planned to be in **start** or in **status** topic?

I think in **start** would be fine, but currently I get it in **status**:
But maybe I'm doing something wrong because also the IP is missing in my topic.

```
home/ems-esp/status -> {"status":"online","version":"1.9.5b50"}
home/ems-esp/start -> 17:56:00 6/Mar/2020
```

I think in this way within status it currently breaks all the availability checks in the sensor definitions:
```
  availability_topic: 'home/ems-esp/status'
  payload_available: "online"
  payload_not_available: "offline"
```

proddy, 2020-03-07:
you're right. I'll change it to `start` and leave the status topic as it was.

closed 2020-03-07 by proddy