emsesp/EMS-ESP #344: Heating Active -doesn`t show correctly 1.9.5.b51

sq4bja, 2020-03-08:
After update to B51 EMS-ESP doesn`t show correctly Heating Active.
I have GB072 with RC310.
Now boiler is heating with flow temperature 32.5C but EMS-ESP shows:

Boiler: Buderus GBx72/Nefit Trendline/Junkers Cerapur/Worcester Greenstar Si/27i (DeviceID: 0x08, ProductID: 123, Version: 06.08)
  Hot tap water: off
  Central heating: off
  Selected flow temperature: 27 C
  Current flow temperature: 32.5 C
  Gas: on
  Boiler pump: on
  Fan: on

Previously on B49 everything works good.
Tom

proddy, 2020-03-08:
That logic for checking if the central heating is running hasn't changed. It's still

"selected flow temp >= 30" where 30 is fixed and set in the file ems.h.

In your case the selected flow temp is 27.

sq4bja, 2020-03-08:
I don't know why, but it happened after the update form B49 to B51. Maybe the controller received some faulty telegrams that cause erroneous readings.

proddy, 2020-03-08:
perhaps. keep an eye on it. if it keeps happening we'll deep dive.

sq4bja, 2020-03-09:
It happened again. I see the flame icon (central heating) on the r310 screen and EMS-ESP shows:
 Boiler: Buderus GBx72/Nefit Trendline/Junkers Cerapur/Worcester Greenstar Si/27i (DeviceID: 0x08, ProductID: 123, Version: 06.08)
  Hot tap water: off
  Central heating: off
 Selected flow temperature: 26 C
  Current flow temperature: 30.8 C

proddy, 2020-03-09:
oh, not good. is "gas on" in `info` too?

sq4bja, 2020-03-09:
Yes:
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off

proddy, 2020-03-09:
looked at the code again and can't figure out why its not changing. Are you compiling the code yourself? If so I'll give you a debug line to add so you can monitor when the change happens.

sq4bja, 2020-03-09:
No - every version comes with official precompiled repository form www panel od EMS-ESP. I reset the boiler. Did not help. Still GAS ON and heating OFF

sq4bja, 2020-03-09:
So, when WW is heating EMS-ESP shows:
 Central heating: active
 Warm Water activated: on

But when WW is ready, boiler is switching Central heating to OFF, but it is still heating (flame on display, and flow temperature is around 31C)

sq4bja, 2020-03-09:
If You have B49 for Wemos D1 mini, please give me a link. I`ll try it again and I will confirm that everything is ok.

proddy, 2020-03-09:
I made a version that dumps out the settings so we can see what is going on. Grab it from [here](https://github.com/proddy/EMS-ESP/releases/download/travis-dev-build/EMS-ESP-1_9_5b55-S-d1_mini.xxx) and upload it. You may need to rename it first to a `.bin` extension. Then start EMS-ESP and capture what you seen on the screen when heating is on/off. You don't need to set verbose logging.

sq4bja, 2020-03-09:
eatingActive=0
XXX debug for sq4bja: wWCurFlow=0, burnGas=0, tapwaterActive=0, selFlowTemp=5, h                                                 eatingActive=0
XXX debug for sq4bja: wWCurFlow=0, burnGas=0, tapwaterActive=0, selFlowTemp=50,                                                  heatingActive=0
XXX debug for sq4bja: wWCurFlow=0, burnGas=0, tapwaterActive=0, selFlowTemp=50,                                                  heatingActive=0
XXX debug for sq4bja: wWCurFlow=0, burnGas=1, tapwaterActive=0, selFlowTemp=50,                                                  heatingActive=1
XXX debug for sq4bja: wWCurFlow=0, burnGas=1, tapwaterActive=0, selFlowTemp=50,                                                  heatingActive=1
XXX debug for sq4bja: wWCurFlow=0, burnGas=1, tapwaterActive=0, selFlowTemp=50,                                                  heatingActive=1
XXX debug for sq4bja: wWCurFlow=0, burnGas=1, tapwaterActive=0, selFlowTemp=50,                                                  heatingActive=1
XXX debug for sq4bja: wWCurFlow=0, burnGas=1, tapwaterActive=0, selFlowTemp=50,                                                  heatingActive=1
XXX debug for sq4bja: wWCurFlow=0, burnGas=1, tapwaterActive=0, selFlowTemp=50,                                                  heatingActive=1
XXX debug for sq4bja: wWCurFlow=0, burnGas=1, tapwaterActive=0, selFlowTemp=50,                                                  heatingActive=1
XXX debug for sq4bja: wWCurFlow=0, burnGas=1, tapwaterActive=0, selFlowTemp=50,                                                  heatingActive=1
XXX debug for sq4bja: wWCurFlow=0, burnGas=1, tapwaterActive=0, selFlowTemp=50,                                                  heatingActive=1
XXX debug for sq4bja: wWCurFlow=0, burnGas=1, tapwaterActive=0, selFlowTemp=50, heatingActive=1
XXX debug for sq4bja: wWCurFlow=0, burnGas=1, tapwaterActive=0, selFlowTemp=50, heatingActive=1
XXX debug for sq4bja: wWCurFlow=0, burnGas=1, tapwaterActive=0, selFlowTemp=50, heatingActive=1
XXX debug for sq4bja: wWCurFlow=0, burnGas=1, tapwaterActive=0, selFlowTemp=50, heatingActive=1
XXX debug for sq4bja: wWCurFlow=0, burnGas=1, tapwaterActive=0, selFlowTemp=50, heatingActive=1
XXX debug for sq4bja: wWCurFlow=0, burnGas=1, tapwaterActive=0, selFlowTemp=50, heatingActive=1

proddy, 2020-03-09:
The data looks ok. And does it match with ‘info’?

sq4bja, 2020-03-09:
Yes. Info shows Heating Active too. The same on Web Status. And MQTT also.

proddy, 2020-03-09:
So it works as expected??

sq4bja, 2020-03-09:
Yes. Thank You. But one thing. If ww is heating, ems-esp shows ww heating indicator ON and central heating ON . I have 1 function boiler with 3d valve. So if ww heating is ON, central heating should be OFF if 3d valve ww is ACTIVE.

proddy, 2020-03-10:
great that it works! thing is, nothing is changed in the code. I just added the single one line debug line for you.

So I like your suggestion. Central Heating should be flagged as "Active" only when
 - selected Flow Temp is > 30 degrees
 - gas/burner is on
 - Warm Water 3-way valve is on

correct?

sq4bja, 2020-03-10:
I don`t know why, but it works. I`ll test it better today. 

Central Heating: 

Temperature trigger is not critical. My flow temperature sometime is
below 30C. 

Central heating should be ACTIVE when: 

- Gas/burner is ON 

- Warm Water 3-way valve is Off 

- (i can`t see on info command now, but in Domoticz i have control named
Boiler Warm Water Heating) -should be OFF 

Central heating should be OFF when: 

- Gas/burner is ON (if GAS is OFF - turn off Central Heating indicator) 

- Warm Water 3-way valve is Active (when GAS is ON)  or OFF (when gas is
off too) 

- (i can`t see currently on info command now, but in Domoticz i have
control named Boiler Warm Water Heating) -should be ON or OFF (when GAS
is OFF) 

So, in RC310 i have a problem with setpoint HC1 thermostat and incorrect
mode name (reverse). I`ll write about it in new issue.

sq4bja, 2020-03-10:
I test it. The problem is flow temperture trigger. Now, winter is wamer, so my boiler set flow temperature to 28-29C, co EMS-ESP can`t indicate control Central Heating. So, as i wrote above - temperature requirement make this problem.

proddy, 2020-03-12:
Using the "Warm water 3-way valve" won't work. It'll always be "off" regardless of whether the heating is active or not.

To fix your problem the solution is to lower the flow temperature threshold to 20 (anything higher than 5 should do it).

I've made the change in b55.

sq4bja, 2020-03-12:
Ok, but im my example Warm Water 3-way valve: is on only, when warm water heating is on. So now in B55 still Centrall Heating is ON at the same time, when Warm Water Heating is ON. But my GB072 heating warm water or central heating - not both.

deqw, 2020-05-26:
If (?) these parameters are mutual exclusive then EMS_Boiler.heatingActive can be Off when  EMS_Boiler.tapwaterActive is on, I think. Something like:

void _checkActive() {
    // hot tap water, using flow to check instead of the burner power
    if (EMS_Boiler.wWCurFlow != EMS_VALUE_INT_NOTSET && EMS_Boiler.burnGas != EMS_VALUE_INT_NOTSET) {
        EMS_Boiler.tapwaterActive = ((EMS_Boiler.wWCurFlow != 0) && (EMS_Boiler.burnGas == EMS_VALUE_BOOL_ON));
    }

    // heating
    if (EMS_Boiler.selFlowTemp != EMS_VALUE_INT_NOTSET && EMS_Boiler.burnGas != EMS_VALUE_INT_NOTSET) {
        EMS_Boiler.heatingActive = !EMS_Boiler.tapwaterActive && ((EMS_Boiler.selFlowTemp >= EMS_BOILER_SELFLOWTEMP_HEATING) && (EMS_Boiler.burnGas == EMS_VALUE_BOOL_ON));
    }
}

proddy, 2020-05-26:
thanks @deqw . added this.

deqw, 2020-06-05:
As a workaround you can set a sensor as follows:

`- platform: template
  sensors:
    heatingactivesensor:
      friendly_name: "Heating Active Sensor"
      value_template: "{{ (is_state('binary_sensor.heating', 'on') or is_state('sensor.boiler_pump', 'on')) and is_state('sensor.warm_water_3_way_valve', 'off') }}"
`

I see home/ems-esp/heating_active only published once (no matter what publish time is set) and on change so when the state is missed you are out of luck, so looking at sensor.boiler_pump at the same time helps with that. Maybe even better leave binary_sensor.heating out at all.

proddy, 2020-06-05:
this has been improved in v2. when there's a change in either the heating or tap water on/off an MQTT message is sent. But your workaround is useful for v1.9

proddy, 2020-06-07:
fix is in v2. needs testing so closing for now.

closed 2020-06-07 by proddy