emsesp/EMS-ESP #345: Boiler temperature is undefined (from > 1.9.5 B42)

sq4bja, 2020-03-08:
After upgrade from b42 to higher i lost boiler temperature. GB072 + R310.
On WWW panel boiler temperature is undefined ℃
On telnet connection it looks like this:

Boiler temperature: ? C
Warm water storage temperature1: ? C
Warm water storage temperature2: 37.6 C

Previously Boiler temperature was the same like WW storage temperature 2.

Tom

sq4bja, 2020-03-08:
[image]

proddy, 2020-03-08:
yes, there's been a lot of back-and-forth on this topic. In previous versions, the boiler temperature was incorrectly specified to keep backward compatibility with Junkers I had to make some changes. It's apparently the temperature of the internal boiler, like on a combi.

I don't think you have a boiler, unless it's external, right?

sq4bja, 2020-03-08:
Yes, i have combi boiler GB072 with external storage tank. Boiler Temperature previously show a temperature od water i storage tank.

proddy, 2020-03-08:
does your RC310 show any boiler temperatures? If so, which ones and what does it call them?

sq4bja, 2020-03-08:
Yes. Look at the picture below. 
[image]

proddy, 2020-03-08:
great and kind you find these back in the `info` command so we can do a mapping?

sq4bja, 2020-03-08:
Ok, i can see it here:
Warm water storage temperature2: 27.3 C
and here:
Warm Water current temperature: 27.3 C

proddy, 2020-03-08:
ok, in 1.9.5b53 I made a few changes and show in the web the DHW temp (which is the same as the warm water current temperature). Also renamed the storage temperature2 and temperature1 to something more meaningful, using 'intern' and 'extern'. hope it makes sense?

sq4bja, 2020-03-08:
Looks OK now. Thanks.
[image]

proddy, 2020-03-08:
I also changed the web interface so you would need to rebuild that too. platformio should have it done it for you, unless you change the scripts.

sq4bja, 2020-03-08:
I`m getting new firmware from Update link on web interface. Is it good?

proddy, 2020-03-08:
Hmmm. I’ll double check.

proddy, 2020-03-09:
think it was just cached in your browser, I couldn't reproduce. Closing this issue now.

closed 2020-03-09 by proddy