emsesp/EMS-ESP #346: Wrong Setpoint temperature and thermostat mode  (RC310)

sq4bja, 2020-03-10:
RC310 with Buderus GB072
When Central Heating Thermostat is in AUTO MODE, and it`s going into night mode (lower temperature mode) HC1 Thermostat still shows:
Setpont room temperature: 21.0C
 
but in RC310 current setpoint is 19,5C

Also thermostat mode switch doen`t work correctly. When i set AUTO - RC310 shows MANUAL, when set DAY - RC310 shows AUTO. 

Also Warm Water Current tap water flow always is 0  (maybe my boiler doesn`t support this function).

proddy, 2020-03-10:
all in 1.9.5b55 correct?

sq4bja, 2020-03-10:
No. It is wrong from a lot of version.

giovanne123, 2020-03-11:
in 1.9.5b54, Buderus G132, RC30 / 2 HCs, ...
seem that I have the same problem, that the Setpoint isn't set right when in night mode:

(Info: Current room temp is ok with ? C because I haven't activated room temp. control for the heating circuits and using outdoor temp. control in my heating system)
```
Heating Circuit 1
   Current room temperature: ? C
   Setpoint room temperature: 24.0 C
   Day temperature: 24.0 C
   Night temperature: 13.0 C
   Vacation temperature: 17.0 C
   Calculated flow temperature: 5 C
   Mode is set to auto
   Mode type is set to night
  Heating Circuit 2
   Current room temperature: ? C
   Setpoint room temperature: 21.0 C
   Day temperature: 21.0 C
   Night temperature: 14.0 C
   Vacation temperature: 17.0 C
   Calculated flow temperature: 5 C
   Mode is set to auto
   Mode type is set to night
```

Also I'm struggling with my HA configs showing the right thermostat data and modes, but maybe I have to adjust my HA configs (climate.yaml, ...) to the newest ems-esp changes? But that is an other task I have to work on...

proddy, 2020-03-12:
@giovanne123 which thermostat do you have? It's can't be an RC30 as I only implemented 1 heating circuit.

@sq4bja I'm trying to reproduce the bug and may need you to do extra testing. I'll let you know.

giovanne123, 2020-03-12:
it is RC30:
```
Thermostat data:
  Thermostat: RC30 (DeviceID: 0x10, ProductID: 67, Version: 02.08)
  Thermostat time is 15:28:05 12/03/2020
  Heating Circuit 1
   Current room temperature: ? C
   Setpoint room temperature: 24.0 C
   Day temperature: 24.0 C
   Night temperature: 13.0 C
   Vacation temperature: 17.0 C
   Calculated flow temperature: 46 C
   Mode is set to auto
   Mode type is set to day
  Heating Circuit 2
   Current room temperature: ? C
   Setpoint room temperature: 21.0 C
   Day temperature: 21.0 C
   Night temperature: 14.0 C
   Vacation temperature: 17.0 C
   Calculated flow temperature: 32 C
   Mode is set to auto
   Mode type is set to day

```

proddy, 2020-05-16:
Think I fixed this in v2, but can't test without someone providing me some telegram logs showing what happens when it goes from auto into night mode and back.

sq4bja, 2020-05-20:
Ok, i can log everything, and if you want, i can do for You the access to get directly data from my boiler (telnet, mqtt and www). GB72 (RC310).

proddy, 2020-05-22:
Thanks @sq4bja , that's kind of you. When I get v2 a little more stable lets experiment with that.

proddy, 2020-06-16:
@sq4bja @giovanne123 I think v2 is stable enough now to try out. I suspect this issue is still there on the RC310 like with 1.9.5 but it'll be easier to debug and trace what is happening. When you have some time could you try it v2? The instructions are in the README on the branch v2. Binaries are in the releases page.

sq4bja, 2020-06-18:
Ok, i'll test it this weekend.

sq4bja, 2020-06-20:
Does esp8266 firmware (BIN) will work with D1mini board?

proddy, 2020-06-20:
> Does esp8266 firmware (BIN) will work with D1mini board?

yes, works with all ESP8266's with 4MB Flash

sq4bja, 2020-06-20:
Ok, it works (i think), but it can`t find EMS device

`ems-esp:/ems$ show
No EMS devices detected. Try using 'scan devices' from the ems menu.

EMS Bus info:
  Tx mode: 1
  Bus protocol: Buderus
  #telegrams received: 338
  #read requests sent: 0
  #write requests sent: 0
  #corrupted telegrams: 0 (0%)
  #tx fails (after 3 retries): 34

Rx Queue is empty

Tx Queue (22 telegrams):
 [158*] READ  Me(0x0B) -> 09(0x09), Version(0x02), data: 20 (offset 0)
 [130 ] READ  Me(0x0B) -> 10(0x10), Version(0x02), data: 20 (offset 0)
 [136 ] READ  Me(0x0B) -> 08(0x08), Version(0x02), data: 20 (offset 0)
 [137 ] READ  Me(0x0B) -> 38(0x38), Version(0x02), data: 20 (offset 0)
 [138 ] READ  Me(0x0B) -> 30(0x30), Version(0x02), data: 20 (offset 0)
 [139 ] READ  Me(0x0B) -> 09(0x09), Version(0x02), data: 20 (offset 0)
 [140 ] READ  Me(0x0B) -> 02(0x02), Version(0x02), data: 20 (offset 0)
 [141 ] READ  Me(0x0B) -> 48(0x48), Version(0x02), data: 20 (offset 0)
 [142 ] READ  Me(0x0B) -> 20(0x20), Version(0x02), data: 20 (offset 0)
 [143 ] READ  Me(0x0B) -> 21(0x21), Version(0x02), data: 20 (offset 0)
 [144 ] READ  Me(0x0B) -> 22(0x22), Version(0x02), data: 20 (offset 0)
 [145 ] READ  Me(0x0B) -> 23(0x23), Version(0x02), data: 20 (offset 0)
 [146 ] READ  Me(0x0B) -> 28(0x28), Version(0x02), data: 20 (offset 0)
 [147 ] READ  Me(0x0B) -> 29(0x29), Version(0x02), data: 20 (offset 0)
 [148 ] READ  Me(0x0B) -> 10(0x10), Version(0x02), data: 20 (offset 0)
 [149 ] READ  Me(0x0B) -> 17(0x17), Version(0x02), data: 20 (offset 0)
 [150 ] READ  Me(0x0B) -> 18(0x18), Version(0x02), data: 20 (offset 0)
 [151 ] READ  Me(0x0B) -> 19(0x19), Version(0x02), data: 20 (offset 0)
 [152 ] READ  Me(0x0B) -> 1A(0x1A), Version(0x02), data: 20 (offset 0)
 [153 ] READ  Me(0x0B) -> 1B(0x1B), Version(0x02), data: 20 (offset 0)
 [154 ] READ  Me(0x0B) -> 11(0x11), Version(0x02), data: 20 (offset 0)
 [159 ] READ  Me(0x0B) -> 08(0x08), ?(0x07), data: 20 (offset 0)
`

sq4bja, 2020-06-20:
Ok, after fiew minutes i get this:
`These EMS devices are currently active:

Thermostat: RC300/RC310/Moduline 3000/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:18.03) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x2A5 0x2B9 0x2A6 0x2BA 0x2A7 0x2BB 0x2A8 0x2BC 0x31D 0x31E
 Subscribed MQTT topics: thermostat_cmd thermostat_cmd_temp thermostat_cmd_mode

EMS Bus info:
  Tx mode: 1
  Bus protocol: Buderus
  #telegrams received: 1150
  #read requests sent: 3
  #write requests sent: 0
  #corrupted telegrams: 0 (0%)
  #tx fails (after 3 retries): 123

Rx Queue is empty

Tx Queue (33 telegrams):
 [18*] READ  Me(0x0B) -> 09(0x09), Version(0x02), data: 20 (offset 0)
 [184 ] READ  Me(0x0B) -> Thermostat(0x10), RCTime(0x06), data: 20 (offset 0)
 [191 ] READ  Me(0x0B) -> 08(0x08), Version(0x02), data: 20 (offset 0)
 [192 ] READ  Me(0x0B) -> 09(0x09), Version(0x02), data: 20 (offset 0)
 [198 ] READ  Me(0x0B) -> Thermostat(0x10), RCTime(0x06), data: 20 (offset 0)
 [199 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A5), data: 20 (offset 0)
 [200 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2B9), data: 20 (offset 0)
 [206 ] READ  Me(0x0B) -> 08(0x08), Version(0x02), data: 20 (offset 0)
 [207 ] READ  Me(0x0B) -> 09(0x09), Version(0x02), data: 20 (offset 0)
 [214 ] READ  Me(0x0B) -> Thermostat(0x10), RCTime(0x06), data: 20 (offset 0)
 [215 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A5), data: 20 (offset 0)
 [216 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2B9), data: 20 (offset 0)
 [222 ] READ  Me(0x0B) -> 08(0x08), Version(0x02), data: 20 (offset 0)
 [223 ] READ  Me(0x0B) -> 09(0x09), Version(0x02), data: 20 (offset 0)
 [230 ] READ  Me(0x0B) -> Thermostat(0x10), RCTime(0x06), data: 20 (offset 0)
 [231 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A5), data: 20 (offset 0)
 [232 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2B9), data: 20 (offset 0)
 [238 ] READ  Me(0x0B) -> 08(0x08), Version(0x02), data: 20 (offset 0)
 [239 ] READ  Me(0x0B) -> 09(0x09), Version(0x02), data: 20 (offset 0)
 [245 ] READ  Me(0x0B) -> Thermostat(0x10), RCTime(0x06), data: 20 (offset 0)
 [246 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A5), data: 20 (offset 0)
 [247 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2B9), data: 20 (offset 0)
 [248 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A7), data: 20 (offset 0)
 [249 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2BB), data: 20 (offset 0)
 [255 ] READ  Me(0x0B) -> 08(0x08), Version(0x02), data: 20 (offset 0)
 [00 ] READ  Me(0x0B) -> 09(0x09), Version(0x02), data: 20 (offset 0)
 [07 ] READ  Me(0x0B) -> Thermostat(0x10), RCTime(0x06), data: 20 (offset 0)
 [08 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A5), data: 20 (offset 0)
 [09 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2B9), data: 20 (offset 0)
 [10 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A7), data: 20 (offset 0)
 [11 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2BB), data: 20 (offset 0)
 [16 ] READ  Me(0x0B) -> 08(0x08), Version(0x02), data: 20 (offset 0)
 [17 ] READ  Me(0x0B) -> 09(0x09), Version(0x02), data: 20 (offset 0)
`

proddy, 2020-06-20:
I see that there are problems with the Tx. Can you remember which tx_mode you had with v1.9?

sq4bja, 2020-06-20:
tx_mode was the same = 1

proddy, 2020-06-20:
that's odd. then try `set tx_mode` to 2 or 4 or 5 and see what happens. This is from the `ems` context/sub-menu. You don't need to restart. The effect should be instant

sq4bja, 2020-06-20:
Ok, everything work perfectly with tx_mode 2.
But if i use multiple telnet connection, when I disconnect second one, the first one works so odd:

ems-esp:/boiler# help
Command not found
ems-esp:/boiler#

I must restart telnet connection to repair it.

proddy, 2020-06-21:
Great that tx_mode 2 works, but also odd since the master is a GB072 (EMS+). @MichaelDvP  any ideas?

And yes, the multi-telnet issue where the commands disappear is a bug. I know why it happens but haven't got around to fixing it yet. I kinda figured not a lot people would have multiple telnet sessions open! I'll add it to the TODO list.

MichaelDvP, 2020-06-21:
@proddy tx_mode 2 is the EMS+ mode, i think this is right. 

@sq4bja if you have 15 minutes time you can try out the timermodes. Goto `ems`, activate `su`, type `show emsbus` to see the requests sent and the fails. Switch `log off`,  than `set tx_mode 10`, wait a minute and check `show emsbus` again. If requests and fails haven't changed, you have waited to short. If you get fails, increase mode (15, 20, 25), wait and check fails. If you get a working mode try to decrease mode in steps 1 until you get fails again, then increase (try larger steps first, my EMS1.0 system works from 6 to 50). Than post the working-range of your system here.

BTW: Are setpoints and modes for your thermostat shown correct now?

proddy, 2020-06-21:
> @proddy tx_mode 2 is the EMS+ mode, i think this is right.

doh, you're right. Too much beer last night.

sq4bja, 2020-06-21:
Hi,
First, show emsbus in tx_mode 2

```
ems-esp:/ems# show emsbus
EMS Bus info:
  Tx mode: 2
  Bus protocol: Buderus
  #telegrams received: 44098
  #read requests sent: 12659
  #write requests sent: 5
  #corrupted telegrams: 0 (0%)
  #tx fails (after 3 retries): 13

Rx Queue is empty
Tx Queue is empty

```

And now tx_mode=10

```
ems-esp:/ems# show emsbus
EMS Bus info:
  Tx mode: 10
  Bus protocol: Buderus
  #telegrams received: 44227
  #read requests sent: 12684
  #write requests sent: 5
  #corrupted telegrams: 0 (0%)
  #tx fails (after 3 retries): 17

Rx Queue is empty

Tx Queue (19 telegrams):
 [115*] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A7), data: 20 (offset 0)
 [83 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2BB), data: 20 (offset 0)
 [84 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A8), data: 20 (offset 0)
 [85 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2BC), data: 20 (offset 0)
 [86 ] READ  Me(0x0B) -> Boiler(0x08), UBAMonitorSlow(0x19), data: 20 (offset 0)
 [87 ] READ  Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: 20 (offset 0)
 [88 ] READ  Me(0x0B) -> Boiler(0x08), UBAParameters(0x16), data: 20 (offset 0)
 [102 ] READ  Me(0x0B) -> Thermostat(0x10), RCTime(0x06), data: 20 (offset 0)
 [103 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A5), data: 20 (offset 0)
 [104 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2B9), data: 20 (offset 0)
 [105 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A6), data: 20 (offset 0)
 [106 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2BA), data: 20 (offset 0)
 [107 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A7), data: 20 (offset 0)
 [108 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2BB), data: 20 (offset 0)
 [109 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A8), data: 20 (offset 0)
 [110 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2BC), data: 20 (offset 0)
 [111 ] READ  Me(0x0B) -> Boiler(0x08), UBAMonitorSlow(0x19), data: 20 (offset 0)
 [112 ] READ  Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: 20 (offset 0)
 [113 ] READ  Me(0x0B) -> Boiler(0x08), UBAParameters(0x16), data: 20 (offset 0)

```

and another tx_mode=15

```
EMS Bus info:
  Tx mode: 15
  Bus protocol: Buderus
  #telegrams received: 44390
  #read requests sent: 12684
  #write requests sent: 5
  #corrupted telegrams: 0 (0%)
  #tx fails (after 3 retries): 40

Rx Queue is empty
[... 40 lines cut ...]
 [252 ] READ  Me(0x0B) -> Boiler(0x08), UBAParameters(0x16), data: 20 (offset 0)
```

So, i don`t know, how to try it is working. My domoticz still getting telegrams with temperatures, but thermostat menu looks so odd:

```
Thermostat: RC300/RC310/Moduline 3000/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:18.03)
  Clock: 15:26:11 21/06/2020
  Heating Circuit 1:
    Current room temperature: 25.6°C
    Setpoint room temperature: 19.5°C
    Mode: auto
    Mode Type: comfort
  Heating Circuit 2:
    Current room temperature: 153.8°C
    Setpoint room temperature: 1.5°C
    Mode: manual
    Mode Type: eco
  Heating Circuit 3:
  Heating Circuit 4:

```
I have only one circuit - circuit 1 - that works OK (i think, it is in summer permanent mode)

Tom

MichaelDvP, 2020-06-21:
Confusing results. Just to make sure: Youre running in mode 2 for a longer time:
```
#read requests sent: 12659
#tx fails (after 3 retries): 13
```
is less than 0.1% fails, not perfect but good.
direct after this output (without waiting)  you set `set tx_mode 10`, and after waiting you get
```
Tx mode: 10
#read requests sent: 12684
#tx fails (after 3 retries): 17
``` 
e.g. 4 fails more, but also 25 successful requests. Confusing: mode 15 have only fails, no successful requests, mode 50 is mixed. Or was the output direct after the mode-change, and the waiting after the output?

> UPDATE: In tx_mode higher then 2 i cant change /boiler/change wwtemp any command doesnt work.

Oh, that's a quick test without waiting.  Can you test some modes between 8 and 25 if a change command works?

circuit 1 values are ok? 
The other circuit should not be displayed and circuit 2 in addition has strange values.

sq4bja, 2020-06-21:
Ok, i`ll test it again, but now EMS-ESB doesn`t answar for telnet connection, and last mqtt telegram goes hour ago.

proddy, 2020-06-21:
also Tom you shouldn't be seeing a HC2 if you don't have one. That's something we need to fix in the software. When you get a chance check with `watch raw 2a6` which will show you info from HC2 and send it over. Thanks for testing!

MichaelDvP, 2020-06-21:
I would prefer a `watch raw 10` to see all thermostat telegrams for all 4 heating circuits and the difference between active hc1, and the inactive hc2, hc3, and hc4.

sq4bja, 2020-06-21:
Ok, another test with tx_mode (just update to newest firmware)
1. After 10 minutes on tx_mode=2

```
 #telegrams received: 396
  #read requests sent: 127
  #write requests sent: 0
  #corrupted telegrams: 0 (0%)
  #tx fails (after 3 retries): 0
Rx Queue is empty
Tx Queue is empty
```

2. After another 5 minutes in tx_mode=8

```
#telegrams received: 572
  #read requests sent: 140
  #write requests sent: 0
  #corrupted telegrams: 0 (0%)
  #tx fails (after 3 retries): 14

Rx Queue is empty

Tx Queue (39 telegrams):
 [14*] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2BB), data: 20 (offset 0)
 [196 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A8), data: 20 (offset 0)
 [197 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2BC), data: 20 (offset 0)
 [209 ] READ  Me(0x0B) -> Boiler(0x08), UBAMonitorSlow(0x19), data: 20 (offset 0)
 [210 ] READ  Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: 20 (offset 0)
 [211 ] READ  Me(0x0B) -> Boiler(0x08), UBAParameters(0x16), data: 20 (offset 0)
 [212 ] READ  Me(0x0B) -> Thermostat(0x10), RCTime(0x06), data: 20 (offset 0)
 [213 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A5), data: 20 (offset 0)
 [214 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2B9), data: 20 (offset 0)
 [215 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A6), data: 20 (offset 0)
 [216 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2BA), data: 20 (offset 0)
 [217 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A7), data: 20 (offset 0)
 [218 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2BB), data: 20 (offset 0)
 [219 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A8), data: 20 (offset 0)
 [220 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2BC), data: 20 (offset 0)
 [233 ] READ  Me(0x0B) -> Boiler(0x08), UBAMonitorSlow(0x19), data: 20 (offset 0)
 [234 ] READ  Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: 20 (offset 0)
 [235 ] READ  Me(0x0B) -> Boiler(0x08), UBAParameters(0x16), data: 20 (offset 0)
 [236 ] READ  Me(0x0B) -> Thermostat(0x10), RCTime(0x06), data: 20 (offset 0)
 [237 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A5), data: 20 (offset 0)
 [238 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2B9), data: 20 (offset 0)
 [239 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A6), data: 20 (offset 0)
 [240 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2BA), data: 20 (offset 0)
 [241 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A7), data: 20 (offset 0)
 [242 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2BB), data: 20 (offset 0)
 [243 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A8), data: 20 (offset 0)
 [244 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2BC), data: 20 (offset 0)
 [255 ] READ  Me(0x0B) -> Boiler(0x08), UBAMonitorSlow(0x19), data: 20 (offset 0)
 [00 ] READ  Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: 20 (offset 0)
 [01 ] READ  Me(0x0B) -> Boiler(0x08), UBAParameters(0x16), data: 20 (offset 0)
 [02 ] READ  Me(0x0B) -> Thermostat(0x10), RCTime(0x06), data: 20 (offset 0)
 [03 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A5), data: 20 (offset 0)
 [04 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2B9), data: 20 (offset 0)
 [05 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A6), data: 20 (offset 0)
 [06 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2BA), data: 20 (offset 0)
 [07 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A7), data: 20 (offset 0)
 [08 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2BB), data: 20 (offset 0)
 [09 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Monitor(0x2A8), data: 20 (offset 0)
 [10 ] READ  Me(0x0B) -> Thermostat(0x10), RC300Set(0x2BC), data: 20 (offset 0)
```
And ems-esp:/boiler# change wwtemp 43   -  doesn't work.

3. After another 5 minutes in tx_mode=10

```
[... 306 lines cut ...]
000+00:59:02.807 N 176: [telegram] [DEBUG] Tx: 0B 90 FF 00 20 01 A6 20 14
000+00:59:03.302 N 177: [telegram] [DEBUG] Tx: 0B 90 FF 00 20 01 A6 20 14
000+00:59:03.355 N 178: [telegram] Rx: 10 0B FF 00 01 A6 91
000+00:59:03.825 N 179: [telegram] [DEBUG] Tx: 0B 90 FF 00 20 01 BA 20 2C
000+00:59:04.223 N 180: [telegram] [DEBUG] Tx: 0B 90 FF 00 20 01 A7 20 16
000+00:59:04.547 N 181: [telegram] [DEBUG] Tx: 0B 90 FF 00 20 01 BB 20 2E
000+00:59:05.020 N 182: [telegram] [DEBUG] Tx: 0B 90 FF 00 20 01 A8 20 08
000+00:59:05.542 N 183: [telegram] [DEBUG] Tx: 0B 90 FF 00 20 01 A8 20 08
000+00:59:05.941 N 184: [telegram] [DEBUG] Tx: 0B 90 FF 00 20 01 BC 20 20
000+00:59:06.985 N 185: [telegram] [DEBUG] Tx: 0B 90 FF 00 20 01 BC 20 20
000+01:00:00.306 N 186: [telegram] [DEBUG] Tx: 0B 88 19 00 20 D0
000+01:00:00.654 N 187: [telegram] [DEBUG] Tx: 0B 88 33 00 20 78
000+01:00:01.077 N 188: [telegram] [DEBUG] Tx: 0B 88 16 00 20 EC
000+01:00:01.450 N 189: [telegram] [DEBUG] Tx: 0B 90 06 00 20 6C
000+01:00:01.973 N 190: [telegram] [DEBUG] Tx: 0B 90 FF 00 20 01 A5 20 12
000+01:00:02.421 N 191: [telegram] [DEBUG] Tx: 0B 90 FF 00 20 01 B9 20 2A
000+01:00:02.819 N 192: [telegram] [DEBUG] Tx: 0B 90 FF 00 20 01 A6 20 14
000+01:00:02.871 N 193: [telegram] Rx: 10 0B FF 00 01 A6 91
000+01:00:03.216 N 194: [telegram] [DEBUG] Tx: 0B 90 FF 00 20 01 BA 20 2C
000+01:00:03.615 N 195: [telegram] [DEBUG] Tx: 0B 90 FF 00 20 01 A7 20 16
000+01:00:04.015 N 196: [telegram] [DEBUG] Tx: 0B 90 FF 00 20 01 BB 20 2E
000+01:00:04.510 N 197: [telegram] [DEBUG] Tx: 0B 90 FF 00 20 01 BB 20 2E
000+01:00:05.058 N 198: [telegram] [DEBUG] Tx: 0B 90 FF 00 20 01 A8 20 08
000+01:00:05.406 N 199: [telegram] [DEBUG] Tx: 0B 90 FF 00 20 01 BC 20 20
```

MichaelDvP, 2020-06-25:
@sq4bja Have you tested a27 in the meantime? 
Is state stable now, setpoint always correct, does it shows only active hcs?

MichaelDvP, 2020-06-25:
I've meant the actual software version v2.0.0a27.

sq4bja, 2020-06-25:
Ok. i'll test it today. Is it possible to upgrade firmware remotely? No i'm using serial port with esptool software.

MichaelDvP, 2020-06-25:
Sure, it's descibed on the [v2 page](https://github.com/proddy/EMS-ESP/tree/v2)
Uploading over WiFi: `espota.py --debug --progress --port 8266 --auth neo -i <IP address> -f <firmware.bin>`

sq4bja, 2020-06-26:
Ok, it looks better

```
Thermostat: RC300/RC310/Moduline 3000/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:18.03)
  Clock: 00:19:46 27/06/2020
  Heating Circuit 1:
    Current room temperature: 25.4°C
    Setpoint room temperature: 0.0°C
    Mode: auto
    Mode Type: eco
    Program is set to Summer mode
    Day temperature: 21.0°C
    Night temperature: 19.5°C
    Target flow temperature: 0°C

```

proddy, 2020-06-27:
great, so all working on the RC310 now or is there still weird things happening?

proddy, 2020-07-06:
if it still shows issues please re-open this.

closed 2020-07-06 by proddy