emsesp/EMS-ESP #348: on a HA restart request a mqtt-publish for topic "start" from the EMS-ESP

giovanne123, 2020-03-12:
Question, is it possible from HA on a HA restart to request a mqtt-publish for topic "start" from the EMS-ESP? If yes, how?

Because topic start is only published once on EMS-ESP booting/restarting.

If I have to restart HA because of an update, I currently have to reboot the EMS-ESP also to retrieve the start topic, otherwise I will stay with "Unknown" status within HA. 

(currently on EMS-ESP 1.9.5b54; HA 0.106.6)

proddy, 2020-03-12:
I need to change the 'start' payload and set the MQTT retain flag, which means that message will always be present in the MQTT broker. I'll make the change now.

giovanne123, 2020-03-12:
Thanks,
for my understanding, is the Retain Flag within the Web-UI meant or does this have a different meaning? I tried to play with it before ;-)

proddy, 2020-03-12:
yes, similar but that setting is only for the specific EMS-ESP topics (like boiler, thermostat etc) and not for the system topics (like start, status)

proddy, 2020-03-12:
fixed in b55.

closed 2020-03-12 by proddy

giovanne123, 2020-03-12:
thanks working :-)