emsesp/EMS-ESP #349: burner minimum power

lsw271, 2020-03-14:
Hello,
I've got too big condensing boiler to my house. Standard burner modulation is between 6,6 - 24kw on central heating and up to 30kw on warm water.
Do you think it's possible to manage "Burner minimum power on heating" parameter? It's in the service menu of my junkers cerapur. 
Now i've got 6.6kw and its the minimum value which i can enter via menu. I would like to change this to lower value. Currently we've got "burner max power" and "burner current power" expressed as a percentage. 
When i changed burner minimum power in the service menu i've got in telnet sth like this:
(00:14:33) Controller -> UBAMaster, type 0x26, telegram: 89 08 26 03 16 (CRC=90) #data=1
(00:14:33) UBAMaster -> All, type 0x26, telegram: 88 00 26 00 01 05 00 16 00 1E 52 64 (CRC=CD) #data=8
(00:14:37) UBAMaster -> All, type 0x2A, telegram: 88 00 2A 00 00 00 00 00 03 00 00 00 E3 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=99) #data=21
(00:14:37) UBAMaster -> All, type 0x2A, telegram: 88 00 2A 00 00 00 00 00 03 00 00 00 E3 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=99) #data=21
(00:14:37) UBAMaster -> All, type 0x34, telegram: 88 00 34 00 31 01 E9 01 E9 A1 00 00 03 00 00 27 ED 00 20 36 00 80 00 (CRC=D6) #data=19

but i don't know how to decode this and what should i send by telnet to check it (the lower value)

proddy, 2020-03-15:
Hi @lsw271 ,  not sure but we could try. When you changed the min power on the Cerapur's menu which value did you select? Was it 22kw by any chance?

If so it looks like the 0x26 telegram could be the one we're after.  And as an example setting to 20kw would be the command `send 89 08 26 03 14` from the telnet console. See if that works, but be careful!

lsw271, 2020-03-16:
@proddy I've done some tests in service menu (with four parameters that are there)and looked up to logs:

Max central heating power (set to 24kw), hex (64) to decimal is 100, maybe as a percent 
```
Controller -> UBAMaster, type 0x26, telegram 89 08 26 07 64 (CRC=EA) #data=1
Controller -> UBAMaster, type 0x27, telegram 89 88 27 0D 01 (CRC=FB) #data=1
UBAMaster -> Controller, type 0x27, telegram 88 09 27 0D 64 (CRC=E2) #data=1
UBAMaster -> All, type 0x26, telegram 88 00 26 00 01 05 00 18 00 1E 52 64 (CRC=2D) #data=8
```
hex to decimal:
88 00 26 00 01 05 00 **18 00 1E 52 64** > 136 | 0 | 38 | 0 | 1 | 5 | 0 | **24 | 30 | 82 | 100** > 24kw | 30kw | 82degrees | 100%

Max warm water power (set to 30kw) hex (64) to decimal is 100
```
Controller -> UBAMaster, type 0x27, telegram 89 08 27 0D 64 (CRC=FA) #data=1
UBAMaster -> All, type 0x27, telegram 88 00 27 00 00 00 03 00 00 00 00 00 00 00 64 28 3C 64 (CRC=65) #data=14
Thermostat -> All, type 0x01A5, telegram 90 00 FF 04 01 A5 26 (CRC=49) #data=1
```

Max flow temperature (set to 82 degrees), hex (52) to decimal is 82
```
Controller -> UBAMaster, type 0x26, telegram: 89 08 26 06 52 (CRC=DE) #data=1
Controller -> UBAMaster, type 0x27, telegram: 89 88 27 0D 01 (CRC=FB) #data=1
UBAMaster -> Controller, type 0x27, telegram: 88 09 27 0D 64 (CRC=E2) #data=1
UBAMaster -> All, type 0x26, telegram: 88 00 26 00 01 05 00 18 00 1E 52 64 (CRC=2D) #data=8
UBAMaster -> All, type 0x18, telegram: 88 00 18 00 26 01 8D 21 00 01 03 24 C0 80 00 01 F0 01 2B 00 00 FF 00 00 00 CC 00 00 00 (CRC=21) #data=25
```

Min central heating and warm water power (set to 6,6kw), hex (16) to decimal is 22. It looks like a 22% of 30kw not 24kw because its a minimum power of both so it will be correct
```
Controller -> UBAMaster, type 0x26, telegram: 89 08 26 03 16 (CRC=90) #data=1
UBAMaster -> All, type 0x26, telegram: 88 00 26 00 01 05 00 16 00 1E 52 64 (CRC=CD) #data=8
```
the second telegram looks like in max central heating power but with another CRC? what is the CRC=CD / CRC = 2D?

When I'm trying to send telegram nothing happens (there is a record in log but in the menu i've got value without changes):
```
System Logging set to Verbose
send 89 08 26 03 16

(00:37:46) Sending raw: 89 08 26 03 16
(00:37:46) Controller -> UBAMaster, type 0x26, telegram: 89 08 26 03 16 (CRC=90) #data=1
(00:37:47) UBAMaster -> All, type 0x34, telegram: 88 00 34 00 2D 01 F0 01 F0 A1 00 00 03 00 00 28 01 00 20 41 00 80 00 (CRC=A2) #data=19
Publishing boiler data via MQTT
(00:37:47) UBAMaster -> All, type 0x18, telegram: 88 00 18 00 26 01 2B 21 00 01 03 20 C0 80 00 01 F0 01 2C 00 00 FF 00 00 00 CC 00 00 00 (CRC=96) #data=25
Publishing boiler data via MQTT
(00:37:48) UBAMaster -> All, type 0x2A, telegram: 88 00 2A 00 00 00 00 00 03 00 00 00 E3 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=99) #data=21
(00:37:48) Thermostat -> UBAMaster, type 0x1A, telegram: 90 08 1A 00 26 64 64 (CRC=FB) #data=3
(00:37:48) Thermostat -> All, type 0x0167, telegram: 90 00 FF 00 01 67 00 00 (CRC=BD) #data=2
(00:37:48) Thermostat -> All, type 0x021D, telegram: 90 00 FF 00 02 1D 00 00 0A 07 (CRC=5A) #data=4
(00:37:53) Thermostat -> UBAMaster, type 0x35, telegram: 90 08 35 00 11 01 (CRC=A9) #data=2
(00:37:53) Thermostat -> UBAMaster, type 0x35, telegram: 90 08 35 03 31 (CRC=72) #data=1
(00:37:54) Thermostat -> All, type 0x01A5, telegram: 90 00 FF 0D 01 A5 01 42 00 62 (CRC=5A) #data=4
Publishing thermostat data via MQTT
```

I think I need to send the second telegram (88 00 26 00 01 05 00 16 00 1E 52 64). I thought it will sends itself as an answer to my telegram but not. It is possible to send from telnet as UBAMaster not controller? before i will try to send the second telegram i would rather ask you, or maybe the sender its defined by telegram? but what with the CRC?

glitter-ball, 2020-03-19:
Changing the burner maximum power is easy - the software just limits the fan speed and, hence, the gas flow. There's no reason not to be able to drop the maximum output all the way down to the minimum.

Changing the minimum output power is a whole different matter. The boiler manufacturer will have proved the combustion is stable and safe at the boiler's stated minimum power - indeed, flue gases will be checked separately at maximum and minimum output to prove this when the boiler is serviced.

Attempting to reduce the output below the intended minimum sounds a bad idea. There's no guarantee that the gas valve will mix the right amount of air at lower settings. Combustion on the burner surface may be unstable or uneven. The fan motor might overheat. All sorts of bad things could happen, so I wouldn't risk it.

I have a 24kW boiler which will modulate down to 30% or 7.2kW. In mild weather, that's too much for the house and the boiler inevitably cycles on and off. There's no easy way to avoid this. Maybe I should have bought a smaller boiler - 18kW would have modulated down to 5.4kW, but is that 1800W going to make a massive difference? Probably not. There will always be times of year when the heat load is less than the boiler's minimum output.

On my boiler, the thermostat doesn't dictate boiler power, anyway. It sends a target flow temperature and the boiler modulates to provide that within its limits.

closed 2020-10-09 by proddy