emsesp/EMS-ESP #354: No tapwater active, while using hot water

FireWizard52, 2020-03-25:
Boiler Type: Nefit Smartline 24 HRC
Thermostat: Moduline 400

It has been noted that the Hot Tap Water indication is not activated, if hot water is used, e.g. during a shower. Instead of the expected Hot Tap Water indication, the Central heating is activated.

Could be a bug.

proddy, 2020-03-26:
The logic is quite simple. It uses the boilers current flow temp and checks if the gas/burner is on. 

- which version of EMS-ESP are you using?
- what is the value of the flow temp (do `info` and look for "Warm Water current tap water flow xx l/min")

FireWizard52, 2020-03-26:
Hello @proddy.

Following answers:

- which version of EMS-ESP are you using?
  Currently version 1.9.5b55-S

- what is the value of the flow temp 

During Tap water use: Warm Water current tap water flow: 0.0 l/min and Gas: on

The Central heating was activated.
As you can see below the room temperature *because of the sun) was much higher than the setpoint.

It was very unlikely, that at the moment I opened the hot water tap, that the heating was also activated.
Thermostat data:
  Thermostat: Moduline 400 (DeviceID: 0x10, ProductID: 78, Version: 03.07)
  Thermostat time is 16:16:44 26/03/2020
  Heating Circuit 1
   Current room temperature: 24.6 C
   Setpoint room temperature: 22.0 C
   Mode is set to auto

For completeness I have attached the complete "info" file.

file:///home/herman/Documents/EMS-ESP_info.txt

Regards

proddy, 2020-03-26:
aw, I see the problem. The EMS-ESP web UI picked up the wrong image. Please discard this special version and upload again. The correct version should be 1.9.5b58

FireWizard52, 2020-03-26:
Hello proddy,

I upgraded to version 1.9.5b58.
However nothing has been changed. Output of info is more or less identical with the previous version, while tapping hot water. So I don't think it is the solution.
The only thing that is different, is that no more debug info is produced.

If you need more info, please let me know.

Regards

proddy, 2020-03-26:
It's because of this:

"Warm Water current tap water flow: 0.0 l/min"

that value is coming from a telegram with type 0x34. Can you do the following and paste back the log so I can see why the value is 0?
`log v`
`boiler read 34`

FireWizard52, 2020-03-26:
Hello proddy,

See below. I hope it is sufficient.
Otherwise a new try.

``Requesting type UBAMonitorWWMessage(0x34) from dest 0x08
(22:21:25) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 03 00 7B 7B 0A 10 65 60 02 16 00 5A 01 BF 01 8A 0F 3D 48 00 C9 00 03 00 (#data=25)
(22:21:26) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 03 0D 7B 7B 0A 10 65 60 02 1C 00 5A 01 C5 01 8F 0F 3D 48 00 C9 00 03 00 (#data=25)
(22:21:27) Sending read of type 0x34 to 0x08, telegram: 0B 88 34 00 20 
(22:21:27) Boiler -> Me, UBAMonitorWWMessage(0x34), telegram: 08 0B 34 00 39 02 21 00 5A 09 00 00 02 00 01 6F F5 02 38 E7 00 (#data=17)
(22:21:27) Boiler -> All, UBADevices(0x07), telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(22:21:27) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 03 1C 7B 77 0A 10 65 60 02 26 00 5A 01 CC 01 91 0F 3D 48 00 C9 00 03 00 (#data=25)
(22:21:28) Boiler -> All, UBADevices(0x07), telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(22:21:28) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 03 2A 7B 6A 0A 10 65 60 02 31 00 5A 01 D5 01 8B 0F 3D 48 00 C9 00 03 00 (#data=25)
(22:21:29) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 03 37 7B 5D 0A 10 65 60 02 3C 00 5A 01 DE 01 8A 0F 30 59 00 CC 00 03 00 (#data=25)
(22:21:30) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 03 42 7B 52 0A 10 64 60 02 47 00 5A 01 E8 00 C9 0F 30 59 00 CC 00 01 00 (#data=25)
(22:21:31) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 08 00 34 00 39 02 4B 00 5A 09 00 00 02 00 01 6F F5 02 38 E7 00 (#data=17)
(22:21:31) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 03 4C 7B 00 02 10 64 60 02 54 00 5A 01 F2 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:21:32) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 03 51 7B 00 02 10 64 60 02 5D 00 5A 01 F9 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:21:32) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 10 08 1A 00 00 00 00 00 (#data=4)
(22:21:33) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 03 52 7B 00 02 10 64 60 02 6A 00 5A 02 04 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:21:34) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 03 4F 7B 00 02 10 64 60 02 73 00 5A 02 0C 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:21:36) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 03 3E 7B 00 02 10 64 60 02 88 00 5A 02 1D 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:21:37) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 03 2E 7B 00 02 10 64 60 02 90 00 5A 02 24 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:21:38) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 03 1C 7B 00 02 10 64 60 02 98 00 5A 02 2B 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:21:39) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 03 09 7B 00 02 10 64 60 02 9D 00 5A 02 31 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:21:40) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 02 F6 7B 00 02 10 60 60 02 A0 00 5A 02 35 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:21:41) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 08 00 34 00 39 02 A2 00 5A 09 00 00 02 00 01 6F F5 02 38 E7 00 (#data=17)
(22:21:41) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 02 E0 7B 00 02 10 60 60 02 A2 00 5A 02 37 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:21:42) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 02 D1 7B 00 02 10 60 60 02 A0 00 5A 02 37 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:21:43) Thermostat -> All, RCTime(0x06), telegram: 10 00 06 00 14 03 16 1A 15 24 03 00 00 00 00 00 00 (#data=13)
(22:21:43) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 02 C3 7B 00 02 10 60 60 02 9E 00 5A 02 36 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:21:44) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 02 B3 7B 00 02 10 60 60 02 99 00 5A 02 33 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:21:45) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 02 A7 7B 00 02 10 60 60 02 93 00 5A 02 2E 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:21:46) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 02 9A 7B 00 02 10 60 60 02 8C 00 5A 02 29 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:21:47) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 02 90 7B 00 02 10 60 60 02 84 00 5A 02 21 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:21:47) Thermostat -> All, Type(0xA2), telegram: 10 00 A2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (#data=13)
(22:21:48) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 02 88 7B 00 02 10 60 60 02 7B 00 5A 02 19 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:21:49) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 02 7F 7B 00 02 10 60 60 02 72 00 5A 02 11 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
Fetching data from EMS devices
Requesting type RC30StatusMessage(0x41) from dest 0x10
Requesting type RC30Set(0xA7) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
Publishing MQTT data for: boiler
[... cut ...]
(22:22:25) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 03 56 7B 00 02 10 64 60 02 5B 00 5A 01 F4 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:22:28) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 03 4A 7B 00 02 10 64 60 02 79 00 5A 02 0D 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:22:29) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 03 3C 7B 00 02 10 64 60 02 83 00 5A 02 14 00 00 0F 30 59 00 CC 00 01 00 (#data=25)
(22:22:30) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 52 03 2A 7B 00 02 10 64 60 02 8C 00 5A 02 1C 00 00 0F 30 59 00 CC 00 01 00 (#data=25)

proddy, 2020-03-26:
The value of the water flow is 0. It's coming from: 
```
telegram: 08 0B 34 00 39 02 21 00 5A 09 00 00 02 00 01 6F F5 02 38 E7 00 (#data=17)
                                                 ^^
```
I don't know why. Just run the tap and check with `info` to see the "Warm Water current tap water flow" values changes.

FireWizard52, 2020-03-26:
Hello proddy,

As expected and as I saw before. It is not changing and stays 0.0 l/min.
Strange as hot water is coming out of the tap.

Regards

FireWizard52, 2020-03-26:
Hi
I noted that the counter Warm Water # starts: counts up. So it is detected in some way

proddy, 2020-03-26:
strange. need to think about this.

kwertie01, 2020-04-06:
Hi,
I can confirm that I have the same issue. I have the exact the same boiler and thermostat.
The flow rate shows always 0.0 l/min and looking at the 0x34 message the data is 00.
It looks like this data is not available with the Nefit Smartline 24 HRC?
Maybe it is possible to use other parameters to determine tap water active like a combination of gas/pump and 3-way valve.

proddy, 2020-04-06:
yes, looks like we need to find a better solution but would need your help. Could you check the following if you have time:
 - when the hot water is running is it shown on the boiler's front-panel and/or the thermostat? This would mean it should be easily detectable.
 - with the hot water running both on and off check with `info` to see the values of 'warm water circulation pump available/type/freq' and the 'warm water 3-way valve'.
 - and lastly, if your German is good take a look at this [PDF](https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT3-Bus_Telegramme.pdf), scroll down to the 0x18 telegram (Message-ID:24_x_y) and see if there's is something we can use in that telegram. It's from Junkers/HT3 but also compatible with EMS+ newer type Buderus boilers.

FireWizard52, 2020-04-06:
Hello,

I have installed a new Moduline 400 thermostat March 14, 2020. A week earlier, I have connected the hardware. In that time I had a (non supported) thermostat Moduline  30. This one was connected with a, so called, RCC module withe the EMS bus. This RCC module acts as a converter between the old iRT protocol (used by the Moduline 30) and the EMS protocol. This combination is not recognized by the EMS-ESP software.

While the Moduline 400 has no indication on its display ,while heating and/or tap water is active, the Moduline 30 has a radiator symbol for heating and a tap for tap water active.
So the boiler indicated on the Moduline 30 whether heating was active, or tap water was active.

In my opinion a telegram exists, if tap water is active.

Regarding the questions.

> when the hot water is running is it shown on the boiler's front-panel and/or the thermostat? This would mean it should be easily detectable.

It is not shown by a LED on the front panel, but not on the Moduline 400.
It is indicated in the Moduline 30, with RCC module.

> with the hot water running both on and off check with info to see the values of 'warm water circulation pump available/type/freq' and the 'warm water 3-way valve'.

See below:

No hot water running:

```
EMS-ESP system status:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: enabled, Shower Alert: disabled
  Saved known device IDs: none

EMS Bus status:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=7975, # CRC errors=21
  Tx: Last poll=2.989 seconds ago, # successful write requests=0

Boiler data:
  Boiler: Sieger BK13,BK15/Nefit Smartline/Buderus GB1x2 (DeviceID: 0x08, ProductID: 64, Version: 03.06)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: on
  Warm Water circulation pump type: charge pump
  Warm Water circulation pump freq: 2x3min
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 55 C
  Warm Water desinfection temperature: 70 C
  Warm Water circulation active: off
  Warm Water set temperature: 57 C
  Warm Water current temperature: 53.8 C
  Warm water temperature (intern): 53.7 C
  Warm water temperature (extern): 57.0 C
  Warm Water current temperature (extern): 57.0 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 146340 times
  Warm Water active time: 65 days 15 hours 9 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 7 C
  Current flow temperature: 61.4 C
  Max boiler temperature: 62.2 C
  Return temperature: 43.8 C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: 1.3 bar
  System service code: 0H (203)
[... 76 lines cut ...]
  Pump modulation: 100 %
  Burner # starts: 238876 times
  Total burner operating time: 490 days 7 hours 20 minutes
  Total heat operating time: 424 days 16 hours 11 minutes
  Total UBA working time: 2729 days 22 hours 36 minutes

Thermostat data:
  Thermostat: Moduline 400 (DeviceID: 0x10, ProductID: 78, Version: 03.07)
  Thermostat time is 12:27:59 06/04/2020
  Heating Circuit 1
   Current room temperature: 24.1 C
   Setpoint room temperature: 22.0 C
   Mode is set to auto

```

You will see that at the moment that hot water is running, the central heating is active.

a

> and lastly, if your German is good take a look at this PDF, scroll down to the 0x18 telegram (Message-ID:24_x_y) and see if there's is something we can use in that telegram. It's from Junkers/HT3 but also compatible with EMS+ newer type Buderus boilers.

I will study it and will come back with an answer.

Regards

kwertie01, 2020-04-08:
I had some time to dive into this issue and gather some more information.

> * when the hot water is running is it shown on the boiler's front-panel and/or the thermostat? This would mean it should be easily detectable.

Indeed as FireWizard52 is saying: when using Warm Water, it is shown by a led indicator at the boiler itself but not on the thermostat.

> * with the hot water running both on and off check with `info` to see the values of 'warm water circulation pump available/type/freq' and the 'warm water 3-way valve'.

I added the by MQTT published values into a influxDB and created some graphs with Grafana.  I made a screenshot of the two tests I did:

1. Start Warm water usage at 19:15 and stop at 19:16
2. Start Central Heating at 19:30 and stop at 19:31

[image]

Looking at this picture it looks like that a combination of Gas:ON and 3-way-valve:ON means that you are using warm water. When using Central heating, only the Gas is ON.

Unfortunately this is not fool proof. Once a while (in my picture below around 80 minutes) the boiler is heating up the water in the storage tank to keep it on the set temperature. So at those times (14:06, 15:26, 16:46, 18:14...) both gas and 3-way-valve are ON bu no warm water usage from the user perspective. This is only to bring the temp in the boiler up to the set level.

[image]
 

> * and lastly, if your German is good take a look at this [PDF](https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT3-Bus_Telegramme.pdf), scroll down to the 0x18 telegram [...]

I took a good look at the document. Very interesting but I made an English translation first so it was easier to read :-)  [EMS Telegram 0x18-v1.0.docx](https://github.com/proddy/EMS-ESP/files/4451254/EMS.Telegram.0x18-v1.0.docx) (maybe it is useful for others?)

For the two tests I captured the raw messages of type 0x18:
[18-central-heating.txt](https://github.com/proddy/EMS-ESP/files/4451294/18-central-heating.txt)
[18-warm-water1.txt](https://github.com/proddy/EMS-ESP/files/4451295/18-warm-water1.txt)

For central heating it is clear that bit0 of byte 9 is 1 during heating.
for warm water it is again not obvious. it looks like the combination of bit1 of byte 9 with bit0 of byte 11 but again this might happen also when the boiler is heated to the set level but I didn't check that yet.

So far my investigation. I hope It helps....

FireWizard52, 2020-04-08:
Hi,

I studied the document, but the header indicates EMS2. Is this the same as EMS?
I could not make much of it, except that there is an indication of "hot water mode"

Further googlen brought me at: https://www.avrfreaks.net/sites/default/files/forum_attachments/EMSbusReceiveExample1.txt

Here I found:
```
0x08,0x18,0x01,0x05,					//#1   0 temperature water out (uitgaand) x 10
0x08,0x18,0x04,0x01,					//#2   1 burner power
0x08,0x18,0x07,0x00,					//#3   2 burner of/off
0x08,0x18,0x07,0x50,					//#4   3 heating pump on/off
0x08,0x18,0x07,0x60,					//#5   4 tap water heating on/off
0x08,0x18,0x0B,0x05,					//#6   5 boiler temperature x 10
0x08,0x18,0x0D,0x05,					//#7   6 temperature water in (cvreturn) x 10
0x08,0x18,0x11,0x06,					//#8   7 water pressure x 10
0x08,0x18,0x12,0x02,					//#9   8 status code 1st letter
0x08,0x18,0x13,0x02,					//#10  9 status code 2nd letter
0x08,0x34,0x01,0x05,					//#11 10 tap water temperature x 10
0x08,0x34,0x05,0x50,					//#12 11 heater boiler on/off (not always available)
0x17,0x91,0x01,0x04,					//#13 12 setpoint room temperature x 2
0x17,0x91,0x02,0x05,					//#14 13 room temperature x 10
0x17,0xA8,0x17,0x81,					//#15 14 setting 0=low, 1=manual, 2=clock
0x17,0xA8,0x1C,0x84,					//#16 15 overruled clock setting x 2 ( 0 = not overruled)
0x17,0xA8,0x1D,0x84					  //#17 16 manual setpoint temperature x 2

```
This corresponds with:

> #1 keteltemp
> #2 branderpower
> #3 brander aan/uit
> #4 cv pomp aan/uit
> #5 warmwater vraag aan/uit
> #6 uitgaande cv watertemp
> #7 retour cv water temp
> #8 cv waterdruk x 10
> #9 status code 1ste letter
> #10 status code 2de letter
> #11 warmwater temp x 10
> #12 boiler verwarming aan/uit
> #13 ingestelde kamer temp
> #14 (thermostaat) huidige kamertemperatuur
> #15 (thermostaat) 0=laag, 1=hand, 2=klok
> #16 (thermostaat) overruled klok instelling
> #17 (thermostaat) ingestelde hand temp

Found at: http://www.nodo-domotica.nl/forum/viewtopic.php?f=35&t=1912

So in my opinion it should be vailable.

Regards

proddy, 2020-04-15:
@FireWizard52 first off, sorry for the delay. Been super busy lately. The perils for working in healthcare.

How you investigated this using Gafana to trace the behaviour is wonderful. And the fact that the boiler/thermostat immediately knows when the hot/warm water is activated makes me think there must be a simpler way to detect this since its a core function of a combi-boiler. I think you may have found it! There's a setting which I forgot to capture in the 0x18 telegram called 'warm water'. I wouldn't have noticed if you hadn't translated the HT3 doc.  Going to try it out now....

kwertie01, 2020-04-15:
Hi Proddy,

Thanks for the compliments. 

As I ended my last post with:
> for warm water it is again not obvious. it looks like the combination of bit1 of byte 9 with bit0 of byte 11 but again this might happen also when the boiler is heated to the set level but I didn't check that yet. 

I did some further investigation in this behavior and I was correct with my assumption that this bit1 of byte 9 (Warm Water Mode) corresponds with the Tap Water Active led on the boiler but it is also activated when the boiler is reheating the water in the boiler storage tank. 

Below a sequence of the reheating of the water in the storage tank while there was NO warm water Usage. If you look at the photo of my boiler you can see that I set my warm water temp to 60 degrees Celsius (dial with tape) and the boiler temp average is indeed kept on 60.39 degrees Celsius looking at the bottom of the Grafana graph.

[image]

(16:22:00) 08 00 18 00 07 02 11 00 00 00 00 00 60 01 D8 02 0A 01 3C 00 00 0F 30 48 00 CB 00 00 00
(16:22:06) 08 00 18 00 07 02 11 00 00 00 00 00 60 01 D8 02 09 01 3C 00 00 0F 30 48 00 CB 00 00 00
(16:22:10) 08 0B 18 00 07 02 11 00 00 00 00 00 60 01 D8 02 09 01 3C 00 00 0F 30 48 00 CB 00 00 00
(16:22:15) 08 00 18 00 07 02 11 00 00 00 00 00 60 01 D7 02 09 01 3C 00 00 0F 30 48 00 CB 00 00 00
(16:22:25) 08 00 18 00 07 02 11 00 00 00 00 00 60 01 D7 02 09 01 3B 00 00 0F 30 48 00 CB 00 00 00
(16:22:35) 08 00 18 00 07 02 10 00 00 00 00 00 60 01 D7 02 09 01 3B 00 00 0F 30 48 00 CB 00 00 00
(16:22:46) 08 00 18 00 07 02 10 00 00 00 00 00 60 01 D7 02 09 01 3B 00 00 0F 30 48 00 CB 00 00 00

WW Mode ON

(16:22:52) 08 00 18 00 57 02 11 7B 00 02 10 4C 60 01 D7 02 09 01 3B 00 00 0F 30 43 01 1B 00 03 00
(16:22:56) 08 00 18 00 57 02 10 7B 00 02 10 6C 60 01 D7 02 07 01 3B 00 00 0F 30 43 01 1B 00 03 00
(16:22:56) 08 00 18 00 57 02 11 7B 00 02 10 6C 60 01 D7 02 09 01 41 00 00 11 30 43 01 1B 00 03 00
(16:22:57) 08 00 18 00 57 02 2C 7B 00 02 10 6D 60 01 D7 02 07 01 C1 00 00 10 30 4C 01 1C 00 03 00
(16:22:58) 08 00 18 00 57 02 3D 7B 00 02 10 6D 60 01 D7 02 07 01 F5 00 00 10 30 4C 01 1C 00 03 00
(16:22:59) 08 00 18 00 57 02 44 7B 00 02 10 6D 60 01 D7 02 07 02 0A 00 67 10 30 4C 01 1C 00 03 00

WW Mode ON and Flame ON

[image]

(16:23:00) 08 00 18 00 57 02 43 7B 33 0A 10 6D 60 01 D7 02 07 02 17 00 8F 10 3D 48 00 C9 00 03 00
(16:23:02) 08 00 18 00 57 02 2C 7B 46 0A 10 65 60 01 D7 02 06 02 24 00 B7 11 3D 48 00 C9 00 03 00
(16:23:02) 08 00 18 00 57 02 19 7B 59 0A 10 65 60 01 D7 02 06 02 2A 00 D7 11 3D 48 00 C9 00 03 00
(16:23:03) 08 00 18 00 57 02 1E 7B 70 0A 10 65 60 01 D7 02 06 02 2B 00 EF 11 3D 48 00 C9 00 03 00
(16:23:04) 08 00 18 00 57 02 32 7B 77 0A 10 65 60 01 D7 02 05 02 26 01 05 11 3D 48 00 C9 00 03 00
(16:23:06) 08 00 18 00 57 02 5D 7B 7A 0A 10 65 60 01 D7 02 05 02 26 01 14 11 3D 48 00 C9 00 03 00
(16:23:06) 08 00 18 00 57 02 63 7B 7A 0A 10 65 60 01 D7 02 05 02 27 01 13 10 3D 48 00 C9 00 03 00
(16:23:07) 08 00 18 00 57 02 7B 7B 7A 0A 10 65 60 01 D7 02 04 02 31 01 19 10 3D 48 00 C9 00 03 00
(16:23:08) 08 00 18 00 57 02 8F 7B 7B 0A 10 65 60 01 D7 02 04 02 3D 01 18 10 3D 48 00 C9 00 03 00
(16:23:09) 08 00 18 00 57 02 9F 7B 7B 0A 10 65 60 01 D7 02 04 02 4D 01 22 10 3D 48 00 C9 00 03 00
(16:23:10) 08 00 18 00 57 02 AD 7B 7B 0A 10 65 60 01 D7 02 04 02 5F 01 26 10 3D 48 00 C9 00 03 00
(16:23:11) 08 0B 18 00 57 02 B6 7B 7B 0A 10 65 60 01 D7 02 05 02 6A 01 26 10 3D 48 00 C9 00 03 00
(16:23:11) 08 00 18 00 57 02 B7 7B 7B 0A 10 65 60 01 D7 02 05 02 6C 01 28 10 3D 48 00 C9 00 03 00
(16:23:12) 08 00 18 00 57 02 C4 7B 7B 0A 10 65 60 01 D7 02 07 02 79 01 29 10 3D 48 00 C9 00 03 00
(16:23:13) 08 00 18 00 57 02 D1 7B 7B 0A 10 65 60 01 D7 02 09 02 87 01 30 10 3D 48 00 C9 00 03 00
(16:23:14) 08 00 18 00 57 02 E0 7B 7B 0A 10 65 60 01 D8 02 0C 02 93 01 2E 10 3D 48 00 C9 00 03 00
(16:23:16) 08 00 18 00 57 02 F3 7B 7B 0A 10 65 60 01 D9 02 0F 02 A2 01 30 10 3D 48 00 C9 00 03 00
(16:23:16) 08 00 18 00 57 02 FD 7B 7B 0A 10 65 60 01 D9 02 11 02 AA 01 36 10 3D 48 00 C9 00 03 00
(16:23:17) 08 00 18 00 57 03 0B 7B 6B 0A 10 64 60 01 DA 02 16 02 B7 00 E1 10 3D 48 00 C9 00 01 00

WW Mode still ON and Flame OFF 

[image]

(16:23:18) 08 00 18 00 57 03 18 7B 4A 02 10 64 60 01 DB 02 19 02 C4 00 00 10 3D 48 00 C9 00 01 00
(16:23:19) 08 00 18 00 57 03 1F 7B 00 02 10 64 60 01 DD 02 1D 02 D0 00 00 0F 3D 48 00 C9 00 01 00
(16:23:23) 08 00 18 00 57 03 10 7B 00 02 10 64 60 01 E4 02 34 02 EA 00 00 0F 3D 48 00 C9 00 01 00
(16:23:24) 08 00 18 00 57 03 09 7B 00 02 10 64 60 01 E4 02 3C 02 EC 00 00 0F 3D 48 00 C9 00 01 00
(16:23:26) 08 00 18 00 57 03 02 7B 00 02 10 64 60 01 E4 02 43 02 EE 00 00 0F 3D 48 00 C9 00 01 00
(16:23:26) 08 00 18 00 57 02 FE 7B 00 02 10 64 60 01 E4 02 47 02 EF 00 00 0F 3D 48 00 C9 00 01 00
(16:23:27) 08 00 18 00 57 02 F9 7B 00 02 10 60 60 01 E4 02 4C 02 EE 00 00 0F 3D 48 00 C9 00 01 00
(16:23:28) 08 00 18 00 57 02 F5 7B 00 02 10 60 60 01 E3 02 51 02 EE 00 00 0F 3D 48 00 C9 00 01 00
(16:23:32) 08 00 18 00 57 02 EC 7B 00 02 10 60 60 01 E4 02 61 02 E7 00 00 0F 3D 48 00 C9 00 01 00
(16:23:36) 08 00 18 00 57 02 EC 7B 00 02 10 60 60 01 E6 02 6B 02 E1 00 00 0F 3D 48 00 C9 00 01 00
(16:23:42) 08 00 18 00 57 02 E7 7B 00 02 10 60 60 01 EC 02 73 02 E0 00 00 10 3D 48 00 C9 00 01 00
(16:23:44) 08 00 18 00 57 02 E5 7B 00 02 10 60 60 01 EF 02 74 02 DF 00 00 0F 3D 48 00 C9 00 01 00
[... 43 lines cut ...]

On the other hand I think I found an other way detecting the real use of warm water.
[... cut ...]
(19:15:10) 08 00 18 00		 57 02 C5 7B 00 02 10 6D 60 02 6E `00 C2` 02 0C 00 61 10 30 4C 01 1C 00 03 00

This temperature drop is not showing up when the boiler is just reheating the storage tank.

So maybe Bit1 of Byte 9 `Warm Water Mode` in combination with byte 15 and 16 `Warm water temperature storage tank sensor 2` does the trick.

So far my 2 cents again and sorry for my long reply.....

**As you stated that you work in the healthcare, I like to end with "Stay Save" and respect for helping all the people in need!**

proddy, 2020-04-19:
the WW storage tank sensors (1&2) differ for type of boiler and also model (Buderus vs Junkers) so some clever logic is needed. I'm still thinking there must be an easier way, just haven't found it yet.

proddy, 2020-10-06:
this should be fixed > 2.1.0b4

closed 2020-10-06 by proddy