emsesp/EMS-ESP #355: Write Support for Bosch CR100/CW100

fwPunsher, 2020-03-27:
Hey guys,

i got a Junkers Cerastar Comfort ZWR and a Bosch CR100 Thermostat.
Sadly writing to the thermostat is currently not supported.
How can I contribute to change that?

proddy, 2020-03-27:
these newer "easy" type of thermostats don't use EMS+ to change values. But I'm thinking the CR100 may work so lets try that.

- which version on EMS-ESP are you using?
- what are the thermostat details that EMS-ESP shows you? product id, etc..
- in telnet do `log t` and let's see what type of traffic is going in and out of the thermostat

fwPunsher, 2020-03-27:
Well that was quick,
Here are some details:
**ESP Version**
`EMS-ESP version 1.9.5b58`
**Thermostat data:**
`Thermostat data:
  Thermostat: Buderus RC200/Bosch CW100/Junkers CW100 (DeviceID: 0x18, ProductID: 157, Version: 41.08)
  Thermostat time is 09:23:53 27/03/2020
  Heating Circuit 1
   Current room temperature: 20.4 C
   Setpoint room temperature: 23.0 C
   Mode is set to day
`
And here is some traffic. Though I cant see any traffic going to the thermostat ...:
```
(10:21:51) Thermostat -> Me, RCTime(0x06), telegram: 98 0B 06 00 14 03 0A 1B 13 35 04 00 18 FF 00 (#data=11)
(10:21:54) Thermostat -> All, Type(0xBF), telegram: 98 00 BF 00 18 9D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (#data=24)
(10:22:01) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 08 01 A5 03 25 03 03 01 03 25 00 C8 (#data=9)
(10:22:07) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 98 08 1A 00 00 00 00 (#data=3)
(10:22:07) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 00 01 A5 00 CF 21 2E 00 00 2E 24 03 25 03 03 01 03 25 00 C8 00 00 11 01 03 (#data=22)
(10:22:07) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 16 01 A5 08 14 00 00 04 00 00 00 00 00 00 4B 00 3C 01 FF 00 03 (#data=18)
(10:22:08) Thermostat -> All, Type(0x021D), telegram: 98 00 FF 00 02 1D 00 00 0A 07 (#data=4)
(10:22:13) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 16 01 A5 08 15 (#data=2)
(10:22:18) Thermostat -> All, RCTime(0x06), telegram: 98 00 06 00 14 03 0A 1B 14 15 04 00 18 FF 00 (#data=11)
(10:22:18) Thermostat -> Boiler, Type(0x23), telegram: 98 08 23 00 00 00 00 (#data=3)
(10:22:32) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 16 01 A5 08 16 (#data=2)
Fetching data from EMS devices
(10:22:51) Thermostat -> Me, RCTime(0x06), telegram: 98 0B 06 00 14 03 0A 1B 14 35 04 00 18 FF 00 (#data=11)
(10:23:01) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 0D 01 A5 03 24 00 C9 (#data=4)
(10:23:07) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 98 08 1A 00 00 00 00 (#data=3)
(10:23:07) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 00 01 A5 00 CF 21 2E 00 00 2E 24 03 24 03 03 01 03 24 00 C9 00 00 11 01 03 (#data=22)
(10:23:07) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 16 01 A5 08 16 00 00 04 00 00 00 00 00 00 4B 00 3C 01 FF 00 03 (#data=18)
(10:23:08) Thermostat -> All, Type(0x021D), telegram: 98 00 FF 00 02 1D 00 00 0A 07 (#data=4)
(10:23:08) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 16 01 A5 08 17 (#data=2)
(10:23:17) Thermostat -> Boiler, UBAFlags(0x35), telegram: 98 08 35 00 11 01 (#data=2)
(10:23:19) Thermostat -> All, RCTime(0x06), telegram: 98 00 06 00 14 03 0A 1B 15 16 04 00 18 FF 00 (#data=11)
(10:23:19) Thermostat -> Boiler, Type(0x23), telegram: 98 08 23 00 00 00 00 (#data=3)
(10:23:46) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 16 01 A5 08 18 (#data=2)
Fetching data from EMS devices
(10:23:51) Thermostat -> Me, RCTime(0x06), telegram: 98 0B 06 00 14 03 0A 1B 15 36 04 00 18 FF 00 (#data=11)
(10:24:01) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 0D 01 A5 03 23 00 CA (#data=4)
(10:24:03) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 04 01 A5 4B (#data=1)
(10:24:04) Thermostat -> Boiler, Type(0x23), telegram: 98 08 23 00 4B 33 00 (#data=3)
(10:24:04) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 98 08 1A 00 4B 33 00 (#data=3)
(10:24:04) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 27 01 A5 01 (#data=1)
(10:24:07) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 98 08 1A 00 4B 33 00 (#data=3)
(10:24:07) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 00 01 A5 00 CF 21 2E 4B 00 2E 24 03 23 03 03 01 03 23 00 CA 00 00 11 01 03 (#data=22)
(10:24:07) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 16 01 A5 08 18 00 00 04 00 00 00 00 00 33 4B 00 3C 01 FF 00 01 (#data=18)
(10:24:08) Thermostat -> All, Type(0x021D), telegram: 98 00 FF 00 02 1D 00 00 0A 07 (#data=4)
(10:24:20) Thermostat -> All, RCTime(0x06), telegram: 98 00 06 00 14 03 0A 1B 16 17 04 00 18 FF 00 (#data=11)
(10:24:20) Thermostat -> Boiler, Type(0x23), telegram: 98 08 23 00 4B 33 00 (#data=3)
Fetching data from EMS devices
(10:24:50) Thermostat -> Me, RCTime(0x06), telegram: 98 0B 06 00 14 03 0A 1B 16 35 04 00 18 FF 00 (#data=11)
(10:25:01) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 0F 01 A5 00 CB (#data=2)
(10:25:01) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 0D 01 A5 03 22 (#data=2)
(10:25:03) Thermostat -> All, Type(0xA2), telegram: 98 00 A2 00 00 00 00 00 00 (#data=5)
(10:25:07) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 98 08 1A 00 4B 33 00 (#data=3)
(10:25:07) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 00 01 A5 00 CF 21 2E 4B 00 2E 24 03 22 03 03 01 03 22 00 CB 00 00 11 01 03 (#data=22)
(10:25:07) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 16 01 A5 08 18 00 00 04 00 00 00 00 00 33 4B 00 3C 01 FF 00 01 (#data=18)
(10:25:07) Thermostat -> All, Type(0x021D), telegram: 98 00 FF 00 02 1D 00 00 0A 07 (#data=4)
(10:25:10) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 16 01 A5 08 19 (#data=2)
(10:25:17) Thermostat -> Boiler, UBAFlags(0x35), telegram: 98 08 35 00 11 01 (#data=2)
(10:25:21) Thermostat -> All, RCTime(0x06), telegram: 98 00 06 00 14 03 0A 1B 17 18 04 00 18 FF 00 (#data=11)
(10:25:21) Thermostat -> Boiler, Type(0x23), telegram: 98 08 23 00 4B 33 00 (#data=3)
(10:25:22) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 98 00 FF 16 01 A5 08 1A (#data=2)
```

fwPunsher, 2020-04-04:
I have read into the Hometop HT3 info I bit. However I am a bit confused now. My thermostat apperently identifies itself sometimes as 0x18 and sometimes as 0x98 could somebody explain this? Also I have read in my thermostat manual that it is possible to connect a second thermostat on the bus and use that one as remote. So there should a possibility of setting parameters like that. Is there somebody with such a setup that could sniff this communication?

proddy, 2020-04-04:
junkers set the 8th bit of the type ID (so 0x18 becomes 0x98). They are just the same so need to worry. Have you tried sending anything to the thermostat yet? `thermostat temp 20` for example.

fwPunsher, 2020-04-04:
Well that's the first thing I tried before authoring this thread. It says my device is unsupported. I also tried those Modem Comands from the Hometop project. However maybe I made a mistake there as the Thermostate didn't react to them at all.

fwPunsher, 2020-04-13:
Hi @proddy  do you have an idea on how to get startet with setting heating modes or setpoint temperatures of the thermostat?
I havent found a working raw command yet.

proddy, 2020-04-14:
Hi @fwPunsher, sorry being very busy these last weeks. We need to see if your thermostat responds to commands like the other type 100s using EMS+. So first try `send 0B 18 FF 08 02 B9 28`  to set the temperature to 20 degrees. If that doesn't work use 8B instead of 0B.

fwPunsher, 2020-04-14:
Both of them didn't change anything. Or will they only update for one specific mode, which doesn't necessary have to be the active one? By the way, if i set the log level to verbose, the telnet connection will be lost within the next minute. This doesn't happen with log t.

fwPunsher, 2020-04-17:
I finally have the commands for the CXY series. norberts1 knew them.
Here they are:

    Auto to manual - mode (CRC is calculated by software)
   ` 8B 18 FF 00 01 B9 00 <CRC>`
    Back from manual to auto-mode
    `8B 18 FF 00 01 B9 FF <CRC>`
    Setup Setpoint temperature for 'Comfort2' to 21.5 degree
    `8B 18 FF 02 01 B9 2B <CRC`> <<-- 2B is 21.5*2:= 43dez or 2B hex.

  Setup Setpoint temperature for 'Manual' to 22.0 degree
  `8B 18 FF 00 01 B9 00 <CRC> `<<-- switching to manual - mode
 ` 8B 18 FF 0A 01 B9 2C <CRC>` <<-- 2C is 22.0*2:= 44dez or 2C hex.

Shall i try to extend the code and do a pull request or would it be better for you to add them yourself?

proddy, 2020-04-17:
yes I saw. I made a typo in my earlier test. I'll add the support this weekend. Thanks for helping out!

proddy, 2020-05-05:
Added to v2.0 (in alpha)/.

closed 2020-05-12 by proddy

giddyhup, 2020-10-13:
@proddy Just to clarify, this means Bosch CW100 thermostats are now fully supported?

proddy, 2020-10-16:
@giddyhup I believe so yes, for a long time.