emsesp/EMS-ESP #356: MQTT don't want to connect (TCP Disconnected)

Chuckame, 2020-03-28:
**Question/Problem**
Actually EMS ESP is working fine, but mqtt do not want to connect.

Actually hassio is used with mqtt addon.
Other component written by me is currently using the same user/password on mqtt, and is working fine.

I created a new user specificly for ems-esp, but not working too. And this new user is working from mqtt CLI !

In telnet, says :
```
[MQTT] Disconnected! (count 3)
[MQTT] Connecting to MQTT...
[MQTT] TCP Disconnected
[MQTT] Disconnected! (count 4)
[MQTT] Connecting to MQTT...
[MQTT] TCP Disconnected
[MQTT] Disconnected! (count 5)
[MQTT] Connecting to MQTT...
[MQTT] TCP Disconnected
[MQTT] Disconnected! (count 6)
[MQTT] Connecting to MQTT...
```

On my mqtt broker, says:
```
New connection from 192.168.1.21 on port 1883.
[ERROR] Auth error with emsesp
```

I've tried with local mqtt, but receive no connection 🤔 

**Device information**
```
info

EMS-ESP system stats:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: enabled, Solar Module: enabled, Mixing Module: disabled
  Shower Timer: disabled, Shower Alert: disabled

EMS Bus stats:
  Bus is connected, protocol: HT3
  Rx: # successful read requests=62, # CRC errors=0
  Tx: Last poll=1.466 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID:0x09 ProductID:95 Vers
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 50 C
  Warm Water desired temperature: 72 C
  Warm Water current temperature: 43.0 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 22677 times
  Warm Water active time: 29 days 7 hours 51 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 0 C
  Current flow temperature: 53.7 C
  Return temperature: 36.0 C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 55 %
  Burner current power: 0 %
  Flame current: -0.1 uA
  System pressure: ? bar
  System service code:  (0)
  Heating temperature setting on the boiler: 73 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 10 %
  Boiler temperature: ? C
  Pump modulation: 0 %
  Burner # starts: 79299 times
  Total burner operating time: 148 days 6 hours 17 minutes
  Total heat operating time: 118 days 22 hours 26 minutes
  Total UBA working time: 2283 days 16 hours 50 minutes

Solar Module stats:
  Solar module: Junkers ISM1 Solar Module (DeviceID:0x30 ProductID:101 Version:23.03)
  Collector temperature: 53.5 C
  Bottom temperature: 49.0 C
  Pump modulation: ? %
  Pump active: off
  Pump working time: 18 days 21 hours 52 minutes
  Energy last hour: 45.1 Wh
  Energy today: ? Wh
  Energy total: ? kWh

Thermostat stats:
  Thermostat: Junkers FR100 (DeviceID:0x10 ProductID:107 Version:15.07)
  Thermostat time is 16:53:28 28/03/2020
  Heating Circuit 1
   Current room temperature: 22.6 C
   Setpoint room temperature: 18.0 C
   Mode is set to manual
   Day Mode is set to night

[MQTT] TCP Disconnected
[MQTT] Disconnected! (count 12)
```

proddy, 2020-03-28:
Hi, which version of EMS-ESP are you using? There were some improvements to MQTT in the dev builds of 1.9.5.

Also does it work if you don't have authentication set on your broker? Are there special characters in the username or password perhaps? Which MQTT broker are you using too? 

I'll see if I can reproduce and find the error but would need a few more details.

Chuckame, 2020-03-28:
Thanks for your rapid answer!

> which version of EMS-ESP are you using? 
git branch: `master`
From `version.h`: 
```
#define APP_VERSION "1.9.4"
```

> Also does it work if you don't have authentication set on your broker?
Not tested for the moment

> Are there special characters in the username or password perhaps?
Only `[a-zA-Z0-9]` in user and password

> Which MQTT broker are you using too?
The mqtt broker is from home assistant add-on store, version 5.1 (internal addon mqtt version: 1.6.3)

I think this may be useful to display esp-ems version from `info` telnet command ?

proddy, 2020-03-28:
Could you upgrade to the latest dev build? You can do this via the web UI (there's a button to switch) or manually upload. The firmware is in https://github.com/proddy/EMS-ESP/releases/tag/travis-dev-build 

I'm not saying it'll fix your problem but at least we'll be on the same build together!

Chuckame, 2020-03-28:
I'm on nodemcuv2, Is this required to recompile with nodemcuv2 board in pio config ?

Chuckame, 2020-03-28:
Oh sorry, don't see the nodemcuv2 build, I'll test it now

Chuckame, 2020-03-28:
The dashboard is not working, only telnet is working !

```
[MQTT] Connecting to MQTT...
[NTP] Internet time: 17:14:49 UTC on 28/3. Local time: 18:14:49 CET
[EMS] Detected new EMS Device with ID 0x08. Fetching version information...
[EMS] Detected new EMS Device with ID 0x09. Fetching version information...
[EMS] Detected new EMS Device with ID 0x10. Fetching version information...
[EMS] Detected new EMS Device with ID 0x30. Fetching version information...
: 95, Version: 20.05)ductIDs a Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT
ductID: 95, Version: 20.05) HT3 Contro
nkers FR100 (DeviceID: 0x10, ProductID: 107, Version: 15.07)
 23.03)viceID: 0x30, ProductID: 101, Version:SM1 Solar Mod
[TELNET] Connected to EMS-ESP version 1.9.5b58. Type ? for commands.
[MQTT] TCP Disconnected
[MQTT] Disconnected! (count 1)
[MQTT] Connecting to MQTT...
```

Chuckame, 2020-03-28:
I have this settings:
```
Current settings:

  wifi_mode=client
  wifi_ssid=HomeNCo
  wifi_password=******************
  mqtt_enabled=on
  mqtt_ip=192.168.1.25
  mqtt_username=espems
  mqtt_password=****************
  mqtt_base=homeassistant
  mqtt_port=1883
  mqtt_keepalive=60
  mqtt_retain=off
  mqtt_qos=0
  mqtt_heartbeat=on
  mqtt_nestedjson=on
  serial=off
  ntp_enabled=on
  ntp_interval=720
  ntp_timezone=2
  log_events=off
  log_ip=
  led=on
  led_gpio=2
  dallas_gpio=14
  dallas_parasite=off
  tx_mode=3
  bus_id=0x0B
  listen_mode=off
  shower_timer=off
  shower_alert=off
  publish_time=0 (automatic)
  master_thermostat=0 (use first detected)
```

Chuckame, 2020-03-28:
Dashboard not working because of pending request: `http://ems-esp.local/fonts/glyphicons-halflings-regular.woff`

Chuckame, 2020-03-28:
Now mqtt apparently connected :
Logs from mqtt server:
```
1585414917: New connection from 192.168.1.25 on port 1883.
[INFO] found espems on Home Assistant
1585414917: New client connected from 192.168.1.25 as c183d3c0-4c47-43bb-a05c-b72d8105fee61585414917254 (p2, c1, k10, u'espems').
1585415602: New connection from 192.168.1.25 on port 1883.
[INFO] found espems on Home Assistant
1585415603: New client connected from 192.168.1.25 as mqtt-explorer-eca1d59b (p2, c1, k60, u'espems').
1585415793: Socket error on client c183d3c0-4c47-43bb-a05c-b72d8105fee61585414917254, disconnecting.
1585416171: Saving in-memory database to /data/mosquitto.db.
```

But still having those logs in telnet:
```
[MQTT] TCP Disconnected
[MQTT] Disconnected! (count 5)
[MQTT] Connecting to MQTT...
[MQTT] TCP Disconnected
[MQTT] Disconnected! (count 6)
[MQTT] Connecting to MQTT...
[MQTT] TCP Disconnected
[MQTT] Disconnected! (count 7)
```

Chuckame, 2020-03-28:
Dashboard is now working, maybe due to cache problems, but mqtt don't want to connect

Chuckame, 2020-03-28:
I cleaned last messages.

Actually installed the last version 1.9.5b58, but still mqtt connection issues.
Nothing is received by the mqtt server, no logs, except my other component connected:
```
1585417182: New connection from 192.168.1.36 on port 1883.
[INFO] found teleinfo on Home Assistant
1585417183: New client connected from 192.168.1.36 as teleinfo (p2, c1, k15, u'teleinfo').
```

I reinstalled the mqtt server, still not working.
Changed the credentials with more simple user/pwd: `emsesp/azerty`, still not working.
```
[MQTT] Connecting to MQTT...
(18:53:01) Boiler -> 0x13, Type(0x05), telegram: 88 13 05 22 00 (#data=1)
(18:53:01) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 00 00 (#data=15)
(18:53:02) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 00 01 A2 37 00 00 22 00 C0 80 00 80 00 01 6D FF FF FF 00 00 00 00 00 00 00 (#data=25)
(18:53:02) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 32 01 A7 01 E9 81 00 00 03 00 00 A4 F7 00 58 95 00 (#data=17)
(18:53:02) Boiler -> All, UBAMonitorSlow(0x19), telegram: 88 00 19 00 80 00 80 00 80 00 FF FF 00 00 01 35 C3 03 41 F9 00 00 00 02 9D 02 00 DD 2E 80 00 (#data=27)
(18:53:07) Thermostat -> Boiler, Version(0x02), telegram: 90 88 02 00 0A (#data=1)
(18:53:07) Boiler -> Thermostat, Version(0x02), telegram: 88 10 02 00 5F 14 05 00 00 00 00 00 00 00 (#data=10)
(18:53:07) Thermostat -> Solar Module, Version(0x02), telegram: 90 B0 02 00 0A (#data=1)
(18:53:07) Solar Module -> Thermostat, Version(0x02), telegram: B0 10 02 00 65 17 03 00 00 00 00 00 65 00 (#data=10)
(18:53:08) Solar Module -> All, ISM1StatusMessage(0x0003), telegram: B0 00 FF 00 00 03 20 00 00 00 00 FB 01 B3 00 00 00 6A 60 (#data=13)
(18:53:09) Solar Module -> All, Type(0xBE), telegram: B0 00 BE 00 30 00 00 00 00 (#data=5)
(18:53:11) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 00 01 A1 37 00 00 22 00 C0 80 00 80 00 01 6D FF FF FF 00 00 00 00 00 00 00 (#data=25)
(18:53:11) Boiler -> 0x13, Type(0x05), telegram: 88 13 05 22 00 (#data=1)
(18:53:12) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 00 32 01 A7 01 E8 81 00 00 03 00 00 A4 F7 00 58 95 00 (#data=17)
[MQTT] TCP Disconnected
[MQTT] Disconnected! (count 11)
```
(In parallel, my other component using mqtt still working)

proddy, 2020-03-30:
so it works now?

big12boy, 2020-03-30:
Just to chip in here, I get the same error for anything newer than 1.9.3 and I also now just tried building it myself, but I do get the same error nonetheless...
I do however receive more information from my MQTT server (OpenHABian Rasperberry Pi with the integrated Moquette Broker).
See the link: [MQTT Log](https://pastebin.com/q00CNRiZ)
I hope it's the same issue, but I'm not fully sure about it... ;)

Chuckame, 2020-03-30:
@proddy yes, I don't know how... Just multiple reboots, some tests, retry, retry, retest, build, rebuild, retest, so back again... Until I built the firmware from my computer on the last available dev version, multiple reboots, some tests, changed mqtt I'm, multiple time and BOUM working!

I think it worked when I set the mqtt ip via telnet.

@big12boy my problem was not the same!! Maybe check your versions, and use the last HA with the last dev build of emsesp?

big12boy, 2020-03-30:
@Chuckame Sorry I wasn't clear enough, I've tried every pre-build bin from 1.9.4 up to the latest dev build and also just now build the latest dev build on my machine. I'm running the latest Stable OpenHAB version with MQTT (there are about 20 devices in my house using this exact MQTT connection for more than a year without any issues). I'm still not certain it might be the same or a related problem...

proddy, 2020-03-30:
@Chuckame @big12boy there have been many changes since 1.9.4 including lots of improvements to how MQTT is being sent. But what you're mentioning is worrying.

@Chuckame what you could try is just download and running mosquitto on your PC (or even RPI#). It's  a single download with one .exe/binary to run. Run it with -v to see the logs and then see if it EMS-ESP connects. Then add a username/password and try again. The quest is to find where and why it fails and the logs should give some information.

big12boy, 2020-03-30:
@proddy I fixed it now and it looks like a combination of failures, I think that the first data send to the broker had a format problem and blocked the topic completely (I tried with mosquitto_sub to listen to messages being send and it was no longer working, when the ESPEMS module connected). I could however restart my OpenHAB Raspi and now it is working and I got the complete queue transmitted to the broker. I do not really get how this happens and it might be an issue with the Broker, but still it is something I haven't discovered in 5+ years working with MQTT (It completely locked the broker and flooded the log with continues error messages).

Chuckame, 2020-03-30:
@proddy I wasn't clear enough me too, it is currently working, but I don't know why, maybe a combination of failures too!

closed 2020-03-30 by Chuckame