emsesp/EMS-ESP #357: Is there technical doc?

Chuckame, 2020-03-30:
Just a generic question, do someone have some lexical documentation about ems and boiler world? Because I'm a noobie and I would be able to understand some basics, like "what is a heat circuit", more explanation about thermostats modes, which parts of our boiler we can activate or not?

proddy, 2020-03-31:
not really. When I find interesting bits I tend to put them into the wiki.

Norbert has a great doc on the EMS telegram formats for the Junkers/HT3 protocol. https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT3-Bus_Telegramme.pdf

PhillyGilly, 2020-03-31:
Hi Chuck. There isn't really anything like this. Although the conversation is in English the underlying technology is German. AFAIK all the serious grunt work on reverse engineering EMS protocol has been done by German speakers (not to say that their English isn't very good too!) In fact it turns out that much of the equipment sold under different brands has the same hardware/software controls. e.g. My UK Worcester Greenstar ERP boiler has the same control circuits as the Buderus Logamax plus, GB192, and Bosch Condens GC9000 (identifying as DeviceID: 0x08, ProductID: 208, Version: 01.04).
This is why we have wW (warm Wasser) for hot water where the UK English would be DHW. The situation isn't helped by the fact that sometimes the German engineers use abbreviations with the initial letters of English words, when abbreviations which aren't quite the same as the ones that a UK native speaker wouldn't use. So, heat(ing) circuit HC is an englishification of Heizungskreis but in the UK this would be called CH (central heating) it's just the other stuff that the boiler provides hot water to for space heating.
I generally roll with the terminology that pops up, but if I do want to use more familiar terms I alias the names in my Home Assistant configuration. i.e.
#02 sensor.dhw_selected_temperature
- platform: mqtt
  state_topic: 'home/ems-esp/boiler_data'
  name: 'DHW Selected Temperature'
  unit_of_measurement: '°C'
  value_template: '{{ value_json.wWSelTemp }}'
  qos: 1
  payload_available: "online"
  payload_not_available: "offline"

If you want/need to understand more, I suggest you download your equipment manuals in English and German and look at the diagrams in the German manuals to see how stuff is labelled. 
Hope that helps.

closed 2020-07-06 by proddy