emsesp/EMS-ESP #360: FW100 not found until I issue autodetect scan

jaryn, 2020-04-03:
**Bug description**
FW100 is never detected automatically (I waited more than 5 minutes) after ems-esp boot. The device though appears after `autodetect scan` is issued.

**Steps to reproduce**
1) power-up ems-esp or reboot it using telnet command
1) `devices` telnet command doesn't show the FW100
1) issue `autodetect scan` telnet command, the FW100 appears (check with `devices` command or with `info` command)

**Expected behavior**
FW100 appears after system start

**Screenshots**
```
Escape character is '^]'.
[UART] Rx/Tx connection established
[WIFI] Connecting to SSID: XXX
[WIFI] Connected to SSID XXX...
[OTA] Listening to firmware updates on ems-esp.local:8266
[SYSTEM] Disabling serial port communication
[NTP] NTP internet time enabled via server pool.ntp.org with timezone 2
[MQTT] Connecting to MQTT...
[MQTT] MQTT connected
[NTP] Internet time: 16:01:42 UTC on 3/4. Local time: 18:01:42 CEST
Shower alert has been set to disabled
Shower timer has been set to disabled
New EMS device recognized as a UBAMaster: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID:0x08 ProductID:95 Version:15.06)
ductID:95 Version:15.06)iceID:0x09 Pro
[TELNET] Connected to EMS-ESP version 1.9.4. Type ? for commands.
System Logging set to None
log t

System Logging set to Thermostat only
de(18:04:23) UBAMaster -> All, type 0x18, telegram: 88 00 18 00 2A 01 A6 64 30 09 03 25 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 02 18 (CRC=2E) #data=25
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
v(18:04:23) UBAMaster -> All, type 0x34, telegram: 88 00 34 08 00 (CRC=88) #data=1
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
ices

These device IDs are on the EMS Bus: 0x08 0x09 0x10
and 2 were recognized by EMS-ESP as:
 Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID:0x08 ProductID:95 Version:15.06)
 Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID:0x09 ProductID:95 Version:15.06)

(18:04:33) UBAMaster -> All, type 0x18, telegram: 88 00 18 00 2A 01 A7 64 30 09 03 25 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 02 18 (CRC=F1) #data=25
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
(18:04:33) UBAMaster -> All, type 0x34, telegram: 88 00 34 08 00 (CRC=88) #data=1
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Fetching data from EMS devices
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
(18:04:36) UBAMaster -> All, type 0x07, telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=ED) #data=15
(18:04:42) Thermostat -> All, type 0x006F, telegram: 90 00 FF 00 00 6F 03 02 00 D2 00 C5 F3 34 00 (CRC=9C) #data=9
(18:04:43) UBAMaster -> All, type 0x18, telegram: 88 00 18 00 2A 01 A5 64 31 09 03 25 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 02 18 (CRC=AB) #data=25
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
(18:04:43) UBAMaster -> All, type 0x07, telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=ED) #data=15
(18:04:43) UBAMaster -> All, type 0x19, telegram: 88 00 19 00 00 73 80 00 80 00 FF FF 00 64 00 16 7A 0E 3B 7D 00 00 00 0E 3B 7D 00 16 7A 80 00 (CRC=F9) #data=27
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
(18:04:44) UBAMaster -> All, type 0x34, telegram: 88 00 34 08 00 (CRC=88) #data=1
[... 103 lines cut ...]
Publishing hot water and heating states via MQTT
Fetching data from EMS devices
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Publishing thermostat data via MQTT
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
(18:05:43) Thermostat -> UBAMaster, type 0x16, telegram: 90 88 16 00 02 (CRC=AF) #data=1
(18:05:44) UBAMaster -> Thermostat, type 0x16, telegram: 88 10 16 00 FF 50 (CRC=8E) #data=2
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
(18:05:44) Thermostat -> All, type 0x006F, telegram: 90 00 FF 00 00 6F 03 02 00 D2 00 C5 F3 34 00 (CRC=9C) #data=9
Publishing thermostat data via MQTT
(18:05:44) Thermostat -> Me, type 0x06, telegram: 90 0B 06 00 14 04 12 03 05 22 05 00 (CRC=55) #data=8
(18:05:45) Thermostat -> Me, type 0x06, telegram: 90 0B 06 00 14 04 12 03 05 22 05 00 (CRC=55) #data=8
(18:05:46) Thermostat -> All, type 0x06, telegram: 90 00 06 00 14 04 12 03 06 04 05 00 (CRC=72) #data=8
```
**Device information**
In the screenshot.
I am quite sure the boiler is Junkers CeraclassExcellence ZSC 35-3 MFA. I can verify if needed.

**Additional context**
Note that I would expect two devices to appear (assuming the boiler has one). For some reason I see the boiler on 2 ids.

jaryn, 2020-04-03:
Seems the FW100 doesn't  always appear after `autodetect scan`. But many times it does. let's say 8/10 cases. Something like that. The EMS-ESP is connected right next to it (daisy-chain topology).

jaryn, 2020-04-03:
> which version are you using?

It is in the "screenshot"
```
[TELNET] Connected to EMS-ESP version 1.9.4. Type ? for commands.
```

I compiled the dev version and the FW100 did appear right after OTA loading the firmware. But after reboot it disappeared and I after scan it appeared.
```
devices

These 2 were recognized by EMS-ESP:
 Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID: 0x08, ProductID: 95, Version: 15.06)
 Controller: HT3 Controller (DeviceID: 0x09, ProductID: 95, Version: 15.06)

...
[SYSTEM] Free Heap: 22824 bytes initially | 9864 bytes used (43%) | 12960 bytes free (56%)
devices

These 2 were recognized by EMS-ESP:
 Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID: 0x08, ProductID: 95, Version: 15.06)
 Controller: HT3 Controller (DeviceID: 0x09, ProductID: 95, Version: 15.06)

devices scan

Requesting EMS bus master for its device list and scanning for external sensors...
[EMS] Detected new EMS Device with ID 0x08. Fetching version information...
[EMS] Detected new EMS Device with ID 0x09. Fetching version information...
[EMS] Detected new EMS Device with ID 0x10. Fetching version information...
New EMS device recognized as a Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID: 0x08, ProductID: 95, Version: 15.06)
New EMS device recognized as a Controller: HT3 Controller (DeviceID: 0x09, ProductID: 95, Version: 15.06)
New EMS device recognized as a Junkers Thermostat: Junkers FW100 (DeviceID: 0x10, ProductID: 105, Version: 13.07)
info

```

 Perhaps there is something going on when doing OTA which is not going on when just rebooting?

jaryn, 2020-04-03:
The behaviour is erratic. I did reset and no FW100 was recognized, though it was detected on the bus (device `0x10`):
```
TLocal time: 00:02:53 CES3 UTC on 3/4.
[TELNET] Connected to EMS-ESP version 1.9.5b58. Type ? for commands.
?

* EMS-ESP version 1.9.5b58
* Hostname: ems-esp (192.168.5.211)
* WiFi SSID: Bezrucova_688 (signal 78%)
* MQTT connected (heartbeat enabled)
*
* Commands:
*  ?/help=show commands, CTRL-D/quit=end telnet session
*  set, system, restart, mqttqueue, kick, save
*  info                                                      show current values deciphered from the EMS messages
*  log <n | b | t | s | m | r | j | v | w [ID] | d [ID]>     logging: none, basic, thermo, solar, mixing, raw, jabber, verbose, watch a type or device
*  publish                                                   publish all values to MQTT
*  refresh                                                   fetch values from the EMS devices
*  devices [scan] | [scan+] | [clear] | [save]               list detected devices, quick scan, deep scan, clear and and save
*  txqueue                                                   show current Tx queue
*  send XX ...                                               send raw telegram data to EMS bus (XX are hex values)
*  thermostat read <type ID>                                 send read request to the thermostat for heating circuit hc 1-4
*  thermostat temp <degrees> [mode] [hc]                     set thermostat temperature. mode is manual,auto,heat,day,night,eco,comfort,holiday,nofrost
*  thermostat mode <mode> [hc]                               set mode (manual,auto,heat,day,night,eco,comfort,holiday,nofrost)
*  boiler read <type ID>                                     send read request to boiler
*  boiler wwtemp <degrees>                                   set boiler warm water temperature
*  boiler wwactive <on | off>                                set boiler warm water on/off
*  boiler wwonetime <on | off>                               set boiler warm water onetime on/off
*  boiler tapwater <on | off>                                set boiler warm tap water on/off
*  boiler flowtemp <degrees>                                 set boiler flow temperature
*  boiler comfort <hot | eco | intelligent>                  set boiler warm water comfort setting

devices

These 2 were recognized by EMS-ESP:
 Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID: 0x08, ProductID: 95, Version: 15.06)
 Controller: HT3 Controller (DeviceID: 0x09, ProductID: 95, Version: 15.06)

devices

These 2 were recognized by EMS-ESP:
 Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID: 0x08, ProductID: 95, Version: 15.06)
 Controller: HT3 Controller (DeviceID: 0x09, ProductID: 95, Version: 15.06)

log v

System Logging set to Verbose
devices

These 2 were recognized by EMS-ESP:
 Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID: 0x08, ProductID: 95, Version: 15.06)
 Controller: HT3 Controller (DeviceID: 0x09, ProductID: 95, Version: 15.06)

devices scan

Requesting EMS bus master for its device list and scanning for external sensors...
Requesting type UBADevices(0x07) from dest 0x08
(00:03:27) Boiler -> 0x13, Type(0x05), telegram: 88 13 05 22 00 (#data=1)
(00:03:27) Boiler -> All, UBAMonitorFast(0x18), telegram: 88 00 18 00 00 00 8A 64 00 00 02 00 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (#data=25)
(00:03:27) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 88 00 34 08 00 (#data=1)
(00:03:28) Sending read of type 0x07 to 0x08, telegram: 8B 88 07 00 20 
(00:03:28) Boiler -> Me, UBADevices(0x07), telegram: 88 0B 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (#data=15)
[EMS] Detected new EMS Device with ID 0x08. Fetching version information...
Requesting type Version(0x02) from dest 0x08
[EMS] Detected new EMS Device with ID 0x09. Fetching version information...
Requesting type Version(0x02) from dest 0x09
[EMS] Detected new EMS Device with ID 0x10. Fetching version information...
Requesting type Version(0x02) from dest 0x10
(00:03:28) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (#data=15)
(00:03:29) Sending read of type 0x02 to 0x08, telegram: 8B 88 02 00 20 
[... 82 lines cut ...]
 Controller: HT3 Controller (DeviceID: 0x09, ProductID: 95, Version: 15.06)

devices

These 2 were recognized by EMS-ESP:
 Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID: 0x08, ProductID: 95, Version: 15.06)
 Controller: HT3 Controller (DeviceID: 0x09, ProductID: 95, Version: 15.06)

devices scan

Requesting EMS bus master for its device list and scanning for external sensors...
[EMS] Detected new EMS Device with ID 0x08. Fetching version information...
[EMS] Detected new EMS Device with ID 0x09. Fetching version information...
[EMS] Detected new EMS Device with ID 0x10. Fetching version information...
New EMS device recognized as a Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID: 0x08, ProductID: 95, Version: 15.06)
New EMS device recognized as a Controller: HT3 Controller (DeviceID: 0x09, ProductID: 95, Version: 15.06)
New EMS device recognized as a Junkers Thermostat: Junkers FW100 (DeviceID: 0x10, ProductID: 105, Version: 13.07)
devices

These 3 were recognized by EMS-ESP:
 Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID: 0x08, ProductID: 95, Version: 15.06)
 Controller: HT3 Controller (DeviceID: 0x09, ProductID: 95, Version: 15.06)
 Thermostat: Junkers FW100 (DeviceID: 0x10, ProductID: 105, Version: 13.07)
```
We can see corrupt telegrams there, but I think I didn't get any with the last release (1.9.4)

proddy, 2020-04-04:
I've seen this behaviour before, it's something in the code that needs to be improved and I've fixed it in version 2.0 so not really planning to back-port it to 1.9.x. Basically what happens when EMS-ESP boots up is
 - sends a request to the master/boiler to show all the connected devices on the EMS bus
 - for each of these devices EMS-ESP sends a personal request for its version details. This will help determine what type and model it is from our database
 - unfortunately that send request fails (due to collision, corruption, something else). So it will not add the device.

a workaround is to do a scan and when you have all the devices save them with `devices save`.

in EMS-ESP 2.0 both the Rx and Tx are asynchronous queues with a more rigid retry logic so if there is a glitch on the wire the last Tx command will be repeated. Well, in theory 'cos I haven't tried it out yet!

jaryn, 2020-04-04:
OK. I am looking forward for the version 2. I didn't see it in the git repo.

proddy, 2020-04-15:
I'll leave this open and we can re-test when I release 2.0 to alpha

jaryn, 2020-04-15:
Looking forward to it!

proddy, 2020-05-10:
if you have a spare ESP8266 can you test your FW100 using v2

jaryn, 2020-05-11:
I think I don't really need a spare. I think I can use the one I have. I use it for experimental purposes and I don't really need it stable especially in summer. I will test it once I get home. Thanks.

closed 2020-06-07 by proddy