emsesp/EMS-ESP #361: Rc20 regulator, unknown? (Version:02.01 ProductID:113 DeviceID:0x17)

pitulek, 2020-04-04:
Hi, why my ems-esp does not recognizes Rc20 regulator? It says 
unknown? (Version:02.01 ProductID:113 DeviceID:0x17),
propably is the id is problem, how to repair that?

my buderus is LOGOMATIC U052-28 and is recognized as :
Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (Version:13.04 ProductID:95 DeviceID:0x08)

thank you very much for your help

Piotr
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/proddy/EMS-ESP/issues) (both open and closed)*
* *searched the [wiki help pages](https://github.com/proddy/EMS-ESP/wiki/Troubleshooting)*

*Completing this template will help developers and contributors help you. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Question**
*A clear and concise description of what the problem/doubt is.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from the telnet session with the logging set to Verbose mode.*

**Additional context**
*Add any other context about the problem here.*

pitulek, 2020-04-05:
I have changed to dev build software and now it says Sieger ES72 (DeviceID: 0x17, ProductID: 113, Version: 02.01) the temp is ok but mode is undefined.

pitulek, 2020-04-05:
pictures of the system
[image]
[image]
[image]

[image]
[image]

proddy, 2020-04-05:
hi, the mode should be coming on telegram type 0xA8 for your RC20. I do see the 0xAE coming in which is used to determine if the mode is in day or night. But I'm not too familiar with this thermostat. Which modes does it actually support?

What you could do to help is `thermostat read a8` and `thermostat read ae`  before and after you have manually changed the mode on your thermostat. And then we can compare the differences and figure out where the information is stored in the telegram.

pitulek, 2020-04-05:
The rc20 suports 3 modes, auto,  night, day.
i cannot read thermostat nothing happen , when i write command /thermostat read a8 / also thermostat read.

night mode teleggrams
0xae
17 00 AE 00 00 00 27 00 CE 00 00 00 (#DATA8)
day mode
17 00 AE 00 80 02 29 00 CE 00 00 00 (#DATA8)

auto(DAY)
17 00 AE 00 04 02 29 00 CE 00 00 00 (#DATA8)

there is no telegram
0xa8

only 0x35 0x1a 0xae

proddy, 2020-04-05:
ok, I can use this info. When you do a `info` command in telnet does it show the mode and mode type? Forget the web interface for now, it's old and I'm busy renewing it.

pitulek, 2020-04-05:
no information about the mode info

proddy, 2020-04-05:
but 'mode type' is shown right? like night or day?

pitulek, 2020-04-05:
no info about mode
only temp  carrent  and set , nothing else

proddy, 2020-04-05:
ok. I'll look into it.

pitulek, 2020-04-05:
if i use the command : thermostat type  setting thermostat mode to unknown

pitulek, 2020-04-05:
night mode log
System Logging set to Thermostat only
(16:05:37) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 17 08 1A 00 00 00 00 00 (#data=4)
(16:05:42) Thermostat -> Boiler, UBAFlags(0x35), telegram: 17 08 35 00 01 01 (#data=2)
(16:05:47) Thermostat -> All, RC20StatusMessage3(0xAE), telegram: 17 00 AE 00 00 00 29 00 CF FF FF 00 (#data=8)
Fetching data from EMS devices
(16:06:11) Thermostat -> Me, telegram: 17 0B 91 00
(16:06:11) Thermostat -> Me, telegram: 17 0B A8 00
(16:06:12) Thermostat -> Me, telegram: 17 0B 06 00
(16:06:37) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 17 08 1A 00 00 00 00 00 (#data=4)
(16:06:42) Thermostat -> Boiler, UBAFlags(0x35), telegram: 17 08 35 00 01 01 (#data=2)
(16:06:48) Thermostat -> All, RC20StatusMessage3(0xAE), telegram: 17 00 AE 00 00 00 29 00 CF FF FF 00 (#data=8)
Fetching data from EMS devices
(16:07:11) Thermostat -> Me, telegram: 17 0B 91 00
(16:07:11) Thermostat -> Me, telegram: 17 0B A8 00
(16:07:12) Thermostat -> Me, telegram: 17 0B 06 00
(16:07:37) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 17 08 1A 00 00 00 00 00 (#data=4)
(16:07:42) Thermostat -> Boiler, UBAFlags(0x35), telegram: 17 08 35 00 01 01 (#data=2)
(16:07:48) Thermostat -> All, RC20StatusMessage3(0xAE), telegram:

pitulek, 2020-04-05:
day mode log

(16:09:23) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 17 08 1A 00 25 64 64 00 (#data=4)
(16:09:23) Thermostat -> Boiler, UBAFlags(0x35), telegram: 17 08 35 00 11 11 (#data=2)
(16:09:30) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 17 08 1A 00 00 00 64 00 (#data=4)
(16:09:33) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 17 08 1A 00 00 00 00 00 (#data=4)
(16:09:33) Thermostat -> All, RC20StatusMessage2(0xAD), telegram: 17 00 AD 00 01 27 29 01 4B 05 01 FF 28 19 0A 02 00 00 (#data=14)
(16:09:48) Thermostat -> All, RC20StatusMessage3(0xAE), telegram: 17 00 AE 00 88 02 29 00 CF FF FF 00 (#data=8)
Fetching data from EMS devices
(16:10:11) Thermostat -> Me, telegram: 17 0B 91 00
(16:10:11) Thermostat -> Me, telegram: 17 0B A8 00
(16:10:11) Thermostat -> Me, telegram: 17 0B 06 00
(16:10:23) Thermostat -> Boiler, UBAFlags(0x35), telegram: 17 08 35 00 11 11 (#data=2)
(16:10:33) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 17 08 1A 00 00 00 00 00 (#data=4)
(16:10:49) Thermostat -> All, RC20StatusMessage3(0xAE), telegram: 17 00 AE 00 88 02 29 00 CF FF FF 00 (#data=8)
Fetching data from EMS devices
(16:11:10) Thermostat -> Me, telegram: 17 0B 91 00
(16:11:10) Thermostat -> Me, telegram: 17 0B A8 00
(16:11:11) Thermostat -> Me, telegram: 17 0B 06 00
(16:11:23) Thermostat -> Boiler, UBAFlags(0x35), telegram: 17 08 35 00 11 11 (#data=2)
(16:11:33) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 17 08 1A 00 00 00 00 00 (#data=4)
(16:11:49) Thermostat -> All, RC20StatusMessage3(0xAE), telegram: 17 00 AE 00 88 02 29 00 CF FF FF 00 (#data=8)
Fetching data from EMS devices
(16:12:10) Thermostat -> Me, telegram: 17 0B 91 00
(16:12:11) Thermostat -> Me, telegram: 17 0B A8 00
(16:12:11) Thermostat -> Me, telegram: 17 0B 06 00
(16:12:24) Thermostat -> Boiler, UBAFlags(0x35), telegram: 17 08 35 00 11 11 (#data=2)
(16:12:33) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 17 08 1A 00 00 00 00 00 (#data=4)
(16:12:34) Thermostat -> Boiler, UBAMaintenanceStatusMessage(0x1C), telegram: 17 88 1C 00 06 (#data=1)
(16:12:34) Boiler -> Thermostat, UBAMaintenanceStatusMessage(0x1C), telegram: 08 17 1C 00 00 00 00 00 00 00 (#data=6)
(16:12:36) Thermostat -> Controller, Type(0x29), telegram: 17 89 29 00 01 (#data=1)
(16:12:36) Controller -> Thermostat, Type(0x29), telegram: 09 17 29 00 7B (#data=1)
(16:12:49) Thermostat -> All, RC20StatusMessage3(0xAE), telegram: 17 00 AE 00 88 02 29 00 CF FF FF 00 (#data=8)

pitulek, 2020-04-05:
AUTO
Fetching data from EMS devices
(16:16:10) Thermostat -> Me, telegram: 17 0B 91 00
(16:16:11) Thermostat -> Me, telegram: 17 0B A8 00
(16:16:11) Thermostat -> Me, telegram: 17 0B 06 00
(16:16:24) Thermostat -> Boiler, UBAFlags(0x35), telegram: 17 08 35 00 11 11 (#data=2)
(16:16:33) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 17 08 1A 00 00 00 00 00 (#data=4)
(16:16:50) Thermostat -> All, RC20StatusMessage3(0xAE), telegram: 17 00 AE 00 0C 02 29 00 CF FF FF 00 (#data=8)
Fetching data from EMS devices
(16:17:10) Thermostat -> Me, telegram: 17 0B 91 00
(16:17:10) Thermostat -> Me, telegram: 17 0B A8 00
(16:17:11) Thermostat -> Me, telegram: 17 0B 06 00
(16:17:24) Thermostat -> Boiler, UBAFlags(0x35), telegram: 17 08 35 00 11 11 (#data=2)
(16:17:34) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 17 08 1A 00 00 00 00 00 (#data=4)
(16:17:50) Thermostat -> All, RC20StatusMessage3(0xAE), telegram: 17 00 AE 00 0C 02 29 00 CF FF FF 00 (#data=8)

pitulek, 2020-04-05:
info
EMS-ESP system status:
  System logging set to Thermostat only
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: disabled, Shower Alert: disabled
  Saved known device IDs: none

EMS Bus status:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=324, # CRC errors=0
  Tx: Last poll=1.250 seconds ago, # successful write requests=3

Boiler data:
  Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID: 0x08, ProductID: 95, Version: 13.04)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water circulation pump type: charge pump
  Warm Water circulation pump freq: 2x3min
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 40 C
  Warm Water desinfection temperature: 70 C
  Warm Water circulation active: off
  Warm Water set temperature: 40 C
  Warm Water current temperature: 42.7 C
  Warm water temperature (intern): ? C
  Warm water temperature (extern): ? C
  Warm Water current temperature (extern): 42.7 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 7545 times
  Warm Water active time: 63 days 6 hours 8 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 0 C
  Current flow temperature: 22.9 C
  Max boiler temperature: ? C
  Return temperature: ? C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 3200.0 uA
  System pressure: ? bar
  System service code: 0H (203)
  Heating temperature setting on the boiler: 60 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 10 %
  Exhaust temperature: ? C
  Pump modulation: 100 %
  Burner # starts: 21381 times
  Total burner operating time: 269 days 17 hours 9 minutes
  Total heat operating time: 206 days 11 hours 0 minutes
  Total UBA working time: 2468 days 21 hours 1 minutes

Thermostat data:
  Thermostat: Sieger ES72 (DeviceID: 0x17, ProductID: 113, Version: 02.01)
  Heating Circuit 1
   Current room temperature: 20.8 C
   Setpoint room temperature: 19.5 C

pitulek, 2020-04-05:
Thermostat -> Me, telegram: 17 0B 91 00
(21:46:58) Thermostat -> Me, telegram: 17 0B A8 00
(21:46:59) Thermostat -> Me, telegram: 17 0B 06 00
(21:47:04) Thermostat -> Boiler, UBASetPoints(0x1A), telegram: 17 08 1A 00 00 00 00 00 (#data=4)
(21:47:06) Thermostat -> Boiler, UBAFlags(0x35), telegram: 17 08 35 00 01 01 (#data=2)
(21:47:25) Thermostat -> All, RC20StatusMessage3(0xAE), telegram: 17 00 AE 00 00 00 26 00 CE 00 00 00 (#data=8)
i found that there is no response from thermostat to commands 0x06, 0xa8, 0x91

as to the 0xAE  17 00 AE 00 00 00 26 00 CE 00 00 00

26 IS set temperature , CE is current temp
and 00 00 before 26 is night mode

8002 is day mode
0400 auto/night
0402 is auto/day

pitulek, 2020-04-08:
maybe it could help too

telegram ad
(22:37:07) Thermostat -> Me, RC20StatusMessage2(0xAD), telegram: 17 0B AD 00 01 26 28 00 4B 05 01 FF 28 19 0A 03 00 00 (#data=14)
(22:41:18) Thermostat -> Me, RC20StatusMessage2(0xAD), telegram: 17 0B AD 00 01 26 28 00 4B 05 01 FF 28 19 0A 03 00 00 (#data=14)
(22:44:53) Thermostat -> Me, RC20StatusMessage2(0xAD), telegram: 17 0B AD 00 01 24 28 00 4B 05 01 FF 28 19 0A 03 00 00 (#data=14)
(22:50:05) Thermostat -> Me, RC20StatusMessage2(0xAD), telegram: 17 0B AD 00 01 24 28 00 4B 05 01 FF 28 19 0A 02 00 00 (#data=14)
(22:52:24) Thermostat -> Me, RC20StatusMessage2(0xAD), telegram: 17 0B AD 00 01 24 28 00 4B 05 01 FF 28 19 0A 01 00 00 (#data=14)

there is information about set temeperatures , mode, and set programm

MichaelDvP, 2020-04-08:
Looks very similar to the 3D/3E telegrams of RC35, (see [https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme](https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegrammel)). 
So AD reads: 17 0B AD 00 01 24 28 00 4B 05 01 FF 28 19 0A 01 00 00
thermostat 17 to emsesp 0B, type AD, offset 00:
01 - heating by radiator
24- nighttemp 18°C
28-daytemp 20°C
00- ?
4B- max flow 75°C
05- min flow 5°C

and telegram AE: 17 00 AE 00 00 00 26 00 CE 00 00 00
00- Bitfield bit 7 manual, bit 2 automatic
00- bitfield bit 1 day-mode (bit 0 summermode?)
26- actual settemp 19°C
00CE - actual roomtemp 20.6°C

But why is actual settemp in AE different from night-settting in AD and manual-bit not set?
Can you check these values by chaning them while logging?

pitulek, 2020-04-08:
Thermostat -> All, RC20StatusMessage2(0xAD), telegram: 17 00 AD 00 01 24 28 00 4B 05 01 FF 28 19 0A 01 00 00 (#data=14
 Thermostat -> All, RC20StatusMessage2(0xAD), telegram: 17 00 AD 00 01 24 28 01 4B 05 01 FF 28 19 0A 01 00 00 (#data=14)
 Thermostat -> All, RC20StatusMessage2(0xAD), telegram: 17 00 AD 00 01 24 28 02 4B 05 01 FF 28 19 0A 01 00 00 (#data=14)

thermostat 17 to emsesp 0B, type AD, offset 00:
01 - heating by radiator
24- nighttemp 18°C
28-daytemp 20°C
00- is mode,  00 night , 01 day, 02 auto
4B- max flow 75°C
05- min flow 5°C
01 -?
FF ?
28 -?
19-?
0A -?
01  set program auto (1-9)
00
00

Thermostat -> All, RC20StatusMessage2(0xAD), telegram: 17 00 AD 00 01 24 28 00 4B 05 01 FF 28 19 0A 01 00 00 (#data=14
 Thermostat -> All, RC20StatusMessage2(0xAD), telegram: 17 00 AD 00 01 24 28 01 4B 05 01 FF 28 19 0A 01 00 00 (#data=14)
 Thermostat -> All, RC20StatusMessage2(0xAD), telegram: 17 00 AD 00 01 24 28 02 4B 05 01 FF 28 19 0A 01 00 00 (#data=14)
  Thermostat -> Me, RC20StatusMessage2(0xAD), telegram: 17 0B AD 00 01 24 28 02 4B 05 01 FF 28 19 0A 01 00 00 (#data=14)
  Thermostat -> Me, RC20StatusMessage2(0xAD), telegram: 17 0B AD 00 01 24 28 00 4B 05 01 FF 28 19 0A 01 00 00 (#data=14) 
  Thermostat -> Me, RC20StatusMessage2(0xAD), telegram: 17 0B AD 00 01 24 28 02 4B 05 01 FF 28 19 0A 05 00 00 (#data=14)

thermostat does not respond  to 0x91  ,0x06 , 0xa8  respnse is
(21:31:29) Thermostat -> Me, telegram: 17 0B 91 00
(21:31:29) Thermostat -> Me, telegram: 17 0B A8 00
(21:31:29) Thermostat -> Me, telegram: 17 0B 06 00

thermostat is buderus rc20 , i can set there time , day number 1-7, programs 1-9, day mode , night , auto using set programs, one time hot water on /off button - that is all 

i am interesting to change set temps remotly from home assistant or domoticz, but there is no A8 COMMAND :(
thanks for your help

pitulek, 2020-04-08:
when i change mode night /day / auto there is automatic telegram from thermostat 

 Thermostat -> All, RC20StatusMessage2(0xAD), telegram: 17 00 AD 03 00 (#data=1) night mode
 Thermostat -> All, RC20StatusMessage2(0xAD), telegram: 17 00 AD 03 01 (#data=1) day 
 Thermostat -> All, RC20StatusMessage2(0xAD), telegram: 17 00 AD 03 02 (#data=1) auto

logng telegram is for request  

Thermostat -> Me, RC20StatusMessage2(0xAD), telegram: 17 0B AD 00 01 24 28 02 4B 05 01 FF 28 19 0A 05 00 00 (#data=14)

pitulek, 2020-04-08:
the short message when i change something 
OB 17 AD 03  --- 00 NIGHT , 01 - DAY MODE, 02 AUTO MODE
0B 17 AD 0B ---- 07 CHANGE SET PROGRAM FOR AUTO  IN THIS CASE NO 7 01-09
0B 17 AD 01 25 -- CHANGE TEMP NIGHT  19,5C
0B 17 AD 02 2A -- CHANGE TEMP DAY 21C

pitulek, 2020-04-08:
I think i found solution

pitulek, 2020-04-08:
so if i send raw  the telegrams it changes mode and set temps and program for auto mode
comands

0B 17 AD 03 --- 00 NIGHT , 01 - DAY MODE, 02 AUTO MODE
0B 17 AD 0B ---- 07 CHANGE SET PROGRAM FOR AUTO IN THIS CASE NO 7 01-09
0B 17 AD 01 25 -- CHANGE TEMP NIGHT 19,5C
0B 17 AD 02 2A -- CHANGE TEMP DAY 21C

please implement it in your soft for this thermostat type

pitulek, 2020-04-09:
AE telegrams

mode night, doing nothing 0000
(09:27:41) Thermostat -> All, RC20StatusMessage3(0xAE), telegram: 17 00 AE 00 00 00 28 00 CE 00 00 00 (#data=8)
mode day , heating hot water 8802

(09:29:41) Thermostat -> All, RC20StatusMessage3(0xAE), telegram: 17 00 AE 00 88 02 2A 00 CE 00 00 00 (#data=8)

mode auto day, heating warm water 0c02

(09:31:42) Thermostat -> All, RC20StatusMessage3(0xAE), telegram: 17 00 AE 00 0C 02 2A 00 CE 00 00 00 (#data=8)

mode auto day , water is hot, gas off change 0c to 04  0402

 Thermostat -> All, RC20StatusMessage3(0xAE), telegram: 17 00 AE 00 04 02 28 00 CF 00 00 00 (#data=8)

day mode , water is hot , gas off 8002

(09:53:47) Thermostat -> All, RC20StatusMessage3(0xAE), telegram: 17 00 AE 00 80 02 28 00 CF 00 00 00 (#data=8)

auto night mode, water is hot, gas off 04 00

Thermostat -> All, RC20StatusMessage3(0xAE), telegram: 17 00 AE 00 04 00 26 00 CF 00 00 00 (#data=8)

MichaelDvP, 2020-04-09:
As far as i understand proddy, he is very busy with his 2.0 built. I have made a [branch](https://github.com/MichaelDvP/EMS-ESP/tree/RC20N) for you for testing. A bin for the bbqkees electronics is included.

pitulek, 2020-04-09:
night mode , water hot  there is 01 at the end of telegram
(10:20:55) Thermostat -> All, RC20StatusMessage3(0xAE), telegram: 17 00 AE 00 00 00 26 00 CE 00 01 00 (#data=8)

pitulek, 2020-04-09:
thank you for your help

pitulek, 2020-04-09:
i do not know how to build bin file in platformio

MichaelDvP, 2020-04-09:
I have compiled already and added to the main dir, you can download the `EMS-ESP-1_9_5b58-d1_mini.bin` and flash it via webinterface or espota.

pitulek, 2020-04-09:
working fine, i can see mode and  thermostat temp works 
i am testing.
thanks!

MrWillibald, 2020-04-09:
Works for my ES72 too, great job @pitulek @MichaelDvP

proddy, 2020-04-14:
> As far as i understand proddy, he is very busy with his 2.0 built. I have made a [branch](https://github.com/MichaelDvP/EMS-ESP/tree/RC20N) for you for testing. A bin for the bbqkees electronics is included.

Hi - yes, been busy with work and some 2.0 updates too. Shall I merge your last PR or do you have a newer build? And thanks BTW!

MichaelDvP, 2020-04-14:
@proddy 
I've made a branch based on your dev, so it's only a small change and not so much work to review. I'll make a extra PR and you can merge this first.

proddy, 2020-04-14:
ok, let me know which one to merge first. thanks for the additions!

MichaelDvP, 2020-04-14:
The changes do not conflict, i have also a merged version local here, if you like onl one PR i can upload it. 
But i think this RC20 is important for many people, thermostats are basic functionality for ems-esp, so check this first.
The other additions i made primary for myself to optimize the heatingcurve. If you want to concentrate on V2.0 you can also skip them.

commit 2eefb7a6 (2020-04-22): Merge pull request #366 from MichaelDvP/RC20N

new Rc20, fix #361
closed 2020-04-30 by proddy