emsesp/EMS-ESP #362: RC20 as Remotecontroller

og9gjz567, 2020-04-09:
Hi,
I've bought a RC20 as Remotecontroller for my RC30 placed at my boiler, to control my temperature remotely. I'm able to control my temperature / modes / etc on the RC20. So I'm sure the communication between RC20 and RC30 works. If I remove the RC20 the RC30 s displays "RC20 missing".

I've updated to the latest dev-build (EMS-ESP version 1.9.5b58) because of the bug #361, but the devices list shows only these entries:
These 4 were recognized by EMS-ESP:
 Boiler: Sieger BK13,BK15/Nefit Smartline/Buderus GB1x2 (DeviceID: 0x08, ProductID: 64, Version: 02.07)
 Controller: BC10/RFM20 Receiver (DeviceID: 0x09, ProductID: 68, Version: 02.00)
 Mixing Module: MM10 Mixing Module (DeviceID: 0x21, ProductID: 69, Version: 02.01)
 Thermostat: RC30 (DeviceID: 0x10, ProductID: 67, Version: 02.08)

If I change the "Raumeinfluss" from 0K to 1K the list of devices grows to these 7 entries:
These 7 were recognized by EMS-ESP:
 Boiler: Sieger BK13,BK15/Nefit Smartline/Buderus GB1x2 (DeviceID: 0x08, ProductID: 64, Version: 02.07)
 Controller: BC10/RFM20 Receiver (DeviceID: 0x09, ProductID: 68, Version: 02.00)
 Thermostat: RC30 (DeviceID: 0x10, ProductID: 67, Version: 02.08)
 Switching Module: WM10 Switch Module (DeviceID: 0x11, ProductID: 71, Version: 02.00)
 unknown?: unknown? (DeviceID: 0x13, ProductID: 82, Version: 01.01)
 Thermostat: Sieger ES72 (DeviceID: 0x19, ProductID: 113, Version: 02.01)
 Mixing Module: MM10 Mixing Module (DeviceID: 0x21, ProductID: 69, Version: 02.01)

As you can see in my last Issue (#248) my system is running with outdoor-influenced weather compensation.
Like glitterball writes: "It sounds as though HC2 is running with outdoor-influenced weather compensation. Unless there is a thermostat or sensor in the area heated by HC2, you have no way of measuring the temperature in that room! Instead, the system is relying on a heating curve and the outside temperature to calculate a suitable value for the flow temperature in HC2." 

I now bought the RC20 so that there is a thermostat in the area heated by HC2.

What kind of debug information do you need?

Thank you very much!

MichaelDvP, 2020-04-10:
I think a good start is to post the info to see the actual setpoints and then identify the telegrams from/to the RC20 `log d 19`  for more than one minute. Next step is to change values on the RC20 and see what changes in the telegrams. 
I expect that the setpoints for roomtemp, nighttemp and so on are synced between master RC30 and remote RC20, so there is no need to read them from RC20. The only new information from RC20 is the current room temperature.

proddy, 2020-04-14:
Strange that the RC20 isn't detected. EMS-ESP should be able to detect all devices on the bus, even if it doesn't recognize them. Did you get an RC20 or the RC20RF? When you find the device_id just set this as the 'master' and it should work.

MichaelDvP, 2020-04-14:
According to Norberts list the device-IDs 0x18, 0x19, 0x1A, 0x1B are remote controls for hc1-4 for a master thermostat at ID 0x10. So RC30 at 0x10 should be set as master (if not set automaticly). 
ID 0x19 is not on the ems_scanDevices list, so may take a minute to identify. 
Since remote control is a special function on RC20/RC20RF there may be some other telegrams to communicate with the master-thermostat. The functions in remote-control-mode are limited to setpoints, currentroomtemp and wwOneTime.

og9gjz567, 2020-04-27:
Hi,
okay thanks for the hint. I've the RC20 connected to the secont HC. So the ID have to be x19. If I take a look to devices output I get:
These 7 were recognized by EMS-ESP:
 Boiler: Sieger BK13,BK15/Nefit Smartline/Buderus GB1x2 (DeviceID: 0x08, ProductID: 64, Version: 02.07)
 Controller: BC10/RFM20 Receiver (DeviceID: 0x09, ProductID: 68, Version: 02.00)
 Mixing Module: MM10 Mixing Module (DeviceID: 0x21, ProductID: 69, Version: 02.01)
 Thermostat: RC30 (DeviceID: 0x10, ProductID: 67, Version: 02.08)
 Switching Module: WM10 Switch Module (DeviceID: 0x11, ProductID: 71, Version: 02.00)
 unknown?: unknown? (DeviceID: 0x13, ProductID: 82, Version: 01.01)
 **Thermostat: Sieger ES72 (DeviceID: 0x19, ProductID: 113, Version: 02.01)**

_One or more devices are not recognized by EMS-ESP. Please report this back in GitHub._

So I guess the Sieger ES72 is my RC20. They seem to be identical in Google Pictures.

Here's my log:
**log d 19**

System Logging set to Device mode
Fetching data from EMS devices
(10:51:16) Thermostat -> Me, telegram: 19 0B 91 00
(10:51:16) Thermostat -> Me, telegram: 19 0B A8 00
(10:51:17) Thermostat -> Me, telegram: 19 0B 06 00
(10:51:30) Thermostat -> All, Type(0xAF), telegram: 19 00 AF 00 00 CB 00 (#data=3)
(10:51:32) Thermostat -> Boiler, UBAFlags(0x35), telegram: 19 08 35 00 00 00 (#data=2)
Fetching data from EMS devices
Publishing MQTT data for: thermostat boiler mixing
(10:52:16) Thermostat -> Me, telegram: 19 0B 91 00
(10:52:17) Thermostat -> Me, telegram: 19 0B A8 00
(10:52:18) Thermostat -> Me, telegram: 19 0B 06 00
(10:52:22) Thermostat -> Controller, Type(0x29), telegram: 19 89 29 00 01 (#data=1)
(10:52:22) Controller -> Thermostat, Type(0x29), telegram: 09 19 29 00 6B (#data=1)
(10:52:31) Thermostat -> All, Type(0xAF), telegram: 19 00 AF 00 00 CB 00 (#data=3)
(10:52:32) Thermostat -> Boiler, UBAFlags(0x35), telegram: 19 08 35 00 00 00 (#data=2)
Fetching data from EMS devices
(10:53:15) Thermostat -> Me, telegram: 19 0B 91 00
(10:53:16) Thermostat -> Me, telegram: 19 0B A8 00
(10:53:17) Thermostat -> Me, telegram: 19 0B 06 00
(10:53:31) Thermostat -> All, Type(0xAF), telegram: 19 00 AF 00 00 CB 00 (#data=3)
(10:53:33) Thermostat -> Boiler, UBAFlags(0x35), telegram: 19 08 35 00 00 00 (#data=2)
log n
System Logging set to None

**log d 19 (Here I set my temperature to 18,5°C) **

System Logging set to Device mode
(10:56:33) Thermostat -> All, Type(0xAF), telegram: 19 00 AF 00 00 CB 00 (#data=3)
(10:56:33) Thermostat -> Boiler, UBAFlags(0x35), telegram: 19 08 35 00 00 00 (#data=2)
**(10:56:50) Thermostat -> Thermostat, RC35Set_HC2(0x47), telegram: 19 10 47 02 25 (#data=1)**
Fetching data from EMS devices
(10:57:17) Thermostat -> Me, telegram: 19 0B 91 00
(10:57:17) Thermostat -> Me, telegram: 19 0B A8 00
(10:57:18) Thermostat -> Me, telegram: 19 0B 06 00
(10:57:34) Thermostat -> Boiler, UBAFlags(0x35), telegram: 19 08 35 00 00 00 (#data=2)
(10:57:34) Thermostat -> All, Type(0xAF), telegram: 19 00 AF 00 00 CB 00 (#data=3)
Fetching data from EMS devices
Publishing MQTT data for: thermostat boiler mixing
(10:58:14) Thermostat -> Me, telegram: 19 0B 91 00
(10:58:15) Thermostat -> Me, telegram: 19 0B A8 00
(10:58:16) Thermostat -> Me, telegram: 19 0B 06 00
log n

MichaelDvP, 2020-04-27:
Ok, lets see: 
the RC20 is set wrongly as master-thermostat and ems-esp asks for telegrams 91, A8 and 06, all are empty, as this is a newer RC20 and is not the master.
As a remote the RC20 sends: 
`19 08 35 00 00 00` tells the boiler no DHW-onetime, if you press the Water-button this should change. 
`19 10 47 02 25` tells the master-thermostat RC30 0x10 that setpoint is daytemp is 18.5°C 
`19 00 AF 00 00 CB 00` publishes to all, i think this is the room-temp 00CB, 20.3°C, is this right?
 
You should update to latest dev 1.9.5b60, this will detect the RC20 as "Sieger ES72/Buderus RC20" and will use the first detected thermostat, which should be RC30 0x10. Otherwise set in terminal:
`set master_thermostat 10 `

Do you know what device 0x13 is? Norberts list says its a universal switch/control, but for what is it used? 

Independently of if it works, can you please  send me a longer complete log (`log v` or `log r` or `log j`). I will try to simulate the remote control RC20 in ems-esp by inserting a room-temperature from a aqara-sensor via mqtt.

og9gjz567, 2020-04-27:
Hi,
thanks for the very fast reply! **I've updated to 1.9.5b60 and now I'm able to set the temp** via mqtt (HA) and console (thermostat temp 20 auto 2)
I've no idea what kind of device 0x13 is. Are you sure, that I've set up my RC20 wrong? I can change the temp and modes on my RC 20 and theses changes will display on the main RC30. I've set up in my HC2 the RC20 as remote without room influence, and in the rc20 I had to declare the HC2, too.

[putty.log](https://github.com/proddy/EMS-ESP/files/4539699/putty.log)

MichaelDvP, 2020-04-27:
@og9gjz567 No, i think the RC20 is set up correctly, i was only curious about the unknown 0x13 device and its purpose. I seen it in your setup before, it is a real device and has nothing to do with the RC20. 

Thanks for the log, But my first approach was not successful. RC30 and RC35 seems to need different messages. I modified ems-esp to answer to the 0x19 poll. If i send `19 00 AF 00 00 CB 00` the RC35 asks for version (i send product 113, v2.01) and than for 5 byte AF `(10 99 AF 00 05`) message, if i answer this with 3 bytes it asks for the missing 2 bytes, if i answer with zeros, nothing happens. The RC35 does not show a room temperature in message 48 and complains if i try to set room influence.  I have to figure out, what RC35 expects in this message and if there other messages RC35 needs to accept the remote temperature. In #278  lmdecroos has a RC20RF with 5 bytes message, but the logs in this thread are very short and i think if i try to simulate the RC20RF (product 93), i also have to simulate the transceiver (product 94, device 18).

MichaelDvP, 2020-04-29:
@og9gjz567 Just for the record: I forgot to set in RC35 the RC20 as remote. I've changed it to RC20  `send 0B 10 47 1A 01` and now it works with sending 3 bytes in AF message (Do you know what the third byte is?). So RC30 and RC35 are not so different, and simulation works (for now with fixed value, i have to add the mqtt for the aqara).
Thank you again for the log.

proddy, 2020-05-16:
Hi @MichaelDvP , looking back at this is there anything we need to implement in v2 regarding the logic for the RC20RF or RC35?

I can add device type 0x13 as a switch, it'll show up but be pretty useless. What do you think?

MichaelDvP, 2020-05-16:
Yes, you can skip the` set masterthermostat` and automate  it. 
Thermostate at id 0x10 is a masterthermostat with 4 heating circuits. For each hc there can be a room-controller (RC20, RC20RF) as remote. These roomcontrollers are on id 0x18 for hc1, 0x19 for hc2 and so on. All remotes sending message-type 0xAF, but there is no need to read this message, since the values are synced and published by the master in his hc-telegrams. 
Thermostat RC20 and other types on id 0x17 can only control a single hc, they can not be combined with a master (0x10) and they can not act as master.  
So if there is id 0x10 it is the master, otherwise 0x17 is in the system controlling only hc1, 0x18 ff can be ignored as thermostat (maybe declared as remote?)

For 0x13 i think it is better to leave it as unknown device as long as you dont know what it is doing. Then you have a better chance that someone with this module reports the function of it.

proddy, 2020-05-16:
wow, I never realized this. It's got me thinking that the way I've implemented it now is probably slightly over complicated in v2. I had assumed you can have multiple thermostats each with multiple heat circuits. So my understanding if I got it correct is...
1. If a 0x10 or 0x17 (like my RC20) is present, it will always be the master.
2. when changing values of a heat circuit such as setting the target temperature, the commands must always be sent to the master (0x10, 0x17)
3. when MQTT commands are sent to change the mode or temperature, they will go to the master
4. 0x18 is always hc1, 0x19 hc2 etc and used for receiving status information, not writing commands specifically

MichaelDvP, 2020-05-16:
Correct, the master thermostat is controlling and ems-esp only taks to the master.

to 4.) 
The remotes sending the measured  roomtemperature to the master  i.e for hc2: (19 10 AF ...) and the master republishes this temperature to all (10 00 47 ..pos 3). The remote can also send setpoints/mode to the master (19 10 47 .. ) and DHW temperatur/onetime to the boiler.  But we can read all this from masterthermostat and UBA. And if we want to change mode, setpoint etc. we send it to the master (0B 10 47 ...). The remotes get their display-information (time, setpoint, etc.) from the broadcasts of the master.

MichaelDvP, 2020-05-16:
Oh, i've found in EMS+ documentatation, that with RC200 it is also possible to control several heatings circuits without mastercontroller (RC300/310). Buderus writes that the normal way is with a RC3xx master and  RC200 as room-remote (as in EMS), but it's possible without master that every RC200 controls one heating circuit, only the first one controls boiler and DHW, the others only the hc.

proddy, 2020-05-18:
thanks for explaining all of that Michael. I've adjusted the login in v2

closed 2020-05-18 by proddy