emsesp/EMS-ESP #364: EMS Bus Connected but Tx is not working.

Snom, 2020-04-13:
Hello,

first of all, thank you for your work. I really like your project :D 
I have a bit trouble connecting to my ems bus as it tells me that Tx is not working. If I connect to the board and enable verbose logging, I can see several logs running through with no errors at all. If I type "info", it shows the correct protocol and that the bus is connected. Rc is successful and read without any CRC errors but Tx has no signal. Is this rather a software issue or did I missed something within my build ? The version I am using is EMS-ESP version 1.9.5b58.

If I switch the Tx mode to 1 it shows the following error: (mode 2 shows no errors)

(00:00:17.986) Sending read of type 0x07 to 0x08, telegram: 0B 88 07 00 20 
** error sending buffer: WDTO

Thank you in advance.

bbqkees, 2020-04-13:
Which boiler and thermostat do you have and which board are you using?
How did you connect it to the bus?

Snom, 2020-04-13:
The boiler is a buderus logano with an RC30. I built the board myself with the provided plans. I connect it via the service jack (3.5 headphone jack) at the service port. So the connection seems to be established and some data seems to be transferred but the Tx is not working properly.

bbqkees, 2020-04-13:
Which Logano?
Which resistor value did you use for the TX resistors?
Often if TX does not work its the wrong resistor value.

proddy, 2020-04-14:
The "WDTO" error (ESP8266 watchdog timeout) error usually happens when the Tx is out of sync and in most cases we've seen is when the board design is faulty, either using cheap/wrong components or wrong values. Check the design!

bbqkees, 2020-04-14:
Indeed in nearly all cases if TX does not work its a hardware issue. Because the TX side is only a few components, the first subject is the TX resistor or the transistor.

Snom, 2020-04-17:
Thank you for your comments. Have you an idea which resistor might cause the issue or do you have any idea what could cause it on the Tx side ? I know its a bit tricky as you cannot see my build but maybe this will help. I checked the plan within my built yesterday and it seems like all parts are at the right spot.

bbqkees, 2020-04-17:
This schematic is the easiest to understand:
[image]

The bottom part is the TX circuit. The 4 resistors above the transistor make a parallel resistor bank.
The resulting value should be around 200-250Ohm. Here its 910/4=227,5Ohm.
In Proddy's circuit its R4/R5 so 380/2=190 Ohm.

Snom, 2020-04-23:
Thank you guys for your help. It seems like it was a problem with the coils in front of the ems bus connection  (L1, L2). It seems like my local electric store gave me either no coils and something else or just bad coils. I directly connected the ems cables to the capacitor. Now I have a Rx and Tx connection. Does it matter if I do not use any coils or should I get some ? 
I will close the ticket now.

closed 2020-04-23 by Snom

bbqkees, 2020-04-24:
The inductors are part of a noise filter.
As the distance between the boiler and the thermostat might be long, so are the wires between them.
Long and unshielded wires pick up interference.
If your board works fine without them, you could leave them out.

Snom, 2020-04-26:
Alright, thank you very much.