emsesp/EMS-ESP #365: Parseable log

VincentSC, 2020-04-13:
Would it be possible to have a log that is easier to save and parse? Cleaning up a new log takes most of time, when I search for manually changed values in the telegrams.

VincentSC, 2020-04-13:
For example to mqtt, 
- source
- target
- EMS/EMS+ command
- payload in 4 forms
  - bits
  - integers
  - hex
  - hex per two

proddy, 2020-04-14:
There's a SysLog feature in EMS-ESP. I don't think I documented it much but you can give it an IP address of a Syslog compatible server (all linux distros' have one and there are plenty of windows small apps). All logs are sent there. MQTT is not the right place for dumping messages.

I actually removed syslog support from my version 2.0 since I didn't think anyone was using it. I may consider adding it again. Let me know how you get on.

higgers, 2020-04-14:
Version 2.0? This sounds exciting. What's going to be in v2.0?

proddy, 2020-04-14:
> Version 2.0? This sounds exciting. What's going to be in v2.0?

bit of the same really, just a total re-write where I cleaned up the code and ported it over to use C++11.  The Rx/Tx are asynchronous queues which should help with packet collision. The web has been re-written in React and Typescript so a lot cleaner. Security is better. Telnet is a real linux-like console. Main thing is the separation between the individual device specific logic and the telegrams making it easily extensible to any other device type. In theory, it could support unlimited thermometers, solar modules etc. But at the end of the day is still does what 1.9 does! But I learned a few tricks along the way. debugging at the moment.....

VincentSC, 2020-04-14:
Just noticed there was a syslog-server - wish I knew before. :) 

I would really like to keep that syslog! Reverse engineering the unsupported telegrams is not really efficient via Telnet (for me), as I really need to watch all variables at the same time. Now I can make a script that listens to port 514, parses the received log-lines, keeps track of any change and print the changed values with some meta.

proddy, 2020-04-15:
great! I'll make sure it's included in 2.0.

closed 2020-04-15 by proddy