emsesp/EMS-ESP #367: Why i do not see vlues from ems-esp in homeassistant

pitulek, 2020-04-14:
Hi!

i have a question , have problem with homeassistant.No values, all are unknown.Do not know the reason why? mqtt broker is working , ems-esp working.

/home/piotr/Pulpit/Zrzut ekranu z 2020-04-14 22-31-04.png

/home/piotr/Pulpit/Zrzut ekranu z 2020-04-14 22-35-22.png

/home/piotr/Pulpit/Zrzut ekranu z 2020-04-14 22-34-29.png

/home/piotr/Pulpit/Zrzut ekranu z 2020-04-14 22-35-10.png

pitulek, 2020-04-14:
[image]
[image]
[image]
[image]

bbqkees, 2020-04-15:
You need to enter the mqqt parameter in configuration.yaml:
`# Example configuration.yaml entry
mqtt:
  broker: IP_ADDRESS_BROKER`

proddy, 2020-04-15:
Also, I check your EMS-ESP settings (in telnet or web). The format of MQTT topics are as default `/home/ems-esp/topic` where `home` is the MQTT base defined in EMS-ESP and the `ems-esp` is the hostname. This is also documented in the [wiki](https://emsesp.github.io/docs/#/MQTT). You can change these obviously but make sure you also change the .yaml files in Home Assistant.

Also use the latest 1.9.5b as it has many more improvements in the MQTT service code.

pitulek, 2020-04-15:
Thank you very much, i have soleved it  by mayself earlier, ems-esp should be. not home/ems-esp.MQTT explorer is useful.
Proddy and bbqkees thank you very much for your help! You do great job.
I am new to this and try to understand how it all works.

closed 2020-04-15 by pitulek