emsesp/EMS-ESP #368: WLAN/Wifi ems-esp - auto fallback to AP mode if lost connection in client mode?

giovanne123, 2020-04-15:
Maybe it is not directly EMS-ESP related and caused by used wifi library or something else.
But maybe someone has an idea.

I have configured EMS-ESP to connect as client to my WLAN (Web UI: Wireless Settings-Mode-Client). Not activated EMS-ESP as an Access Point itself.

But I was wondering why I sometimes see the AP ems-esp and can connect to it without passwort.
I think it could be somehow related to if the EMS-ESP has lost connection to WLAN and can't reconnect and then brings up the own AP. (like when doing the initial setup of Wifi)
Maybe somehow a fallback within the Wifi library?
But I think it is not that fine to connect to it without authentification.
I haven't tried to reproduce it so far.
I restart the EMS-ESP and he connects again to the WLAN and the AP of the ems-esp is gone.

Does someone know something about that or had seen the same behaviour?

proddy, 2020-04-15:
yes, this is a known intermittent problem and I've seen it myself a few times. It's fixed in version 2.0 (hopefully) which has proper AP fallback provisioning. It'll be released soon for testing. Thanks for reporting!

dutchrazor, 2020-04-18:
I have not seen this in some time, though my HASS was also down after a failed update.

Wouldn't it be nice to have the option of automatically restarting on a set schedule , to prevent any kind of problems that only pop up after a few weeks of usage?

proddy, 2020-04-19:
I think I just need to re-write the network logic to something like
 - if a wifi ssid/password has been set and it fails to connect, fallabck to AP mode
 - but keep trying to reconnect to the user's ssid every miniute

I don't like the scheduled restarts. You could get Home Assistant to restart EMS-ESP at scheduled times by sending an MQTT message with a 'restart' payload.

giovanne123, 2020-04-19:
I'm not sure about the details when I saw the ems-esp AP in my Wifi environment.
But as I can remember, I think the EMS-ESP was also accessible / connected to the WLAN still/again, because for restarting I connected to telnet over wifi and restarted the EMS-ESP. Also HA was still receiving MQTT messages from the EMS-ESP. After restart the ems-esp AP was gone.

Can the ESP8266/Wemos be running in mode client and AP at the same time?

Then I think, that maybe the EMS-ESP lost the connection to my WLAN, opened/started the AP but also after a while reconnects to WLAN in client mode.
Because of that I was wondering and found it not that fine that you can connect to the EMS-ESP's AP without auth. and that the EMS-ESP is in parallel connected to the WLAN...

MichaelDvP, 2020-04-19:
@giovanne123 the esp can run in both modes (see Wifi.mode(WIFI_AP_STA)), but i don't think this happens here. The EMS-ESP uses a lib [jw](https://github.com/xoseperez/justwifi) and create a fallback to AP in startup, but switch off the fallback on connect. So after successful connect the AP should not appear.  
Only a reboot (planed or unplaned) makes a AP possible if the connection failed. If it happens again check the [system] for reboot time and reason. I think you had a crash and reconnect problems on reboot. I'm not sure, but i expect that the AP disappear after successful reconnect. Maybe it was on the scanned-network-list, but not accessible any more after the reconnect.  

@proddy on a fallback AP it should not be possible to readout any parameters from ems-esp. The AP should be limited to set a new Station SSID and password and, if new parameters are set, makes a factory reset to all other parameters.

proddy, 2020-04-29:
I've seen EMS-ESP connect to a WiFi SSID and also appear as an AP. I'm not sure why or able to reproduce it so never looked into it. Version 2.0 will have my new code that fixes all of this.

closed 2020-04-29 by proddy