emsesp/EMS-ESP #369: Shower boolean for MQTT

dutchrazor, 2020-04-18:
I hooked up the shower fan to Home Assistant via ESPHome.

Now I was wondering if it would be possible to use the shower detection feature already present in the code to have a "Shower active" boolean via MQTT?

One could also look at the tap water usage, but since the MQTT update rate is much lower than the EMS rate I think it would be more accurate if it is done on the ESP. ( shower duration already seems to work everytime)

proddy, 2020-04-19:
It's easy to add and I'll do it for version 2.0 which I'm currently updating. And at the same time clean up the code a little. I was thinking then of a single topic called shower_status with JSON payload {'on' or 'off', duration}. Would that work for your ESPhome implementation?

dutchrazor, 2020-04-19:
Sounds good! Then I can simply put a trigger on it and potentially also add the mirror heater on it.

Thanks.

proddy, 2020-04-29:
It's implemented, now testing

proddy, 2020-05-11:
in v2

dutchrazor, 2021-01-23:
Hello,

Sorry for the late testing.. 

I installed stable v2.2.0 and tested the shower boolean. Unfortunately it does not seem to show up as a binary sensor in home assistant.

I see for example homeassistant/binary_sensor/tapwater_active on startup, and after turning on the shower for a while  the ems-esp/shower_data and ems-esp/shower_active topics. But the binary sensor topic for shower_active seems to not appear.

Here is a complete list of the topics available. 

Is there something I could be doing wrong?

`ems-esp/boiler_data
ems-esp/boiler_data_ww
ems-esp/heartbeat
ems-esp/heating_active
ems-esp/info
ems-esp/shower_active
ems-esp/shower_data
ems-esp/status
ems-esp/tapwater_active
ems-esp/thermostat_data
homeassistant/binary_sensor/ems-esp/heating_active/config
homeassistant/binary_sensor/ems-esp/tapwater_active/config
homeassistant/climate/ems-esp/thermostat_hc1/config
homeassistant/sensor/ems-esp/boiler/config
homeassistant/sensor/ems-esp/boiler_UBAuptime/config
homeassistant/sensor/ems-esp/boiler_auxElecHeatNrgConsDHW/config
homeassistant/sensor/ems-esp/boiler_auxElecHeatNrgConsHeating/config
homeassistant/sensor/ems-esp/boiler_auxElecHeatNrgConsTotal_/config
homeassistant/sensor/ems-esp/boiler_boilHystOff/config
homeassistant/sensor/ems-esp/boiler_boilHystOn/config
homeassistant/sensor/ems-esp/boiler_boilTemp/config
homeassistant/sensor/ems-esp/boiler_burnGas/config
homeassistant/sensor/ems-esp/boiler_burnMaxPower/config
homeassistant/sensor/ems-esp/boiler_burnMinPeriod/config
homeassistant/sensor/ems-esp/boiler_burnMinPower/config
homeassistant/sensor/ems-esp/boiler_burnStarts/config
homeassistant/sensor/ems-esp/boiler_burnWorkMin/config
homeassistant/sensor/ems-esp/boiler_coolingStarts_/config
homeassistant/sensor/ems-esp/boiler_curBurnPow/config
homeassistant/sensor/ems-esp/boiler_curFlowTemp/config
homeassistant/sensor/ems-esp/boiler_exhaustTemp/config
homeassistant/sensor/ems-esp/boiler_fanWork/config
homeassistant/sensor/ems-esp/boiler_flameCurr/config
homeassistant/sensor/ems-esp/boiler_heatPump/config
homeassistant/sensor/ems-esp/boiler_heatWorkMin/config
homeassistant/sensor/ems-esp/boiler_heatingActivated/config
homeassistant/sensor/ems-esp/boiler_heatingStarts/config
homeassistant/sensor/ems-esp/boiler_heatingTemp/config
homeassistant/sensor/ems-esp/boiler_ignWork/config
homeassistant/sensor/ems-esp/boiler_maintenance/config
homeassistant/sensor/ems-esp/boiler_maintenanceMessage/config
homeassistant/sensor/ems-esp/boiler_nrgConsCompCooling/config
homeassistant/sensor/ems-esp/boiler_nrgConsCompHeating/config
homeassistant/sensor/ems-esp/boiler_nrgConsCompTotal/config
homeassistant/sensor/ems-esp/boiler_nrgConsCompWw/config
homeassistant/sensor/ems-esp/boiler_nrgConsTotal/config
homeassistant/sensor/ems-esp/boiler_nrgSuppCooling/config
homeassistant/sensor/ems-esp/boiler_nrgSuppHeating/config
homeassistant/sensor/ems-esp/boiler_nrgSuppTotal_/config
homeassistant/sensor/ems-esp/boiler_nrgSuppWw/config
homeassistant/sensor/ems-esp/boiler_outdoorTemp/config
homeassistant/sensor/ems-esp/boiler_pumpDelay/config
homeassistant/sensor/ems-esp/boiler_pumpMod/config
homeassistant/sensor/ems-esp/boiler_pumpMod2/config
homeassistant/sensor/ems-esp/boiler_pumpModMax/config
homeassistant/sensor/ems-esp/boiler_pumpModMin/config
homeassistant/sensor/ems-esp/boiler_retTemp/config
homeassistant/sensor/ems-esp/boiler_selBurnPow/config
homeassistant/sensor/ems-esp/boiler_selFlowTemp/config
homeassistant/sensor/ems-esp/boiler_serviceCodeNumber/config
homeassistant/sensor/ems-esp/boiler_setBurnPow/config
homeassistant/sensor/ems-esp/boiler_setFlowTemp/config
homeassistant/sensor/ems-esp/boiler_switchTemp/config
homeassistant/sensor/ems-esp/boiler_sysPress/config
homeassistant/sensor/ems-esp/boiler_upTimeCompCooling/config
homeassistant/sensor/ems-esp/boiler_upTimeCompHeating/config
homeassistant/sensor/ems-esp/boiler_upTimeCompWw/config
homeassistant/sensor/ems-esp/boiler_upTimeControl/config
homeassistant/sensor/ems-esp/boiler_wWActivated/config
homeassistant/sensor/ems-esp/boiler_wWActive/config
homeassistant/sensor/ems-esp/boiler_wWChargeType/config
homeassistant/sensor/ems-esp/boiler_wWCharging/config
homeassistant/sensor/ems-esp/boiler_wWCirc/config
homeassistant/sensor/ems-esp/boiler_wWCircPump/config
homeassistant/sensor/ems-esp/boiler_wWCircPumpMode/config
homeassistant/sensor/ems-esp/boiler_wWCurFlow/config
homeassistant/sensor/ems-esp/boiler_wWCurTemp/config
homeassistant/sensor/ems-esp/boiler_wWCurTemp2/config
homeassistant/sensor/ems-esp/boiler_wWDisinfecting/config
homeassistant/sensor/ems-esp/boiler_wWDisinfectionTemp/config
homeassistant/sensor/ems-esp/boiler_wWHeat/config
homeassistant/sensor/ems-esp/boiler_wWOneTime/config
homeassistant/sensor/ems-esp/boiler_wWRecharging/config
homeassistant/sensor/ems-esp/boiler_wWSelTemp/config
homeassistant/sensor/ems-esp/boiler_wWSetPumpPower/config
homeassistant/sensor/ems-esp/boiler_wWSetTemp/config
homeassistant/sensor/ems-esp/boiler_wWStarts/config
homeassistant/sensor/ems-esp/boiler_wWStarts2/config
homeassistant/sensor/ems-esp/boiler_wWTempOK/config
homeassistant/sensor/ems-esp/boiler_wWType/config
homeassistant/sensor/ems-esp/boiler_wWWorkM/config
homeassistant/sensor/ems-esp/boiler_wwBufferTemperature/config
homeassistant/sensor/ems-esp/boiler_wwMixTemperature/config
homeassistant/sensor/ems-esp/boiler_wwStorageTemp1/config
homeassistant/sensor/ems-esp/boiler_wwStorageTemp2/config
homeassistant/sensor/ems-esp/status/config
homeassistant/sensor/ems-esp/thermostat/config
homeassistant/sensor/ems-esp/thermostat_errorcode/config
homeassistant/sensor/ems-esp/thermostat_hc1_hatemp/config
homeassistant/sensor/ems-esp/thermostat_hc1_mode/config
homeassistant/sensor/ems-esp/thermostat_hc1_seltemp/config
homeassistant/sensor/ems-esp/thermostat_time/config`

proddy, 2021-01-23:
I'll need to check the code again

proddy, 2021-01-23:
Note the binary sensor in HA is only created when the shower is first detected, and it's not 'retained' so if you delete or restart the MQTT broker it will  be lost.

rkuijer, 2021-02-16:
Dear @proddy, I don't see any config for the shower_active and the shower_data.
[image]
The payload is there but no config.
I'm running version v2.2.1b2

proddy, 2021-02-16:
I think this is because it tries to send out the MQTT config to HA before the connection with MQTT has been established. I'll need to try it out and put in a fix.

commit 6f170133 (2021-02-20): fix MQTT payloads being rejected when MQTT is connecting (fixes #369)
commit 82af92c9 (2021-02-21): queue mqtt even if its not connected yet (fixes #369)
closed 2021-02-21 by proddy