emsesp/EMS-ESP #371: Compilation issue with latest platform version

moustic999, 2020-04-21:
Dear,

with the update to latest Espressif 8266 2.4.0 I get compilation issue.
I tried back with the 2.3.0 and it works fine.

here is the log for 2.4.0 with errors :

> Executing task: C:\Users\guest\.platformio\penv\Scripts\platformio.exe run <

Processing release (board: d1_mini; framework: arduino; platform: espressif8266)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
** Building web...
node ./tools/webfilesbuilder/node_modules/gulp/bin/gulp.js --silent --cwd ./tools/webfilesbuilder
Creating file: index.html.gz
Creating file: glyphicons-halflings-regular.woff.gz
Creating file: myesp.html.gz
Creating file: required.css.gz
Creating file: required.js.gz
Creating file: myesp.js.gz
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 2.4.0 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20603.200130 (2.6.3)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.191122 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <CircularBuffer> 1.3.3 #98e638a
|-- <OneWire> 2.3.5 #8539273
|-- <ESP Async WebServer> 1.2.3 #f13685e
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson> 6.15.1 #daa279d
|-- <ESPAsyncUDP> #697c75a
|-- <uuid-common> 1.1.0
|-- <uuid-log> 2.1.1
|   |-- <uuid-common> 1.1.0
|-- <uuid-syslog> 2.0.4
|   |-- <uuid-common> 1.1.0
|   |-- <uuid-log> 2.1.1
|   |   |-- <uuid-common> 1.1.0
|   |-- <ESP8266WiFi> 1.0
|-- <JustWifi> 2.0.2
|   |-- <ESP8266WiFi> 1.0
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|-- <EEPROM_Rotate> 0.9.2
|   |-- <EEPROM> 1.0
|-- <ArduinoJson> 6.15.1 #daa279d
|-- <ESPAsyncTCP> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <Ticker> 1.0
Building in release mode
LINKFLAGS = -T eagle.flash.4m1m.ld -Os -nostdlib -Wl,--no-check-sections -Wl,-static -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read -u app_entry -u _printf_float -u _scanf_float -u _DebugExceptionVector -u _DoubleExceptionVector -u _KernelExceptionVector -u _NMIExceptionVector -u _UserExceptionVectors       
Compiling .pio\build\release\src\MyESP.cpp.o
Compiling .pio\build\release\src\Ntp.cpp.o
Compiling .pio\build\release\src\TelnetSpy.cpp.o
Compiling .pio\build\release\src\TimeLib.cpp.o
In file included from C:\Users\guest\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include/user_interface.h:31:0,
                 from src\TelnetSpy.cpp:12:
C:\Users\guest\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include/ipv4_addr.h:29:23: fatal error: lwip/init.h: No such file or directory
 #include <lwip/init.h>
                       ^
compilation terminated.
In file included from .pio\libdeps\release\ArduinoJson\src/ArduinoJson/Configuration.hpp:121:0,
                 from .pio\libdeps\release\ArduinoJson\src/ArduinoJson.hpp:7,
                 from .pio\libdeps\release\ArduinoJson\src/ArduinoJson.h:9,
                 from src\MyESP.h:14,
                 from src\MyESP.cpp:10:
C:\Users\guest\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:286:26: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://platformio.org/lib/search?query=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
                          ^
compilation terminated.
In file included from C:\Users\guest\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/ESP8266WiFi.h:31:0,
                 from src\Ntp.h:10,
                 from src\Ntp.cpp:5:
C:\Users\guest\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/IPAddress.h:27:23: fatal error: lwip/init.h: No such file or directory
 #include <lwip/init.h>
                       ^
compilation terminated.
In file included from src\TimeLib.h:9:0,
                 from src\TimeLib.cpp:6:
C:\Users\guest\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:286:26: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://platformio.org/lib/search?query=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
                          ^
compilation terminated.
*** [.pio\build\release\src\TelnetSpy.cpp.o] Error 1
*** [.pio\build\release\src\Ntp.cpp.o] Error 1
*** [.pio\build\release\src\TimeLib.cpp.o] Error 1
*** [.pio\build\release\src\MyESP.cpp.o] Error 1
===================================================================== [FAILED] Took 14.68 seconds =====================================================================

Environment    Status    Duration
-------------  --------  ------------
travis         IGNORED
esp12e         IGNORED
d1_mini        IGNORED
nodemcuv2      IGNORED
nodemcu        IGNORED
debug          IGNORED
release        FAILED    00:00:14.678
================================================================ 1 failed, 0 succeeded in 00:00:14.678 ================================================================ 
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

proddy, 2020-04-21:
Is this EMS-ESP 1.9.4 from the Master branch? I haven't touched that since last year - 1.9.5 is the latest dev with all the fixes. I really should just merge it back into Master soon.

moustic999, 2020-04-21:
this is the latest dev, I always use the dev branch

moustic999, 2020-04-21:
version returned with worikin framework 2.3.0 is EMS-ESP-1_9_5b58-d1

proddy, 2020-04-21:
I can't reproduce? I'll take another look.

proddy, 2020-04-21:
tried it on clean Win10 and Ubuntu, compiles fine. Think it must be something in your environment?