emsesp/EMS-ESP #372: Solar Module Energy Total

PhillyGilly, 2020-04-24:
Due to the unseasonally sunshine over the UK my solar system has generated a lot of hot water but the Energy Total reported has disappeared.
[image]
The data is in telegram 0x28E at positions 10+11 but it isn't being used.
[log24042020.txt](https://github.com/proddy/EMS-ESP/files/4526810/log24042020.txt)

proddy, 2020-04-24:
found the error, made a fix, but can't test it. It's b60

PhillyGilly, 2020-04-24:
Hello @proddy I've downloaded the latest version from Github 
The test says Version 1.9.5b60 but the file name is EMS-ESP-dev-d1_mini1-9-5b58.
The 0x28E telegram is sent once per hour so I'll feed back in a bit.
Thanks.

MichaelDvP, 2020-04-24:
Are you sure that "Energy Total" is 16 bit, that's roll over at only 6553 kWh?  I think it's 32 bit at pos 8. Also for "Energy today" in Wh a 16 bit in pos 6 seems to small, 32 bit at pos 4 is also possible.

proddy, 2020-04-24:
@MichaelDvP I think you're right. It must be 32-bit (a 4 byte code). Any way we can confirm this?

proddy, 2020-04-24:
@PhillyGilly ignore the b58 bin file, it was leftover from the last PR. The bug was how the data is rendered on the screen and in the MQTT. After you've uploaded the new firmware it should be shown correctly when doing an `info`

MichaelDvP, 2020-04-24:
> Any way we can confirm this

I don't have a solar module. Maybe @PhillyGilly can test if a write to byte 9 changes the total energy in his thermostat display (not in ems-esp)
`send 0B 30 FF 09 02 8E 01`
should set byte9 to 1 and 0x00018346 give a display of 9914.2 kWh
`send 0B 30 FF 09 02 8E 00`
to revert back to actual value 
@proddy, please check for correct commands, i'm not so familiar with EMS+ (why SM100 sends B0 instead of 30 as src)

PhillyGilly, 2020-04-24:
> @PhillyGilly ignore the b58 bin file, it was leftover from the last PR. The bug was how the data is rendered on the screen and in the MQTT. After you've uploaded the new firmware it should be shown correctly when doing an `info`

Sorry no joy here.
[log24042020-2.txt](https://github.com/proddy/EMS-ESP/files/4528586/log24042020-2.txt)

PhillyGilly, 2020-04-24:
> I don't have a solar module. Maybe @PhillyGilly can test if a write to byte 9 changes the total energy in his thermostat display (not in ems-esp)
> `send 0B 30 FF 09 02 8E 01`
> should set byte9 to 1 [...]

@MichaelDVP That sounds like a plan for later when I can set up my laptop nearer to the local display in my plant room. 
Meanwhile I just copied the telegram structure from the Wiki and I agree that the total is probably rolling over. I will try to catch the next 0x28e and then compare the delta today to the delta total and see it that helps.

MichaelDvP, 2020-04-24:
No, it doesn't help, the higher bytes are always 00, you need more panels solar or more sun ;-)
Reading is with 32 bit:
`B0 00 FF 00 02 8E 00 00 00 00 00 00 00 00 00 00 83 46`
`-- -- -- -- -- --  00 00 00 00` last hour 0.0 Wh
 `-- -- -- -- -- --  -- -- -- -- 00 00 00 00` Today 0 Wh
 `-- -- -- -- -- --  -- -- -- -- -- -- -- --00 00 83 46` Total 3360.6 kWh
and 
`B0 00 FF 00 02 8E 00 00 2C 79 00 00 0A AB 00 00 83 61`
`-- -- -- -- -- --  00 00 2C 79` last hour 1138.5 Wh
 `-- -- -- -- -- --  -- -- -- -- 00 00 0A AB` Today 2731 Wh
 `-- -- -- -- -- --  -- -- -- -- -- -- -- -- 00 00 83 61` Total 3363.3 kWh

The telegram starts with B0, is it a Junkers SM100?
Then the send command has to be 
`send 8B 30 FF 09 02 8E 01`
But you will see in the telegram if writing works. The telegram after the command should be 
`B0 00 FF 00 02 8E 00 00 2C 79 00 00 0A AB 00 01 83 61`
 `-- -- -- -- -- --  -- -- -- -- -- -- -- -- -- ^^`

PhillyGilly, 2020-04-24:
> No, it doesn't help, the higher bytes are always 00, you need more panels solar or more sun ;-)
> Reading is with 32 bit:
> `B0 00 FF 00 02 8E 00 00 00 00 00 00 00 00 00 00 83 46`
> `-- -- -- -- -- -- 00 [...]

The solar module is branded as a Bosch MS200 but I think it is a Junkers item.
Thank you for the explanation about the four bytes. That's really clear. 
[SM200analysis.xlsx](https://github.com/proddy/EMS-ESP/files/4529454/SM200analysis.xlsx)
My (more laborious) Excel comes to the same conclusion that Energy total is being correctly shown in bytes 10+11. So the data is in the telegram but not getting any further.

I will try the send method to see if bytes 8+9 are used for energy totals above 6554kWh when the system isn't running.

MichaelDvP, 2020-04-25:
@PhillyGilly i looked up the data of solar panels and calculated that 16bit fields for last hour and today will allow a maximum system of 12 panels. Since the leading bytes stays zero in night/standby and day/working i amm sure that all fields are 32 bit.
@proddy b60 rendering does not help. _setValue() masks out in line 398 the high values. 
But short/ushort values for missing sensors can be 0x7D00, 0x8000 and 0x8300, i have all these sensor values in the boiler fast telegram 0x18 and rendering should display all this values as "?". 

My boiler monitor-fast:
`08 0B 18 00 20 02 02 64 00 01 01 00 00 02 58 7D 00 80 00 00 00 FF 30 59 00 00 FF 00 00 83 00`
Sensors in pos 9 (0x0258), pos 11 (0x7D00), pos 13 (0x8000) and pos 25 (0x8300).

proddy, 2020-04-25:
@MichaelDvP so we need to adjust the code to support 4-byte 32-bits for the energy calculations and also change the _setValue function in ems.cpp to only show in invalid value if its 7D00, 8000 and 8300?

I'll also need to make these modifications to v2 on EMS-ESP. I'm still fixing some memory leaks. If you want to take a peek let me know and I'll give you access to the project in github.

MichaelDvP, 2020-04-25:
If you like, i can make this change to 1.9.5b60 and make a PR, so you can concentrate n v2.  
And yes, i'm very interrested to take a peek at v2,

PhillyGilly, 2020-04-25:
@MichaelDvP I ran the suggested tests today.
Here is my local display at 12:29
[image]
I ran the command send 0B 30 FF 09 02 8E 00 AND I also ran 08 30 FF 09 02 8E 00 to cover both bases. There was no change in the local display. From previous checking, I think the local display only updates hourly and as we were looking for the 0x28E telegram which is published just after the hour bar, I started logging at 12:58 and captured this 
(13:03:05) Solar Module -> All, SM100Energy(0x028E), telegram: B0 00 FF 00 02 8E 00 00 41 3A 00 00 08 19 00 00 83 A9 (#data=12) 
The local display also disappointed.
[image]

With this amount of sunshine I will probably get up to 6555kWhthis year so we will know if it rolls over or not :-)

MichaelDvP, 2020-04-25:
Since it is Junkers protokoll you have to send with first MSB set
`send 8B 30 FF 09 02 8E 01`
and
`send 8B 30 FF 09 02 8E 00`
turn log v on before to see the response. Maybe the telegram is nor writeable.

PhillyGilly, 2020-04-26:
> Since it is Junkers protokoll you have to send with first MSB set
> `send 8B 30 FF 09 02 8E 01`
> and
> `send 8B 30 FF 09 02 8E 00`
> turn log v on before to see the response. Maybe the telegram is nor writeable.

@MichaelDvP thank you for the codes.
I wasn't able to set up my laptop near the local display so I sent the first code from my PC, set log v, and than went up five flights of stairs to see if there was any impact on the local display. There wasn't it read 3383.7kWh. I waited until after the hour bar and checked again, it read 3385.6kWh. FWIW I noticed that the today value changed from 1.7kWh to 3.6kWh at the same time.  When I came back to my PC I stopped the logging and send the second command.
However there was no 0x"*E telegram in the log because it only contained the last three minutes data and I was too slow. But I think you are right that this telegram is not writable.

MichaelDvP, 2020-04-27:
@PhillyGilly ok, no write to this telegram, it's just am monitor message.
But anyway, reading the telegram with 32 bits can do no harm. You can try [SM_energy built](https://github.com/MichaelDvP/EMS-ESP/tree/SM_energy), if it works i'll make a PR.

PhillyGilly, 2020-04-27:
@MichaelDvP Bingo! Thank you.

Solar Module data:                                                                Solar module: SM200 Solar Module (DeviceID: 0x30, ProductID: 164, Version: 25.04)                                                                               Collector temperature (TS1): 29.8 C                                             Bottom temperature (TS2): 47.6 C                                                Bottom temperature (TS5): 30.4 C                                                Pump modulation: 0 %                                                            Valve (VS2) status: off                                                         Pump (PS1) active: off                                                          Energy last hour: 0.0 Wh                                                        Energy today: 0 Wh                                                              Energy total: 13.2 kWh

MichaelDvP, 2020-04-27:
Ups, the _setValue  for uint32_t reads only 24 bits. I'll look if we need an new function for 32 bit.

MichaelDvP, 2020-04-27:
Should work now (time-values are all 3 bytes long, i added a function for 4 bytes).

PhillyGilly, 2020-04-27:
(13:02:20) Solar Module -> All, SM100Energy(0x028E), telegram: B0 00 FF 00 02 8E 00 00 6C B6 00 00 12 F9 00 00 84 93 (#data=12)
Solar Module data:                                                                Solar module: SM200 Solar Module (DeviceID: 0x30, ProductID: 164, Version: 25.04)                                                                               Collector temperature (TS1): 65.9 C                                             Bottom temperature (TS2): 48.7 C                                                Bottom temperature (TS5): 55.1 C                                                Pump modulation: 100 %                                                          Valve (VS2) status: on                                                          Pump (PS1) active: on                                                           Energy last hour: 2783.0 Wh                                                     Energy today: 4857 Wh                                                           Energy total: 3393.9 kWh

@MichaelDvP That's really good. Thank you.
One last question, why is Energy today zero decimal places when both other values are one decimal?

MichaelDvP, 2020-04-27:
@PhillyGilly both values are in Wh, Energy total is in kWh. Since the sun shine up to 10-12 hours a day, the day field should hold a larger amount of Wh, that's why bosch stores the day-value without decimals.

PhillyGilly, 2020-04-28:
That makes some sense.
FWIW I checked back to #267 and looked at the values in December.

https://github.com/proddy/EMS-ESP/issues/267#issuecomment-569908825

proddy, 2020-04-28:
@MichaelDvP @PhillyGilly , I just wanted to let you guys know I appreciate all the work you're doing on fine-tuning the solar module logic. I am watching and porting all the great changes Michael is doing to my version 2.0 which is coming along nicely.

commit 1a5be8e4 (2020-04-28): Merge pull request #376 from MichaelDvP/SM_energy

SM200 energy to 32 bit, fixes #372
commit 487cda08 (2020-04-28): SM200 energy to 32 bit, fixes #372
closed 2020-04-30 by proddy