emsesp/EMS-ESP #373: Add thermostat summer/holiday mode to MQTT topic

proddy, 2020-04-25:
As requested by @majdzik84, include the summer mode in the `thermostat_data` MQTT topic for RC35/30 type models.

Question: how best to represent this? 
1. program=summer | holiday
2. summermode = true | false and holidaymode = true | false
3. something else?

majdzik84, 2020-04-25:
2. summermode = true | false

MichaelDvP, 2020-04-25:
Why not using the existing fields "mode" or (imho better) "modetype"?
`{"mode":"auto","modetype":"day"|"night"|"summer"|"holiday"|"party"}`

majdzik84, 2020-04-25:
It seems to me that it should not be classified in this way because summer mode can occur both day and night. I have never used holiday and party but I suppose it works in a similar way.

Personally, I'd rather have information that 
mode: auto, modetype: day | night is on
in addition, the stove works in summer mode

MichaelDvP, 2020-04-25:
AFAIK:
- summer-mode is switched on if outside temperature reaches a given level. Than heating and flow is turned off, regardless of day/night setting. It overrides mode settings and modetype. If outside temperature falls, mode and modetype are recovered. 
- holiday-mode sets the temperature to holiday-temperature regardless of day or night, it overrides mode setting.
- party-mode sets day-temp for one night, it overrides the auto-night-setting.

We can change the mode by mqtt or telnet, but not modetype, this is info from the thermostate. So i think modetype is the right field.

majdzik84, 2020-04-25:
Thanks for the clarification.

MichaelDvP, 2020-04-27:
Just to complete/correct the explanation, i've checked manual and ems-doku:
- summer-mode is triggert by the damped outdoor temperature (telegram A3, pos 0 from the thermostat). The switching temperature is set in hc-setting (3D, 47, etc.) pos 0x16.
- holiday-mode is triggert by date, you can set start and end of holiday, The dates are in telegram 3F from thermostat in pos 0x57 - 0x5C: holiday-start: day, month, year, holiday-end: day, month, year. 
- party-mode is triggert by setting a countdown-time in hours in telegram 3F, pos 0x56 and sets daytemp for this time. After this time the thermostat goes back to setting before (auto,day,night).
- pause-mode is triggert by  setting a countdown-time in hours in telegram 3F, pos 0x55 and sets nighttemp for this time. After this time the thermostat goes back to setting before (auto,day,night).

proddy, 2020-05-13:
I've added summer/holiday mode to mqtt in v2 as suggested above. 

@MichaelDvP the logic in ems.cpp:1265 that ignores the set_point temperature if there is no HC active (e.g. in summer mode) - is this a good thing or not?

MichaelDvP, 2020-05-13:
If you mean: 
`    if (EMS_RxTelegram->data[14] == 0x00) { `
this is needed, because otherwise the function `_getHeatingCircuit(EMS_RxTelegram)` will set the hc active. But this data is the calculated flow setpoint and in summer mode it is going to minimum, which is set in telegram 0x3D, pos 16 (default set to 5°C). According to manual this parameter is only active if RC35 is set to weather-controlled, i'm not sure what it is in room-controlled mode.
I think it's much better only to use pos 0 (heating-type) in 0x3D to determine if a hc is active, than you can skip these lines. (remember: Junkers logic uses also the `_getHeatingCircuit` to dertermine the active hc.

proddy, 2020-05-15:
ok, I'll think of how to do this properly in v2. The only drawback with 0x3D is that you have to query it.

closed 2020-05-15 by proddy