emsesp/EMS-ESP #375: Add relay on esp pin to control hot water pump

pitulek, 2020-04-28:
Hi!

I have Logomax 052u-28 and there is external 200 liters boiler for hot water,also  for the hot water external pump.
Is it possible to add  for any pin of esp8266 to control it  by relay to turn on /off the pump.
Your soft is working very good for thermostat rc20n, but i can't control  programm for auto  by mqtt AD 0B XX.  
Thank you very much for your response.
[image]

proddy, 2020-04-28:
Yes, it would be possible. Code-wise its pretty simple. I suggest controlling on/off via MQTT. The tricky thing is connecting the wires from BBQKees' gateway board to an external relay board (5v, GND, GPIO). Easiest I think would be just to use the existing D5 (used currently for external dallas sensors) as these wires are externally exposed. Have a think about what you want to do and we'll see if we can make it.

MichaelDvP, 2020-04-29:
I'm not sure that a bus powered device has enough spare power to drive a relais. Also you need a extra isolation case, as you are handling with 230VAC power. Wouldn't it be easier and cheaper to buy a sonoff or smart plug and flash it with tasmota? With some of the smart-plugs you can also monitor the pump power consumption. (I personally like the very small Gosund SP111, working good with tasmota, also tuya-convert works). 
I like the idea to have a equivalent to the telnet-send-command in mqtt. It makes it flexible to change setting not used commenly. I's easy to add, I suggest the topic generic_cmd. For your purpose its than ` {"cmd":"send","data":"0B 10 AD 0B XX"}`

proddy, 2020-04-29:
thanks @MichaelDvP, yes I was thinking of a separate relay board but you're right using an independent smart board running tasmota is much more versatile.

I've added the 'send' command to MQTT in EMS-ESP v2. Check it out.

pitulek, 2020-04-29:
Thank you very much for your help. You are great!

MichaelDvP, 2020-04-29:
I thought of such a relais:
[image]
It's 3.3V and i measured the current, it draws 140 mA, but i don't know what EMS and the esp-reglator can drive.
@proddy v2 changes are not pushed yet? 
I added to the 1.9.5, @pitulek you can test at [my forc](https://github.com/MichaelDvP/EMS-ESP/tree/mqtt_send) to see if it fit your demands.

pitulek, 2020-04-29:
I have relay module like this dedicated for ESP WEMOS D1 MINI with transitor to drive it but no transoptor there(could be not safe for EMS module,if something go wrong with 230v and thank you for this notice. Second module is with transoptor, and can have separate vcc, it is safe with transoptor.Just all i need is state on/off it  the mqqt topic, and comannd to change the state on pin of Wemos d1 mini.
I can do it with extra esp wemos with espeasy soft, but it could be nice to have all in one :)

Also i have question and sorry tht i bother you
How to send command from homeassistant? : 

type: button
tap_action:
  action: call-service
  service: mqtt.publish
  topic: ems-esp\thermostat_cmd
  payload: {\ "cmd\":\"daytemp\", \"data\":\ 20\ }
i always get this :

required key not provided @ data['topic']

thermostat is working fine, but single command not, what i am doing wrong?

The script from wiki onetime water does not work for me too, get the same message.

Thank you for your help and time.
[image]
[image]

proddy, 2020-04-29:
for the MQTT publish from HA make sure the payload is formatted correctly:

`payload: "{\"cmd\":\"daytemp\",\"data\":20}"`

MichaelDvP, 2020-04-29:
@pitulek my concerns were not about isolation in/out of the relais, but interference of 230VAC with human body. 
Anyway, if you want to use this relais you can test my forc: i've added gpio4 (D2) and gpio5 (D1) to generic_cmd {"cmd":"gpio4", "data":1} to switch on(1)/off(0). Other free gpios are 1, 3, 12 if you do not change LED and Dallas. I do not check collision with LED and Dallas.
I've never used HA, but topic path is normaly "home/ems-esp/thermostat_cmd" ?

bbqkees, 2020-04-30:
You could indeed use D5 for a relay and for that purpose also the JST header on the Gateway.
The JST header outputs D5, GND and 3V3. 
The bus cannot handle another relay if the Gateway is powered via the bus.
If its powered via USB or the service jack its no problem.
A bus powered Gateway can control a relay board with an optocoupler as long as the coil of the relay is powered externally.
Also keep in mind the 3V3 of the JST header is usually not enough for a relay as they need 5V.

pitulek, 2020-04-30:
Thank you very much for all! So i try to test it now.

bbqkees, 2020-04-30:
Be careful with those relay boards when switching a 230VAC or other mains voltage!
F.i. the Wemos compatible relay board is very dangerous.
There is not enough isolation space between the high and low level voltages and the PCB of coarse exposes the mains voltage on the solder joints.
[image]

Actually, all those boards are dangerous when used without an enclosure.

pitulek, 2020-05-04:
Thank you! I use board with optocoupler and much bigger with separate voltage 5v, and put all in enclosure.I  know how dengerous is 230v for human body.

proddy, 2020-05-16:
closing this. if you have any results to share @pitulek please do!

closed 2020-05-16 by proddy