emsesp/EMS-ESP #377: Bosch Compress 6000AW

Vikingn, 2020-04-29:
Hi ! I have tried to connect my heatpump today and get some info, but lack some. Saw it was another case with some similiar.
Autodetect show this
New EMS device recognized as a UBAMaster: Nefit Enviline (DeviceID:0x08 ProductID:172 Version:01.09)
New EMS device recognized as a Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID:0x10 ProductID:158 Version:31.03)
New EMS device recognized as a Gateway: Web Gateway KM200 (DeviceID:0x48 ProductID:189 Version:04.06)
[image]

EMS-ESP system stats:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: disabled, Shower Alert: disabled

EMS Bus stats:
  Bus is connected, protocol: HT3
  Rx: # successful read requests=408, # CRC errors=0
  Tx: Last poll=4.374 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: Nefit Enviline (DeviceID:0x08 ProductID:172 Version:01.09)
  Hot tap water: running
  Central heating: off
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water selected temperature: ? C
  Warm Water desired temperature: ? C
  Warm Water current temperature: ? C
  Warm Water current tap water flow: ? l/min
  Warm Water # starts: ? times
  Warm Water 3-way valve: off
  Selected flow temperature: 27 C
  Current flow temperature: 25.1 C
  Return temperature: ? C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 11 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: ? bar
  System service code:
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Outside temperature: 5.3 C
  Boiler temperature: ? C
  Pump modulation: 10 %
  Burner # starts: 0 times
  Total burner operating time: 0 days 0 hours 0 minutes
  Total heat operating time: 0 days 0 hours 0 minutes
  Total UBA working time: 1575 days 19 hours 28 minutes

Thermostat stats:
  Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID:0x10 ProductID:158 Version:31.03)
  Thermostat time is 22:50:22 29/04/2020
  Heating Circuit 1
   Current room temperature: ? C
   Setpoint room temperature: 20.5 C
   Mode is set to manual
   Day Mode is set to day

Do you need the verbose logg ?

proddy, 2020-04-30:
Could you upgrade to 1.9.5 and see if more information is captured? Also what type of information are you looking for?

Vikingn, 2020-04-30:
Its 

Warm Water activated: ?
Warm Water circulation pump available: ?
Warm Water selected temperature: ? C
Warm Water desired temperature: ? C
Warm Water current temperature: ? C
Return temperature: ? C
System pressure: ? bar
Heating temperature setting on the boiler: ? C
Boiler temperature: ? C

If you look in this document http://husdata.se/Download/o309fq2jio/H1%20Manual%2020.01.pdf

En see for model Rego 2000 or type I  see on page 8,9 and bottom on page 11 what should be possible to get out.
see the pic here how it can look like in home assistant and what temps etc you can get.
http://husdata.se/produkter.asp?prodid=18

This is a link to a Bosch Compress 7000 in Finland

http://husdata.se/pop_pulse.asp?ID_User=1379&DS=2&View=Pulse

See different temps, speed and if the pump is heating the house or the tap water.

Vikingn, 2020-05-01:
I have updated to 1.9.6b0
It give me a little more. This is what I get now:
When listen to mqtt in Hassio
{
    "selFlowTemp": 30,
    "selBurnPow": 17,
    "curBurnPow": 25,	This should be  Actual modulation in %
    "pumpMod": 1,
    "outdoorTemp": 3.7,
    "curFlowTemp": 35.3, This should be temp to the heating
    "wWOnetime": "on",
    "wWDesinfecting": "on",
    "wWReady": "on",
    "wWRecharge": "on",
    "wWTempOK": "on",
    "wWCirc": "off",
    "burnGas": "off",
    "flameCurr": 0,
    "heatPmp": "on",
    "fanWork": "off",
    "ignWork": "off",
    "wWHeat": "off",
    "UBAuptime": 2271353, Minutes since installed
    "burnStarts": 0,
    "burnWorkMin": 0,
    "heatWorkMin": 0
}

Changed to Auto mode on the app then I get Comfort
{
    "hc1": {
        "seltemp": 21,
        "currtemp": -3276.8,  This is because room sensor is optrional and I dont have it
        "mode": "auto",       Comment or manual gives Auto or Opt in the Bosch ios app
        "modetype": "comfort"
    }
}
The red marking is not working
[image]
[image]

Vikingn, 2020-05-01:
Here is what I get in telnet

EMS-ESP system status:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: disabled, Shower Alert: disabled
  Saved known device IDs: none

EMS Bus status:
  Bus is connected, protocol: HT3
  Rx: # successful read requests=14, # CRC errors=0
  Tx: Last poll=3.849 seconds ago, # successful write requests=0

Boiler data:
  Boiler: Nefit Enviline (DeviceID: 0x08, ProductID: 172, Version: 01.09)
  Hot tap water: running
  Central heating: off
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water circulation pump type: 3-way pump
  Warm Water circulation pump freq: 5x3min
  Warm Water selected temperature: ? C
  Warm Water desinfection temperature: ? C
  Warm Water circulation active: ?
  Warm Water set temperature: ? C
  Warm Water current temperature: ? C
  Warm water temperature (intern): ? C
  Warm water temperature (extern): ? C
  Warm Water current temperature (extern): ? C
  Warm Water current tap water flow: ? l/min
  Warm Water # starts: ? times
  Warm Water 3-way valve: off
  Selected flow temperature: 29 C
  Current flow temperature: 26.0 C
  Max boiler temperature: ? C
  Return temperature: ? C
  Gas: off
  Warm water mode: on
  Boiler pump: ?
  Fan: ?
  Ignition: ?
  Circulation pump: ?
  Burner selected max power: 28 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: ? bar
  System service code:
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Exhaust temperature: ? C
  Pump modulation: ? %
  Burner # starts: ? times
  Total UBA working time: 1577 days 9 hours 35 minutes

Thermostat data:
  Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID: 0x10, ProductID: 158, Version: 31.03)
  Thermostat time is 12:56:54 01/05/2020
  Heating Circuit 1
   Current room temperature: --3276.-8 C
   Setpoint room temperature: 21.0 C
   Mode is set to manual

on devices

These 3 were recognized by EMS-ESP:
 Boiler: Nefit Enviline (DeviceID: 0x08, ProductID: 172, Version: 01.09)
 Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID: 0x10, ProductID: 158, Version: 31.03)
 Gateway: Web Gateway KM200 (DeviceID: 0x48, ProductID: 189, Version: 04.06)

Tried this scan too

devices scan

Requesting EMS bus master for its device list and scanning for external sensors...
[EMS] Detected new EMS Device with ID 0x08. Fetching version information...
[EMS] Detected new EMS Device with ID 0x10. Fetching version information...
[EMS] Detected new EMS Device with ID 0x48. Fetching version information...
New EMS device recognized as a Bosch Boiler: Nefit Enviline (DeviceID: 0x08, ProductID: 172, Version: 01.09)
New EMS device recognized as a Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID: 0x10, ProductID: 158, Version: 31.03)
New EMS device recognized as a Gateway: Web Gateway KM200 (DeviceID: 0x48, ProductID: 189, Version: 04.06)

devices scan+

Started deep scan of EMS bus for our known devices. This can take up to 10 seconds...
New EMS device recognized as a Bosch Boiler: Nefit Enviline (DeviceID: 0x08, ProductID: 172, Version: 01.09)
New EMS device recognized as a Gateway: Web Gateway KM200 (DeviceID: 0x48, ProductID: 189, Version: 04.06)
New EMS device recognized as a Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID: 0x10, ProductID: 158, Version: 31.03)

devices save

The device IDs 08 48 10 will be automatically scanned when EMS-ESP boots up.

Something more you need ?

proddy, 2020-05-01:
well, there's a lot of information to work through here, and to be honest, this is not on top of my to-do list. As the only developer on this hobby project, I'm spending the little time I have on the core of EMS-ESP which will eventually become version 2.0. So I suggest if you have time try and understand the code a little and EMS+ logic and then slowly experiment with sniffing out the telegrams. Also consult @VincentSC who's done some investigation in the Enviline before. When we know which telegrams hold which data it's easy to add to the code base. Forget all the data values that are shown now and only concentrate on the information you're really interested in.

Vikingn, 2020-05-01:
Oh ! Thaught you was intrested in Circuit & ESP8266 code to read and write from Bosch compatible boilers, thermostats & devices that use the EMS/EMS+ bus  as the header of this project.

Can you ask him for me ? I dont see a way to contact @VincentSC  here ?

proddy, 2020-05-01:
Bosch has many brands and more than a hundred different models. Their protocols are not publically documented so all my work has been trial and error over the last 1.5 years, with the much-appreciated help from others that actually own these devices (I don't!). My time is purely voluntary and very limited. The software is free to use and extend. There is also no personal gain or profit in it for me, and only the kind appreciation from others keeps this project active.

You'll find Vincent's notes in the earlier reply I posted yesterday.

closed 2020-05-01 by proddy

VincentSC, 2020-05-01:
As my country is in a lockdown and I have my kids around all the time, I've not made much progress.

Vikingn, 2020-05-01:
> As my country is in a lockdown and I have my kids around all the time, I've not made much progress.
Its lock down ere also. Im back to work 2 days a week now after 4 weeks at home.
Do you have some info how you have worked with this before ?  Did you read what I have written ?
Understand that you do this in free time. Do you just log the verbose log first ?
What tool do you use ?

Vikingn, 2020-05-01:
We will try to  find the information in thee log.
Understand you do this in spare time.

Same for me.. and I have a friend that is retired that help me.

He is good in programming end communicatioin.

So hope we got further.

I will update you when we have some more..

Hope to hear from Vincentr.

Sendt fra E-post<https://go.microsoft.com/fwlink/?LinkId=550986> for Windows 10