emsesp/EMS-ESP #378: 1.9.5 MQTT topic - home/ems-esp/start

giovanne123, 2020-04-30:
Within 1.9.5 (master), can it be that the MQTT topic: home/ems-esp/start isn't send after reboot/system start? 
Was introduced with #348 (retain flag)

Isn't updated for my currently ({"version":"1.9.5b58","IP":"192.168.0.129","boottime":"2020-04-15T18:36:52"})

Version 1.9.5 is display right within telnet and WebUI and other MQTT topics are send correctly.

proddy, 2020-04-30:
strange. let me check...

proddy, 2020-04-30:
I do get the start topic sent. If i delete it (using MQTT Explorer) and reboot EMS-ESP it pops up again. But I did notice the retain is not working as it should.

giovanne123, 2020-04-30:
ok, I get the start topic also send after new subscribe and reboot.
But I still get old payload {"version":"1.9.5b58","IP":"192.168.0.129","boottime":"2020-04-15T18:36:52"}
instead of version: 1.9.5 or boottime: today's date&time
hm...

giovanne123, 2020-05-01:
nothing changed but after testing again today and after restarting of everything (HA; EMS-ESP; MQTT republish; ...) now the start topic reports the right data ;-)

closed 2020-05-01 by giovanne123