emsesp/EMS-ESP #379: Grab current EMS-ESP version information from github

giovanne123, 2020-05-01:
I would like to display the release/version info of EMS-ESP grabbed from github on e.g. HA.
- display current released version, e.g.: 1.9.5
- display current dev/beta version, e.g.: 1.9.6b0

So that I'm able to see directly in HA if there is a new release or dev available and don't have to go to the EMS-ESP github.

What would be the best way? Grab some url on github and analyse that?
Someone already done something similiar?

proddy, 2020-05-01:
In HA install the excellent [HACS](https://hacs.xyz/) and add the plugin [GitHub Entity Row](https://github.com/benct/lovelace-github-entity-row). 

Or use shields.io like https://img.shields.io/github/v/release/proddy/EMS-ESP.svg?include_prereleases@sort=semver

Or use the json from the URL https://api.github.com/repos/proddy/EMS-ESP/releases/latest and parse it like I do in the EMS-ESP web interface

You could also just watch releases in GitHub and it'll send you an email.

giovanne123, 2020-05-02:
OK - thanks. I will look which solution will fits best for me.
So can be closed.

proddy, 2020-05-02:
I was thinking about this. They might be an easier way in HA using sensors. For example:

sensor.yaml

```yaml
  - platform: rest
    resource: https://api.github.com/repos/proddy/EMS-ESP/releases
    name: Version
    value_template: '{{ value_json.tag_name }}'
```

For the dev version, it's a little harder and you'll need to use the body since the version is not stored (yet).

```yaml
  - platform: rest
    resource: https://api.github.com/repos/proddy/EMS-ESP/releases/tags/travis-dev-build
    name: Dev Version
    value_template: '{{ value_json.body }}'
```

I haven't tried it though. Let us know if it works!

giovanne123, 2020-05-04:
it is working, maybe there is some better regex for dev, but working for me:
```
- platform: rest
  scan_interval: 3600
  resource: https://api.github.com/repos/proddy/EMS-ESP/releases
  name: "EMS-ESP Version @ Github (latest)"
  value_template: '{{ value_json[0].tag_name }}'

- platform: rest
  scan_interval: 3600
  resource: https://api.github.com/repos/proddy/EMS-ESP/releases/tags/travis-dev-build
  name: EMS-ESP Version @ Github (dev)"
  value_template: '{{ value_json.body[8:16] }}'
```

closed 2020-05-04 by giovanne123