emsesp/EMS-ESP #382: Telegram type 0xE9

nhmariend, 2020-05-03:
Do you plan to interpret  0xE9 ?

From the log from my Bosch Compress 6000 heatpump 

###############################################################################
Type 0xE9: DHW status
Sent every 10 seonds
1 = DHW setpoint (with setpoint = "off": 0x0A = 10 ° C)              -- 63
2 - 3 = hot water actual value * 10                                  -- 57,5
4 - 5 = hot water actual value * 10                                  -- 56,1
10 = setpoint thermal disinfection                                   -- 65
11 = setpoint daily heating temperature                              -- 60
13 =
 bit 0 : DHW preparation activated by automatic program
 bit 2 : DHW preparation activated by one-time charge
 bit 3 : DHW preparation activated by thermal disinfection
 bit 4 : DHW preparation active (always 1 if bit 0, 2 or 3 also 1)
14 =
 bit 0 : Circulation day mode (or by single charge / th. disinfection activated)
 bit 2 : circulating pump on
 bit 4 : WW-preparation (when switching to daytime operation sometimes short 0, then again 1)
 bit 5 : WW preparation
15 - 17 = Warmwater preparations
18 - 20 = # hours water heating
21 = Warming water (0x64 = 100%)
24 = Set value of time program (if set = "off": 0x00)                -- 63
26 = same as byte 1, in C

UBAMaster -> All, type 0xE9, telegram:
88 00 E9 00
3F 02 3F 02 31 00 00 00 00 41 3C 00 00 00 00 00 00 00 00 00 00 00 00 3F (CRC=E4) #data=24
63 57.5  56.1              65 60                                     63

proddy, 2020-05-04:
Yes, its on the list (see https://github.com/proddy/EMS-ESP/issues/330#issuecomment-590532488) and you're information helps. I'll add it to the version 2.0 as I'm no longer extending the 1.9.x branch

nhmariend, 2020-05-04:
Nice !! We will see what we we get then.  Do you know when you will have the version 2.0 ?

proddy, 2020-05-04:
Version 2.0 is still in development, I'm adding bits slowly. Most of the core has been done and next is to add the solar and mixing modules. Then finally merge with the new Web code. It kinda works but still very much in alpha stage. If you're comfortable with C++ and building and don't mind buggy code then you're welcome to try it out...

proddy, 2020-05-04:
@nhmariend there's a lot of data in the E9 telegram. Which values are important for you to read out? Adding each one means adding to the Telnet, Web UI and MQTT so I only want to add the information that is useful.

MichaelDvP, 2020-05-05:
It looks like the 0xE9 is a replacement for the 0x34 and 0x33 telegrams of gas/oil-boilers. Some of the values are declared already. 
@nhmariend can you check your numbering: telegram count starts with 1, bits start with 0, but bit 1 always missing? Byte 26 should be 25 is this the only typo? Also check the descriptions, i.e. is it #hours or #minutes as in other boilers in 18-20?
I have difficulties to understand the difference between 13b4, 14b4 and 14b5. Also some other values are not clear to me.

proddy, 2020-05-10:
Hi @MichaelDvP the numbering @nhmariend was referring too was taken from the wiki that I wrote (https://emsesp.github.io/docs/#/MC110-controller?id=type-0xe9-dhw-status). If it looks incorrect let me know as I started coding this in v2 to support the 0xE9 telegram type.

MichaelDvP, 2020-05-10:
I have no heatpump, my boiler sends telegram 0x34 for DHW. But the description of 
> 26 = same as byte 1, in C

seems not correct if the example telegram states: `... (CRC=E4) #data=24`

Also bitfields counts 0,2,3... with
> bit 2 : DHW preparation activated by one-time charge
> bit 3 : DHW preparation activated by thermal disinfection

whereas these values are in 0x34, byte 5 at bit 1, and 2.

proddy, 2020-05-10:
E9 has a different structure it seems. I guess we'll never know until someone actually tests this out, like Vincent in https://github.com/proddy/EMS-ESP/issues/330#issuecomment-590565988

Anyway I've just pushed the change to the v2. When I've added the mixing logic I'll upload it to this repo. I'm still confused between the different circuits of the MM. It has both heating circuits and warm water circuits so need to decide on the right way to support this.

nhmariend, 2020-05-10:
I have ordered a cable and will upgrade another tool that can read/write to my heatpump.
Then I get help from a friend so we will see what happen. We want to use the ems-esp to log all communication. You can have the log too. My friend has made a software to read the log and make it easier to understand what the code could be. I hope to get the cable soon so I get more data.
Come back to you then @MichaelDvP 
I think my system can be used for cooling in the summer too, but I havent used it for that.
Need a radiator with a fan then. Not that many days with that heat here.

bbqkees, 2020-05-10:
It seems according to Norbertds documentation with those MM modules you can set in theory a heating address HC1 to HC8 and HC9 and HC10. For DHW you can set 1 to 8.
But by checking the MM100 installation manual and the Nefit Systeemwijzer, it practically appears to be limited to HC1 to HC4 and for a DHW boiler only address 9 and 10.

MichaelDvP, 2020-05-11:
> I'm still confused between the different circuits of the MM. It has both heating circuits and warm water circuits so need to decide on the right way to support this.

Yes it's confusing, also that the different circuits unlike thermostat are not in one device. 
It can be up to 10 devices with device-ID 0x20 .. 0x29, where 0x20 .. 0x27 are heating circuits and 0x28, 0x29 DHW-buffers. Each device-ID is sending different message-types (i.e. device 0x20: message 0x01D7, device 0x28: message 0x0231). 
In 1.9.5 the mixing is handled as one device with diffent circuits, that's wrong. Right way is to handle them as different devices.

proddy, 2020-05-11:
@bbqkees @MichaelDvP I've moved the Mixing discussion to a new thread so we can keep this for 0xE9. Would you mind commenting on https://github.com/proddy/EMS-ESP/issues/386?

closed 2020-05-11 by proddy