emsesp/EMS-ESP #385: MQTT message in boiler_data should always be valid JSON object?

zegerk, 2020-05-10:
When hooking up the MQTT I ran into some issues since the data in the boiler_data message can be a single string ("offline"), which is not a valid JSON object. Wouldn't it be better to always put a valid JSON object there?

proddy, 2020-05-10:
I've noticed this too sometimes, and it causes havok with Home Assistant. I'm not sure why this is happening. The "offline" payload is only used in the last will topic which is "status" so should not affect any of the other messages (like boiler_data). I'll see if I can reproduce on the latest v2 code

proddy, 2020-05-12:
I've been monitoring EMS-ESP v2 and haven't seen this appear for a while.

MichaelDvP, 2020-05-13:
I think in 1.9.5 the function `MyESP::_mqttQueue` should not add to queue if there is no payload. The boiler payload is null if not connected, maybe than the last-will payload is sended instead.
In v2 the functions are different, but i have not completely understand how it works. (i'm more asm/c programmer, but my c++ knowledge is very basic).

proddy, 2020-05-13:
probably you're right. This is fixed in v2. I re-wrote the whole mqtt service and haven't seen this problem again.

closed 2020-05-13 by proddy