emsesp/EMS-ESP #386: Mixing Module device logic

proddy, 2020-05-11:
For v2 I want to re-design how the mixing modules are handled.

From what @bbqkees and @michaeldvp were saying in https://github.com/proddy/EMS-ESP/issues/382 each module represents its own heating circuit (HC) and warm water controller (DHW).

What I would like confirming so I complete the coding is

- does each MM register itself with a unique device id (0x20 to 0x29), and have a version and product id?
- each MM can either be a HC (0x20-0x27) or one of two DHW controllers (0x28, 0x29), but not both?
- how does a device distinguish itself from a heating circuit or a DHW controller/buffer? Do they have different product ids or is the device_id determining which is what?
- are the words HC and DHW correct or should they be named something else?
- for the HC we capture now the pump modulation %, valve status and set flow temperature. Is this the most important data to fetch?
- for the DHW we capture also the pump modulation %, temperature status and the flow temperature. Same as the HC. Is this the right information and naming?
- do we need to send commands to the MM?

MichaelDvP, 2020-05-11:
> does each MM register itself with a unique device id (0x20 to 0x29), and have a version and product id?

yes, see #299 
`These device IDs are on the EMS Bus: 0x08 0x09 0x10 0x20 0x29 0x30
and 8 were recognized by EMS-ESP as:
 UBAMaster: Buderus Logamax plus/GB192/Bosch Condens GC9000 (DeviceID:0x08 ProductID:208 Version:01.04)
 Controller: W-B ErP Boiler Control Panel (DeviceID:0x09 ProductID:209 Version:01.04)
 Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID:0x10 ProductID:158 Version:33.03)
 Mixing Module: MM200 Mixing Module (DeviceID:0x29 ProductID:161 Version:29.04)
 Mixing Module: MM200 Mixing Module (DeviceID:0x28 ProductID:161 Version:29.04)
 Solar Module: SM200 Solar Module (DeviceID:0x30 ProductID:164 Version:25.04)
 Mixing Module: MM200 Mixing Module (DeviceID:0x21 ProductID:161 Version:29.04)
 Mixing Module: MM200 Mixing Module (DeviceID:0x20 ProductID:161 Version:29.04)`
>    each MM can either be a HC (0x20-0x27) or one of two DHW controllers (0x28, 0x29), but not both?
>    how does a device distinguish itself from a heating circuit or a DHW controller/buffer? Do they have different product ids or is the device_id determining which is what?

each module have a rotary switch to set the device id.  (MM200 have two switches, MM 400 four switches) 
>    for the HC we capture now the pump modulation %, valve status and set flow temperature. Is this the most important data to fetch?

current flowtemp, setpoint flowtemp and pump mod are important, valve status changes rapidly, byte 1 contains also valve movement (bit 0 opening, bit1 closing)    
 >   for the DHW we capture also the pump modulation %, temperature status and the flow temperature. Same as the HC. Is this the right information and naming?

It's not the same as in hc. The DHW-telegram is also on Norberts list, but with different lenght as logged in #299 .  I'm not sure which values are important.
>   do we need to send commands to the MM?

for mixing HC i think not. The only value is temperature setpoint, and this is calculated by the thermostat and sent often, so our value will be overwritten. For DHW i don't know. 
There is also the possiblility to use mixers as switch in unmixed circuits, But i don't know the configuration and messages for that. 
([MM100 manual in different languages](https://webservices.buderus.at/download/pdf/file/6720813621.pdf) )

proddy, 2020-05-11:
that was helpful Michael, thanks. I've implemented a basic version in mixing.cpp in v2. Haven't simulated it yet.

MichaelDvP, 2020-05-15:
While testing something other i found out, that also MM10 has a address-switch. I switched the positions 2,3,4 and logged what happend, here is the result:

(15:03:19) 22 00 AA 00 FF 0C 0A 11 0A 32 
(15:03:19) 22 00 AB 00 00 01 19 00 00 03 02 
(15:03:19) 22 00 02 00 45 02 01 
New EMS device recognized as a Mixing Module: MM10 Mixing Module (DeviceID: 0x22, ProductID: 69, Version: 02.01)
(15:03:19) 22 00 01 00 AA 25 30 20 63 66 65 01 02 19 15 AA AA AA 06 00 00 00 00 00 00 00 00 00 00 00 00 
(15:03:20) 0B A2 AB 00 20 
(15:03:20) 22 0B AB 00 00 01 19 00 00 03 02 
(15:03:21) 10 22 AC 00 00 00 00 

(15:03:32) 23 00 AA 00 FF 0C 0A 11 0A 32 
(15:03:32) 23 00 AB 00 00 01 19 00 00 03 02 
(15:03:33) 23 00 02 00 45 02 01 
New EMS device recognized as a Mixing Module: MM10 Mixing Module (DeviceID: 0x23, ProductID: 69, Version: 02.01)
(15:03:33) 23 00 01 00 AA 25 30 20 63 66 65 01 02 19 15 AA AA AA 06 00 00 00 00 00 00 00 00 00 00 00 00 
(15:03:34) 0B A3 AB 00 20 
(15:03:34) 23 0B AB 00 00 01 19 00 00 03 02 
(15:03:34) 10 23 AC 00 00 00 00 

(15:03:54) 21 00 AA 00 FF 0C 0A 11 0A 32 
(15:03:54) 21 00 AB 00 00 01 19 00 00 03 02 
(15:03:54) 21 00 02 00 45 02 01 
(15:03:55) 21 00 01 00 AA 25 30 20 63 66 65 01 02 19 15 AA AA AA 06 00 00 00 00 00 00 00 00 00 00 00 00 
(15:03:55) 10 21 AC 00 1C 64 01 

The mixing module changes only the address to another hc, but telegram type stays 0xAB. 
You need a additional logic to assign the messages to the hc. 
(0xAC is from thermostat to mixer setting setpoint-temp and pumpMod, 0xAA is Mixer-parameters, on/off at 0, valve openingtime at 1 (0C ==120sec))

proddy, 2020-05-16:
nice, so it behaves like the MM100. Is it

0x20 - heating circuit/position 1
0x21 - heating circuit/position 2
0x22 - heating circuit/position 3
0x23 - heating circuit/position 4

?

I had only implemented 0xAB. Is there anything worth capturing the 0xAA (mixing parameters) other than the on/off at pos 0 and the opening time at pos 1?

MichaelDvP, 2020-05-16:
Not exactly like MM100, MM100 also changes messages-type, MM10 does not. And EMS (RC30/35) does not accept a mixer on 0x20, hc1 is unmixed on these thermostats. But you can use the same logic for both types. 

The other parameters in 0xAA are not know to me. In RC35 settings i can set only Mixer on/off, Valve time (120sec) and boiler offset (5 °C). The valve time (time the valve needs a signal to change from cold to hot)  is hardware dependent and is set one time at installation, 
Changing the boiler offset could be interesting but i think it's internal of RC35 and not published on the bus: By changing it on RC35 i can see nothing in the log.

proddy, 2020-05-16:
ok got it. I've added the logic to v2. So in your real life example with your
`Mixing Module -> All, type 0xAB, telegram: 21 00 AB 00 2D 01 BE 64 04 01 00 (CRC=15) #data=7` with device ID 0x21 this would be shown as Heating Controller 2. Ok?

MichaelDvP, 2020-05-16:
Yes, i have only one heating circuit which is hc2.
(RC35 sends only type 0x47/48 for hc2, mixer is id 0x21 and my simulated remote control sends as id 0x19).

proddy, 2020-05-16:
added. will be in next v2 build when I've merged in the web component

closed 2020-05-16 by proddy