emsesp/EMS-ESP #387: unknown controller DeviceID:0x09 ProductID:89

marcusb, 2020-05-19:
I have a Buderus GB142 boiler, it seems to be working fine with EMS-ESP.
It reports an unknown controller device on the bus, in addition to the boiler. From other issues I assume it's the control panel of the boiler (that sits on a retractable drawer).

(Not a bug but maybe it should be added to the database.)

```
devices

These device IDs are on the EMS Bus: 0x08 0x09
and 2 were recognized by EMS-ESP as:
 Sieger BK13,BK15/Nefit Smartline/Buderus GB1x2 (DeviceID:0x08 ProductID:64 Version:03.06)
 unknown? (DeviceID:0x09 ProductID:89 Version:02.00)

You have a device is that is not yet known by EMS-ESP. Please report this as a GitHub issue so we can expand the EMS device library.

info

EMS-ESP system stats:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: disabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: disabled, Shower Alert: disabled

EMS Bus stats:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=68, # CRC errors=0
  Tx: Last poll=3.13 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: Sieger BK13,BK15/Nefit Smartline/Buderus GB1x2 (DeviceID:0x08 ProductID:64 Version:03.06)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: on
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 50 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: ? C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 0 times
  Warm Water active time: 0 days 0 hours 0 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 7 C
  Current flow temperature: 59.9 C
  Return temperature: 56.4 C
  Gas: off
  Boiler pump: off
  Fan: on
  Ignition: off
  Circulation pump: on
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: 1.2 bar
  System service code: 0H (203)
  Heating temperature setting on the boiler: 75 C
  Boiler circuit pump modulation max power: 95 %
  Boiler circuit pump modulation min power: 50 %
  Boiler temperature: 61.9 C
  Pump modulation: 0 %
  Burner # starts: 20897 times
  Total burner operating time: 335 days 15 hours 18 minutes
  Total heat operating time: 335 days 15 hours 18 minutes
  Total UBA working time: 1662 days 1 hours 13 minutes
```

proddy, 2020-05-19:
Thanks for reporting this. It is indeed the built-in controller. I'll include in the DB.

MichaelDvP, 2020-05-23:
Please also add in v2 
`{ 72, DeviceType::BOILER, F("MC10"), DeviceFlags::EMS_DEVICE_FLAG_NONE},`
(My boiler get lost, Device 115 is double instead)

proddy, 2020-05-23:
ok. should the MC10 be a DeviceType::MIXING or DeviceType::BOILER ?

MichaelDvP, 2020-05-23:
It's a boiler (Buderus GB125 with MC10), in 19.5 it is recogized as MC10 Module, in v2 unknown.

proddy, 2020-05-23:
ok. added. I'll push it to the v2 branch shortly.

proddy, 2020-05-23:
@MichaelDvP if it's marked as a BOILER then I assume it has a deviceID of 0x08. If it's also responding to the mixing IDs (0x20-0x29) then I will need also to add it as a recognized mixing unit.

MichaelDvP, 2020-05-23:
@proddy it is the same devices as in 1.9.5 master, ems_devices.h, line 257
In v2, device_library.h it is missing, i think a simple typo, as lines 27 and 30 are identical.

proddy, 2020-05-23:
ok got it, my mistake. I was also mixed up with the MM10 and MC10. Thanks for correcting me

closed 2020-06-13 by proddy