emsesp/EMS-ESP #388: Problem using wall thermostat

deqw, 2020-05-24:
In my ui-lovelace I have defined a thermostat control:

`type: thermostat
     entity: climate.thermostat`

When using the wall attached Moduline 300 thermostat the thermostat control slider is correctly reflecting the changes made on the Moduline 300. These changes are at the same time visible in the EMS-ESP Dashboard page. No problem there.

However, when using once the thermostat control slider to set the temperature the Moduline 300 can no longer be used to make temperature changes. The by control set temperature is also not changed on the Moduline 300. In the EMS-ESP Dashboard the correct set temperature is visible each time using the control slider.

To make the Moduline operable again I can change the mode, which was on' manual', to 'off' and back to 'manual' again. This mode change can be done from the homeassistant Overview page, which means the Moduline 300 is still on the bus. 

Don't know if I'm dealing with a bug or misconfiguration here? Below some device information:

Nefit Topline/Buderus GB162 (Version:03.06 ProductID:115 DeviceID:0x08)
BC10 Base Controller (Version:01.03 ProductID:114 DeviceID:0x09)
RC20/Moduline 300 (Version:03.03 ProductID:77 DeviceID:0x17)

proddy, 2020-05-24:
It's quite possible a miss-configuration. It's tricky setting it up correctly in HA in 1.9.x.  In the next version it uses HA MQTT discovery which simplifies things significantly. 

Anyway I suspect HA is sending out the wrong MQTT messages. Check if your `climate.yaml` matches the one in the [doc](https://emsesp.github.io/docs/integrations/homeassistant/climate.yaml). Also use MQTTExplorer to see what is being sent.

There's some more doc in the wiki [here](https://emsesp.github.io/docs/#/Home-Assistant)

deqw, 2020-05-24:
Thanks for your swift reply!

First I did an update to EMS-ESP firmware 1.9.5. After the update I see I now have lost the MQTT messages on the System Status page. Only visible now: ‘MQTT’  ‘MQTT is connected’  ‘MQTT heartbeat is enabled’, but not the messages that are normally shown below. Using Chrome browser. Like you advised I installed MQTT Explorer, so no big problem there. Great tip.

My climate.yaml is like the one in the documentation, with one exception: I have not set, and so do not use, the ‘home’ base prefix anywhere (using it didn’t work for me). So all my references are like: "ems-esp/thermostat_data” which is working fine elsewhere eg. using in Node Red without any problem.

Feedback from the heater/bus when changing the temperature with the Moduline 300 is quite slow. Publish time in EMS-ESP is set to 1 sec now for testing purpose, but of course that does not impact the Moduline. 

What remains is that changing the temperature (seltemp) is not reflected in the Moduline 300 manual temperature setting. The set ‘seltemp’ is however immediately visible when selecting the info button on the Moduline. To get the Moduline operable again you have to push the manual button once (on the Moduline or webcontrol, doesn’t matter) to get the Moduline 300 set temperature out on the bus again (can see in MQTT Explorer that otherwise thermostat_data is not set anymore).

proddy, 2020-05-24:
- I removed the showing the mqtt messages. It was taking up too much code and memory and not really a core function of EMS-ESP. It's documented too in the changelog and also the release notes (https://github.com/proddy/EMS-ESP/releases/tag/1.9.5)
- the home prefix was a bug that was fixed in 1.9.5, so all good hopefully if you decide to use it again
- the lag from the moduline 300 is normal. It just takes the thermostat some time to adjust before sending the signals to the boiler and sometimes the MQTT misses the change. I've tested this in version 2.0 and it's around 3 seconds response time, worst case.
- the changing of the temp when in different modes is a known problem that seems to keep coming back. I'll ask BBQKees if I can borrow his Moduline 300 again and run some tests. Then squish the bug.

thanks for reporting this feedback

deqw, 2020-05-24:
To be in line with the documentation I added the home base prefix again everywhere where needed and can confirm that it is working now with 1.9.5.

Will let it rest for now and will be awaiting further updates/2.0. Thanks for the great work!

closed 2020-06-07 by proddy