emsesp/EMS-ESP #389: RC10/Moduline 100 (DeviceID: 0x17, ProductID: 79, Version: 01.02) not showing in MQTT HA

Tommybear1979, 2020-05-25:
**Question**
I've got it all working (boiler temps etc) except for the Thermostat. It's not getting updated in Home Assistant. Boiler seems to work. Added the example config. Values show up in telnet.
What am i doing wrong?

Screenshot added of the telnet info.
 Thermostat: RC10/Moduline 100 (DeviceID: 0x17, ProductID: 79, Version: 01.02)

configuration.yaml :
 `climate: !include climate.yaml`
climate.yaml: 
`  - platform: mqtt
    name: Thermostat
    modes:
      - "auto"
      - "heat"
      - "off"
    mode_command_topic: "home/ems-esp/thermostat_cmd_mode1"
    temperature_command_topic: "home/ems-esp/thermostat_cmd_temp1"
    mode_state_topic: "home/ems-esp/thermostat_data"
    current_temperature_topic: "home/ems-esp/thermostat_data"
    temperature_state_topic: "home/ems-esp/thermostat_data"
    mode_state_template: "{% if value_json.hc1.mode in ['manual', 'day'] %} heat {% elif value_json.hc1.mode in ['night', 'off'] %} off {% else %} auto {% endif %}"
    current_temperature_template: "{{ value_json.hc1.currtemp }}"
    temperature_state_template: "{{ value_json.hc1.seltemp }}"
    temp_step: 0.5`

proddy, 2020-05-25:
Hi Tom. The HA climate.,yaml looks ok so the problem must be somewhere else. Things to check:

- make sure the MQTT base is set to 'home'. This was a problem when upgrading from earlier versions.
- download http://mqtt-explorer.com/ and see what information is being sent from EMS-ESP and that the topics match
- look into the HA logs to see if any data is malformed

If you're still stuck ping me on https://gitter.im/EMS-ESP/community# and I'll see if I can help you real-time. I'm taking a few days of this week to finish up version 2.0

Tommybear1979, 2020-05-25:
Found it!
I needed to remove 'hc1' from the topics.
Thanks! I bought you a couple of coffee's this morning :-)
`
    
  - platform: mqtt
    name: Thermostat
    modes:
      - "auto"
      - "heat"
      - "off"
    mode_command_topic: "home/ems-esp/thermostat_cmd_mode1"
    temperature_command_topic: "home/ems-esp/thermostat_cmd_temp1"
    mode_state_topic: "home/ems-esp/thermostat_data1"
    current_temperature_topic: "home/ems-esp/thermostat_data1"
    temperature_state_topic: "home/ems-esp/thermostat_data1"
    mode_state_template: "{% if value_json.mode in ['manual', 'day'] %} heat {% elif value_json.mode in ['night', 'off'] %} off {% else %} auto {% endif %}"
    current_temperature_template: "{{ value_json.currtemp }}"
    temperature_state_template: "{{ value_json.seltemp }}"
    temp_step: 0.5`

closed 2020-05-25 by Tommybear1979

proddy, 2020-05-25:
nice, and thanks for the cuppa :)