emsesp/EMS-ESP #390: Bosch Compress 600AW Missing Heating circut 1

nhmariend, 2020-05-27:
We have used version 2.0.0.a9
Recieving boiler data, no thermostat data to MQTT.

EMS Bus protocol: HT3, #telegrams received: 23500, #Read requests sent: 1645, #Write requests sent: 0, #CRC errors: 8 (0%)

ems-esp:/ems# show devices
These EMS devices are currently active:

Thermostat: RC300/RC310/Moduline 3000/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:03.03) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x2A5 0x2B9 0x2A6 0x2BA 0x2A7 0x2BB 0x2A8 0x2BC
 These MQTT topics are registered: thermostat_cmd thermostat_cmd_temp thermostat_cmd_mode

Boiler: Bosch Enviline/Compress 6000AW (DeviceID:0x08 ProductID:172, Version:09.09)
 This Boiler will respond to telegram type IDs: 0x18 0x19 0x34 0x1C 0x2A 0x33 0x14 0x35 0x15 0x16 0x1A 0xD1 0xE4 0xE5 0xE9 0xE3 0xE5
 These MQTT topics are registered: boiler_cmd boiler_cmd_wwactivated boiler_cmd_wwonetime boiler_cmd_wwcirculation boiler_cmd_wwtemp

ems-esp:/thermostat# show
Thermostat: RC300/RC310/Moduline 3000/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:03.03)
 Clock: 21:42:14 27/05/2020
 Heating Circuit 3:
  Current room temperature: ?°C
  Setpoint room temperature: ?°C
 Heating Circuit 4:
  Current room temperature: ?°C
  Setpoint room temperature: ?°C
 Heating Circuit 2:
  Current room temperature: ?°C
  Setpoint room temperature: ?°C

ems-esp:/boiler# show
Boiler: Bosch Enviline/Compress 6000AW (DeviceID:0x08 ProductID:172, Version:09.09)
  Selected flow temperature: 24°C
  Warm Water selected temperature: 60°C
  Central heating: off
  Warm Water activated: off
  Warm Water circulation pump available: off
  Warm Water circulation pump type: 3-way pump
  Warm Water circulation pump freq: 5x3min
  Warm Water selected temperature: 60°C
  Warm Water disinfection temperature: 65°C
  Warm Water circulation active: off°C
  Warm Water set temperature: 63°C
  Warm Water current temperature: 58.1°C
  Warm water temperature (intern): ?°C
  Warm water temperature (extern): ?°C
  Warm Water current temperature (extern): 48.7°C
  Warm Water current tap water flow: ?l/min
  Warm Water # starts: 0
  Warm Water active time: 0 days 0 hours 0 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 24°C
  Current flow temperature: 23.9°C
  Max boiler temperature: ?°C
  Return temperature: ?°C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 1%
  Burner current power: 0%
  Flame current: 0.0uA
  System pressure: ?bar
  Heating temperature setting on the boiler: ?°C
  Boiler circuit pump modulation max power: ?%
  Boiler circuit pump modulation min power: ?%
  Outside temperature: 13.2°C
  Exhaust temperature: ?°C
  Pump modulation: 10%
  Burner # starts: 0
  Total burner operating time: 0 days 0 hours 0 minutes
  Total heat operating time: 0 days 0 hours 0 minutes
  Heat Pump modulation: 0%

proddy, 2020-05-27:
Hi Niels, great that you're trying out v2. To help fix this bug could you
 - tell me how active heating circuits you have. Is it 1,  3 or 4?
 - send me some raw data. Use `log trace full 0x10` or just `log trace`.

nhmariend, 2020-05-27:
Its 2 circuits awailable I think, but I just use 1 now.

Have warm water in a floor at livingroom and kitchen.

But I have prepared to add radiators in to other rooms and maybe one more.

I added this system 4 years ago when I changed windows etc and it was
something wrong with the floor.
So when we get new floor we include water for heating.

I will log tomorrow.

nhmariend, 2020-05-30:
[ems28may.txt](https://github.com/proddy/EMS-ESP/files/4705469/ems28may.txt)

It includes this regarding Thermostat

Thermostat(0x10), Version (0x02),  data: 01
Thermostat(0x10) -> Me(0x0B), RCTime (0x06),  data: 14 05 15 1C 1E 19 03 01 10 FF 00
Thermostat(0x10) -> Boiler(0x08), UBADevices (0x07),  data: 0E
Thermostat(0x10) -> 48(0x48), UBADHWStatus (0x269), data: 69 08 4F 00 00 00 02 00 00 00 02 00 03 00 03 00 03
Thermostat(0x10) -> 48(0x48), ? (0x26A), data: 6A 08 4F 00 00 00 02 00 00 00 02 00 03 00 03 00 03
Thermostat(0x10) -> 48(0x48), ? (0x26B), data: 6B 08 4F 00 00 00 02 00 00 00 02 00 03 00 03 00 03
Thermostat(0x10) -> 48(0x48), ? (0x26C), data: 6C 08 4F 00 00 00 02 00 00 00 02 00 03 00 03 00 03
Thermostat(0x10) -> 48(0x48), ? (0x26D), data: 6D 08 4F 00 00 00 02 00 00 00 02 00 03 00 03 00 03
Thermostat(0x10) -> 48(0x48), ? (0x29B), data: 9B 00 47 00 00 00 01 00 00 00 02 00 03 00 06 00 03
Thermostat(0x10) -> 48(0x48), UBAMaintenanceStatus(0x29C), data: 9C 00 43 00 00 00 01 00 00 00 02 00 03 00 06 00 03
Thermostat(0x10) -> 48(0x48), ? (0x29D), data: 9D 00 43 00 00 00 01 00 00 00 02 00 03 00 06 00 03
Thermostat(0x10) -> 48(0x48), ? (0x29E), data: 9E 00 43 00 00 00 01 00 00 00 02 00 03 00 06 00 03
Thermostat(0x10) -> 48(0x48), ? (0x2AF), data: AF 08 0F 00 00 00 14 00 00 00 19 00 00 00 26 00 00
Thermostat(0x10) -> 48(0x48), ? (0x2B0), data: B0 08 0B 00 00 00 14 00 00 00 19 00 00 00 4B 00 00
Thermostat(0x10) -> 48(0x48), ? (0x2B1), data: B1 08 0B 00 00 00 14 00 00 00 19 00 00 00 4B 00 00
Thermostat(0x10) -> 48(0x48), ? (0x2B2), data: B2 08 0B 00 00 00 14 00 00 00 19 00 00 00 4B 00 00
Thermostat(0x10) -> All(0x00), ? (0xBF),  data: 10 9E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Tried to answer on the mail from the github, but not shureif that works

Niels

proddy, 2020-06-01:
thanks for the logs. I've noticed you have a device 0x48 which is probably your gateway/modem (KM200) and is sending all the thermostat data for your heating circuits (2A6, 2A7, 2A8) for hc 2, 3 and 4.

The other ranges 26A->26D, 29B->29E and 2B0->2B2 are also interesting. It looks like a completely different range and not similar to the RC300 monitor telegrams we're used too.

First thing is to get EMS-ESP to recognize this gateway. I'll need to run some simulations here first and do a new build for that so to make it easier to recognize unknown devices, instead of just ignoring them. Then I'll send you a new version.

In the meantime would you mind capturing some more logs, this time the raw data with `log watch raw` ?

nhmariend, 2020-06-01:
Hi Proddy !
I will log more !
In ww.Husdata.se they can this controller rego2000
I have a device from Husdata that is connected to a raspberry PI.
You can see what I can get out here my pump is then model I
http://husdata.se/Download/o309fq2jio/H1%20Manual%2020.01.pdf
But that one doesnt give me all I want and its not open.YOu see the
addresses is not the same as the telegrams.
Some info is possible to read on the monitor on the heatpump.
So I think your project is much better.

How long time do you think I should log ?  Should I change some settings ?
Best regards
Niels Henrik Mariendal

proddy, 2020-06-01:
that H1 is interesting. Just log for 1 minute, should be enough. I'd like to know why some of those telegrams have no data (i.e. where is says <empty>)

nhmariend, 2020-06-01:
Hi Proddy !

This is the ones I can choose from

off [full|raw] [trace ID]
emerg [full|raw] [trace ID]
crit [full|raw] [trace ID]
alert [full|raw] [trace ID]
err [full|raw] [trace ID]
warning [full|raw] [trace ID]
notice [full|raw] [trace ID]
info [full|raw] [trace ID]
trace [full|raw] [trace ID]
debug [full|raw] [trace ID]
all [full|raw] [trace ID]

So I selected all
Niels

=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2020.06.01 14:20:35 =~=~=~=~=~=~=~=~=~=~=~=

┌──────────────────────────────────────────┐
│ [1mEMS-ESP version 2.0.0a9_ralm01[22m               │
│ [0;92m[4mhttps://github.com/proddy/EMS-ESP[0m        │
│                                          │
│ type [4mhelp[0m to show available commands     │
└──────────────────────────────────────────┘

[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ log wa
[full|raw] [trace ID]
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ log warning[K[K[K[K[KTCH RAW
Invalid log level
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ ?
Command not found
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ HE[K[Khelp
boiler
ems
exit
help
log [level] [full|raw] [trace ID]
mqtt
refresh
show
show version
su
system
thermostat
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ log watch raw
Invalid log level
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ log 
off [full|raw] [trace ID]
emerg [full|raw] [trace ID]
crit [full|raw] [trace ID]
alert [full|raw] [trace ID]
err [full|raw] [trace ID]
warning [full|raw] [trace ID]
notice [full|raw] [trace ID]
info [full|raw] [trace ID]
trace [full|raw] [trace ID]
debug [full|raw] [trace ID]
all [full|raw] [trace ID]
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ log all raw
Log level = ALL
Tracing all telegrams
Displaying raw bytes = off
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:32.667 D 0: [telegram] [0;36mNew Rx [#73] telegram, length 0
[0m[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:32.693 D 1: [telegram] [0;36mNew Rx [#74] telegram, length 0
[0m[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:32.707 D 2: [telegram] [0;36mNew Rx [#75] telegram, length 0
[0m[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:32.719 D 3: [telegram] [0;36mSending read Tx [#93], telegram: 8B C8 02 00 20 46
[0m[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:32.770 D 4: [emsesp] [0;36mLast Tx read successful
[... 451 lines cut ...]
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ 000+01:38:46.566 T 456: [emsesp] Thermostat(0x10) -> 48(0x48), ?(0x50A), data: <empty>
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ 000+01:38:46.566 T 457: [emsesp] 48(0x48) -> Thermostat(0x10), ?(0x508), data: <empty>
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ 000+01:38:46.566 T 458: [emsesp] Thermostat(0x10) -> 48(0x48), ?(0x507), data: <empty>
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ 000+01:38:46.566 T 459: [emsesp] 48(0x48) -> Thermostat(0x10), ?(0x509), data: <empty>
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ 000+01:38:46.566 T 460: [emsesp] Thermostat(0x10) -> 48(0x48), ?(0x508), data: <empty>
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ 000+01:38:46.566 T 461: [emsesp] 48(0x48) -> Thermostat(0x10), ?(0x50A), data: <empty>
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ 000+01:38:46.566 T 462: [emsesp] Thermostat(0x10) -> 48(0x48), ?(0x509), data: <empty>
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:46.916 D 463: [telegram] [0;36mNew Rx [#79] telegram, length 0
[0m[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:46.931 D 464: [telegram] [0;36mNew Rx [#80] telegram, length 0
[0m[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:47.366 T 465: [emsesp] 48(0x48) -> Thermostat(0x10), RC300Monitor(0x2A6), data: <empty> @offset 2
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ 000+01:38:47.366 T 466: [emsesp] Thermostat(0x10) -> 48(0x48), ?(0x50A), data: <empty> @offset 2
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:47.444 D 467: [telegram] [0;36mNew Rx [#81] telegram, length 0
[0m[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:47.459 D 468: [telegram] [0;36mNew Rx [#82] telegram, length 0
[0m[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:47.484 D 469: [telegram] [0;36mNew Rx [#83] telegram, length 0
[0m[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:47.499 D 470: [telegram] [0;36mNew Rx [#84] telegram, length 0
[0m[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:47.524 D 471: [telegram] [0;36mNew Rx [#85] telegram, length 0
[0m[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:47.539 D 472: [telegram] [0;36mNew Rx [#86] telegram, length 0
[0m[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:47.564 D 473: [telegram] [0;36mNew Rx [#87] telegram, length 0
[0m[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:47.579 D 474: [telegram] [0;36mNew Rx [#88] telegram, length 0
[0m[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:47.604 D 475: [telegram] [0;36mNew Rx [#89] telegram, length 0
[0m[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:47.619 D 476: [telegram] [0;36mNew Rx [#90] telegram, length 0
[0m[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:47.796 D 477: [telegram] [0;36mNew Rx [#91] telegram, length 0
[0m[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:47.811 D 478: [telegram] [0;36mNew Rx [#92] telegram, length 0
[0m[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+01:38:47.836 D 479: [telegram] [0;36mNew Rx [#93] telegram, length 0
[0m[

nhmariend, 2020-06-06:
Does the logg look ok ?

proddy, 2020-06-06:
unfortunately not. I cleaned up the log (See attached) and analyzed it but can't decipher the telegram types 26A-26D, and 29A-29E as below. There are no temperature values in those telegrams.

Thermostat(0x10) -> 48(0x48), ? (0x26A), data: 6A 08 4F 00 00 00 02 00 00 00 02 00 03 00 03 00 03
Thermostat(0x10) -> 48(0x48), ? (0x26B), data: 6B 08 4F 00 00 00 02 00 00 00 02 00 03 00 03 00 03
Thermostat(0x10) -> 48(0x48), ? (0x26C), data: 6C 08 4F 00 00 00 02 00 00 00 02 00 03 00 03 00 03
Thermostat(0x10) -> 48(0x48), ? (0x26D), data: 6D 08 4F 00 00 00 02 00 00 00 02 00 03 00 03 00 03
Thermostat(0x10) -> 48(0x48), ? (0x29B), data: 9B 00 47 00 00 00 01 00 00 00 02 00 03 00 06 00 03
Thermostat(0x10) -> 48(0x48), ? (0x29D), data: 9D 00 43 00 00 00 01 00 00 00 02 00 03 00 06 00 03
Thermostat(0x10) -> 48(0x48), ? (0x29E), data: 9E 00 43 00 00 00 01 00 00 00 02 00 03 00 06 00 03
Thermostat(0x10) -> 48(0x48), ? (0x2AF), data: AF 08 0F 00 00 00 14 00 00 00 19 00 00 00 26 00 00
Thermostat(0x10) -> 48(0x48), ? (0x2B0), data: B0 08 0B 00 00 00 14 00 00 00 19 00 00 00 4B 00 00
Thermostat(0x10) -> 48(0x48), ? (0x2B1), data: B1 08 0B 00 00 00 14 00 00 00 19 00 00 00 4B 00 00
Thermostat(0x10) -> 48(0x48), ? (0x2B2), data: B2 08 0B 00 00 00 14 00 00 00 19 00 00 00 4B 00 00

What I'll do as add a new gateway device to EMS-ESP v2 and then you can try that out. I'll update this issue when the version is ready for downloading.

nhmariend, 2020-06-06:
That sounds good ! I will try that version then.

proddy, 2020-06-06:
Try 2.0.0.a15 at https://github.com/proddy/EMS-ESP/tree/v2. Capture some logs and send 'em over for further analysis.

nhmariend, 2020-06-06:
I will do that !!

Thanks

nhmariend, 2020-06-08:
ems-esp:/$ show
Boiler: Bosch Enviline/Compress 6000AW (DeviceID:0x08 ProductID:172, Version:01.                          09)
  Central heating: off
  Warm Water activated: off
  Warm Water charging type: 3-way valve
  Warm Water circulation pump available: off
  Warm Water circulation pump freq: 5x3min
  Warm Water circulation active: off
  Warm Water disinfection temperature: 65°C
  Warm Water selected temperature: 60°C
  Warm Water set temperature: 63°C
  Warm Water current temperature (intern): 53.7°C
  Warm water storage temperature (intern): ?°C
  Warm Water current temperature (extern): 52.7°C
  Warm water storage temperature (extern): ?°C
  Warm Water current tap water flow: ?l/min
  Warm Water # starts: 0
  Warm Water active time: 0 days 0 hours 0 minutes
  Warm Water charging: off
  Warm Water disinfecting: off
  Selected flow temperature: 29°C
  Current flow temperature: 23.4°C
  Max boiler temperature: ?°C
  Return temperature: ?°C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Burner selected max power: 13%
  Burner current power: 0%
  Flame current: 0.0uA
  System pressure: ?bar
  Heating temperature setting on the boiler: ?°C
  Boiler circuit pump modulation max power: ?%
  Boiler circuit pump modulation min power: ?%
  Outside temperature: 15.8°C
  Exhaust temperature: ?°C
  Pump modulation: 10%
  Burner # starts: 0
  Total burner operating time: 0 days 0 hours 0 minutes
  Total heat operating time: 0 days 0 hours 0 minutes
  Heat Pump modulation: 0%

Thermostat: RC300/RC310/Moduline 3000/CW400/Sense II (DeviceID:0x10, ProductID:1                          58, Version:31.03)
  Clock: 18:22:01 08/06/2020

Then you have the log
[ems8june.txt](https://github.com/proddy/EMS-ESP/files/4747290/ems8june.txt)

Does it look good ?

nhmariend, 2020-06-08:
ems-esp:/ems# show devices
These EMS devices are currently active:

Boiler: Bosch Enviline/Compress 6000AW (DeviceID:0x08 ProductID:172, Version:01.09)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x18 0x19 0x34 0x1C 0x2A 0x33 0x14 0x35 0x15 0x16 0x1A 0xD1 0xE4 0xE5 0xE9 0xE3 0xE5
 Subscribed MQTT topics: boiler_cmd boiler_cmd_wwactivated boiler_cmd_wwonetime boiler_cmd_wwcirculation boiler_cmd_wwtemp

Thermostat: RC300/RC310/Moduline 3000/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:31.03) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x2A5 0x2B9 0x2A6 0x2BA 0x2A7 0x2BB 0x2A8 0x2BC
 Subscribed MQTT topics: thermostat_cmd thermostat_cmd_temp thermostat_cmd_mode

Gateway Module: KM200 (DeviceID:0x48, ProductID:189, Version:04.06)

proddy, 2020-06-08:
It's still hard to see what is happening. I'm not 100% certain the thermostat is sending the information to your modem. Could you grab the latest version (a16 now!) and do
 `log trace raw 10`

this will only capture the information being sent to and from the thermostat. Fetch about 2 mins worth of data and also make a note of what the temperatures should roughly be for each of your heating circuits. Then the hunt begins....

nhmariend, 2020-06-08:
Yes, I will do that !

nhmariend, 2020-06-08:
Here are the result

man. 8. jun. 2020 kl. 22:35 skrev Niels Henrik Mariendal <
nhmariend@gmail.com>:

> Yes, I will do that !
>
>
>> It's still hard to see what is happening. I'm not 100% certain the
>> thermostat is sending the information to your modem. Could you grab the
>> latest version (a16 now!) and do
>> log trace raw 10
>>
>> this will only capture the information being sent to and from the
>> thermostat. Fetch about 2 mins worth of data and also make a note of what
>> the temperatures should roughly be for each of your heating circuits. Then
>> the hunt begins....

log trace raw 10
Log level = TRACE
Tracing only telegrams that match a device ID or telegram type of 0x10
Displaying telegrams as raw bytes
ems-esp:/$ 000+00:03:11.294 T 0: [telegram] Rx: C8 90 F7 00 01 FF 04 07 31
ems-esp:/$ 000+00:03:11.312 T 1: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:03:11.335 T 2: [telegram] Rx: C8 90 F7 00 01 FF 04 08 3E
ems-esp:/$ 000+00:03:11.351 T 3: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:03:11.375 T 4: [telegram] Rx: C8 90 F7 00 01 FF 04 09 3F
ems-esp:/$ 000+00:03:11.393 T 5: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:03:11.415 T 6: [telegram] Rx: C8 90 F7 00 01 FF 04 0A 3C
ems-esp:/$ 000+00:03:11.432 T 7: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:03:11.638 T 8: [telegram] Rx: C8 90 F7 02 01 FF 01 A6 BA
ems-esp:/$ 000+00:03:11.655 T 9: [telegram] Rx: 90 48 F7 02 28
ems-esp:/$ 000+00:03:11.679 T 10: [telegram] Rx: C8 90 F7 02 01 FF 01 A7 BB
ems-esp:/$ 000+00:03:11.696 T 11: [telegram] Rx: 90 48 F7 02 28
ems-esp:/$ 000+00:03:11.719 T 12: [telegram] Rx: C8 90 F7 02 01 FF 01 A8 B4
ems-esp:/$ 000+00:03:11.735 T 13: [telegram] Rx: 90 48 F7 02 28
ems-esp:/$ 000+00:03:12.094 T 14: [telegram] Rx: C8 90 F7 00 01 FF 04 07 31
ems-esp:/$ 000+00:03:12.111 T 15: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:03:12.135 T 16: [telegram] Rx: C8 90 F7 00 01 FF 04 08 3E
ems-esp:/$ 000+00:03:12.151 T 17: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:03:12.175 T 18: [telegram] Rx: C8 90 F7 00 01 FF 04 09 3F
ems-esp:/$ 000+00:03:12.191 T 19: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:03:12.215 T 20: [telegram] Rx: C8 90 F7 00 01 FF 04 0A 3C
ems-esp:/$ 000+00:03:12.232 T 21: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:03:12.718 T 22: [telegram] Rx: C8 90 F7 00 01 FF 04 0A 3C
ems-esp:/$ 000+00:03:12.736 T 23: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:03:12.759 T 24: [telegram] Rx: C8 90 F7 00 01 FF 04 07 31
ems-esp:/$ 000+00:03:12.775 T 25: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:03:12.799 T 26: [telegram] Rx: C8 90 F7 00 01 FF 04 08 3E
ems-esp:/$ 000+00:03:12.815 T 27: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:03:12.839 T 28: [telegram] Rx: C8 90 F7 00 01 FF 04 09 3F
ems-esp:/$ 000+00:03:12.856 T 29: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:03:13.094 T 30: [telegram] Rx: C8 90 F7 02 01 FF 01 A6 BA
ems-esp:/$ 000+00:03:13.112 T 31: [telegram] Rx: 90 48 F7 02 28
ems-esp:/$ 000+00:03:13.135 T 32: [telegram] Rx: C8 90 F7 02 01 FF 01 A7 BB
ems-esp:/$ 000+00:03:13.152 T 33: [telegram] Rx: 90 48 F7 02 28
ems-esp:/$ 000+00:03:13.175 T 34: [telegram] Rx: C8 90 F7 02 01 FF 01 A8 B4
ems-esp:/$ 000+00:03:13.191 T 35: [telegram] Rx: 90 48 F7 02 28
ems-esp:/$ 000+00:03:13.663 T 36: [telegram] Rx: C8 90 F7 00 01 FF 04 07 31
ems-esp:/$ 000+00:03:13.680 T 37: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:03:13.703 T 38: [telegram] Rx: C8 90 F7 00 01 FF 04 08 3E
ems-esp:/$ 000+00:03:13.719 T 39: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:03:13.743 T 40: [telegram] Rx: C8 90 F7 00 01 FF 04 09 3F
ems-esp:/$ 000+00:03:13.759 T 41: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:03:13.783 T 42: [telegram] Rx: C8 90 F7 00 01 FF 04 0A 3C
ems-esp:/$ 000+00:03:13.799 T 43: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:03:14.263 T 44: [telegram] Rx: C8 90 F7 00 01 FF 04 09 3F
ems-esp:/$ 000+00:03:14.279 T 45: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:03:14.303 T 46: [telegram] Rx: C8 90 F7 00 01 FF 04 0A 3C
ems-esp:/$ 000+00:03:14.319 T 47: [telegram] Rx: 90 48 F7 00 2A
[... 806 lines cut ...]
ems-esp:/$ 000+00:04:13.793 T 854: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:04:14.067 T 855: [telegram] Rx: C8 90 F7 0A 01 FF 02 00 9A
ems-esp:/$ 000+00:04:14.081 T 856: [telegram] Rx: 90 48 F7 0A 20
ems-esp:/$ 000+00:04:14.107 T 857: [telegram] Rx: C8 90 F7 00 01 FF 02 00 3A
ems-esp:/$ 000+00:04:14.121 T 858: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:04:14.339 T 859: [telegram] Rx: C8 90 F7 00 01 FF 02 00 3A
ems-esp:/$ 000+00:04:14.354 T 860: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:04:15.787 T 861: [telegram] Rx: C8 90 F7 00 02 FF 02 00 22
ems-esp:/$ 000+00:04:15.801 T 862: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:04:15.827 T 863: [telegram] Rx: C8 90 F7 02 01 FF 01 A6 BA
ems-esp:/$ 000+00:04:15.841 T 864: [telegram] Rx: 90 48 F7 02 28
ems-esp:/$ 000+00:04:15.867 T 865: [telegram] Rx: C8 90 F7 02 01 FF 01 A7 BB
ems-esp:/$ 000+00:04:15.881 T 866: [telegram] Rx: 90 48 F7 02 28
ems-esp:/$ 000+00:04:15.907 T 867: [telegram] Rx: C8 90 F7 02 01 FF 01 A8 B4
ems-esp:/$ 000+00:04:15.921 T 868: [telegram] Rx: 90 48 F7 02 28
ems-esp:/$ 000+00:04:15.947 T 869: [telegram] Rx: C8 90 F7 00 01 FF 04 07 31
ems-esp:/$ 000+00:04:15.961 T 870: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:04:15.987 T 871: [telegram] Rx: C8 90 F7 00 01 FF 04 08 3E
ems-esp:/$ 000+00:04:16.002 T 872: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:04:16.027 T 873: [telegram] Rx: C8 90 F7 00 01 FF 04 09 3F
ems-esp:/$ 000+00:04:16.041 T 874: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:04:16.243 T 875: [telegram] Rx: C8 90 F7 00 01 FF 04 0A 3C
ems-esp:/$ 000+00:04:16.258 T 876: [telegram] Rx: 90 48 F7 00 2A
ems-esp:/$ 000+00:04:16.283 T 877: [telegram] Rx: C8 90 F7 01 02 FF 02 00 32
ems-esp:/$ 000+00:04:16.297 T 878: [telegr

proddy, 2020-06-09:
> also make a note of what the temperatures should roughly be for each of your heating circuits

did you capture those too?

nhmariend, 2020-06-09:
Hi Proddy !

No ! Just logged yesterday. I can do that today. I have just HC1 in use
now, but has prepared use of one more later.
So I should get temp for HC1
The temp there should be low now because the temp is 15-20 degrees outside.

I tried to change setting on the Temp. But you should at least se the
setting 20.0 degrees.
Im back home in about 1 hour, can look more into this then.

Niels Henrik

proddy, 2020-06-16:
doesn't like any of the thermostat data is being sent to the KM200 unless I'm missing something.

nhmariend, 2020-06-16:
Yes, we will do that !

nhmariend, 2020-06-16:
New log using your latest version 
log trace

[ems16juni1.txt](https://github.com/proddy/EMS-ESP/files/4788132/ems16juni1.txt)

proddy, 2020-06-16:
LOL, I forgot to tell you we moved some of the commands around. You'll need to do a `watch on` to see the incoming telegrams.

nhmariend, 2020-06-16:
Is it better now ?

[ems16juni2.txt](https://github.com/proddy/EMS-ESP/files/4788900/ems16juni2.txt)

nhmariend, 2020-06-16:
Here is what I get now in Home assistant via mqtt

{
    "rssid": 100,
    "uptime": "000+04:54:00.319",
    "freemem": 78,
    "mqttpublishfails": 0
}

QoS: 1 - Retain: false
Meldingen 18 ble mottatt på Bosch/ems-esp/thermostat_data klokken 23:50:

{
    "hc1": {
        "seltemp": 20,
        "daytemp": 19,
        "nighttemp": 13.5,
        "mode": "manual",
        "modetype": "day"
    },
    "hc2": {
        "seltemp": 0,
        "currtemp": -2866.09,
        "daytemp": 21,
        "nighttemp": 15
    },
    "hc3": {
        "seltemp": 0,
        "currtemp": -2866.09,
        "daytemp": 21,
        "nighttemp": 15
    },
    "hc4": {
        "seltemp": 0,
        "currtemp": -2866.09,
        "daytemp": 21,
        "nighttemp": 15
    }
}

QoS: 0 - Retain: false
Meldingen 17 ble mottatt på Bosch/ems-esp/boiler_data klokken 23:50:

{
    "wWSelTemp": 60,
    "wWDisinfectionTemp": 65,
    "selFlowTemp": 22,
    "selBurnPow": 0,
    "curBurnPow": 0,
    "pumpMod": 10,
    "wWCircPump": "off",
    "wWCiPuType": "valve",
    "wWCirc": "off",
    "outdoorTemp": 18.8,
    "wWCurTmp": 52.6,
    "curFlowTemp": 27.3,
    "wWActivated": "off",
    "wWOnetime": "off",
    "wWDesinfecting": "off",
    "wWReady": "off",
    "wWRecharge": "off",
    "wWTempOK": "off",
    "burnGas": "off",
    "flameCurr": 0,
    "heatPmp": "on",
    "fanWork": "off",
    "ignWork": "off",
    "wWHeat": "off",
    "wWStarts": 0,
    "wWWorkM": 0,
    "burnStarts": 0,
    "burnWorkMin": 0,
    "heatWorkMin": 0,
    "pumpmodulation": 0
}

QoS: 0 - Retain: false
Meldingen 16 ble mottatt på Bosch/ems-esp/thermostat_data klokken 23:50:

{
    "hc1": {
        "seltemp": 20,
        "daytemp": 19,
        "nighttemp": 13.5,
        "mode": "manual",
        "modetype": "day"
    },
    "hc2": {
        "seltemp": 0,
        "currtemp": -2866.09,
        "daytemp": 21,
        "nighttemp": 15
    },
    "hc3": {
        "seltemp": 0,
        "currtemp": -2866.09,
        "daytemp": 21,
        "nighttemp": 15
    },
    "hc4": {
        "seltemp": 0,
        "currtemp": -2866.09,
        "daytemp": 21,
        "nighttemp": 15
    }
}

proddy, 2020-06-16:
I can see a few bugs that need fixing. What would really help is the raw data:
```
$ ems
$ watch raw 10
```
and leave it for 2 mins

nhmariend, 2020-06-17:
Here you have a new logfile
[ems17juni1.txt](https://github.com/proddy/EMS-ESP/files/4793518/ems17juni1.txt)

proddy, 2020-06-17:
great, and with the latest build does it look better? what does `show` show you?

nhmariend, 2020-06-18:
Now I have a new file here.
The MQTT is strange, It change modetypes
[Strange data  to MQTT.txt](https://github.com/proddy/EMS-ESP/files/4799358/Strange.data.to.MQTT.txt)
[ems18juni1.txt](https://github.com/proddy/EMS-ESP/files/4799359/ems18juni1.txt)

 etc on HC!,2 etc.
Ifyou look at the values

proddy, 2020-06-18:
thanks. I'll work on fixing that.

proddy, 2020-06-19:
I couldn't reproduce - are you sure you're using the latest builds? version a23 or a24?

nhmariend, 2020-06-19:
I was 22a, will test 24a later today or tomorrow!

Come back with new files then

proddy, 2020-06-19:
also do a `watch raw 90` from the ems menu. We need to find out where those values were coming from. It wasn't in the dump you sent previously. Look for longer telegrams starting with "90 0B FF 00 01... " followed by either a A6, A7 or A8

nhmariend, 2020-06-19:
Here are the files using a24
[ems19june2.txt](https://github.com/proddy/EMS-ESP/files/4806814/ems19june2.txt)
[ems19june1.txt](https://github.com/proddy/EMS-ESP/files/4806815/ems19june1.txt)

proddy, 2020-06-19:
thanks. I used the data you provided and simulated the RC300 but don't get those strange values you are seeing (like large negative current temperature values). So I suspect something else is wrong. I'll take a closer look tomorrow.

nhmariend, 2020-06-20:
Mqtt into Home assistant look like this now
[ems20june-mqtt.txt](https://github.com/proddy/EMS-ESP/files/4807573/ems20june-mqtt.txt)
If its a large . temp it could be that I dont have it, not in use or like room temp is optional. The mqtt looks better now in hc1,2,3 etc but now its not current temp on them.

proddy, 2020-06-20:
ok, looking better. A few more things
 - we need to see the HC1, HC2 and HC3 telegrams to fix the current room temperature. Do a `log info` and then `watch raw 2a6`. See what comes in on HC2. Then do the same for `2a5`. Also notice what the actual values should be on your thermostat for each heating circuit so we can compare the values.
 - instead of building the .bin each time you can take the latest firmware builds directly from https://github.com/proddy/EMS-ESP/releases/download/travis-v2-build/EMS-ESP-dev-esp8266.bin and use the instructions in the README to upload. You could create a .BAT script that automatically does this which may help speed up testing.

nhmariend, 2020-06-20:
I have logged them

When I watch the mqtt for HC1 that one is correct 
"hc1": {
"seltemp": 20, This is the one that I see in the Bosch app ! When I change the temp in the app it also change here !
"daytemp": 19,
"nighttemp": 13.5,
"mode": "manual", This works ! Its called opt in the app If I change to auto in the app also called auto here the modetype change to comfort. But the temp doesnt show the same as in the app.
"modetype": "day"

I have just one HC in use. Its ready to use one more but need to install radiator and connect water to the heatpump.

So I think its just the 2a5 that is intresting for me now ?

[ems20june-2a5-1.txt](https://github.com/proddy/EMS-ESP/files/4807822/ems20june-2a5-1.txt)
[ems20june-2a6-1.txt](https://github.com/proddy/EMS-ESP/files/4807823/ems20june-2a6-1.txt)

In this version 2.0 I have started to get error messages H01-5504 and A01-5503 It was not there in 1.95, This it was canbus error. Is it some setting that is needed to adjust?

Have a pic of how it look like inhome assistant

[image]
Varme means Heating/Manual mode

proddy, 2020-06-22:
Could you try 2.0.0a26 ?

nhmariend, 2020-06-22:
upgrading it now

nhmariend, 2020-06-22:
Here are some info from 2.0.0a26
[june22-2a5.txt](https://github.com/proddy/EMS-ESP/files/4814520/june22-2a5.txt)

I get this
{
    "hc1": {
        "seltemp": 18,  This is the same as I have on my Bosch app
        "daytemp": 126,        Strange value !
        "nighttemp": 13.5,
        "circuitcalctemp": 20,
        "mode": "manual",    This didnt change when I change Auto/Opt
        "modetype": "holiday" This is calles Opt in the app /when changed to Auto on the app it show summer here.
    }
}

[june22-10.txt](https://github.com/proddy/EMS-ESP/files/4814553/june22-10.txt)

The error messages in maintenance is now gone !!  So no error on the EMS/Canbus.

nhmariend, 2020-06-23:
I trythe a27now !

See that the error is back H01-5504 and A01-5503 Think its EMS/Canbus problem. That was no error in a26

[ems23june-10.txt](https://github.com/proddy/EMS-ESP/files/4820464/ems23june-10.txt)

proddy, 2020-06-23:
just to re-cap, the problems are from your RC300 thermostat where you're not seeing correct values for the current temperature, mode and modetype. This is what I'm focusing on now. 

Those errors, are they coming from the UBA/Master/Boiler? Do you know what they mean? If you use telnet and type `show` do you also see these values under the service code? These values are in telegram type 0x18 so perhaps send a couple of example telegrams

nhmariend, 2020-06-23:
Hi ! 
Its a little late here but here you have some info.

In the servicemanual its 5503 means bad connection or error on Canbus or noise from electrisity(powercable too near the canbus calble) and the other is if the error has been ther for some period.

Watching incoming telegrams, displayed as raw bytes
Filtering only telegrams that match a device ID or telegram type of 0x18
000+06:53:16.733 N 66: [telegram] Rx: C8 88 18 0B 02 7C

ems-esp:/ems$ show
These EMS devices are currently active:

Boiler: Bosch Enviline/Compress 6000AW (DeviceID:0x08 ProductID:172, Version:01.09)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x18 0x19 0x34 0x1C 0x2A 0x33 0x14 0x35 0x15 0x16 0x1A 0xD1 0xE3 0xE4 0xE5 0xE9
 Subscribed MQTT topics: boiler_cmd boiler_cmd_wwactivated boiler_cmd_wwonetime boiler_cmd_wwcirculation boiler_cmd_wwtemp

ems-esp:/$ show
Boiler: Bosch Enviline/Compress 6000AW (DeviceID:0x08 ProductID:172, Version:01.09)
  Central heating: off
  Warm Water activated: off
  Warm Water charging type: 3-way valve
  Warm Water circulation pump available: off
  Warm Water circulation pump freq: 5x3min
  Warm Water circulation active: off
  Warm Water disinfection temperature: 65°C
  Warm Water selected temperature: 60°C
  Warm Water set temperature: 63°C
  Warm Water current temperature (intern): 54.4°C
  Warm Water current temperature (extern): 53.7°C
  Warm Water # starts: 0
  Warm Water active time: 0 days 0 hours 0 minutes
  Warm Water charging: off
  Warm Water disinfecting: off
  Selected flow temperature: 20°C
  Current flow temperature: 25.8°C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Burner selected max power: 0%
  Burner current power: 0%
  Flame current: 0.0uA
  Outside temperature: 18.7°C
  Pump modulation: 10%
  Pump modulation2: 0%
  Burner # starts: 0
  Total burner operating time: 0 days 0 hours 0 minutes
  Total heat operating time: 0 days 0 hours 0 minutes

Thermostat: RC300/RC310/Moduline 3000/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:31.03)
  Clock: 00:18:19 24/06/2020
  Heating Circuit 1:
    Setpoint room temperature: 17.5°C
    Mode: manual
    Mode Type: day
    Program is set to Holiday mode
    Day temperature: 19.0°C
    Night temperature: 13.5°C
    Target flow temperature: 20°C

Gateway Module: KM200 (DeviceID:0x48, ProductID:189, Version:04.06)

Thermostat: RC300/RC310/Moduline 3000/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:31.03) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x2A5 0x2B9 0x2A6 0x2BA 0x2A7 0x2BB 0x2A8 0x2BC 0x31D 0x31E
 Subscribed MQTT topics: thermostat_cmd thermostat_cmd_temp thermostat_cmd_mode

Gateway Module: KM200 (DeviceID:0x48, ProductID:189, Version:04.06)

EMS Bus info:
  Tx mode: 2
  Bus protocol: HT3
  #telegrams received: 20843
  #read requests sent: 1678
  #write requests sent: 0
  #corrupted telegrams: 1 (0%)
  #tx fails (after 3 retries): 0

Rx Queue is empty

Tx Queue is empty

nhmariend, 2020-06-24:
Do you want me to log some more info ?

proddy, 2020-06-24:
not yet, haven't had time to look yet.

nhmariend, 2020-06-24:
Thats ok. 
When Ijust log in I get this
001+01:27:02.233 N 7: [telegram] Rx: C8 88 18 0B 02 7C
001+01:28:37.260 N 8: [telegram] Rx: C8 88 18 0B 02 7C
001+01:30:16.286 N 9: [telegram] Rx: C8 88 18 0B 02 7C
001+01:31:57.712 N 10: [telegram] Rx: C8 88 18 0B 02 7C

proddy, 2020-06-25:
> When just log in I get this
> 001+01:27:02.233 N 7: [telegram] Rx: C8 88 18 0B 02 7C
> 001+01:28:37.260 N 8: [telegram] Rx: C8 88 18 0B 02 7C
> 001+01:30:16.286 N 9: [telegram] Rx: C8 88 18 0B 02 7C
> 001+01:31:57.712 N 10: [telegram] Rx: C8 88 18 0B 02 7C

This is just the modem (km200) asking the boiler for the selected warm water temperature.

proddy, 2020-06-25:
I've been looking at your latest logs and trying to figure out why the current temperature from the thermostat is not showing. in the telegrams it's shown as 'unset'. What I need is actual data from the thermostat like you did last time, with also the exact active room temperature value as show on your RC300 so then I can try and find where the data stored.

nhmariend, 2020-06-25:
I can do that !
Sit out in the sin for late dinner.

What command do you want me to use ?

Watch raw 10 ?

proddy, 2020-06-25:
> Sit out in the sin for late dinner.

haha. exactly. It's also nice weather in Gjøvik right? My home office hit 28 degrees inside today, too hot to work. My brain is melting.

> What command do you want me to use ?
> Watch raw 10 ?

yes, let's start with `log info` and `watch raw 10` and leave it for a few minutes. Also remember to write down what the actual thermostat shows as the room temperature.

by the way, have you also tried using the MQTT with `ha` mode? With this you shouldn't need any additional sensor.yaml o climate.yaml type files and build your own lovelace dashboards dynamically.

nhmariend, 2020-06-25:
The roomtemp is 17.5 degrees
And it set as opt/not Auto in the Bosch app
Water is set to 63 degrees
Whe I start the temp is 27.4 in room temp feeding
Actual water temp 52.9
Out 22.7
[ems25june-10.txt](https://github.com/proddy/EMS-ESP/files/4833759/ems25june-10.txt)
Is it possible e-mail you directly?
Not tried that ha mode. Some more info on that ?

nhmariend, 2020-06-25:
[image]
[image]
[image]
Some pics taken now for you

proddy, 2020-06-25:
beautiful there. Definitely on my bucket list to visit. For chat go to https://gitter.im/EMS-ESP/community

nhmariend, 2020-06-25:
I tried to log in there and sent a message. last picwas wrong. Light here this time a year, almost no wind. Was aroud 30 degrees today.

nhmariend, 2020-06-26:
[ems26june-10.txt](https://github.com/proddy/EMS-ESP/files/4837302/ems26june-10.txt)
Here are a log with your a29

nhmariend, 2020-06-26:
[ems26june-Info.txt](https://github.com/proddy/EMS-ESP/files/4837427/ems26june-Info.txt)
Also a log info

proddy, 2020-06-27:
still can't find the room temperature (17.5 degrees) in any of those log files you sent, unfortunately.

nhmariend, 2020-06-28:
[image]
[image]

See the temp here in HA The one I marked with red isnt there because I dont have it. Its optional to read room temp.

proddy, 2020-06-28:
ok, so what needs to be changed/fixed? If the current room temperature is not in the EMS telegrams EMS-ESP can't pick it up and send it.

[5 more replies]

closed 2020-09-23 by proddy