emsesp/EMS-ESP #391: unknown devices for Bosch Condens 5000i

poesel, 2020-05-30:
Hi,

I have a Bosch Condens 5000i boiler that is not recognized by the firmware (1.9.5)

> devices
> 
> These 3 were recognized by EMS-ESP:
>  unknown?: unknown? (DeviceID: 0x08, ProductID: 195, Version: 05.02)
>  unknown?: unknown? (DeviceID: 0x09, ProductID: 241, Version: 01.05)
>  Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID: 0x10, ProductID: 158, Version: 18.06)
> 
> One or more devices are not recognized by EMS-ESP. Please report this back in GitHub.

> info
> 
> EMS-ESP system status:
>   System logging set to None
>   LED: on, Listen mode: off
>   Boiler: disabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: disabled
>   Shower Timer: disabled, Shower Alert: disabled
>   Saved known device IDs: none
> 
> EMS Bus status:
>   Bus is connected, protocol: HT3
>   Rx: # successful read requests=277, # CRC errors=141
>   Tx: Last poll=1.422 seconds ago, # successful write requests=0
> 
> 
> Thermostat data:
>   Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID: 0x10, ProductID: 158, Version: 18.06)
>   Thermostat time is 16:36:19 30/05/2020
>   Heating Circuit 1
>    Current room temperature: 21.5 C
>    Setpoint room temperature: 20.0 C
>    Mode is set to manual
> 

Since this is a pretty new device I guess it would work with EMS 2 but when I set it to mode 'EMS+' nothing works at all.

proddy, 2020-06-01:
thanks. I expect if we add the boiler to the known device library it should pick up the values. If you're comfortable with building and uploading the firmware yourself you can try it out.

poesel, 2020-06-01:
Ok, I think I can build the firmware. How and where do I add the boiler?

proddy, 2020-06-01:
add this line to the top of `ems_devices.h` with the rest of the boilers:

```
    {195, EMS_DEVICE_TYPE_BOILER, "Bosch Condens 5000i", EMS_DEVICE_FLAG_NONE},
```

if its too much hassle I can make a new build quicky and put it in https://github.com/proddy/EMS-ESP/releases/tag/travis-dev-build

poesel, 2020-06-01:
I wouldn't call line 270 'top of ems_devices.h' but ok, found it. :)

This line: 'Finally use the build targets in PlatformIO's to build & upload.' (from https://emsesp.github.io/docs/#/Building-firmware)
is a bit confusing if you are new to PlatformIO. I found it by going through the commands with ctrl-shift-p and searching:

Build is ctrl-alt-b
Upload ctrl-alt-u

During the first build it complained about the existing /release directory. After deletion it worked.
During upload I had a request to allow incoming network connections. This probably killed a timeout and I had to repeat it.
Apart from that build & upload worked flawlessly.

poesel, 2020-06-01:
That is get from info. Better but still a lot of question marks.

> info
> 
> EMS-ESP system status:
>   System logging set to None
>   LED: on, Listen mode: off
>   Boiler: enabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: disabled
>   Shower Timer: disabled, Shower Alert: disabled
>   Saved known device IDs: none
> 
> EMS Bus status:
>   Bus is connected, protocol: HT3
>   Rx: # successful read requests=42, # CRC errors=16
>   Tx: Last poll=2.9 seconds ago, # successful write requests=0
> 
> Boiler data:
>   Boiler: Bosch Condens 5000i (DeviceID: 0x08, ProductID: 195, Version: 05.02)
>   Hot tap water: running
>   Central heating: off
>   Warm Water activated: ?
>   Warm Water circulation pump available: ?
>   Warm Water circulation pump type: 3-way pump
>   Warm Water circulation pump freq: 5x3min
>   Warm Water selected temperature: ? C
>   Warm Water desinfection temperature: ? C
>   Warm Water circulation active: off
>   Warm Water set temperature: ? C
>   Warm Water current temperature: ? C
>   Warm water temperature (intern): ? C
>   Warm water temperature (extern): ? C
>   Warm Water current temperature (extern): ? C
>   Warm Water current tap water flow: ? l/min
>   Warm Water # starts: ? times
>   Warm Water 3-way valve: off
>   Selected flow temperature: 0 C
>   Current flow temperature: 23.0 C
>   Max boiler temperature: ? C
>   Return temperature: ? C
>   Gas: off
>   Warm water mode: off
>   Boiler pump: off
>   Fan: off
>   Ignition: off
>   Circulation pump: off
>   Burner selected max power: 0 %
>   Burner current power: 0 %
>   Flame current: 0.0 uA
>   System pressure: ? bar
>   System service code:
>   Heating temperature setting on the boiler: ? C
>   Boiler circuit pump modulation max power: ? %
>   Boiler circuit pump modulation min power: ? %
>   Outside temperature: --3276.-8 C
>   Exhaust temperature: ? C
>   Pump modulation: 0 %
>   Burner # starts: 47 times
>   Total burner operating time: 0 days 9 hours 45 minutes
>   Total heat operating time: 0 days 3 hours 5 minutes
>   Total UBA working time: 13 days 0 hours 50 minutes
> 
> Thermostat data:
>   Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID: 0x10, ProductID: 158, Version: 18.06)
>   Thermostat time is 12:06:38 01/06/2020
>   Heating Circuit 1
>    Current room temperature: 21.3 C
>    Setpoint room temperature: 20.0 C
>    Mode is set to manual

There is also still an unknown device:

> devices
> 
> These 3 were recognized by EMS-ESP:
>  Boiler: Bosch Condens 5000i (DeviceID: 0x08, ProductID: 195, Version: 05.02)
>  unknown?: unknown? (DeviceID: 0x09, ProductID: 241, Version: 01.05)
>  Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID: 0x10, ProductID: 158, Version: 18.06)

proddy, 2020-06-01:
> I wouldn't call line 270 'top of ems_devices.h' but ok, found it. :)
>
> This line: 'Finally use the build targets in PlatformIO's to build & upload.' (from [...]

Great. PlatformIO and Visual Studio Code is a great combo if you're doing a lot of development. I think in your case it would have just been easier to follow the CLI way and first do the `npm install` and then the `pio run -t upload`.

proddy, 2020-06-01:
> That is what I get from info. Better but still a lot of question marks.

Ok, that's progress. Check if the Tx is working (use `info`). EMS-ESP sends read requests to the boiler to return it's information so probably that is not working. Try the tx_mode that supports HT3 (make sure you reboot). If it still doesn't work try turning on verbose logging in the telnet and sending some log output over. `log v`

> There is also still an unknown device:
> > unknown?: unknown? (DeviceID: 0x09, ProductID: 241, Version: 01.05)

This is the onboard controller in your boiler. It doesn't really do anything. EMS-ESP doesn't need it. You can add this to the `ems_devices.h` if you want as
``
    {241, EMS_DEVICE_TYPE_CONTROLLER, "Condens 5000 Controller", EMS_DEVICE_FLAG_NONE},
``

poesel, 2020-06-01:
I'm using VSC from time to time but with MicroPython. So I don't need to build anything. :)
I've added the line for the controller and now have no unknown devices.
Also switched to HT3 which I think was successful. But still the same number of '?' (I think).

> info
> 
> EMS-ESP system status:
>   System logging set to None
>   LED: on, Listen mode: off
>   Boiler: enabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: disabled
>   Shower Timer: disabled, Shower Alert: disabled
>   Saved known device IDs: none
> 
> EMS Bus status:
>   Bus is connected, protocol: HT3
>   Rx: # successful read requests=28, # CRC errors=6
>   Tx: Last poll=1.352 seconds ago, # successful write requests=0
> 
> Boiler data:
>   Boiler: Bosch Condens 5000i (DeviceID: 0x08, ProductID: 195, Version: 05.02)
>   Hot tap water: running
>   Central heating: off
>   Warm Water activated: ?
>   Warm Water circulation pump available: ?
>   Warm Water circulation pump type: 3-way pump
>   Warm Water circulation pump freq: 5x3min
>   Warm Water selected temperature: ? C
>   Warm Water desinfection temperature: ? C
>   Warm Water circulation active: off
>   Warm Water set temperature: ? C
>   Warm Water current temperature: ? C
>   Warm water temperature (intern): ? C
>   Warm water temperature (extern): ? C
>   Warm Water current temperature (extern): ? C
>   Warm Water current tap water flow: ? l/min
>   Warm Water # starts: ? times
>   Warm Water 3-way valve: off
>   Selected flow temperature: 0 C
>   Current flow temperature: 27.7 C
>   Max boiler temperature: ? C
>   Return temperature: ? C
>   Gas: off
>   Warm water mode: off
>   Boiler pump: off
>   Fan: off
>   Ignition: off
>   Circulation pump: off
>   Burner selected max power: 0 %
>   Burner current power: 0 %
>   Flame current: 0.0 uA
>   System pressure: ? bar
>   System service code:
>   Heating temperature setting on the boiler: ? C
>   Boiler circuit pump modulation max power: ? %
>   Boiler circuit pump modulation min power: ? %
>   Outside temperature: --3276.-8 C
>   Exhaust temperature: ? C
>   Pump modulation: 0 %
>   Burner # starts: 47 times
>   Total burner operating time: 0 days 9 hours 45 minutes
>   Total heat operating time: 0 days 3 hours 5 minutes
>   Total UBA working time: 13 days 3 hours 3 minutes
> 
> Thermostat data:
>   Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID: 0x10, ProductID: 158, Version: 18.06)
>   Thermostat time is 14:19:14 01/06/2020
>   Heating Circuit 1
>    Current room temperature: 21.8 C
>    Setpoint room temperature: 20.0 C
>    Mode is set to manual
[... 68 lines cut ...]
> (14:25:34) Sending read of type 0x18 to 0x08, telegram: 8B 88 18 00 20
> (14:25:34) Corrupt telegram: FE CC 9C 80 20
> (14:25:35) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (#data=15)
> (14:25:35) Sending read of type 0x19 to 0x08, telegram: 8B 88 19 00 20
> (14:25:35) Boiler -> Me, telegram: 88 0B 19 00
> (14:25:36) Sending read of type 0x33 to 0x08, telegram: 8B 88 33 00 20
> (14:25:36) Boiler -> Me, telegram: 88 0B 33 00
> (14:25:36) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (#data=15)
> (14:25:36) Sending read of type 0x16 to 0x08, telegram: 8B 88 16 00 20
> (14:25:36) Boiler -> Me, telegram: 88 0B 16 00
> (14:25:36) Sending read of type 0x14 to 0x08, telegram: 8B 88 14 00 20
> (14:25:36) Boiler -> Me, UBATotalUptimeMessage(0x14), telegram: 88 0B 14 00 00 49 DE (#data=3)
> (14:25:37) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 09 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (#data=15)
> (14:25:38) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 01 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (#data=15)
> (14:25:38) Boiler -> All, UBADevices(0x07), telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (#data=15)
> (14:25:42) Boiler -> All, UBAMonitorFast2(0xE4), telegram: 88 00 E4 00 10 20 2D 2D 00 CB 00 01 16 00 00 00 02 01 06 00 00 80 00 00 00 10 00 01 16 00 00 (#data=27)
> (14:25:42) Boiler -> All, UBAMonitorFast2(0xE4), telegram: 88 00 E4 23 00 09 60 00 99 00 00 (#data=7)
> (14:25:43) Boiler -> All, HeatPumpMonitor1(0xE3), telegram: 88 00 E3 00 04 00 00 00 00 00 00 00 00 00 00 01 16 00 64 52 00 00 00 00 (#data=20)
> (14:25:43) Boiler -> All, Type(0xE9), telegram: 88 00 E9 00 3C 01 F1 01 F1 80 00 01 49 46 3C 00 00 02 00 01 90 00 00 23 00 00 00 3C 03 00 (#data=26)
> (14:25:43) Corrupt telegram: F2 80 E0 FC
> (14:25:52) Boiler -> All, UBAMonitorFast2(0xE4), telegram: 88 00 E4 00 10 20 2D 2D 00 CB 00 01 16 00 00 00 02 01 07 00 00 80 00 00 00 10 00 01 16 00 00 (#data=27)
> (14:25:52) Boiler -> All, UBAMonitorFast2(0xE4), telegram: 88 00 E4 23 00 09 60 00 99 00 00 (#data=7)
> (14:25:53) Boiler -> All, HeatPumpMonitor1(0xE3), telegram: 88 00 E3 00 04 00 00 00 00 00 00 00 00 00 00 01 16 00 64 52 00 00 00 00 (#data=20)
> (14:25:53) Boiler -> All, Type(0xE9), telegram: 88 00 E9 00 3C 01 F1 01 F1 80 00 01 49 46 3C 00 00 02 00 01 90 00 00 23 00 00 00 3C 03 00 (#data=26)
>

MichaelDvP, 2020-06-02:
The 0xE9 message is missing. There was a discussion about it in #382, but left a confusion about bit/byte numbering. You can add it to ems.cpp: 
In line 3250 add `{EMS_DEVICE_UPDATE_FLAG_HEATPUMP, EMS_TYPE_HPMonitorWW, "HeatPumpMonitorWW", _process_HPMonitorWWMessage},` 
and somewhere near line 1029 add this function (If something is wrong, check the byte positions): 
/**
 * HPMonitorWWMessage - type 0xE9 - warm water monitor of heatpumps. 24 bytes long
 * received every 10 seconds
 */
void _process_HPMonitorWWMessage(_EMS_RxTelegram * EMS_RxTelegram) {
    _setValue(EMS_RxTelegram, &EMS_Boiler.wWSetTmp, 0);
    _setValue(EMS_RxTelegram, &EMS_Boiler.wWCurTmp, 1);
    _setValue(EMS_RxTelegram, &EMS_Boiler.wWCurTmp2, 3);
    _setValue(EMS_RxTelegram, &EMS_Boiler.wWDesinfectTemp, 9);
    _setValue(EMS_RxTelegram, &EMS_Boiler.wWOneTime, 12, 2);
    _setValue(EMS_RxTelegram, &EMS_Boiler.wWDesinfecting, 12, 3);
    _setValue(EMS_RxTelegram, &EMS_Boiler.wWRecharging, 12, 4);
    _setValue(EMS_RxTelegram, &EMS_Boiler.wWCirc, 13, 2);
    _setValue(EMS_RxTelegram, &EMS_Boiler.wWStarts, 14);
    _setValue(EMS_RxTelegram, &EMS_Boiler.wWWorkM, 17);
//    _setValue(EMS_RxTelegram, &EMS_Boiler.wWCurFlow, 9);
//    _setValue(EMS_RxTelegram, &EMS_Boiler.wWReadiness, 5, 3);
//    _setValue(EMS_RxTelegram, &EMS_Boiler.wWTemperatureOK, 5, 5);
}

proddy, 2020-06-02:
0xE9 is implemented, see:
https://github.com/proddy/EMS-ESP/blob/cfeb1ce2916af2605b4c54602936904cc123235e/src/boiler.cpp#L67

MichaelDvP, 2020-06-02:
@proddy yes, i know it's in v2, line-numbers and code is for 1.9.5 and @poesel can now verify the correct positions.

poesel, 2020-06-02:
I've added the line at 3250 and added the function behind line 1029.
Unfortunately I get this build error:

Compiling .pio\build\release\src\ems.cpp.o
src\ems.cpp:3271:39: error: 'EMS_TYPE_HPMonitorWW' was not declared in this scope
     {EMS_DEVICE_UPDATE_FLAG_HEATPUMP, EMS_TYPE_HPMonitorWW, "HeatPumpMonitorWW", _process_HPMonitorWWMessage},
                                       ^
*** [.pio\build\release\src\ems.cpp.o] Error 1

MichaelDvP, 2020-06-02:
Declare in `ems_device.h` around line 43 `#define EMS_TYPE_HPMonitorWW 0xE9`

poesel, 2020-06-02:
Ok, did that and it compiled and deployed without problems. Thanks.
Upon first boot the boiler wasn't recognized. I thought it was because of EMS mode so I switched to EMS+ and restarted. Unfortunately that didn't work either:

> 
> devices
> 
> No devices were recognized. This may be because Tx is disabled or failing.
> 
> info
> 
> EMS-ESP system status:
>   System logging set to None
>   LED: on, Listen mode: off
>   Boiler: disabled, Thermostat: disabled, Solar Module: disabled, Mixing Module: disabled
>   Shower Timer: disabled, Shower Alert: disabled
>   Saved known device IDs: none
> 
> EMS Bus status:
>   Bus is connected, protocol: HT3
>   Rx: # successful read requests=0, # CRC errors=2
>   Tx: no signal
> 

I switched back to EMS and to my surprise:

> 
> devices
> 
> These 2 were recognized by EMS-ESP:
>  Boiler: Bosch Condens 5000i (DeviceID: 0x08, ProductID: 195, Version: 05.02)
>  Controller: Condens 5000 Controller (DeviceID: 0x09, ProductID: 241, Version: 01.05)
> 
> info
> 
> EMS-ESP system status:
>   System logging set to None
>   LED: on, Listen mode: off
>   Boiler: enabled, Thermostat: disabled, Solar Module: disabled, Mixing Module: disabled
>   Shower Timer: disabled, Shower Alert: disabled
>   Saved known device IDs: none
> 
> EMS Bus status:
>   Bus is connected, protocol: HT3
>   Rx: # successful read requests=11, # CRC errors=3
>   Tx: Last poll=1.392 seconds ago, # successful write requests=0
> 
> Boiler data:
>   Boiler: Bosch Condens 5000i (DeviceID: 0x08, ProductID: 195, Version: 05.02)
>   Hot tap water: running
>   Central heating: off
>   Warm Water activated: ?
>   Warm Water circulation pump available: ?
>   Warm Water circulation pump type: 3-way pump
>   Warm Water circulation pump freq: 5x3min
>   Warm Water selected temperature: ? C
>   Warm Water desinfection temperature: 70 C
>   Warm Water circulation active: off
>   Warm Water set temperature: 60 C
>   Warm Water current temperature: 60.2 C
>   Warm water temperature (intern): ? C
>   Warm water temperature (extern): ? C
>   Warm Water current temperature (extern): 60.2 C
>   Warm Water current tap water flow: ? l/min
>   Warm Water # starts: 451 times
>   Warm Water active time: 0 days 0 hours 39 minutes
>   Warm Water 3-way valve: off
>   Selected flow temperature: 0 C
>   Current flow temperature: 50.6 C
>   Max boiler temperature: ? C
>   Return temperature: ? C
>   Gas: off
>   Warm water mode: off
>   Boiler pump: off
>   Fan: off
>   Ignition: off
>   Circulation pump: off
>   Burner selected max power: 0 %
>   Burner current power: 0 %
>   Flame current: 0.0 uA
>   System pressure: ? bar
>   System service code:
>   Heating temperature setting on the boiler: ? C
>   Boiler circuit pump modulation max power: ? %
>   Boiler circuit pump modulation min power: ? %
>   Outside temperature: --3276.-8 C
>   Exhaust temperature: ? C
>   Pump modulation: 0 %
>   Burner # starts: 51 times
>   Total burner operating time: 0 days 10 hours 36 minutes
>   Total heat operating time: 0 days 3 hours 5 minutes
>   Total UBA working time: 14 days 11 hours 9 minutes
> 

But still lots of '?'.
Hmm - so I went back to EMS+ and tried twice but TX is not working. No result.

MichaelDvP, 2020-06-03:
EMS is a slow bus and after restart it takes some time (up to a minute) to collect all data. There are allways some '?', because not all boilers/compressors have the same parameters. 
It's up to you to check if there are values that are displayed on the heating control, but not in emsesp. Than you can log the messages and change a value on the heating-control (remember the time of change), Later search the log what have changed at this time. As you can see [here](https://emsesp.github.io/docs/#/MC110-controller) there are some positions unknown in the telegrams. If you have identified an wrong/new parameter-position don't forget to tell it proddy.
Stay with tx_mode ems, if there are successful read requests it works.

poesel, 2020-06-03:
Thanks for the help!

btw the page you linked to has 2 typos: one 'r' is missing in 'controller' and I guess in line 3 it should be 'byte' instead of 'bit'

closed 2020-06-03 by poesel

proddy, 2020-06-03:
glad you got it working. I fixed the typo on the wiki too !