emsesp/EMS-ESP #395: syslog errors on v2: reboots and incorrect timings

proddy, 2020-06-05:
I logged with syslog to see the tx-errors and there is a another strange thing. I get reboots every 2 hours (mark is set to 2h, can it be that?) and always after 4 complete tx-errors and the first error of 5th. But the time between retries is very long, seems the counter is'nt cleared in between.  

Another thing: [system] and [network] logs with local time (mest), [emseesp] (and also [boiler], [thermostat]) logs with utc.
[syslog.txt](https://github.com/proddy/EMS-ESP/files/4734214/syslog.txt)

_Originally posted by @MichaelDvP in https://github.com/proddy/EMS-ESP/issues/380#issuecomment-639281947_

proddy, 2020-06-05:
@MichaelDvP been running for a while and can't reproduce. Have the internal mark set to 30 mins and not seeing any restarts. I'll keep testing.

MichaelDvP, 2020-06-06:
I had the same 2h-restarts with syslog disabled, I now suspect the `EMS-DEBUG` flag, that was set. Testing now with disabled flag.

MichaelDvP, 2020-06-06:
The time-settings depends on the network connection, the first messages before connected send without time, the receiving syslog.d inserts the localtime. After connection the emsesp sends gmt-time: `syslog.ccp` line 394. For me i commented out this line and my syslog now shows a consistent timeline. 
I close this issue, the reboots are not caused by syslog.

closed 2020-06-06 by MichaelDvP

proddy, 2020-06-06:
I made the change in syslog as you suggested.