emsesp/EMS-ESP #397: Add BufferBoilerTemp and MixTemperature for Bosch 9000i

suysh, 2020-06-08:
Hello,
I found 2 parameters in my thermostat that are not read in the EMS-ESP, and I found the locations in the telegrams.
Can you add the 2 parameters from the following telegram:
`(21:10:43) 88 00 2A 00 00 00 00 00 00 00 00 00 D2 00 00 80 00 00 [01 9D] 80 00 [02 79] 00`

The value [01 9D] = 41.3°C corresponds to MixTemperature (mengertemperatuur) on my thermostat in the menu Device/Burner, but I think it's for warmWater use.
The value [02 79] = 63.3°C corresponds to BufferBoilerTemperature (bufferboilertemperatuur) on my thermostat in the menu Solar
For the moment it's unclear to what these values correspond in my system, but when I can log them it will be visible and I will report back with the exact function.

All information about my system:
```
EMS-ESP system status:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: enabled, Solar Module: enabled, Mixing Module: enabled
  Shower Timer: disabled, Shower Alert: disabled
  Saved known device IDs: none

EMS Bus status:
  Bus is connected, protocol: HT3
  Rx: # successful read requests=5376, # CRC errors=0
  Tx: Last poll=3.350 seconds ago, # successful write requests=0

Boiler data:
  Boiler: Buderus Logamax plus/GB192/Bosch Condens GC9000 (DeviceID: 0x08, ProductID: 208, Version: 01.04)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water circulation pump type: 3-way pump
  Warm Water circulation pump freq: 2x3min
  Warm Water comfort setting: Eco
  Warm Water selected temperature: 60 C
  Warm Water desinfection temperature: 70 C
  Warm Water circulation active: off
  Warm Water set temperature: 58 C
  Warm Water current temperature: 42.3 C
  Warm water temperature (intern): 42.3 C
  Warm water temperature (extern): 56.5 C
  Warm Water current temperature (extern): 56.5 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 682 times
  Warm Water active time: 4 days 21 hours 29 minutes
  Warm Water 3-way valve: on
  Selected flow temperature: 5 C
  Current flow temperature: 48.5 C
  Max boiler temperature: ? C
  Return temperature: 32.7 C
  Gas: off
  Warm water mode: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: 1.1 bar
  System service code:  (203)
  Heating temperature setting on the boiler: 82 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 10 %
  Outside temperature: 13.8 C
  Exhaust temperature: ? C
  Pump modulation: 0 %
  Burner # starts: 2484 times
  Total burner operating time: 42 days 18 hours 29 minutes
  Total heat operating time: 37 days 21 hours 0 minutes
  Total UBA working time: 399 days 0 hours 16 minutes

Solar Module data:
  Solar module: SM100 Solar Module (DeviceID: 0x30, ProductID: 163, Version: 27.05)
  Collector temperature (TS1): 18.0 C
  Bottom temperature (TS2): 51.9 C
  Bottom temperature (TS5): --3276.-8 C
  Pump modulation: 0 %
  Valve (VS2) status: off
  Pump (PS1) active: off
  Energy last hour: 0.0 Wh
  Energy today: 1012 Wh
  Energy total: 83.6 kWh

Thermostat data:
  Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID: 0x10, ProductID: 158, Version: 18.06)
  Thermostat time is 21:18:07 08/06/2020
  Heating Circuit 1
   Current room temperature: 23.8 C
   Setpoint room temperature: 14.0 C
   Mode is set to manual
  Heating Circuit 2
   Current room temperature: 23.8 C
   Setpoint room temperature: 14.0 C
   Mode is set to manual

Mixing module data:
  Mixing Module: MM100 Mixing Module (DeviceID: 0x21, ProductID: 160, Version: 24.05)
  Mixing Circuit 1
   Current pump modulation: 0 %
   Current valve status: 0
  Mixing Circuit 2
   Current flow temperature: 27.5 C
   Current pump modulation: 0 %
   Current valve status: 0
  Warm Water Circuit 1
   Current pump status: 0
  Warm Water Circuit 2
   Current pump status: 0

```
Kind regards, Hans

proddy, 2020-06-08:
Sure, it's easy to add. But....I'm no longer adding new functionality to version 1.9.x so you'll have to switch to version 2.0 which is still in 'alpha' but kinda works. It's in a separate v2 branch and you'll need to compile and upload yourself for now. If you need help let me know. I'll add the new variables tomorrow.

suysh, 2020-06-09:
Thanks.
Do you have a document step-by-step how to compile on a windows machine? (I never did)
Can I upload the newly compiled file via the 'update firmware' menu item? And upload again the stable version in the same way?

proddy, 2020-06-09:
There is no Web UI in version 2 just either use PlatformIO to build and upload or maybe easier if I provide a build and you follow the steps on https://emsesp.github.io/docs/#/Uploading-firmware?id=via-usb

And its always handy to have a spare Wemos ESP8266 if you have one, so you can switch between v1.0 and v2.0. I haven't tested v2 with HT3 or Solar yet so you'll be the first, which is great but also a little scary

suysh, 2020-06-09:
If you can compile and provide me the bin that would be very helpfull.
I have some spare ESP's to try.

proddy, 2020-06-09:
@suysh can you give me some example data so I can run some simulations. They are coming from the thermostat right? (device id 0x10)

also noticed the strange bottom temp on the Solar (value --3276.-8 C) so would be interesting if that is fixed in v2.

suysh, 2020-06-09:
@proddy I think the data is coming from the solar module.
The strange value is maybe because I have only one temperature sensor.
My solar is recently filled and I have to do some more research about which values are present.
How you prefer the example data? Added normal and raw.
```
System Logging set to Solar Module only
(16:55:17) Solar Module -> All, Type(0x0266), telegram: B0 00 FF 00 02 66 01 62 00 13 40 (#data=5)
(16:55:18) Solar Module -> All, SM100Monitor(0x0262), telegram: B0 00 FF 00 02 62 02 7B (#data=2)
Fetching data from EMS devices
(16:55:47) Solar Module -> All, SM100Monitor(0x0262), telegram: B0 00 FF 00 02 62 02 84 02 3C 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 (#data=24)
(16:55:47) Solar Module -> All, SM100Monitor(0x0262), telegram: B0 00 FF 18 02 62 80 00 (#data=2)
(16:55:47) Solar Module -> All, Type(0x0263), telegram: B0 00 FF 00 02 63 80 00 80 00 00 00 80 00 80 00 80 00 00 80 00 (#data=15)
(16:55:48) Solar Module -> All, SM100Status(0x0264), telegram: B0 00 FF 02 02 64 00 04 00 00 00 00 00 00 00 00 64 (#data=11)
(16:55:48) Solar Module -> All, Type(0x0266), telegram: B0 00 FF 00 02 66 01 62 00 13 40 (#data=5)
(16:55:48) Solar Module -> All, Type(0x0268), telegram: B0 00 FF 00 02 68 0C 00 (#data=2)
(16:55:49) Solar Module -> All, SM100Status2(0x026A), telegram: B0 00 FF 02 02 6A 03 00 00 03 03 00 00 00 03 (#data=9)
(16:55:49) Solar Module -> All, Type(0x02A9), telegram: B0 00 FF 00 02 A9 80 00 (#data=2)
(16:55:50) Solar Module -> All, Type(0x02AA), telegram: B0 00 FF 00 02 AA 00 00 03 64 00 (#data=5)
(16:55:50) Solar Module -> All, SM100Monitor(0x0262), telegram: B0 00 FF 00 02 62 02 85 (#data=2)
(16:55:56) Solar Module -> Me, SM100Monitor(0x0262), telegram: B0 0B FF 00 02 62 02 87 02 3C 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 (#data=25)
(16:56:17) Solar Module -> All, SM100Monitor(0x0262), telegram: B0 00 FF 00 02 62 02 8F (#data=2)
(16:56:18) Solar Module -> All, Type(0x0266), telegram: B0 00 FF 00 02 66 01 62 00 13 40 (#data=5)
Fetching data from EMS devices
Publishing MQTT data for: thermostat boiler mixing solar
(16:56:48) Solar Module -> All, SM100Monitor(0x0262), telegram: B0 00 FF 00 02 62 02 99 02 3C 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 (#data=24)
(16:56:48) Solar Module -> Boiler, UBAFlags(0x35), telegram: B0 08 35 02 00 (#data=1)
(16:56:48) Solar Module -> All, SM100Monitor(0x0262), telegram: B0 00 FF 18 02 62 80 00 (#data=2)
(16:56:49) Solar Module -> All, Type(0x0263), telegram: B0 00 FF 00 02 63 80 00 80 00 00 00 80 00 80 00 80 00 00 80 00 (#data=15)
(16:56:49) Solar Module -> All, SM100Status(0x0264), telegram: B0 00 FF 02 02 64 00 04 00 00 00 00 00 00 00 00 64 (#data=11)
(16:56:50) Solar Module -> All, Type(0x0266), telegram: B0 00 FF 00 02 66 01 62 00 13 40 (#data=5)
(16:56:50) Solar Module -> All, Type(0x0268), telegram: B0 00 FF 00 02 68 0C 00 (#data=2)
(16:56:50) Solar Module -> All, SM100Status2(0x026A), telegram: B0 00 FF 02 02 6A 03 00 00 03 03 00 00 00 03 (#data=9)
(16:56:51) Solar Module -> All, Type(0x02A9), telegram: B0 00 FF 00 02 A9 80 00 (#data=2)
(16:56:51) Solar Module -> All, Type(0x02AA), telegram: B0 00 FF 00 02 AA 00 00 03 64 00 (#data=5)
(16:56:53) Solar Module -> Me, SM100Monitor(0x0262), telegram: B0 0B FF 00 02 62 02 9B 02 3C 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 (#data=25)
(16:57:14) Solar Module -> All, SM100Status(0x0264), telegram: B0 00 FF 09 02 64 64 (#data=1)
(16:57:15) Solar Module -> All, SM100Status2(0x026A), telegram: B0 00 FF 0A 02 6A 04 (#data=1)
(16:57:17) Solar Module -> All, SM100Status(0x0264), telegram: B0 00 FF 09 02 64 0F (#data=1)
(16:57:18) Solar Module -> All, SM100Monitor(0x0262), telegram: B0 00 FF 00 02 62 02 A3 (#data=2)
(16:57:19) Solar Module -> All, Type(0x0266), telegram: B0 00 FF 00 02 66 01 62 00 13 40 (#data=5)
(16:57:44) Solar Module -> All, SM100Monitor(0x0262), telegram: B0 00 FF 00 02 62 02 AD (#data=2)
Fetching data from EMS devices
(16:57:49) Solar Module -> All, SM100Monitor(0x0262), telegram: B0 00 FF 00 02 62 02 AF 02 3C 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 (#data=24)
(16:57:49) Solar Module -> All, SM100Monitor(0x0262), telegram: B0 00 FF 18 02 62 80 00 (#data=2)
(16:57:50) Solar Module -> All, Type(0x0263), telegram: B0 00 FF 00 02 63 80 00 80 00 00 00 80 00 80 00 80 00 00 80 00 (#data=15)
(16:57:50) Solar Module -> All, SM100Status(0x0264), telegram: B0 00 FF 02 02 64 00 04 00 00 00 00 00 0F 00 00 64 (#data=11)
(16:57:51) Solar Module -> All, Type(0x0266), telegram: B0 00 FF 00 02 66 01 62 00 13 40 (#data=5)
(16:57:51) Solar Module -> All, Type(0x0268), telegram: B0 00 FF 00 02 68 0C 00 (#data=2)
(16:57:51) Solar Module -> All, SM100Status2(0x026A), telegram: B0 00 FF 02 02 6A 03 00 00 03 03 00 00 00 04 (#data=9)
(16:57:52) Solar Module -> All, Type(0x02A9), telegram: B0 00 FF 00 02 A9 80 00 (#data=2)
(16:57:52) Solar Module -> All, Type(0x02AA), telegram: B0 00 FF 00 02 AA 00 00 03 64 00 (#data=5)
(16:57:55) Solar Module -> Me, SM100Monitor(0x0262), telegram: B0 0B FF 00 02 62 02 B1 02 3C 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 (#data=25)
(16:58:10) Solar Module -> All, SM100Monitor(0x0262), telegram: B0 00 FF 00 02 62 02 B7 (#data=2)
(16:58:13) Solar Module -> All, SM100Status(0x0264), telegram: B0 00 FF 09 02 64 11 (#data=1)
(16:58:20) Solar Module -> All, Type(0x0266), telegram: B0 00 FF 00 02 66 01 62 00 13 40 (#data=5)
```
```
[... cut ...]
(17:00:03) A0 00 FF 00 01 D7 00 00 00 80 00 00 00 00 03
(17:00:03) A1 08 1A 00 00 00 00
(17:00:03) A1 00 FF 00 01 55 00
(17:00:03) A1 00 FF 00 01 D8 00 02 00 01 1F 00 00 00 03
(17:00:04) 88 00 18 00 05 01 0C 00 00 00 00 40 40 01 5F 01 F4 01 05 00 00 0C 00 00 00 CB 00 00 00 00 00
(17:00:04) 88 00 18 1B 00 00 00 00 00 00 00 00 00 00 00
(17:00:04) 88 00 2A 00 00 00 00 00 00 00 00 00 D2 00 00 80 00 00 01 08 80 00 02 47 00
(17:00:05) 88 00 34 00 3A 01 5F 01 F5 21 00 00 04 00 00 1B 89 00 02 AA 00 80 00
```

proddy, 2020-06-09:
These messages come from the boiler (the source is 0x08 or 0x88) and part of the MC10Status telegram. Luckily this was already added in v2 so I just included the two extra parameters. You can try it out.

- download the firmware from https://github.com/proddy/EMS-ESP/releases/download/travis-dev-build/EMS-ESP-dev-esp8266-2.0a17.xxx
- install Python 3.8 on your Windows machine (https://www.python.org/downloads/)
- install esptool with `pip install esptool`
- connect an ESP8266 to the USB. Figure out which COM port it's on (windows device manager). E.g. COM6
- `esptool.py -p COM6 -b 921600 write_flash 0x00000 EMS-ESP-dev-esp8266-2.0a17.xxx`

when finished, use putty or something similar to connect to the Serial port with baud 115200. You should see the EMS-ESP menu. Do `su` with password `neo`, then `system`, and then `set wifi ssid` and `set wifi password`. After it boots connect it back to the EMS (or mount it on a BBQKees gateway) and you're all set. See the README for more commands and how to log.

suysh, 2020-06-11:
Hello,

I wrote the image to the board, can open it in putty, change parameters.
But after a restart I cant connect via USB. I see that it connects to my wireless router and receive an IP address.
That’s all fine, but I can’t reach my board via USB for updating other parameters like mqtt etc...

I reflashed my board and as long I don’t do a reset it’s fine to connect...
I liked the webinterface to update parameters.

Kind regards, Hans

MichaelDvP, 2020-06-11:
After you have successful connected to your wifi, the com-port is used for ems and you can only reach the ems-esp by telnet over wifi.

suysh, 2020-06-11:
Yes, that's the trick... Thanks !!
Now connect it to EMS-bus and test

suysh, 2020-06-11:
Hello, I did not get it fully functional. See some logs below.
I receive a lot of Tx operation failed.
```
EMS Bus info:
  Bus protocol: HT3
  #telegrams received: 309
  #read requests sent: 0
  #write requests sent: 0
  #incomplete telegrams: 1 (0%)
  #tx fails (after 3 retries): 107
Rx Queue is empty
Tx Queue is empty
```

```
000+00:07:33.438 N 125: [emsesp] Unrecognized EMS device with device ID 0x16 with product ID 220. Please report on GitHub.
000+00:07:35.345 E 126: [emsesp] Last Tx operation failed after 3 retries. Ignoring request.
000+00:07:35.988 N 127: [emsesp] Unrecognized EMS device with device ID 0x16 with product ID 220. Please report on GitHub.
000+00:07:37.294 E 128: [emsesp] Last Tx operation failed after 3 retries. Ignoring request.
000+00:07:37.313 N 129: [emsesp] Unrecognized EMS device with device ID 0x16 with product ID 220. Please report on GitHub.
000+00:07:40.419 E 130: [emsesp] Last Tx operation failed after 3 retries. Ignoring request.
000+00:08:01.720 E 131: [emsesp] Last Tx operation failed after 3 retries. Ignoring request.
000+00:08:03.750 E 132: [emsesp] Last Tx operation failed after 3 retries. Ignoring request.
000+00:08:05.820 E 133: [emsesp] Last Tx operation failed after 3 retries. Ignoring request.
000+00:08:08.125 E 134: [emsesp] Last Tx operation failed after 3 retries. Ignoring request.
000+00:08:10.850 E 135: [emsesp] Last Tx operation failed after 3 retries. Ignoring request.
000+00:08:12.820 E 136: [emsesp] Last Tx operation failed after 3 retries. Ignoring request.
000+00:08:14.970 E 137: [emsesp] Last Tx operation failed after 3 retries. Ignoring request.
000+00:08:30.364 N 138: [emsesp] Unrecognized EMS device with device ID 0x16 with product ID 220. Please report on GitHub.
000+00:08:31.865 N 139: [emsesp] Unrecognized EMS device with device ID 0x16 with product ID 220. Please report on GitHub.
000+00:08:33.045 E 140: [emsesp] Last Tx operation failed after 3 retries. Ignoring request.
000+00:08:33.064 N 141: [emsesp] Unrecognized EMS device with device ID 0x16 with product ID 220. Please report on GitHub.
```

```
Boiler: Junkers Logamax plus/GB192/Condens GC9000 (DeviceID:0x08 ProductID:208, Version:01.04)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water charging type: 3-way valve
  Warm Water circulation pump available: off
  Warm Water circulation pump freq: 2x3min
  Warm Water circulation active: off
  Warm Water comfort setting: Eco
  Warm water mix temperature: 27.6°C
  Warm water buffer boiler temperature: 75.7°C
  Warm Water disinfection temperature: 70°C
  Warm Water selected temperature: 60°C
  Warm Water set temperature: 58°C
  Warm Water current temperature (intern): 34.7°C
  Warm water storage temperature (intern): 34.7°C
  Warm Water current temperature (extern): 48.8°C
  Warm water storage temperature (extern): 48.7°C
  Warm Water current tap water flow: 0.0l/min
  Warm Water # starts: 684
  Warm Water active time: 4 days 21 hours 54 minutes
  Warm Water charging: on
  Warm Water disinfecting: off
  Selected flow temperature: 5°C
  Current flow temperature: 28.2°C
  Max boiler temperature: ?°C
  Return temperature: 26.3°C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Burner selected max power: 0%
  Burner current power: 0%
  Flame current: 0.0uA
  System pressure: 1.7bar
  System service code:  (203)
  Heating temperature setting on the boiler: 82°C
  Boiler circuit pump modulation max power: 100%
  Boiler circuit pump modulation min power: 10%
  Outside temperature: 20.2°C
  Exhaust temperature: ?°C
  Pump modulation: 0%
  Burner # starts: 2487
  Total burner operating time: 42 days 18 hours 56 minutes
  Total heat operating time: 37 days 21 hours 2 minutes

Thermostat: RC300/RC310/Moduline 3000/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:18.06)
  Clock: 20:31:25 11/06/2020

Mixing Module: MM100 (DeviceID:0x20, ProductID:160, Version:24.05)

Mixing Module: MM100 (DeviceID:0x21, ProductID:160, Version:24.05)

Solar Module: SM100 (DeviceID:0x30, ProductID:163, Version:27.05)
  Collector temperature (TS1): ?°C
  Bottom temperature (TS2): ?°C
  Bottom temperature (TS5): ?°C
  Pump modulation: ?%
  Valve (VS2) status: ?
  Pump (PS1) active: ?
  Energy last hour: ?Wh
  Energy today: ?Wh
  Energy total: ?kWh
```

proddy, 2020-06-11:
do

`su`
`ems`
`set tx_mode 3`

which switches to Junkers/Bosch/HT3 mode for sending. Eventually, we'll just have one but for now, you'll need to set it manually (only once).

suysh, 2020-06-11:
Yes, did that
```
000+00:04:06.279 E 52: [emsesp] Last Tx operation failed after 3 retries. Ignoring request.
000+00:04:08.803 E 53: [emsesp] Last Tx operation failed after 3 retries. Ignoring request.
000+00:04:11.279 E 54: [emsesp] Last Tx operation failed after 3 retries. Ignoring request.
000+00:04:13.204 E 55: [emsesp] Last Tx operation failed after 3 retries. Ignoring request.
ems-esp:/ems# set tx_mode 3
Tx mode = 3
ems-esp:/ems# help
exit
help
log [level] [pretty | raw] [trace ID]

```
You can see in the log
```
EMS Bus info:
  Bus protocol: HT3
  #telegrams received: 309
  #read requests sent: 0
  #write requests sent: 0
  #incomplete telegrams: 1 (0%)
  #tx fails (after 3 retries): 107
Rx Queue is empty
Tx Queue is empty
```

suysh, 2020-06-11:
I switched back to my old ESP and there everything is again working.
Overlooked my connections with the new board, nothing strange there.
Was running over the commands in the 2.0 version... trying until the 'format' function.
Probably it is doing what is says :-) Do I have to overwrite back with USB ?

proddy, 2020-06-11:
The problem is in the new UART code. I suspect on the Rx side. What I'll do is bring the old code back from 1.9 and build a switch to test old vs new. I'll let you know when it's done.

@MichaelDvP is working on this piece and although it works flawlessly on his system it's yet 100% on EMS 1.0 and HT3. But he's close!

MichaelDvP, 2020-06-12:
I think rx works, it's been 309 telegrams with valid crc, also boiler-ems telegrams are received correct, but ems+ messages are not decoded right (@proddy : missing type_id in telegram.cpp, line 297, i'll make a pr if all is fixed), all tx fails, and there is a strange device 0x16. 
I've added the ems+ decode, put retry to top of queue and add a debug-message for tx echos. 
@suysh:  can you flash this file and set putty logging to file with printable output.
Type `show`, goto `ems`, type `show`, then type `log debug raw`. Please switch the different tx-modes, type `set tx_mode 1`, wait a bit until sending,  `set tx_mode 2`, wait, `set tx_mode 3`, wait, `set tx_mode 4` while logging. Post the log here, 
[firmware.zip](https://github.com/proddy/EMS-ESP/files/4769026/firmware.zip)

PS: The format function clears the settings including wifi-settings, so you have to configure new over usb-serial-terminal if you want to keep the old firmware. Later, if connected, you can update firmware over espota `python espota.py -i ems-esp -p 8266 -a neo -f firmware.bin -r`

proddy, 2020-06-12:
@MichaelDvP yes the new rx works fine, but I've noticed it leaves something behind on the uart that somehow affects Tx. With tx_mode 1 on v2, which is essentially the same code from v1.9, I'm also getting a lot of Tx retries and a few failures. I haven't looked into why yet. I saw your changes and it looks fine so happy if you want to push the PR.

suysh, 2020-06-12:
@MichaelDvP The logs : 
[log.txt](https://github.com/proddy/EMS-ESP/files/4770115/log.txt)

MichaelDvP, 2020-06-12:
First look, mode 2 works and the long delays are needed for ems+, should be added to esp32.
Some values are missing, do they come up if it's longer in mode 2? 
If you restart in mode 2 to clear the stats, are there fails after some time?  
Retries seems to toggle the first bit, so first retry can not work.  
I think you have only hc1 and hc2, 3+4 should not be displayed, right?

@proddy v1.9.5 switches rx-timeout of, the new uart keeps it in factory state. For my understanding it only affects the interrupt, but maybe it affects the hardware, try to set `USC1(EMSUART_UART) = 0;` 
Other possibilities are: set 1,5 or 2 stopbits to make the uart wait an additional bittime (works on my system), Or wait in the interrupt for rx goes idle before reading the fifo. `while((USS(EMSUART_UART) >> USRXD) == 0);`

suysh, 2020-06-12:
@MichaelDvP Some values are also missing in V1 (don't have them in my boiler), but there are missing in V2 and have them in V1. From the solar all is present on ems except the TS5 temperature.
Correct I have 2 circuits hc1 + hc2
The log for tx_mode = 2
[putty.log](https://github.com/proddy/EMS-ESP/files/4770728/putty.log)

MichaelDvP, 2020-06-12:
Found an issue in solar energy and puzzled values in mixer, look if it's better now.
RC300 setpoint was set at 3 positions, i take only the one in MonitorMessage, since this should be the actual setpoint. Check if it is correct.
[firmware.zip](https://github.com/proddy/EMS-ESP/files/4771376/firmware.zip)

suysh, 2020-06-12:
Both info/show from V1 and V2 with only 1 minute difference so the values should be same. There are some differences.
[V2.txt](https://github.com/proddy/EMS-ESP/files/4771586/V2.txt)
[V1.txt](https://github.com/proddy/EMS-ESP/files/4771587/V1.txt)

MichaelDvP, 2020-06-13:
@suysh the missing values for solar takes a bit to come up, the missing dot in energy values is added. I've changed tx_mode 2 to burn less cpu and to be better portable to esp32. Can you check if it is working. 
I'm a bit confused by the missing temperature value in mixer 1, what does this mixer do? 
Have you checked mqtt? All ok?
[firmware.zip](https://github.com/proddy/EMS-ESP/files/4774265/firmware.zip)

If all is ok, can you do a longer (10 min) debug raw log, there is this device 0x16, i'm curious to see what it is sending on the bus, seems to be very rare. Do you know what it is?

@proddy can you also check mode 2 on your system. I'll push the source to my github.

proddy, 2020-06-13:
@MichaelDvP I'm working on some changes to the core too so I will merge with yours and test tx_mode 2 (although I have EMS1.0).

That device ID of 0x16 is strange, I can't find it anywhere in the ht3 or buderus documentation. We would need to see the results of a ` boiler/read 7` (EMS_TYPE_UBADevices) to see if it really does exist and also ping it with `ems\send telegram 0B 96 02 00 20` to get the version details.

proddy, 2020-06-13:
@MichaelDvP I'm getting now a 100% success rate with Tx, using tx_mode 1 if I add back the `USC1(EMSUART_UART) = 0`. The other tx modes are still failing. I'll checkin my changes shortly and then we can merge with yours. Or if you prefer I can port them manually over?

MichaelDvP, 2020-06-13:
We can merge later. My system is also EMS 1.0 and all modes working. Maybe it is uncritical because all components are on the boiler and the ems-bus lines are less than 1m (no roomcontroler, only weather controlled), No emc problems, no reflections, etc. 
Can you check with less delay, set `emsTxWait = 5 * EMSUART_BIT_TIME * 11;` for 11 bittimes. 
Also you can check with break ended by timer in the isr uncommend the last lines but set the timer to `timer1_write(5 * EMSUART_TX_BRK_WAIT`.

proddy, 2020-06-13:
I tried both those and still the same. I'll check in what I have. tx_mode 1 works fine on my system. The EMS-ESP gateway is 3m from the boiler connected via the EMS wires. I think we should move this conversation back to UART github issue so we can help Hans with his solar & mixer issues :-)

suysh, 2020-06-14:
I tested the version from yesterday morning.
The logs : 
[putty.log](https://github.com/proddy/EMS-ESP/files/4776998/putty.log)
There are a lot of warnings, I copied some below.
```
ems-esp:/$ 000+00:08:51.720 D 1862: [emsesp] No telegram type handler found for ID 0x255 (src 0x20, dest 0x00)
ems-esp:/$ log level000+00:08:28.780 D 1827: [emsesp] No telegram type handler found for ID 0x3AA (src 0x30, dest 0x00)
ems-esp:/$ log 000+00:08:28.176 D 1824: [emsesp] No telegram type handler found for ID 0x3A9 (src 0x30, dest 0x00)
ems-esp:/$ 000+00:08:27.322 D 1818: [emsesp] No telegram type handler found for ID 0x368 (src 0x30, dest 0x00)
ems-esp:/$ 000+00:08:27.136 D 1815: [emsesp] No telegram type handler found for ID 0x366 (src 0x30, dest 0x00)
ems-esp:/$ 000+00:08:26.279 D 1809: [emsesp] No telegram type handler found for ID 0x363 (src 0x30, dest 0x00)
ems-esp:/$ log 0 pret000+00:08:24.591 D 1800: [emsesp] No telegram type handler found for ID 0xBF (src 0x20, dest 0x00)
ems-esp:/$ log 0000+00:08:23.932 N 1797: [emsesp] Unrecognized EMS device with device ID 0x16 with product ID 220. Please report on GitHub.
ems-esp:/$ log000+00:08:22.777 N 1783: [emsesp] Unrecognized EMS device with device ID 0x09 with product ID 224. Please report on GitHub.

```

suysh, 2020-06-14:
Now the ems interface is working, I want to continue to update the mqtt.
There is no mqtt in the telnet menu for setting the mqtt parameters?
Is there a difference in the method how the values via mqtt are transmitted ?

MichaelDvP, 2020-06-15:
The warnings are normal, not all type-ids re known for now
0xBF is a error message from a device, mixers and solar sending this every minute.
0x255 is from mixer a single byte message, always 0, 
The other messages from solar 0x3xx are not known. If you know a value in your system that is not displayed in ems-esp, we can search the messages for this value.
Device ID 0x09 is the controller in your 9000i, the contoller is only pollig the devices and send no messages himself, we'll add the controller, but it's only cosmetic.
The device 0x16 seems to send nothing, it's only registered in the boiler. Do you have any extra module installed to the bus, like ASM10, PM10, EM10, GM10, DM10, DRM10, UM10, VM10,..  

mqtt is a submenu. in telnet type `mqtt`, than `help` for commands and `set` for settings (`su` is needed to change settings and to show full commands).

suysh, 2020-06-16:
OK. Thanks for explaining. Looked over the `mqtt` menu, or forgot the `su`.
Not all possible menu combinations are described, but that's normal for a debugging version.

No, don't know any other installed modules. Let me know if I can help by testing a new version.
When V2 is stable enough to run fulltime, I will search for missing values in my thermostat.

proddy, 2020-06-16:
> Not all possible menu combinations are described, but that's normal for a debugging version.

A full list of commands are described in the v2 branch page in GitHub (https://github.com/proddy/EMS-ESP/tree/v2). If you have any ideas on how to make EMS-ESP more intuitive please let us know. Because telnet is insecure I wanted to prevent anyone from hacking into a home network and blowing up the boiler, so hid the commands under 'su'. This was a request from a user 

> When V2 is stable enough to run fulltime

I think v2 is stable enough now. I've been running it for a while in production with no issues. It has the functionality as 1.9 plus more. When the web UI is finished I'll move to a beta. You can take the latest builds directly from github [here](https://github.com/proddy/EMS-ESP/releases/tag/travis-v2-build).

suysh, 2020-06-16:
:-) Aha, didn't found the section of v2 before, indeed there is everything completely described.
Used the lastest version `EMS-ESP version 2.0.0a20`
I tried to get the mqtt to work in v2 but nothing gets published. Values are the same as my V1 interface. Can I see on telnet if there are values received from ems and transmitted to mqtt?
```
MQTT is enabled
IP = 10.0.0.22
Username = <unset>
Password = <unset>
Port = 1883
Base = ems-esp
QOS = 1
Retain Flag = disabled
Format for JSON = nested
Heartbeat = enabled
Publish time = 10 seconds
```

MichaelDvP, 2020-06-16:
After setting the parameters it is a good idea to restart the ems-esp (system/restart). There is a `reconnect` command in mqtt section, but with my mqtt-broker (ioBroker) it doesn't work in all cases, `restart` works always. With `show` you can see if you are connected, `log debug` lists also the mqtt publishes. The `publish` command tries to publish all at one time, for me with lots of sensors that gives errors at the third publish (boiler_data). Better to wait a few seconds with logging on to see the publishes.

suysh, 2020-06-16:
Found the problem, mqtt is publishing with topic 'ems-esp/<base>/sm_data'
I need to clear the base for using topic 'ems-esp/sm_data'
Now I'm receiving data via mqtt, but with errors. Sometimes the data is not correct. collectortemp is fluctuating, energylasthour is huge.
`log debug raw` is not working in telnet?
Any idea?
```
[20:06:46] openhabian@openhab:~$ mosquitto_sub -v -t 'ems-esp/sm_data'
ems-esp/sm_data {"collectortemp":0.3,"bottomtemp":81.9,"pumpmodulation":0,"pump":"off","valvestatus":"off","energylasthour":1671168,"energytoday":920,"energytotal":106.3}
ems-esp/sm_data {"collectortemp":0.3,"bottomtemp":81.9,"pumpmodulation":0,"pump":"off","valvestatus":"off","energylasthour":1671168,"energytoday":920,"energytotal":106.3}
ems-esp/sm_data {"collectortemp":0.3,"bottomtemp":81.9,"pumpmodulation":0,"pump":"off","valvestatus":"off","energylasthour":1671168,"energytoday":920,"energytotal":106.3}
ems-esp/sm_data {"collectortemp":46,"bottomtemp":81.9,"pumpmodulation":0,"pump":"off","valvestatus":"off","energylasthour":1671168,"energytoday":920,"energytotal":106.3}
ems-esp/sm_data {"collectortemp":46,"bottomtemp":81.9,"pumpmodulation":0,"pump":"off","valvestatus":"off","energylasthour":1671168,"energytoday":920,"energytotal":106.3}
ems-esp/sm_data {"collectortemp":45.7,"bottomtemp":81.9,"pumpmodulation":0,"pump":"off","valvestatus":"off","energylasthour":0,"energytoday":920,"energytotal":106.3}
ems-esp/sm_data {"collectortemp":0.3,"bottomtemp":81.9,"pumpmodulation":0,"pump":"off","valvestatus":"off","energylasthour":1671168,"energytoday":920,"energytotal":106.3}
ems-esp/sm_data {"collectortemp":45.5,"bottomtemp":81.9,"pumpmodulation":0,"pump":"off","valvestatus":"off","energylasthour":1671168,"energytoday":920,"energytotal":106.3}
ems-esp/sm_data {"collectortemp":45.5,"bottomtemp":81.9,"pumpmodulation":0,"pump":"off","valvestatus":"off","energylasthour":1671168,"energytoday":920,"energytotal":106.3}
ems-esp/sm_data {"collectortemp":45.4,"bottomtemp":81.9,"pumpmodulation":0,"pump":"off","valvestatus":"off","energylasthour":1671168,"energytoday":920,"energytotal":106.3}
ems-esp/sm_data {"collectortemp":45.4,"bottomtemp":81.9,"pumpmodulation":0,"pump":"off","valvestatus":"off","energylasthour":1671168,"energytoday":920,"energytotal":106.3}
ems-esp/sm_data {"collectortemp":45.2,"bottomtemp":81.9,"pumpmodulation":0,"pump":"off","valvestatus":"off","energylasthour":0,"energytoday":920,"energytotal":106.3}
ems-esp/sm_data {"collectortemp":0.3,"bottomtemp":81.9,"pumpmodulation":0,"pump":"off","valvestatus":"off","energylasthour":1671168,"energytoday":920,"energytotal":106.3}
ems-esp/sm_data {"collectortemp":0.3,"bottomtemp":81.9,"pumpmodulation":0,"pump":"off","valvestatus":"off","energylasthour":1671168,"energytoday":920,"energytotal":106.3}
ems-esp/sm_data {"collectortemp":0.3,"bottomtemp":81.9,"pumpmodulation":0,"pump":"off","valvestatus":"off","energylasthour":1671168,"energytoday":920,"energytotal":106.3}
ems-esp/sm_data {"collectortemp":44.9,"bottomtemp":81.9,"pumpmodulation":0,"pump":"off","valvestatus":"off","energylasthour":1671168,"energytoday":920,"energytotal":106.3}
ems-esp/sm_data {"collectortemp":44.9,"bottomtemp":81.9,"pumpmodulation":0,"pump":"off","valvestatus":"off","energylasthour":1671168,"energytoday":920,"energytotal":106.3}
```

suysh, 2020-06-16:
Log seems to be split in log and watch command. The log below is done during receiving different collectortemp values.
[log.txt](https://github.com/proddy/EMS-ESP/files/4788417/log.txt)

suysh, 2020-06-16:
The telegrams with the collectortemp solar data:
```
000+01:00:09.259 N 235: [telegram] Rx: B0 0B FF 00 02 62 01 44 03 30 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 33
000+01:00:30.686 N 262: [telegram] Rx: B0 00 FF 00 02 62 01 44 03 30 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 AA
000+01:00:30.873 N 263: [telegram] Rx: B0 00 FF 18 02 62 80 00 B8
000+01:01:09.510 N 322: [telegram] Rx: B0 0B FF 00 02 62 01 44 03 30 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 33
000+01:01:31.787 N 355: [telegram] Rx: B0 00 FF 00 02 62 01 44 03 30 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 AA
000+01:01:32.407 N 356: [telegram] Rx: B0 00 FF 18 02 62 80 00 B8
```

proddy, 2020-06-16:
thanks for the raw data. I'll add it to test data and try and simulate what is happening. 

good job finding the `watch` command. A lot of commands have moved around since v1.9.

MichaelDvP, 2020-06-17:
> Now I'm receiving data via mqtt, but with errors. Sometimes the data is not correct. collectortemp is fluctuating, energylasthour is huge.

Funny combination of three issues in these lines:
```
000+00:29:12.056 D 623: [telegram] Sending read Tx [#102], telegram: 8B B0 FF 00 20 00 03 20 9D
000+00:29:12.096 D 624: [emsesp] Last Tx read successful
000+00:29:12.096 N 625: [telegram] Rx: B0 0B FF 00 00 03 DB
000+00:29:12.096 D 626: [emsdevice] Processing ISM1StatusMessage...
```
I'll make a pr.

proddy, 2020-06-17:
I'm fixing telegram.cpp and testing, so please wait for my fixes before your PR. The EMS+ logic is wrong.

MichaelDvP, 2020-06-17:
Ups, to late, but it's only a small change.

suysh, 2020-06-22:
Is the EMS+ logic error already fixed?
Keep using Tx-mode 2 ? From help : `Try different tx_mode settings. 1 is default for EMS 1.0. EMS 2.0 is 2 and HT3/Bosch/Junkers is 3`

proddy, 2020-06-22:
yes, tx_mode 2 from EMS 2.0 should work on your system. Also try modes 4 and 5. Does this from the `ems` context/menu and use `show` to see if you have an Tx failures.

@MichaelDvP was thinking of resetting all the counts (#received, #sent, #errors etc) after the `set tx_mode` command is issued so we can see impact?

suysh, 2020-06-22:
tx_mode 2 is working. It's a good idea to reset the error count.
```
EMS Bus info:
  Tx mode: 4
  Bus protocol: HT3
  #telegrams received: 604
  #read requests sent: 129
  #write requests sent: 0
  #corrupted telegrams: 0 (0%)
  #tx fails (after 3 retries): 29
```
```
EMS Bus info:
  Tx mode: 5
  Bus protocol: HT3
  #telegrams received: 981
  #read requests sent: 266
  #write requests sent: 0
  #corrupted telegrams: 0 (0%)
  #tx fails (after 3 retries): 36
```

proddy, 2020-06-22:
> tx_mode 2 is working. It's a good idea to reset the error count.

added this in 2.0.0a26

suysh, 2020-06-22:
Only `intelligent` is accepted as value for `change comfort`
```
ems-esp:/boiler# change comfort intelligent
ems-esp:/boiler# change comfort eco
Invalid value. Must be hot, eco or intelligent
ems-esp:/boiler# change comfort hot
Invalid value. Must be hot, eco or intelligent
```
OK thanks for the update. I'm running behind `EMS-ESP version 2.0.0a25`

MichaelDvP, 2020-06-22:
> Only intelligent is accepted as value for change comfort

Yes and no, the other values are accepted, but give also  a errormessage. 

Can you check is the setpoit of your RC300 are displayed correct, also if you change mode or change acual value temporary on the controller.

suysh, 2020-06-22:
The `change wwtemp <xx>` is not working.
When changing the warmwater temperature on my thermostat I found the set values in the ems telegrams. Hope this helps:
Set to 56°C:
```
000+00:25:37.924 N 157: [telegram] Rx: 90 08 35 03 [38] 7B
000+00:25:38.402 N 158: [telegram] Rx: 90 00 FF 00 02 1B [38] 1E
```
Set to 57°C:
```
000+00:27:45.330 N 203: [telegram] Rx: 90 08 35 03 [39] 7A
000+00:27:45.357 N 204: [telegram] Rx: 90 00 FF 00 02 1B [39] 1F
```
Set to 55°C:
```
000+00:33:57.160 N 230: [telegram] Rx: 90 00 FF 00 02 1B [37] 11
000+00:33:59.681 N 232: [telegram] Rx: 90 08 35 03 [37] 74
```

suysh, 2020-06-22:
For the comfort I only have 2 setting: eco / hot
If the ww is set to eco and I do `change comfort intelligent` the thermostat goes to hot
if the ww is set to hot and I do `change comfort eco` the thermostat keeps hot
if the ww is set to hot and I do `change comfort hot` the thermostat keeps hot

suysh, 2020-06-22:
For the solar : collectortemp and bottomtemp values are OK.
The `pumpmodulation` and `pump` behaves strange. The logic from v1 was working correct.
Pump is off while modulation is not zero. Data from solar is transmitted more frequently than in v1.

suysh, 2020-06-22:
To set the boiler temp, maybe the Tx telegram is also important: (set to 56°C)
```
000+01:09:04.536 N 73: [telegram] Rx: 90 08 35 03 [38] 7B
000+01:09:05.823 N 80: [telegram] Rx: 90 00 FF 00 02 1B [38] 1E
000+01:09:06.370 N 81: [telegram] [DEBUG] Tx: 8B 90 FF 00 20 01 BB 20 [38]
```

suysh, 2020-06-22:
Set back to 57°C, but this time no Tx:
```
000+01:14:21.691 N 183: [telegram] Rx: 90 08 35 03 [39] 7A
000+01:14:22.176 N 184: [telegram] Rx: 90 00 FF 00 02 1B [39] 1F
```
Set to 58°C:
```
000+01:18:20.424 N 265: [telegram] Rx: 90 08 35 03 [3A] 79
000+01:18:20.900 N 266: [telegram] Rx: 90 00 FF 00 02 1B [3A] 1C
```

suysh, 2020-06-22:
I started testing at 16h, from then the pump-value is frequently going on and off.
The 100% modulation values are also not fitting in the regulation.
[image]
```
ems-esp/sm_data {"collectortemp":96.3,"bottomtemp":89.4,"pumpmodulation":0,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":96.3,"bottomtemp":89.4,"pumpmodulation":0,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":97.5,"bottomtemp":89.4,"pumpmodulation":0,"pump":"on","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":97.5,"bottomtemp":89.4,"pumpmodulation":0,"pump":"on","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":98.3,"bottomtemp":89.4,"pumpmodulation":0,"pump":"on","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":98.3,"bottomtemp":89.4,"pumpmodulation":0,"pump":"on","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":98.3,"bottomtemp":89.4,"pumpmodulation":0,"pump":"on","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":99.3,"bottomtemp":89.4,"pumpmodulation":100,"pump":"off","valvestatus":"on","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":99.8,"bottomtemp":89.4,"pumpmodulation":15,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":99.8,"bottomtemp":89.4,"pumpmodulation":15,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":100.7,"bottomtemp":89.4,"pumpmodulation":15,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":100.7,"bottomtemp":89.4,"pumpmodulation":15,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":101.3,"bottomtemp":89.4,"pumpmodulation":15,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":101.3,"bottomtemp":89.4,"pumpmodulation":17,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":102.3,"bottomtemp":89.4,"pumpmodulation":17,"pump":"on","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":102.3,"bottomtemp":89.4,"pumpmodulation":17,"pump":"on","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":103.2,"bottomtemp":89.4,"pumpmodulation":19,"pump":"on","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":103.3,"bottomtemp":89.4,"pumpmodulation":19,"pump":"on","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":103.3,"bottomtemp":89.4,"pumpmodulation":19,"pump":"on","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":104.3,"bottomtemp":89.4,"pumpmodulation":21,"pump":"on","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":104.7,"bottomtemp":89.4,"pumpmodulation":21,"pump":"on","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":104.7,"bottomtemp":89.4,"pumpmodulation":21,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":105.6,"bottomtemp":89.4,"pumpmodulation":23,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":105.6,"bottomtemp":89.4,"pumpmodulation":23,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":106.3,"bottomtemp":89.4,"pumpmodulation":23,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":106.3,"bottomtemp":89.4,"pumpmodulation":25,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":107.2,"bottomtemp":89.4,"pumpmodulation":25,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":107.3,"bottomtemp":89.4,"pumpmodulation":25,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":108,"bottomtemp":89.4,"pumpmodulation":25,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":108.3,"bottomtemp":89.4,"pumpmodulation":27,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":108.3,"bottomtemp":89.4,"pumpmodulation":27,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
[... cut ...]
ems-esp/sm_data {"collectortemp":112.8,"bottomtemp":89.4,"pumpmodulation":39,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":111.3,"bottomtemp":89.4,"pumpmodulation":41,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
ems-esp/sm_data {"collectortemp":111.1,"bottomtemp":89.4,"pumpmodulation":41,"pump":"off","valvestatus":"off","energylasthour":1426,"energytoday":5359,"energytotal":127}
```

proddy, 2020-06-22:
the `change wwtemp` and `change comfort`, did they work correctly in v1.9? It'll help identify the root cause of the problem. 

I'll also look into the solar pump modulation on/off values since this is bug in v2 if it worked correctly in 1.9.

suysh, 2020-06-22:
Yes, it works in v1.9.

suysh, 2020-06-22:
The `boiler wwtemp xx` is not working in v1.9

proddy, 2020-06-22:
> The 100% modulation values are also not fitting in the regulation.

The code for for pump modulation is the same as in 1.9. The 100% will be in a telegram like `30 00 FF 09 02 64 64` (x64 = 100%). Perhaps we just have to ignore 100% values?

suysh, 2020-06-22:
> > The 100% modulation values are also not fitting in the regulation.
> 
> The code for for pump modulation is the same as in 1.9. The 100% will be in a telegram like `30 00 FF 09 02 64 64` (x64 = 100%). Perhaps we just have to ignore 100% values?

At first sight, it looks like there was no reason to go to 100% modulation. What I noticed is that the solar is giving much more mqtt strings in v2 as in v1.9
Probably the 100% is correct but it's only for a very short period selected, as the mqtt is publishing all modified data (very frequently) it is transmitted.

MichaelDvP, 2020-06-22:
Async motors have only very low starting torque, it's commen to start with boost, a short time with 100% , than go down to desired rpm. It's a sampling reason that you dont see if every time. 

The DHW temp is normaly set in message 33, your boiler have it in 35. My boiler does not have pos 3 in 35 (its only 2 bytes long), maybe it's possible to send both telegrams, but better to set flags for the different boiler types.

suysh, 2020-06-23:
Can you change the timing when data is transmitted via mqtt from solar? In v1.9 it was slower (approx every 2 minutes ?)

[12 more replies]

closed 2020-07-31 by proddy