emsesp/EMS-ESP #413: (v2) setting warm water mode/comfort (hot, eco, intelligent) can only bet set via Thermostat and not Boiler on some systems

proddy, 2020-06-25:
Some boilers are unable to set the ww mode, it has to be done via the thermostat. E.g @MichaelDvP found out that sending a value to telegram type 0x37 with (0=eco, 1=comfort, 2=intelligent) works fine on RC35 and probably the RC300 range of Buderus thermostats too.

We need to decide if this is true for all brands, and whether to clone or copy the logic between boiler and thermostat devices classes.

SpaceTeddy, 2020-06-26:
may you send me the full hex string and I will ckeck this for my Sieger BK15 with es73 Thermostat. Can I test this with V1 as well?

BTW: WW activation / deactivation ist not working with my system -> only tested with V1

proddy, 2020-06-26:
@SpaceTeddy does your boiler have settings like hot, eco, comfort etc? I couldn't find it in Norberts HT3 list.

SpaceTeddy, 2020-06-26:
absolutly not sure, but I can set WW only to ECO or AUTO by knob. I guess that AUTO means, that it is controlled by the external es73 controller. Therefore it is in AUTO position. EMS-ESP info screen shows me HOT, so  it may possible, that these three modes are not possible with my system.

Bonusbartus, 2020-06-27:
Aren't the values incorrect here in the first place?
`// 1=hot, 2=eco, 3=intelligent
void Boiler::set_warmwater_mode(const uint8_t comfort) {
    uint8_t set;
    if (comfort == 1) {
        LOG_INFO(F("Setting boiler warm water to hot"));
        set = 1;
    } else if (comfort == 2) {
        LOG_INFO(F("Setting boiler warm water to eco"));
        set = 0;
    } else if (comfort == 3) {
        LOG_INFO(F("Setting boiler warm water to intelligent"));
        set = 2;
    } else {
        return; // do nothing
    }
    write_command(EMS_TYPE_UBAParameterWW, 9, comfort);
    // some boilers do not have this setting, than it's done by thermostat
    // Test for RC35, but not a good way, we are here in boiler context.
    // EMSESP::send_write_request(0x37, 0x10, 2, &set, 1, 0); // for RC35, maybe work also on RC300
}`

IIRC the values should be 0x00,  0xDB and 0xEC.
Also it could be necessary to also switch WW , to OFF and ON.... Don't remember...

MichaelDvP, 2020-06-27:
@Bonusbartus 
Tested on my system (GB125, MC10, BC10, RC35, MM10) , but no luck. I can only change WW mode on the RC35 with 0x37 telegram.
BTW: the first time i see that the controller (BC10) are sending telegrams and revert all my changes to the boiler
1. Test set to EC, than toggle off/on
```
send telegram "0B 08 33 09 EC"
001+05:32:28.265 N 3: [telegram] Rx: 08 00 33 00 08 FF 30 FB FF 28 FF 07 46 EC 00 BB
001+05:32:28.524 N 4: [telegram] Rx: 09 08 33 09 00 0E
001+05:32:38.548 N 5: [telegram] Rx: 08 00 33 00 08 FF 30 FB FF 28 FF 07 46 00 00 7A
send telegram "0B 08 33 01 00"
001+05:33:57.018 N 7: [telegram] Rx: 08 00 33 00 08 00 30 FB FF 28 FF 07 46 00 00 04
001+05:33:57.278 N 8: [telegram] Rx: 09 08 33 01 FF E1
001+05:34:03.107 N 9: [telegram] Rx: 08 0B 33 00 08 FF 30 FB FF 28 FF 07 46 00 00 3F
send telegram "0B 08 33 01 FF"
001+05:34:34.856 N 11: [telegram] Rx: 08 00 33 00 08 FF 30 FB FF 28 FF 07 46 00 00 7A
```
2. Test: set off, than change to EC, than back to on
```
send telegram "0B 08 33 01 00"
001+05:35:03.109 N 12: [telegram] Rx: 08 0B 33 00 08 00 30 FB FF 28 FF 07 46 00 00 41
001+05:35:12.150 N 14: [telegram] Rx: 09 08 33 01 FF E1
001+05:35:22.177 N 15: [telegram] Rx: 08 00 33 00 08 FF 30 FB FF 28 FF 07 46 00 00 7A
send telegram "0B 08 33 09 EC"
001+05:35:35.654 N 16: [telegram] Rx: 08 00 33 00 08 FF 30 FB FF 28 FF 07 46 EC 00 BB
001+05:35:35.913 N 17: [telegram] Rx: 09 08 33 09 00 0E
send telegram "0B 08 33 01 FF"
001+05:35:55.621 N 18: [telegram] Rx: 08 00 33 00 08 FF 30 FB FF 28 FF 07 46 00 00 7A
001+05:36:03.108 N 19: [telegram] Rx: 08 0B 33 00 08 FF 30 FB FF 28 FF 07 46 00 00 3F
```

proddy, 2020-06-27:
> Aren't the values incorrect here in the first place?
> `// 1=hot, 2=eco, 3=intelligent void Boiler::set_warmwater_mode(const uint8_t comfort) { uint8_t set; if (comfort == 1) { LOG_INFO(F("Setting [...]

@Bonusbartus was right, at least on my Nefit trendline boiler these work immediately:
`send telegram "b 8 33 9 0` = Hot (value 0x00)
`send telegram "b 8 33 9 ec` = Intelligent (value 0xEC)
`send telegram "b 8 33 9 d8` = Eco (value 0xD8)

closed 2020-07-06 by proddy