emsesp/EMS-ESP #414: MQTT output doesn't match WIKI document

bennybubble, 2020-06-27:
Hi,
Two days ago I upgraded to 1.9.5 from 1.9.4 and noticed that some of the published topics do not match the WIKI page.
I assumed (I know this could lead to wrong things) that the WIKI page reflects the latest 'official' version, being 1.9.5.

Below a screenshot from MQTT explorer abt 20 minutes after restart:
[image]

The WIKI says that **status** returns version, IP etc but it only shows "online" while I would expect this with the **start** topic.
I went through all the config pages in the WEBUI but couldn't find anything that relates to this.

Thanks for your help!

-ben

proddy, 2020-06-27:
strange, there should be a topic called 'start' with the IP and version in the payload. If NTP is enabled it will send the start topic when it has time (for boottime), if not it will send it immediately.

bennybubble, 2020-06-27:
I checked the NTP settings and they are correct: the status page in the WEBUI show sthe correct time.
After two hours still no correct status message:

[image]

-ben

proddy, 2020-06-27:
'status' will always show either online or offline. It's used as the MQTT last will and also automation systems (like HA) to see if the device is still active. The IP and version is in topic called 'start'.

bennybubble, 2020-06-27:
Thanks for updating the WIKI, I restarted esp-ems several times but it still doesn't publish the start topic.
Anything else I can try?

proddy, 2020-06-27:
that's very strange. I'll try and reproduce here. It's been a while since I touched the 1.9 code. 

You could try switching to version 2.0. It's still in development but pretty stable. I haven't touched any of the 1.9 code for a while now.

proddy, 2020-06-27:
Did a quick test, with ntp enabled I didn't get the start topic. In telnet after a `set ntp_enabled off` and `restart` it came up again. Can you try that?

bennybubble, 2020-06-28:
Yes, this works !
Afterwards I changed it back to on again and now I see the complete start message including version!

[image]

Thanks, Ben

proddy, 2020-06-28:
great, still, it's probably a bug in the code somewhere but since we'll all moving to v2 I won't spend the time hunting out. With this workaround, it should be ok for now, Thanks for reporting this.

closed 2020-06-28 by proddy

bennybubble, 2020-06-28:
Had to do a restart of mes-esp (ntp_enabled was on) and no start message anymore.
I'll wait for the official V2.0 and give that a try!

Thanks anyway!