emsesp/EMS-ESP #418: (v2) tRC35 - thermostat show doesn't show temp set by change temp

susisstrolch, 2020-07-02:
Thermostat: RC35 (DeviceID:0x10, ProductID:86, Version:01.11)

Tried to set mode to summer via telnet.
>su
>thermostat
>show
```
Thermostat: RC35 (DeviceID:0x10, ProductID:86, Version:01.11)
  Clock: 10:10:59 02/07/2020
  Warm Water mode: auto
  Damped Outdoor temperature: 19°C
  Temp sensor 1: 27.2°C
  Temp sensor 2: 27.8°C
  Offset int. temperature: 0.0°C
  Min ext. temperature: -12°C
  Building: medium
  Heating Circuit 1:
    Setpoint room temperature: 0.0°C
    **Mode: night**
    **Mode Type: night**
    Day temperature: 22.0°C
    Night temperature: 18.0°C
    **Holiday temperature: 17.0°C**
    Offset temperature: 0.0°C
    Design temperature: 77°C
    **Summer temperature: 9°C**
    Target flow temperature: 5°C
```
> change temp 9 holiday
> show
```
Thermostat: RC35 (DeviceID:0x10, ProductID:86, Version:01.11)
  Clock: 10:17:57 02/07/2020
  Warm Water mode: auto
  Damped Outdoor temperature: 19°C
  Temp sensor 1: 27.2°C
  Temp sensor 2: 27.8°C
  Offset int. temperature: 0.0°C
  Min ext. temperature: -12°C
  Building: medium
  Heating Circuit 1:
    Setpoint room temperature: 0.0°C
    **Mode: auto**
    **Mode Type: day**
    **Program is set to Summer mode**
    Day temperature: 22.0°C
    Night temperature: 18.0°C
    **Holiday temperature: 17.0°C**
    Offset temperature: 0.0°C
    Design temperature: 77°C
    Summer temperature: 9°C
    Target flow temperature: 5°C
```
Mode changed from night to day, no change to holiday temperature
(maybe the 9° are out of range for the RC35 - can't also go below 15° on RC35 console). 
Also the "Program changed to summer mode" occurred - w/o request to change the current mode.

Next try:
> set temp 15 holiday
> show
```
Thermostat: RC35 (DeviceID:0x10, ProductID:86, Version:01.11)
  Clock: 10:18:57 02/07/2020
  Warm Water mode: auto
  Damped Outdoor temperature: 19°C
  Temp sensor 1: 27.2°C
  Temp sensor 2: 27.8°C
  Offset int. temperature: 0.0°C
  Min ext. temperature: -12°C
  Building: medium
  Heating Circuit 1:
    Setpoint room temperature: 0.0°C
    **Mode: auto**
    **Mode Type: day**
    **Program is set to Summer mode**
    Day temperature: 22.0°C
    Night temperature: 18.0°C
    **Holiday temperature: 9.0°C**
    Offset temperature: 0.0°C
    Design temperature: 77°C
    Summer temperature: 9°C
    Target flow temperature: 5°C
```
Console show holiday temperature as 9°C,  even 15°C was requested.

MichaelDvP, 2020-07-02:
You can not set summermode. This mode is set automacally by the RC  if the damped outdoor temperature is above the summer temperature. If you change the summer temperature you can trigger the mode switch. But the a31 set all temperatures with factor 2, but summertemp is factor 1 (also designtemp and nofrosttemp has to be changed to factor 1. I'll check it on my RC35 and make a pr).
If summer mode is activated, setpoint room temperature goes to zero.

Also for holiday mode: to set this mode you have to set the time for holiday begin and end, than the controller switches to holiday mode.  Setting of holiday/party-times is not implemented yet (type-ID 0x3F, pos 87 ff.)

The show command is updated on every read, if you change a temperature you have to wait the next read to get the new value.

proddy, 2020-07-06:
think we can close this after your latest update Michael?

MichaelDvP, 2020-07-07:
Temp changes working for me on RC35 and displayed a bit faster, mode change also, but mode-type takes some time until it is set and published. 
If @susisstrolch agree you can close.

susisstrolch, 2020-07-07:
You can close that one.

closed 2020-07-07 by proddy