emsesp/EMS-ESP #421: (v2) New Web UI

proddy, 2020-07-05:
First port to the new web framework is in the [v2_web](https://github.com/proddy/EMS-ESP/tree/v2_web) branch.

Tested on ESP32. There are still some issues on the ESP8266 due to lack of heap memory.

A lot of the console commands have moved or been removed so check out the README.

MichaelDvP, 2020-07-06:
Very nice interface. Tested with the Wemos-esp32. First observertions:
- Typo in `EMSESPSettingsController.tsx` line 163 should be `value={data.syslog_mark_interval}` (and 0 for off is missing)
 I think it's better to put the syslog server below shower-settings to the other syslog settings. 
- i can't get mqtt to use `home/ems-esp` as before, setting Client-Id to `home/ems-esp` does not work, it ignores home/ and use the same path as Client-Id  `ems-esp`
- when using the console i get some strange effects, after a few minutes some sensors disappear, then the hc2 of thermostat disappear, a bit later the thermostat doesn't update anymore, but showing old data for time, building, etc. But no tx-errors in the log. Without using the console everything works good. 
- as in the alpha-versions with esp32: after OTA (espota) it often doesn't connect to ems-bus, i need to restart and everything is fine. I havn't tested web-upload yet.

proddy, 2020-07-06:
thanks for the feedback.
- I've made the changes to the settings.
- I dropped the mqtt_base. It was confusing some people before so I checked other systems (like tasmota, home assistant, espurna, esphome) and thought just using the hostname should suffice. If you think it's needed I can add it back.
- Strange that the console is not refreshing. I'll look into it
- during the OTA the UART is disabled, and I never re-enable it since I assumed the ESP device automatically gets booted. Perhaps it's a soft boot. You shouldn't see this problem anymore in the v2_web but let me know if its still a problem.

MichaelDvP, 2020-07-06:
> I dropped the mqtt_base

Ok, that's explaining it. I don't need the base, it was only to for changing between different moduls. I've changed now my readout script and reset the base on the other 8266 with a32.

> Strange that the console is not refreshing

It's not he only the console is not refreshing, The values also disappearing from mqtt after opening the console. 

> You shouldn't see this problem anymore in the v2_web but let me know if its still a problem.

I've see it also in v2_web (one time for now). It's not a problem since i know i have to reboot. Only the frst times i tested uart changes on esp32 were: change code - ota - not working - change code - ota -  ...  - reset all code - ota - not working - ?

proddy, 2020-07-06:
@MichaelDvP since v2_web has a different networking & settings core I'm thinking of merging it back into v2 now that the web is pretty stable and still keeping the git history. It does mean we all need to move over to the new v2 code. Let me know if you're ok with this and when a good time is.

MichaelDvP, 2020-07-07:
I thought the extra branch is only until 8266 is ok, and my 8266 is running stable since a hour. Also i don't have the console issue with 8266, i'll clean flash the esp32 and look if it is still there (it was ota over v2 on the esp32). I tried factory reset from web, but all settings stay, it does not work as expected. 
Also i have problems with sending mqtt cmds, seems there is something wrong with mqtt subscriptions. Mqtt publish works good.

proddy, 2020-07-07:
I'll look into why the Factory Reset is not working on the ESP32. The MQTT subscribe I know what the problem is - I just forgot to finish off the coding and rushed it.

MichaelDvP, 2020-07-07:
After clean flash the factory reset works on esp32.

FredericMa, 2020-07-08:
I get the following error while building for ESP8266:

```
lib\framework\NTPSettingsService.cpp: In member function 'void NTPSettingsService::configureNTP()':
lib\framework\NTPSettingsService.cpp:68:34: error: 'setTZ' was not declared in this scope
     setTZ(_state.tzFormat.c_str());
```

proddy, 2020-07-08:
> I get the following error while building for ESP8266:
> 
> ```
> lib\framework\NTPSettingsService.cpp: In member function 'void NTPSettingsService::configureNTP()':
> lib\framework\NTPSettingsService.cpp:68:34: error: 'setTZ' was not declared in this scope
>      setTZ(_state.tzFormat.c_str());
> ```

Is your pio environment up to date?

FredericMa, 2020-07-08:
> Is your pio environment up to date?

Hmm... didn't think about that. Good hint! I'll check later today when I'm home.

FredericMa, 2020-07-08:
Updated the pio env and now it compiles. Thanks!

bbqkees, 2020-07-09:
My build of v2_web here fails immediately on two Win 10 machines with the error 'ImportError: No module named pathlib:'.
I'm using Python 3.8 as interpreter in pio and verified pathlib is there so that's not the problem.
Spent some time on it but can't figure it out. Any ideas?

proddy, 2020-07-09:
Do you have npm installed?  https://nodejs.org/en/download/

bbqkees, 2020-07-09:
Yes that's installed.
The non-web v2 build fine a week ago and 1.9.5 today as well.

MichaelDvP, 2020-07-09:
With the actual code on 8266 the web is accessible only for the first AP-configuring, after connecting to station i can't open webpage. Gone back to b3 (commit cdfb04f77cb417f3b8d534befd7cd34ea590e04b) and web worked again.

proddy, 2020-07-09:
> With the actual code on 8266 the web is accessible only for the first AP-configuring, after connecting to station i can't open webpage. Gone back to b3 (commit cdfb04f77cb417f3b8d534befd7cd34ea590e04b) and web worked again.

I noticed that too last night. I’m out for a few days so can’t fix it.

FredericMa, 2020-07-09:
It isn't completely broken. I'm running a build from the last commit and I noticed that after a restart or after not accessing the UI for a while, I need to refresh a few times before the UI shows up. Sometimes one refresh is enough.
Once loaded, navigating through the pages is no issue.
Sometimes it also causes a restart of the ESP8266.

MichaelDvP, 2020-07-09:
I tested some builds and it's something in commit https://github.com/proddy/EMS-ESP/commit/cef80dff125e1db5222152015eb5ff5c4daa81ff 
But i don't understand exactly what this change do an what causes the issue.

proddy, 2020-07-09:
I think in the ap code. I’ll take a look later.

proddy, 2020-07-09:
> Yes that's installed.
> The non-web v2 build fine a week ago and 1.9.5 today as well.

You have Python 2.7 still installed on your win10 machine which is being picked up by the build subsystem. The web code needs 3.8. Just remove Python 2.7 and restart pio and it should work again.

MichaelDvP, 2020-07-11:
> Once loaded, navigating through the pages is no issue.
Sometimes it also causes a restart of the ESP8266.

Yes, i can reproduce this with actual built, if a browser window stays open all works fine. And i also had this issue (rare) with the older built, so it's probably not the mentioned commit. Maybe when closing the browser session is not cleaned up proper.

MichaelDvP, 2020-07-14:
The console `su`-command excepts any password. It's in line 352 of `console.cpp`, The line should be:
`if (securitySettings.jwtSecret.equals(password.c_str())) {`

bbqkees, 2020-07-14:
Removing 2.7 did the trick.
With the firmware on a new ESP8266 the ESP crashes/reboots when accessing the AP. 
Setting the wifi details in pio and uploading the firmware again the ESP can be accessed.

The new web interface is great! It is so fast, even on the ESP8266.

Some things that I noticed:
I had to manually scan the bus for the Moduline 400 thermostat to be detected (was perhaps the same on 1.9.5, can't remember).
What is the MQTT client ID for?
The option to set the MQTT topic base is missing.
Some MQTT settings are in the MQTT options, but other MQTT setting are in the general EMS-ESP settings.

How long is a certain EMS bus status retained?
If you get a few TX errors, are these displayed until a reboot?

proddy, 2020-07-14:
I'll look at fixing the AP crash this weekend. I shouldn't have checked-in untested code before going on vacation ;-)

The Moduline 400 should be automatically detected. I suspect its the Tx Mode setting. In the latest changes I've been unable to find a mode that gives no errors (before 22 was working) on our EMS 1.0 Nefit Trendlines.

I decided to drop the mqtt base after looking into what tasmota and other firmwares do. And I will move all the MQTT settings under one web page eventually too when I know everything is working.

The bus status is calculated every few seconds. If more than 5% of the telegrams sent fail after 3 retries it's considered a fail. This includes both read and write actions. When EMS-ESP is booted the first Tx will be the one asking the UBA Master/Boiler for what device IDs he has on the bus.

FredericMa, 2020-07-16:
Probably I'm doing something wrong but it looks like the generic thermostat command doesn't work anymore.
I'm trying to send payload `{"hc2":{"mode":"heat","heattemp":"22.5"}}` to topic `ems-esp/thermostat_cmd`.
It used to work with the `v2` branch but then the topic was `home/ems-esp/thermostat_cmd`.
I've set logging to debug but nothing shows up after publishing a message.

proddy, 2020-07-16:
@FredericMa okay, I'll try and simulate here. A lot has changed in the MQTT code so it's most likely a bug.

MichaelDvP, 2020-07-17:
For me the `thermostat_cmd` is working. @FredericMa the topic used is the wifi-hostname, not the mqtt-Client-ID, check if you are using the right topic.

FredericMa, 2020-07-17:
The hostname is also set at `ems-esp`.
Also the published data from ems-esp is using `ems-esp/`

FredericMa, 2020-07-17:
Now I think of it, it rebooted my server which also restarted mosquito. Maybe reconnecting to mosquito doesn't resubscribe. Will check later today.

PhillyGilly, 2020-07-17:
Although I'm running very well on 1.9.6b0 I'm looking forward to trying v2 on my ESP.

proddy, 2020-07-17:
@MichaelDvP branch v2_web merged back into v2

FredericMa, 2020-07-17:
I can confirm the above.
Restarted ems-esp, `thermostat_cmd` works perfect.
Restart mosquitto, `thermostat_cmd` doesn't work anymore.
Restart ems-esp again, all fine again.

I do see ems-esp reconnecting in the log:
```
000+00:00:41.695 I 51: [mqtt] MQTT connected
```

proddy, 2020-07-17:
If you restart the MQTT broker mosquitto it will loose all its subscriptions. There may be a way to make the 'retain', not sure but I'll look into it. If you're using home assistant do you also use the MQTT format called 'Home Assistant' which will give you MQTT discovery support. Which reminds me I'd better start on updating the documentation on how this works.

FredericMa, 2020-07-18:
AFAIK it is up to the client to resubscribe when reconnecting. Or at least that's how I do it for my mqtt projects at home.
I'm using home assistant but didn't use the built in discovery yet.

proddy, 2020-07-18:
@FredericMa if you disable the MQTT 'clean session' flag the broker should remember the sessions, including previous subscriptions. Can you try this? I'll change to 'true' as default like in EMS-ESP v1.9. See also http://www.steves-internet-guide.com/mqtt-clean-sessions-example/.

FredericMa, 2020-07-19:
I tried the code from commit 0d7c1421887ff93d6d16e9779a6b83adc97aaf16 but I'm still experiencing the same issue.
I still need to restart ems-esp after restarting mosquitto in order to send commands.

proddy, 2020-07-19:
ok, I'll build in the re-subscribe function.

FredericMa, 2020-07-19:
I tested the latest code but there is an issue with the resubscribe topics. They contain double ems-esp prefixes: 
```
000+00:11:56.031 I 207: [mqtt] MQTT connected
000+00:11:56.031 D 208: [mqtt] Publishing topic ems-esp/info (#379, attempt #1, pid 1)
000+00:11:56.282 D 209: [mqtt] Publishing topic ems-esp/status (#380, attempt #1, pid 1)
000+00:11:56.533 D 210: [mqtt] Subscribing to topic: ems-esp//cmd
000+00:11:56.784 D 211: [mqtt] Subscribing to topic: ems-esp/ems-esp/boiler_cmd
000+00:11:57.035 D 212: [mqtt] Subscribing to topic: ems-esp/ems-esp/boiler_cmd_wwactivated
000+00:11:57.286 D 213: [mqtt] Subscribing to topic: ems-esp/ems-esp/boiler_cmd_wwonetime
000+00:11:57.537 D 214: [mqtt] Subscribing to topic: ems-esp/ems-esp/boiler_cmd_wwcirculation
000+00:11:57.788 D 215: [mqtt] Subscribing to topic: ems-esp/ems-esp/boiler_cmd_wwtemp
000+00:11:57.993 D 216: [thermostat] Performing a thermostat publish (device ID 0x10)
000+00:11:58.039 D 217: [mqtt] Subscribing to topic: ems-esp/ems-esp/thermostat_cmd
000+00:11:58.290 D 218: [mqtt] Subscribing to topic: ems-esp/ems-esp/thermostat_cmd_temp
000+00:11:58.541 D 219: [mqtt] Subscribing to topic: ems-esp/ems-esp/thermostat_cmd_mode
```

I guess because of this fix `clean-session` can be `true` again, right?

proddy, 2020-07-19:
Could also be something I missed in code. I'll check again.

MichaelDvP, 2020-07-21:
> Could also be something I missed in code. I'll check again.

Yes, i think in line 77 of `mqtt.cpp` you call `make_topic(` , but the function already contains the full topic (see line 402).

proddy, 2020-07-21:
> > Could also be something I missed in code. I'll check again.
> 
> Yes, i think in line 77 of `mqtt.cpp` you call `make_topic(` , but the function already contains the full topic (see line 402).

It was a bug somewhere else, fixed and tested in b7. I also extended the `show mqtt` command to list the subscribed topics.

FredericMa, 2020-07-21:
I can confirm that the MQTT subscriptions on reconnect are working as expected now.

closed 2020-07-30 by proddy